elFarto/nvidia-vaapi-driver #3: ERROR: Include dir /usr/include/cuda does not exist (need help building on Arch)

linxlan, 2022-01-04:
I'm trying to build on Arch Linux but got an error. I have package "cuda" installed, but it looks like I'm missing some more dependencies.

```
The Meson build system
Version: 0.60.3
Source dir: /home/user/Downloads/nvidia-vaapi-driver
Build dir: /home/user/Downloads/nvidia-vaapi-driver/build
Build type: native build
Project name: nvidia-vaapi-driver
Project version: 0.1
C compiler for the host machine: ccache cc (gcc 11.1.0 "cc (GCC) 11.1.0")
C linker for the host machine: cc ld.bfd 2.36.1
Host machine cpu family: x86_64
Host machine cpu: x86_64
Library m found: YES
Library EGL found: YES
Library cuda found: YES
Checking for function "cuInit" with dependency -lcuda: YES 

meson.build:10:4: ERROR: Include dir /usr/include/cuda does not exist.

A full log can be found at /home/user/Downloads/nvidia-vaapi-driver/build/meson-logs/meson-log.txt
```

[meson-log.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/7807858/meson-log.txt)

helloer, 2022-01-04:
Change `/usr/include/cuda` to `/opt/cuda/include` in https://github.com/elFarto/nvidia-vaapi-driver/blob/6d285a0f04f2bc5641eb2c8be0dbbc4c601a95ee/meson.build#L10

linxlan, 2022-01-04:
It works, thanks. Now I got more errors. I will be grateful if you can help.

meson install -C build
```
ninja: Entering directory `/home/user/Downloads/nvidia-vaapi-driver/build'
[2/12] Compiling C object nvidia_drv_video.so.p/src_vabackend.c.o
FAILED: nvidia_drv_video.so.p/src_vabackend.c.o 
ccache cc -Invidia_drv_video.so.p -I. -I.. -I/opt/cuda/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/local/include -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -g -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_vabackend.c.o -MF nvidia_drv_video.so.p/src_vabackend.c.o.d -o nvidia_drv_video.so.p/src_vabackend.c.o -c ../src/vabackend.c
../src/vabackend.c:17:10: fatal error: drm/drm_fourcc.h: No such file or directory
   17 | #include <drm/drm_fourcc.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
[8/12] Compiling C object nvidia_drv_video.so.p/src_h264.c.o
../src/h264.c: In function ‘computeH264CudaCodec’:
../src/h264.c:114:5: warning: ‘VAProfileH264Baseline’ is deprecated [-Wdeprecated-declarations]
  114 |     if (profile == VAProfileH264Baseline || profile == VAProfileH264ConstrainedBaseline || profile == VAProfileH264Main || profile == VAProfileH264High) {
      |     ^~
In file included from /usr/include/va/va_backend.h:32,
                 from ../src/vabackend.h:5,
                 from ../src/h264.c:1:
/usr/include/va/va.h:501:5: note: declared here
  501 |     VAProfileH264Baseline va_deprecated_enum = 5,
      |     ^~~~~~~~~~~~~~~~~~~~~
../src/h264.c: At top level:
../src/h264.c:134:5: warning: ‘VAProfileH264Baseline’ is deprecated [-Wdeprecated-declarations]
  134 |     .supportedProfiles = { VAProfileH264Baseline, VAProfileH264ConstrainedBaseline, VAProfileH264Main, VAProfileH264High }
      |     ^
In file included from /usr/include/va/va_backend.h:32,
                 from ../src/vabackend.h:5,
                 from ../src/h264.c:1:
/usr/include/va/va.h:501:5: note: declared here
  501 |     VAProfileH264Baseline va_deprecated_enum = 5,
      |     ^~~~~~~~~~~~~~~~~~~~~
[11/12] Compiling C object nvidia_drv_video.so.p/src_vp9.c.o
ninja: build stopped: subcommand failed.
Could not rebuild /home/user/Downloads/nvidia-vaapi-driver/build
```

helloer, 2022-01-04:
Change `#include <drm/drm_fourcc.h>` to `#include <libdrm/drm_fourcc.h>` in https://github.com/elFarto/nvidia-vaapi-driver/blob/6d285a0f04f2bc5641eb2c8be0dbbc4c601a95ee/src/vabackend.c#L17

linxlan, 2022-01-04:
It works now, thanks.

ManuLinares, 2022-01-04:
Why is this issue closed? The bug isn't fixed.

At least, post the PATCH in the README for arch instructions.

eli-schwartz, 2022-01-04:
https://mesonbuild.com/Dependencies.html#cuda

This should not be manually coding arbitrary directories and find_library calls. Use meson's official cuda dependency lookup.

dawidgarus, 2022-01-04:
I created a fix for it: https://github.com/elFarto/nvidia-vaapi-driver/pull/9

eli-schwartz, 2022-01-05:
@dawidgarus

That fix hardcodes two arbitrary directories instead of using Meson's cuda support.

elFarto, 2022-01-05:
Using Meson's CUDA support is difficult, as it wants to link with libcudart (the full runtime), and we only need libcuda (the one the display driver installs). I don't really want to require users install all the CUDA dependencies just for this.
It may be possible to use Mesons built in support and modify the dependency after, but I'm just not that familiar with it.

philipl, 2022-01-05:
Easier to fix #1. I still want to work on this, but need to find the time.

elFarto, 2022-01-09:
Closing this in favour of #1

closed 2022-01-09 by elFarto