elFarto/nvidia-vaapi-driver #4: Not working with 495.46 driver on wayland

X-dark, 2022-01-04:
When running vainfo I get:

```
$ NVD_LOG=1 vainfo 
[30879-30879] ../src/vabackend.c:1522       __vaDriverInit_1_0 Initing NVIDIA VA-API Driver
libEGL warning: DRI2: failed to create dri screen
[30879-30879] ../src/export-buf.c:  56                reconnect Reconnecting to stream
[30879-30879] ../src/export-buf.c:  66                reconnect cuda error 'invalid resource handle' (400)
```

Is there any more debug I could do to help understand?

elFarto, 2022-01-04:
That's not unexpected, I don't have any Wayland stuff setup at the moment so I haven't tested with it.

You could try moving the `eglDebugMessageControlKHR` call to just after the first `eglInitialize` call in the `initExporter` function. That might show exactly what is failing.

I'll see about improving the error checking/logging around that initialisation code.

NXij, 2022-01-04:
nvdec does not work on wayland yet afaik

krakow10, 2022-01-04:
> nvdec does not work on wayland yet afaik
Last time I checked it worked fine with mpv

Whitecatkeke, 2022-01-04:
Does mpv works with nvdec-copy but not nvdec? Just tried to use nvdec with mpv, and I saw the same error "libEGL warning: DRI2: failed to create dri screen". Nvdec-copy works well.

philipl, 2022-01-04:
That probably means your egl libraries aren't set up correctly.

elFarto, 2022-01-04:
I've pushed some changes to the EGL initialisation code to make it a little more robust and log more information, can you try reproducing it with the latest version?

Whitecatkeke, 2022-01-04:
> I've pushed some changes to the EGL initialisation code to make it a little more robust and log more information, can you try reproducing it with the latest version?

Thanks. I think it is working now with Firefox. Strangely, mpv triggers vaapi successfully, however, the videos are green with error (nvidia-drm.modeset=1 has been set up to enable Wayland)

[vo/gpu/opengl] before video texture creation: OpenGL error INVALID_OPERATION.
[vo/gpu/opengl] after creating texture: OpenGL error INVALID_OPERATION.

DatCrazyDelphox, 2022-01-04:
> > I've pushed some changes to the EGL initialisation code to make it a little more robust and log more information, can you try reproducing it with the latest version?
> 
> Thanks very much. I think it is working now with Firefox. Strangely, mpv triggers vaapi successfully, however, the videos are green with error (nvidia-drm.modeset=1 has been set up to enable Wayland)
> 
> ```
> [vo/gpu/opengl] before video texture creation: OpenGL error INVALID_OPERATION.
> [vo/gpu/opengl] after creating texture: OpenGL error INVALID_OPERATION.
> ```
> 
> 

Tô fix this mpv error you should try setting hwdec to vaapi-copy instead, then it should work

Whitecatkeke, 2022-01-04:
> > > I've pushed some changes to the EGL initialisation code to make it a little more robust and log more information, can you try reproducing it with the latest version?
> > 
> > 
> > Thanks very much. I think it is working now with Firefox. Strangely, mpv triggers vaapi successfully, however, the videos are green with error (nvidia-drm.modeset=1 has been set up to enable Wayland)
> > ```
> > [vo/gpu/opengl] before video texture creation: OpenGL error INVALID_OPERATION.
> > [vo/gpu/opengl] after creating texture: OpenGL error INVALID_OPERATION.
> > ```
> 
> To fix this mpv error you should try setting hwdec to vaapi-copy instead, then it should work

Vaapi-copy works like a charm. Could I ask why vaapi does not work? Thanks very much.

elFarto, 2022-01-04:
hwdec=vaapi does work, but you'll need a nightly build of mpv due to a omission in the OpenGL/VA-API code that wasn't using the DRM modifiers. There's also a test.sh script in the source tree that has some extra options you can pass in to mpv to make it work with more codecs (ffmpeg disables hardware decode for some codecs).

Whitecatkeke, 2022-01-04:
Thanks very much for your explanation.

DatCrazyDelphox, 2022-01-04:
> Vaapi-copy works like a charm. Could I ask why vaapi does not work? Thanks very much.

Might be because nvdec currently doesn't have a dma-buf interface to send data directly to the gpu or something like that...

X-dark, 2022-01-06:
> I've pushed some changes to the EGL initialisation code to make it a little more robust and log more information, can you try reproducing it with the latest version?

Thank you. vainfo is not complaining anymore but some errors are reported by NVD_LOG:

```
❯ NVD_LOG=1 vainfo 
[53380-53380] ../src/vabackend.c:1522       __vaDriverInit_1_0 Initing NVIDIA VA-API Driver
[53380-53380] ../src/export-buf.c:  90          findCudaDisplay Found 3 EGL devices
[53380-53380] ../src/export-buf.c:  94          findCudaDisplay Got EGL_CUDA_DEVICE_NV value '0' from device 0
[53380-53380] ../src/export-buf.c: 121             initExporter Got EGLDisplay from CUDA device
[53380-53380] ../src/export-buf.c:  56                reconnect Reconnecting to stream
vainfo: VA-API version: 1.13 (libva 2.13.0)
vainfo: Driver version: VA-API NVDEC driver
[53380-53380] ../src/vabackend.c: 202              vaToCuCodec vaToCuCodec: Unknown codec: 2
[53380-53380] ../src/vabackend.c: 202              vaToCuCodec vaToCuCodec: Unknown codec: 3
[53380-53380] ../src/vabackend.c: 202              vaToCuCodec vaToCuCodec: Unknown codec: 4
[53380-53380] ../src/vabackend.c: 202              vaToCuCodec vaToCuCodec: Unknown codec: 12
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileAV1Profile0            :	VAEntrypointVLD
[53380-53380] ../src/vabackend.c:1511              nvTerminate In nvTerminate
[53380-53380] ../src/export-buf.c:  36                    debug [EGL] eglStreamImageConsumerConnectNV: EGL_BAD_STATE_KHR error: In EGL Access Table::stream2.consumer.disconnect: Consumer handle does not match reservation (0x5aee4e81 vs 0x5aee4e89).
```

elFarto, 2022-01-06:
That EGL_BAD_STATE_KHR error can be ignored, it's harmless. I've fixed it in master.

elFarto, 2022-01-09:
Closing issue as it appears to be resolved now.

closed 2022-01-09 by elFarto