elFarto/nvidia-vaapi-driver #5: Chrome Support

luigifcruz, 2022-01-04:
Does this library support Chrome? If not, what would take to add support? 

Awesome library. Looking forward to testing it. I'm happy to contribute!

elFarto, 2022-01-04:
I haven't tried it with Chrome yet, I wasn't aware that VA-API support had been added, or is it a patched version?

It shouldn't be too hard to add support, it's unlikely to do anything drastically different from Firefox.

hf29h8sh321, 2022-01-04:
Running with `--enable-features=UseVaapiDecoder` prints the following errors:

```[419021:419021:0104/124417.354326:ERROR:vaapi_wrapper.cc(1070)] : FillProfileInfo_Locked failed for va_profile VAProfileH264Main and entrypoint VAEntrypointVLD
[419021:419021:0104/124417.356714:ERROR:vaapi_wrapper.cc(1070)] : FillProfileInfo_Locked failed for va_profile VAProfileH264High and entrypoint VAEntrypointVLD
[419021:419021:0104/124417.358761:ERROR:vaapi_wrapper.cc(1070)] : FillProfileInfo_Locked failed for va_profile VAProfileH264ConstrainedBaseline and entrypoint VAEntrypointVLD
[419021:419021:0104/124417.360671:ERROR:vaapi_wrapper.cc(1070)] : FillProfileInfo_Locked failed for va_profile VAProfileVP8Version0_3 and entrypoint VAEntrypointVLD
[419021:419021:0104/124417.362997:ERROR:vaapi_wrapper.cc(1070)] : FillProfileInfo_Locked failed for va_profile VAProfileVP9Profile0 and entrypoint VAEntrypointVLD```

luigifcruz, 2022-01-04:
@elFarto I think it's a flag that needs to be enabled on compile time. But I think most distributions have it enabled. Not sure how well it works with the new Ozone layer.

urbenlegend, 2022-01-04:
Getting the same errors as @hf29h8sh321 in Brave, which has VA-API enabled by default in their Linux builds.

Would love Chromium-based browser support here as well. Thanks!

robsterooni, 2022-01-04:
Anyone have any idea where a change would have to be made for this to work?   Looks like the error is logged from chromium itself, but I suspect the issue lies within some part of vaapi not bring implemented in this lib.  I will do some digging on this when I get some time and would appreciate any pointers anyone may have.

elFarto, 2022-01-04:
I'm focusing more on getting the library working well under Firefox, so this isn't a high priority at the moment.

The first thing to look at is where the FillProfileInfo_Locked message is logged, and see what it's looking for from the VA-API implementation. Once you can see why it's failing (likely a missing attribute, or function not implemented) we can look about fixing that.

robsterooni, 2022-01-04:
Cool, thanks, I will start there once I have repro-d the errors logged in this issue.   TBH i am hoping its just incorrect chromium flags as they are a mess and constantly changing.   Congrats on the first release of this btw - its a potential game changer until Vulkan saves us all.

helloer, 2022-01-04:
> The first thing to look at is where the FillProfileInfo_Locked message is logged, and see what it's looking for from the VA-API implementation. Once you can see why it's failing (likely a missing attribute, or function not implemented) we can look about fixing that.

Looks like the problem is with https://github.com/elFarto/nvidia-vaapi-driver/blob/5ddf7cadad170813ad6a4b17457020b4a3173d3a/src/vabackend.c#L452  chrome looks for VAConfigAttribRTFormat attribute [here](https://source.chromium.org/chromium/chromium/src/+/main:media/gpu/vaapi/vaapi_wrapper.cc;l=1176-1239) and right now it doesn't return such attributes

you can bypass this by hard coding values (very hacky)
```c
attrib_list->type == VAConfigAttribRTFormat;
attrib_list->value = VA_RT_FORMAT_YUV420;
*num_attribs = 1;
```
moreover vaCreateContext will fail because chrome doesn't pass [VASurfaceIDs](https://source.chromium.org/chromium/chromium/src/+/main:media/gpu/vaapi/vaapi_wrapper.cc;l=2056?q=vaCreateContext&ss=chromium) to vaCreateContext. Also chrome uses `vaPutSurface` which is not implemented.

philipl, 2022-01-04:
If chromium (Note that Google never build Chrome with vaapi support) is still using `vaPutSurface` in their X11 backend, then that makes everything slow anyway. It's probably not hard to implement, but it's also not zero-copy. The wayland backend must be using dma-buf passing if it's not going a CPU copy.

puyoxyz, 2022-01-05:
chrome support would be cool for playing stadia in 4k

elFarto, 2022-01-05:
Ok, the missing attribute is simple to fix, the surface IDs not being passed in is a bit more problematic, and the use of vaPutSurface is just weird. 

I think we can work around the surface IDs not being passed in, as we don't really need them, they're just assigned a picture number to be used later. This could be move to a dynamic allocation. We would then just need to come up with some reasonable defaults for the missing ulNumDecodeSurfaces and ulNumOutputSurfaces fields, which shouldn't be too hard (I already had a note to look at that anyway).

vaPutSurface is going to be hard to get working if they're using it. vaPutImage is also difficult to work around. As noted above we don't really have any surfaces, they're all managed for us, so to 'put' something into a surface is not an action we can do. We could try stubbing those methods to return successful but not actually do anything and see what that breaks. 

I noticed that their DMA-BUF export code seems to throw away the DRM modifiers, and that's a blocker. The NVIDIA driver requires they be passed into the import method (it's the same thing I've fixed in Firefox and MPV).

I also noticed some protected playback stuff, that's almost certainly never going to work, hopefully it doesn't use that by default.

helloer, 2022-01-05:
Looks like vaPutSurface will not be used when you start chromium with `--use-gl=egl` but there is another problem - chromium uses [VAProfileNone with VAEntrypointVideoProc as entrypoint](https://source.chromium.org/chromium/chromium/src/+/main:media/gpu/vaapi/vaapi_video_decode_accelerator.cc;l=715) which is currently not supported by the driver.

elFarto, 2022-01-24:
I've made a couple of changes to the library to try and support Chrome better, but I'm not having much luck testing it. Chrome gives me a weird EGL error when I try and start it. You'll likely need to use NVD_LOG=filename as it stdout doesn't get displayed.

helloer, 2022-01-25:
```
[36080-36080] ../src/export-buf.c: 141          findCudaDisplay Found 3 EGL devices
[36080-36080] ../src/export-buf.c: 145          findCudaDisplay Got EGL_CUDA_DEVICE_NV value '0' from device 0
[36080-36080] ../src/export-buf.c: 157          findCudaDisplay Checking device file: /dev/dri/renderD128
[36080-36080] ../src/export-buf.c: 164          findCudaDisplay Got DRM_IOCTL_GET_CAP ioctl response: 0 1
[36080-36080] ../src/export-buf.c: 204             initExporter Got EGLDisplay from CUDA device
[36080-36080] ../src/export-buf.c: 102                reconnect Reconnecting to stream
[36080-36080] ../src/vabackend.c: 446           nvCreateConfig got profile: 7 with 1 attributes
[36080-36080] ../src/vabackend.c: 465           nvCreateConfig got config attrib: 0 0 1
[36080:36232:0125/114726.887900:VERBOSE2:vaapi_video_decode_accelerator.cc(483)] DecodeTask(): Decoder requesting a new set of surfaces
[36080:36080:0125/114726.888755:VERBOSE2:vaapi_video_decode_accelerator.cc(578)] InitiateSurfaceSetChange():  |requested_num_pics_| = 6; |requested_num_reference_frames_| = 6
[36080:36080:0125/114726.888900:VERBOSE2:vaapi_video_decode_accelerator.cc(644)] TryFinishSurfaceSetChange(): Requesting 6 pictures of size: 1920x1088 and visible rectangle = 0,0 1920x1080
[36080:36080:0125/114726.890074:ERROR:vaapi_video_decode_accelerator.cc(724)] Failed to initialize VppVaapiWrapper
[36080:36080:0125/114726.890213:VERBOSE1:vaapi_video_decode_accelerator.cc(144)] NotifyError(): Notifying of error 4
[36080:36080:0125/114726.891818:VERBOSE2:vaapi_video_decode_accelerator.cc(1041)] Cleanup(): Destroying VAVDA
[36080-36080] ../src/vabackend.c:1621              nvTerminate In nvTerminate
[36080-36080] ../src/export-buf.c:  54          releaseExporter Releasing exporter, 0 outstanding frames
[36080-36080] ../src/export-buf.c:  71          releaseExporter Done releasing frames
[36080-36080] ../src/export-buf.c:  81          releaseExporter Done releasing EGLImages
```
`LIBVA_DRIVER_NAME=nvidia NVD_LOG=1 chromium --enable-features=VaapiVideoDecoder --enable-logging=stderr --loglevel=0 --vmodule=vaapi_wrapper=4,vaapi_video_decode_accelerator=4 --use-gl=egl`

Testing on 441c03d, I think it still fails because of missing VAProfileNone https://github.com/elFarto/nvidia-vaapi-driver/issues/5#issuecomment-1005603180

Tatsh, 2022-01-25:
Google Chrome Beta crashes a few times with `--use-gl=egl` and otherwise the above command makes no difference from before.

Tatsh, 2022-01-27:
Chrome Beta actually disables GPU entirely when this is loaded. Definitely something wrong.

elFarto, 2022-01-28:
@helloer A couple of questions. Which chromium build are you running? and are you running this under Wayland?

I'm getting these errors at the moment, attempting to run it under X11:
```
[21490:21490:0128/205707.729988:ERROR:gpu_init.cc(453)] Passthrough is not supported, GL is egl, ANGLE is 
[21490:21490:0128/205707.748983:ERROR:sandbox_linux.cc(374)] InitializeSandbox() called with multiple threads in process gpu-process.
[21490-21490] ../src/export-buf.c:  52                    debug [EGL] eglCreateWindowSurface: EGL_BAD_MATCH error: In eglCreateWindowSurface: Native and EGL formats are incompatible
[21490:21490:0128/205707.777189:ERROR:gl_surface_egl.cc(1510)] eglCreateWindowSurface failed with error EGL_BAD_MATCH
```

helloer, 2022-01-28:
> Which chromium build are you running? and are you running this under Wayland?
Chromium 97.0.4692.99 Arch Linux under X11. Maybe it's due to https://github.com/archlinux/svntogit-packages/blob/packages/chromium/trunk/unexpire-accelerated-video-decode-flag.patch?

elFarto, 2022-01-29:
I've spent a few hours working on this today and managed to get...not far. I've discovered that XFCE's display composition is what was causing the the EGL_BAD_MATCH errors (I think). Disabling that got me to helloer's error. I hacked in some code to pretend we support VAProfileVideoProc, then got to [this](https://source.chromium.org/chromium/chromium/src/+/refs/tags/97.0.4692.99:ui/gfx/linux/gpu_memory_buffer_support_x11.cc;drc=7aadec9885f60ddcac901e44fca4211c69e2dfe5;l=129) line.

I've checked the code in on the chrome_support branch.

timocapa, 2022-04-03:
> EGL_BAD_MATCH

I get the same thing on Gnome 42 on X11 with a GTX 1080 on 510.60.02

Not sure if I need `--use-gl=egl`, if I drop it the error goes too

If I open `chrome://gpu`, I get this:

```
[100114-100114] ../src/vabackend.c:1642       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x1bb4012f3800 10
[100114-100114] ../src/vabackend.c:1645       __vaDriverInit_1_0 Non-DRM display type detected, defaulting to GPU ID 0. Use NVD_GPU to pick a specific GPU.
[100114-100114] ../src/export-buf.c: 164       findGPUIndexFromFd Looking for GPU index: 0
[100114-100114] ../src/export-buf.c: 175       findGPUIndexFromFd Found 4 EGL devices
[100114-100114] ../src/export-buf.c: 184       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
[100114-100114] ../src/export-buf.c: 136 checkModesetParameterFromFd Unable to check nvidia_drm modeset setting
[100114-100114] ../src/export-buf.c: 208       findGPUIndexFromFd Selecting EGLDevice 0
[100114-100114] ../src/export-buf.c: 270             initExporter Driver doesn't support 16-bit surfaces
[100114-100114] ../src/export-buf.c: 103                reconnect Reconnecting to stream
[100114-100114] ../src/vabackend.c: 456           nvCreateConfig got profile: 6 with 1 attributes
[100114-100114] ../src/vabackend.c: 477           nvCreateConfig got config attrib: 0 0 1
[100114-100114] ../src/vabackend.c:1345 nvQuerySurfaceAttributes with 4 (nil) -1431655766
[100114-100114] ../src/vabackend.c:1345 nvQuerySurfaceAttributes with 4 0x1bb401640180 5
[100114-100114] ../src/vabackend.c: 456           nvCreateConfig got profile: 6 with 0 attributes
[100114-100114] ../src/vabackend.c: 456           nvCreateConfig got profile: 7 with 1 attributes
[100114-100114] ../src/vabackend.c: 477           nvCreateConfig got config attrib: 0 0 1
[100114-100114] ../src/vabackend.c:1345 nvQuerySurfaceAttributes with 4 (nil) -1431655766
[100114-100114] ../src/vabackend.c:1345 nvQuerySurfaceAttributes with 4 0x1bb401640180 5
[100114-100114] ../src/vabackend.c: 456           nvCreateConfig got profile: 7 with 0 attributes
[100114-100114] ../src/vabackend.c: 456           nvCreateConfig got profile: 13 with 1 attributes
[100114-100114] ../src/vabackend.c: 477           nvCreateConfig got config attrib: 0 0 1
[100114-100114] ../src/vabackend.c:1345 nvQuerySurfaceAttributes with 4 (nil) -1431655766
[100114-100114] ../src/vabackend.c:1345 nvQuerySurfaceAttributes with 4 0x1bb401640180 5
[100114-100114] ../src/vabackend.c: 456           nvCreateConfig got profile: 13 with 0 attributes
[100114-100114] ../src/vabackend.c: 456           nvCreateConfig got profile: 14 with 1 attributes
[100114-100114] ../src/vabackend.c: 477           nvCreateConfig got config attrib: 0 0 1
[100114-100114] ../src/vabackend.c:1345 nvQuerySurfaceAttributes with 9 (nil) -1431655766
[100114-100114] ../src/vabackend.c:1345 nvQuerySurfaceAttributes with 9 0x1bb401640180 5
[100114-100114] ../src/vabackend.c: 456           nvCreateConfig got profile: 14 with 0 attributes
[100114-100114] ../src/vabackend.c: 456           nvCreateConfig got profile: 19 with 1 attributes
[100114-100114] ../src/vabackend.c: 477           nvCreateConfig got config attrib: 0 0 1
[100114-100114] ../src/vabackend.c:1345 nvQuerySurfaceAttributes with 10 (nil) -1431655766
[100114-100114] ../src/vabackend.c:1345 nvQuerySurfaceAttributes with 10 0x1bb401640180 5
[100114-100114] ../src/vabackend.c: 456           nvCreateConfig got profile: 19 with 0 attributes
[100114-100114] ../src/vabackend.c:1626              nvTerminate Terminating 0x1bb4012f3800
[100114-100114] ../src/export-buf.c:  65          releaseExporter Releasing exporter, 0 outstanding frames
[100114-100114] ../src/export-buf.c:  82          releaseExporter Done releasing frames

```

pc00per, 2022-07-01:
Hoping this could work on chrome some day. There's already [vdpau backend implementation of va-api](https://github.com/xuanruiqi/vdpau-va-driver-vp9)  working perfectly on chromium. It currently supports all h264 profiles & vp9-profile0 & uses VDAVideoDecoder.
But as we know NVDEC is more efficient & closer to nvidia's proprietary implementation as VDPAU is dead outdated.

elFarto, 2022-07-24:
I've reworked some stuff and got Chromium on X11 (with desktop GL) support sort of working, but it's quite a hack and very inefficient, and the resulting image isn't quite the correct colour due to the library having to do the YUV -> RGB conversion. I would not recommend it for daily use.
Chromium on X11 with EGL won't work until NVIDIA support the DRI3 open method in their X11 driver, there's nothing I can do to work around that.

However, I'm having a problem getting Chromium on Wayland (with EGL) to even initialise VA-API. I'm using this command on Fedora 36/GNOME 42:
```
 NVD_LOG=1 chromium-freeworld --enable-features=VaapiVideoDecoder --password-store=basic --use-gl=egl --enable-logging=stderr --vmodule=gpu_video_decode_accelerator_factory=4,vaapi_wrapper=4,vaapi_video_decode_accelerator=4 --disable-gpu-sandbox --disable-features=UseChromeOSDirectVideoDecoder
```
But I see an output early on about DRM Prime not being supported. This seems to be the result of the wl_drm protocol on the display server returning the capabilities of 0. I can't find what's setting that value.

Can some of you try launching chromium with VA-API and NVD_LOG=1 and see if it initialises the library on startup? Also, could you note which wayland compositor you're using. You don't specifically need this branch to test it, chromium will at least initialise the regular version.

pc00per, 2022-07-24:
@elFarto Did you test your patches on xwayland with ozone enabled? I don't think egl will help with this because it has lot of other issues such as reduced gpu performance. Even if you make it work, no one gonna have chromium on egl backend.

elFarto, 2022-07-24:
--ozone-platform=wayland is automatically passed with chromium-freeworld. Just tried regular chromium too, slightly different error:
```
[57047:57047:0724/154417.696624:WARNING:wayland_drm.cc(96)] Failed to get drm magic
```
rather than:
```
[55021:55021:0724/154043.238392:WARNING:wayland_drm.cc(96)] Drm prime capability is not supported
```

timocapa, 2022-07-24:
```
timo@mauve ~> NVD_LOG=1 google-chrome-stable --enable-features=VaapiVideoDecoder --password-store=basic --use-gl=egl --enable-logging=stderr --vmodule=gpu_video_decode_accelerator_factory=4,vaapi_wrapper=4,vaapi_video_decode_accelerator=4 --disable-gpu-sandbox --disable-features=UseChromeOSDirectVideoDecoder --ozone-platform=wayland
[32525:32525:0724/164745.040269:WARNING:wayland_object.cc(95)] Binding to gtk_shell1 version 4 but version 5 is available.
[32525:32525:0724/164745.040300:WARNING:wayland_object.cc(95)] Binding to zwp_pointer_gestures_v1 version 1 but version 3 is available.
[32525:32525:0724/164745.040310:WARNING:wayland_object.cc(95)] Binding to zwp_linux_dmabuf_v1 version 3 but version 4 is available.
[32525:32525:0724/164745.040446:WARNING:wayland_drm.cc(96)] Drm prime capability is not supported
``` 
same thing on gchrome

pc00per, 2022-07-24:
@elFarto I meant xwayland btw. Let ozone choose X11 as ozone backend & see if it really works. I'm not on pc, as soon as I'm home, will try this myself.
`env GDK_BACKEND=x11 NVD_LOG=1 chromium`

Korothi, 2022-12-28:
```
[filip@PC-ARCH ~]$ NVD_LOG=1 chromium
     12752.030663233 [19273-19273] ../src/vabackend.c:2083       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
     12752.030673783 [19273-19273] ../src/vabackend.c:2092       __vaDriverInit_1_0 Now have 0 (0 max) instances
     12752.030677513 [19273-19273] ../src/vabackend.c:2118       __vaDriverInit_1_0 Selecting Direct backend
     12752.036261359 [19273-19273] ../src/direct/direct-export-buf.c:  90      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD128
     12752.036268689 [19273-19273] ../src/direct/nv-driver.c: 190            init_nvdriver Initing nvdriver...
     12752.036271839 [19273-19273] ../src/direct/nv-driver.c: 195            init_nvdriver Got dev info: a00 1 2 6
     12752.036337147 [19273-19273] ../src/direct/nv-driver.c: 256            init_nvdriver NVIDIA kernel driver version: 525.60.11
     12752.158712890 [19273-19273] ../src/vabackend.c: 637           nvCreateConfig got profile: 6 with 1 attributes
     12752.158720960 [19273-19273] ../src/vabackend.c: 658           nvCreateConfig got config attrib: 0 0 1
     12752.158723480 [19273-19273] ../src/vabackend.c:1749 nvQuerySurfaceAttributes with 4 (8) (nil) -1431655766
     12752.158726170 [19273-19273] ../src/vabackend.c:1749 nvQuerySurfaceAttributes with 4 (8) 0xd9800da6040 7
     12752.159820070 [19273-19273] ../src/vabackend.c:1807 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
     12752.159825800 [19273-19273] ../src/vabackend.c: 637           nvCreateConfig got profile: 6 with 0 attributes
     12752.159833889 [19273-19273] ../src/vabackend.c: 637           nvCreateConfig got profile: 7 with 1 attributes
     12752.159835809 [19273-19273] ../src/vabackend.c: 658           nvCreateConfig got config attrib: 0 0 1
     12752.159837439 [19273-19273] ../src/vabackend.c:1749 nvQuerySurfaceAttributes with 4 (8) (nil) -1431655766
     12752.159839319 [19273-19273] ../src/vabackend.c:1749 nvQuerySurfaceAttributes with 4 (8) 0xd9800da6040 7
     12752.160945249 [19273-19273] ../src/vabackend.c:1807 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
     12752.160950289 [19273-19273] ../src/vabackend.c: 637           nvCreateConfig got profile: 7 with 0 attributes
     12752.160954439 [19273-19273] ../src/vabackend.c: 637           nvCreateConfig got profile: 13 with 1 attributes
     12752.160956119 [19273-19273] ../src/vabackend.c: 658           nvCreateConfig got config attrib: 0 0 1
     12752.160957599 [19273-19273] ../src/vabackend.c:1749 nvQuerySurfaceAttributes with 4 (8) (nil) -1431655766
     12752.160959269 [19273-19273] ../src/vabackend.c:1749 nvQuerySurfaceAttributes with 4 (8) 0xd9800da6040 7
     12752.161912982 [19273-19273] ../src/vabackend.c:1807 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
     12752.161917682 [19273-19273] ../src/vabackend.c: 637           nvCreateConfig got profile: 13 with 0 attributes
     12752.161921252 [19273-19273] ../src/vabackend.c: 637           nvCreateConfig got profile: 17 with 1 attributes
     12752.161922922 [19273-19273] ../src/vabackend.c: 658           nvCreateConfig got config attrib: 0 0 1
     12752.161924412 [19273-19273] ../src/vabackend.c:1749 nvQuerySurfaceAttributes with 8 (8) (nil) -1431655766
     12752.161926312 [19273-19273] ../src/vabackend.c:1749 nvQuerySurfaceAttributes with 8 (8) 0xd9800da6040 7
     12752.162876905 [19273-19273] ../src/vabackend.c:1807 nvQuerySurfaceAttributes Returning constraints: width: 144 - 8192, height: 144 - 8192
     12752.162881585 [19273-19273] ../src/vabackend.c: 637           nvCreateConfig got profile: 17 with 0 attributes
     12752.162885285 [19273-19273] ../src/vabackend.c: 637           nvCreateConfig got profile: 14 with 1 attributes
     12752.162887065 [19273-19273] ../src/vabackend.c: 658           nvCreateConfig got config attrib: 0 0 1
     12752.162888565 [19273-19273] ../src/vabackend.c:1749 nvQuerySurfaceAttributes with 9 (8) (nil) -1431655766
     12752.162890125 [19273-19273] ../src/vabackend.c:1749 nvQuerySurfaceAttributes with 9 (8) 0xd9800da6040 7
     12752.163841158 [19273-19273] ../src/vabackend.c:1807 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
     12752.163845938 [19273-19273] ../src/vabackend.c: 637           nvCreateConfig got profile: 14 with 0 attributes
     12752.163850088 [19273-19273] ../src/vabackend.c: 637           nvCreateConfig got profile: 19 with 1 attributes
     12752.163851918 [19273-19273] ../src/vabackend.c: 658           nvCreateConfig got config attrib: 0 0 1
     12752.163853398 [19273-19273] ../src/vabackend.c:1749 nvQuerySurfaceAttributes with 10 (8) (nil) -1431655766
     12752.163855148 [19273-19273] ../src/vabackend.c:1749 nvQuerySurfaceAttributes with 10 (8) 0xd9800da6040 7
     12752.164923828 [19273-19273] ../src/vabackend.c:1807 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
     12752.164929398 [19273-19273] ../src/vabackend.c: 637           nvCreateConfig got profile: 19 with 0 attributes
     12752.164933578 [19273-19273] ../src/vabackend.c: 637           nvCreateConfig got profile: 32 with 1 attributes
     12752.164935298 [19273-19273] ../src/vabackend.c: 658           nvCreateConfig got config attrib: 0 0 1
     12752.164936868 [19273-19273] ../src/vabackend.c:1749 nvQuerySurfaceAttributes with 11 (8) (nil) -1431655766
[... cut ...]
src/gbm_drv_common.c:56: GBM-DRV error (get_bytes_per_component): Unknown or not supported format: 538982482

src/gbm_drv_common.c:56: GBM-DRV error (get_bytes_per_component): Unknown or not supported format: 943215175

src/gbm_drv_common.c:56: GBM-DRV error (get_bytes_per_component): Unknown or not supported format: 842094158

src/gbm_drv_common.c:56: GBM-DRV error (get_bytes_per_component): Unknown or not supported format: 842094169

src/gbm_drv_common.c:56: GBM-DRV error (get_bytes_per_component): Unknown or not supported format: 808530000

```

Korothi, 2022-12-28:
above was wayland. this is x11:
```
[filip@PC-ARCH ~]$ chromium
[4150:4323:1228/133407.891383:ERROR:vaapi_wrapper.cc(2291)] vaCreateContext failed, VA error: resource allocation failed
[4150:4323:1228/133407.891454:ERROR:vaapi_video_decoder.cc(1206)] failed creating VAContext
[4150:4150:1228/133407.913800:ERROR:shared_image_factory.cc(575)] Could not find SharedImageBackingFactory with params: usage: Gles2|Raster|DisplayRead|Scanout, format: RED_8, share_between_threads: 0, gmb_type: shared_memory
[4150:4150:1228/133407.913890:ERROR:shared_image_factory.cc(575)] Could not find SharedImageBackingFactory with params: usage: Gles2|Raster|DisplayRead|Scanout, format: RG_88, share_between_threads: 0, gmb_type: shared_memory
[4150:4150:1228/133407.931197:ERROR:shared_image_factory.cc(575)] Could not find SharedImageBackingFactory with params: usage: Gles2|Raster|DisplayRead|Scanout, format: RED_8, share_between_threads: 0, gmb_type: shared_memory
[4150:4150:1228/133407.931300:ERROR:shared_image_factory.cc(575)] Could not find SharedImageBackingFactory with params: usage: Gles2|Raster|DisplayRead|Scanout, format: RG_88, share_between_threads: 0, gmb_type: shared_memory
[4150:4150:1228/133407.937296:ERROR:shared_image_factory.cc(575)] Could not find SharedImageBackingFactory with params: usage: Gles2|Raster|DisplayRead|Scanout, format: RED_8, share_between_threads: 0, gmb_type: shared_memory
[4150:4150:1228/133407.937393:ERROR:shared_image_factory.cc(575)] Could not find SharedImageBackingFactory with params: usage: Gles2|Raster|DisplayRead|Scanout, format: RG_88, share_between_threads: 0, gmb_type: shared_memory
[4150:4150:1228/133407.943439:ERROR:shared_image_factory.cc(575)] Could not find SharedImageBackingFactory with params: usage: Gles2|Raster|DisplayRead|Scanout, format: RED_8, share_between_threads: 0, gmb_type: shared_memory
[4150:4150:1228/133407.943607:ERROR:shared_image_factory.cc(575)] Could not find SharedImageBackingFactory with params: usage: Gles2|Raster|DisplayRead|Scanout, format: RG_88, share_between_threads: 0, gmb_type: shared_memory
[4150:4150:1228/133407.950084:ERROR:shared_image_factory.cc(575)] Could not find SharedImageBackingFactory with params: usage: Gles2|Raster|DisplayRead|Scanout, format: RED_8, share_between_threads: 0, gmb_type: shared_memory
[4150:4150:1228/133407.955803:ERROR:shared_image_factory.cc(575)] Could not find SharedImageBackingFactory with params: usage: Gles2|Raster|DisplayRead|Scanout, format: RG_88, share_between_threads: 0, gmb_type: shared_memory
[4150:4150:1228/133407.956207:ERROR:shared_image_factory.cc(575)] Could not find SharedImageBackingFactory with params: usage: Gles2|Raster|DisplayRead|Scanout, format: RED_8, share_between_threads: 0, gmb_type: shared_memory
[4150:4150:1228/133407.956268:ERROR:shared_image_factory.cc(575)] Could not find SharedImageBackingFactory with params: usage: Gles2|Raster|DisplayRead|Scanout, format: RG_88, share_between_threads: 0, gmb_type: shared_memory
[4150:4150:1228/133408.058509:ERROR:shared_image_factory.cc(575)] Could not find SharedImageBackingFactory with params: usage: Gles2|Raster|DisplayRead|Scanout, format: RED_8, share_between_threads: 0, gmb_type: shared_memory
[4150:4150:1228/133408.058591:ERROR:shared_image_factory.cc(575)] Could not find SharedImageBackingFactory with params: usage: Gles2|Raster|DisplayRead|Scanout, format: RG_88, share_between_threads: 0, gmb_type: shared_memory
[4150:4150:1228/133408.211753:ERROR:shared_image_factory.cc(575)] Could not find SharedImageBackingFactory with params: usage: Gles2|Raster|DisplayRead|Scanout, format: RED_8, share_between_threads: 0, gmb_type: shared_memory
[4150:4150:1228/133408.211824:ERROR:shared_image_factory.cc(575)] Could not find SharedImageBackingFactory with params: usage: Gles2|Raster|DisplayRead|Scanout, format: RG_88, share_between_threads: 0, gmb_type: shared_memory

```

elFarto, 2022-12-28:
It does look like there have been some changes to Chromium recently. The X11 backend is now trying to use DMA-BUFs which is good. However, there's another limitation, it requires a single fd be exported for both planes, which we can't currently do (I might be able to make the direct backend do it, but it runs into issues with smaller videos due to different modifiers). I did try some experiments with it, but didn't get too far.

The Chromium issue is [here](https://bugs.chromium.org/p/chromium/issues/detail?id=974438), however it's been open a while now, so I don't know if it'll get fixed.

xuanruiqi, 2023-02-15:
Sorry, but just asking: is the Chromium issue still blocking further work on Chromium support?

elFarto, 2023-02-15:
It's mostly blocking it. I think I can make some changes to get it to get this driver to only use a single fd, but in my quick test I did there were more issues after making it return both planes in a single fd.

xuanruiqi, 2023-02-16:
Thanks. I'm actually maintaining libva-vdpau-driver-vp9, but I don't really understand how things work and recently things are not working...

elFarto, 2023-02-18:
> Thanks. I'm actually maintaining libva-vdpau-driver-vp9, but I don't really understand how things work and recently things are not working...

Chromium has made changes lately to use DMA-BUF, which might make the VDPAU based driver impossible to support. I'm not sure if the previous method (vaPutSurface I think) is still supported.

xuanruiqi, 2023-02-18:
Thanks for clarifying. Since I've always considered the VDPAU driver to only be a transitional solution, and it seems that we have to move on now. Let me know if I can help with Chrome support here in any way. I can code C but unfortunately I have no idea how the encoding/decoding stuff works...

wx5391805, 2023-03-02:
I'm testing the driver on Chromium and I get the error:
`ERROR:vaapi_wrapper.cc(1442)]  Unsupported va_profile: VAProfileNone`
Viewed the source code, I'm pretty sure Chromium uses VA-API via VAProfileNone, but it's not included in my vainfo, any suggestions? Thanks a lot.

```
libva info: VA-API version 1.18.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/aarch64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
     84548.552233874 [26606-26606] ../src/vabackend.c:2165       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
     84548.552241074 [26606-26606] ../src/vabackend.c:2173       __vaDriverInit_1_0 ````````````[wx] isDrm = 1,3
     84548.552246034 [26606-26606] ../src/vabackend.c:2176       __vaDriverInit_1_0 Now have 0 (0 max) instances
     84548.552250154 [26606-26606] ../src/vabackend.c:2199       __vaDriverInit_1_0 Selecting EGL backend
     84548.557059879 [26606-26606] ../src/export-buf.c: 132       findGPUIndexFromFd Defaulting to CUDA GPU ID 0. Use NVD_GPU to select a specific CUDA GPU
     84548.557077199 [26606-26606] ../src/export-buf.c: 149       findGPUIndexFromFd Looking for GPU index: 0
     84548.559117881 [26606-26606] ../src/export-buf.c: 161       findGPUIndexFromFd Found 3 EGL devices
     84548.559241001 [26606-26606] ../src/export-buf.c: 170       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
     84548.559289801 [26606-26606] ../src/export-buf.c: 208       findGPUIndexFromFd Selecting EGLDevice 0
     84548.560856123 [26606-26606] ../src/export-buf.c: 279         egl_initExporter Driver doesn't support 16-bit surfaces
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.18 (libva 2.18.0.pre1)
vainfo: Driver version: VA-API NVDEC driver [egl backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileAV1Profile0            : VAEntrypointVLD
     84548.688397582 [26606-26606] ../src/vabackend.c:2075              nvTerminate Terminating 0xaaaaf39809a0
     84548.688405782 [26606-26606] ../src/export-buf.c:  61      egl_releaseExporter Releasing exporter, 0 outstanding frames
     84548.688412102 [26606-26606] ../src/export-buf.c:  78      egl_releaseExporter Done releasing frames
     84548.688416582 [26606-26606] ../src/vabackend.c:2089              nvTerminate Now have 0 (0 max) instances
```

wx5391805, 2023-03-06:
> Chromium v110 nuked `--use-gl=desktop` completely lmao. All of your work here is useless now xD.

On the contrary lmao, as x11 in libva is deprecated and chromium begins to rely on drm,
The `vdpau-va-driver` is broken. This drive is the future solution, which is of great significance.

xuanruiqi, 2023-03-08:
New Chrome seems to be using VAProfileNone...

wx5391805, 2023-03-09:
I don't think I've ever been this close to success LOL. I can run this drive(direct backend) in Chromium 102 vaapi  X11 backend(X11 avoids VAProfileNone). Also, I fake a  Success in vaPutSurface but do nothing. Interesting, video draws something like my linux desktop I guess it's from my X11 or gpu buffer ... @elFarto make any progress in vaPutSurface ?

elFarto, 2023-03-11:
vaPutSurface only works in the chrome-support branch. That code will never be merged as it's a complete hack to make it work (requiring me to manually copy each frame pixel-by-pixel to get it in the correct layout). In addition, vaPutSurface isn't used with DMA-BUF, so it's of no use to us going forward.

VAProfileNone was normally only used for VPP, which I didn't think it used with DMA-BUF. I'll have to look into what's actually been changed recently.

andrewathalye, 2023-03-13:
It even _almost_ works in the Flatpak version of Chrome, but as before seems to be stuck with some sort of profile / format issue.

retrixe, 2023-04-30:
Just noting, Chrome has replaced VDAVideoDecoder (which is used when you `--disable-features=UseChromeOSDirectVideoDecoder`) with VaapiVideoDecoder which supports new codecs e.g. AV1 and is supposed to be more efficient. It works out of the box with no flags on Intel systems currently (like my tiger lake laptop).

List of related flags: https://chromium.googlesource.com/chromium/src/media/+/master/base/media_switches.cc#720

VDAVideoDecoder hasn't been removed yet, but it's deprecated, not receiving new features and probably will be removed at some point soon.

retrixe, 2023-05-27:
> @retrixe You're right. But is it only for Intel integrated GPUs, or Radeon/RDNA and nvidia too? Because there are VPA drivers for all of those right...

This only works on Intel GPUs, I think it's incompatible with the Mesa VA-API drivers, and I haven't tested whether or not nvidia-vaapi-driver works since it's not working with my hybrid GPU last I checked. Using the flag `VaapiIgnoreDriverChecks` enables VaapiVideoDecoder but it doesn't function correctly unless using the Intel media driver.

retrixe, 2023-05-31:
On my NVIDIA Optimus system, running:
`❯ NVD_BACKEND=direct NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia switcherooctl launch google-chrome-unstable --enable-features=VaapiVideoDecodeLinuxGL,VaapiIgnoreDriverChecks`
results in the following output when playing a video.

<details>
<summary>Output</summary>
      2149.102151039 [22788-22788] ../src/vabackend.c:2165       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
      2149.102192282 [22788-22788] ../src/vabackend.c:2174       __vaDriverInit_1_0 Now have 0 (0 max) instances
      2149.102202691 [22788-22788] ../src/vabackend.c:2200       __vaDriverInit_1_0 Selecting Direct backend
      2149.110785202 [22788-22788] ../src/backend-common.c:  31            isNvidiaDrmFd Invalid driver for DRM device: i915
      2149.110835836 [22788-22788] ../src/direct/direct-export-buf.c:  85      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD129
      2149.110844105 [22788-22788] ../src/direct/nv-driver.c: 217            init_nvdriver Initing nvdriver...
      2149.110853115 [22788-22788] ../src/direct/nv-driver.c: 222            init_nvdriver Got dev info: 100 1 0 0
      2149.111012812 [22788-22788] ../src/direct/nv-driver.c: 283            init_nvdriver NVIDIA kernel driver version: 530.41.03
      2149.228292296 [22788-22788] ../src/vabackend.c: 579    nvGetConfigAttributes Got here with profile: 6 == 4
      2149.228317082 [22788-22788] ../src/vabackend.c: 647           nvCreateConfig got profile: 6 with 1 attributes
      2149.228321961 [22788-22788] ../src/vabackend.c: 668           nvCreateConfig got config attrib: 0 0 1
      2149.228326007 [22788-22788] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 4 (8) (nil) -1431655766
      2149.228328181 [22788-22788] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 4 (8) 0x7b0022a01c0 8
      2149.230361517 [22788-22788] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      2149.230378631 [22788-22788] ../src/vabackend.c: 647           nvCreateConfig got profile: 6 with 0 attributes
      2149.230401025 [22788-22788] ../src/vabackend.c: 579    nvGetConfigAttributes Got here with profile: 7 == 4
      2149.230403142 [22788-22788] ../src/vabackend.c: 647           nvCreateConfig got profile: 7 with 1 attributes
      2149.230404756 [22788-22788] ../src/vabackend.c: 668           nvCreateConfig got config attrib: 0 0 1
      2149.230406877 [22788-22788] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 4 (8) (nil) -1431655766
      2149.230408379 [22788-22788] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 4 (8) 0x7b0022a01c0 8
      2149.232746285 [22788-22788] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      2149.232788822 [22788-22788] ../src/vabackend.c: 647           nvCreateConfig got profile: 7 with 0 attributes
      2149.232812168 [22788-22788] ../src/vabackend.c: 579    nvGetConfigAttributes Got here with profile: 13 == 4
      2149.232818800 [22788-22788] ../src/vabackend.c: 647           nvCreateConfig got profile: 13 with 1 attributes
      2149.232825216 [22788-22788] ../src/vabackend.c: 668           nvCreateConfig got config attrib: 0 0 1
      2149.232832387 [22788-22788] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 4 (8) (nil) -1431655766
      2149.232838901 [22788-22788] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 4 (8) 0x7b0022a01c0 8
      2149.240768025 [22788-22788] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      2149.240784451 [22788-22788] ../src/vabackend.c: 647           nvCreateConfig got profile: 13 with 0 attributes
      2149.240802172 [22788-22788] ../src/vabackend.c: 579    nvGetConfigAttributes Got here with profile: 17 == 8
      2149.240804573 [22788-22788] ../src/vabackend.c: 647           nvCreateConfig got profile: 17 with 1 attributes
      2149.240806588 [22788-22788] ../src/vabackend.c: 668           nvCreateConfig got config attrib: 0 0 1
      2149.240847351 [22788-22788] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 8 (8) (nil) -1431655766
      2149.240849614 [22788-22788] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 8 (8) 0x7b0022a01c0 8
      2149.245043712 [22788-22788] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 144 - 8192, height: 144 - 8192
      2149.245084966 [22788-22788] ../src/vabackend.c: 647           nvCreateConfig got profile: 17 with 0 attributes
      2149.245111338 [22788-22788] ../src/vabackend.c: 579    nvGetConfigAttributes Got here with profile: 14 == 9
      2149.245118635 [22788-22788] ../src/vabackend.c: 647           nvCreateConfig got profile: 14 with 1 attributes
      2149.245125446 [22788-22788] ../src/vabackend.c: 668           nvCreateConfig got config attrib: 0 0 1
      2149.245132493 [22788-22788] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 9 (8) (nil) -1431655766
      2149.245138706 [22788-22788] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 9 (8) 0x7b0022a01c0 8
      2149.247742800 [22788-22788] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      2149.247863082 [22788-22788] ../src/vabackend.c: 647           nvCreateConfig got profile: 14 with 0 attributes
[... cut ...]
[22788:23206:0531/193209.133471:ERROR:vaapi_wrapper.cc(2429)] vaCreateContext failed, VA error: resource allocation failed
[22788:23206:0531/193209.133561:ERROR:vaapi_video_decoder.cc(1242)] failed creating VAContext
      2165.358849528 [22788-23206] ../src/vabackend.c:2075              nvTerminate Terminating 0x7b007cece00
      2165.358988571 [22788-23206] ../src/vabackend.c:2089              nvTerminate Now have 0 (0 max) instances

</details>

Maybe someone with an NVIDIA desktop can test this out?

Tatsh, 2023-05-31:
> On my NVIDIA Optimus system, running `NVD_BACKEND=direct NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia switcherooctl launch google-chrome-unstable --enable-features=VaapiVideoDecodeLinuxGL,VaapiIgnoreDriverChecks` results in the following output when playing a video.
> 
> Output
> Maybe someone with an NVIDIA desktop GPU can test this out? I'm not sure if this issue is caused by `Invalid driver for DRM device: i915` which might be related to #11 (but with Chrome instead of Firefox), `chrome://gpu` does report the correct codecs being supported: [image] However, it fails to use the GPU for decoding as aforementioned in the logs.

On my desktop with a RTX 3080 Ti, nothing is displayed under *Video Acceleration Information*.

[image]

RyuKay24, 2023-06-04:
I dont know if this could change anything but nvidia has improved dma-buf compatibility on the new beta drivers, changelog: [535.43.02BETA](https://www.nvidia.co.uk/download/driverResults.aspx/205050/en-uk)

gilvbp, 2023-06-28:
@RyuKay24 using 535 (final-drivers) gives the error:

':gbm_wrapper.cc(258)] Failed to export buffer to dma_buf: Arquivo ou diretório inexistente'

retrixe, 2023-07-14:
@asklow it doesn't work, only on Intel GPUs so far (doesn't work with Mesa AMD VA-API either)

Chrome is able to detect it and list the supported codecs, but it fails to use it and falls back to software decode (see logs above)

Wlnfr, 2023-07-31:
Desktop GT1030, Brave Version 1.56.14 Chromium: 115.0.5790.114 
Flags: `--enable-features=VaapiVideoDecodeLinuxGL,VaapiIgnoreDriverChecks`
`about:gpu` shows supported codecs
[image]
But when I try playing any video, I get same error as retrixe did [here](https://github.com/elFarto/nvidia-vaapi-driver/issues/5#issuecomment-1570303157)
<details>
<summary>Output</summary>
<br>

      2159.527255382 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c:2171       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
      2159.527280780 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c:2180       __vaDriverInit_1_0 Now have 0 (0 max) instances
      2159.527302593 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c:2203       __vaDriverInit_1_0 Selecting EGL backend
      2159.537066012 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/export-buf.c: 132       findGPUIndexFromFd Defaulting to CUDA GPU ID 0. Use NVD_GPU to select a specific CUDA GPU
      2159.537088926 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/export-buf.c: 149       findGPUIndexFromFd Looking for GPU index: 0
      2159.539754382 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/export-buf.c: 161       findGPUIndexFromFd Found 3 EGL devices
      2159.539882922 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/export-buf.c: 170       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
      2159.539890387 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/export-buf.c: 191       findGPUIndexFromFd Selecting EGLDevice 0
      2159.541704662 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/export-buf.c: 260         egl_initExporter Driver supports 16-bit surfaces
      2159.738520681 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c: 580    nvGetConfigAttributes Got here with profile: 6 == 4
      2159.738541927 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c: 648           nvCreateConfig got profile: 6 with 1 attributes
      2159.738550862 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c: 669           nvCreateConfig got config attrib: 0 0 1
      2159.738558537 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c:1765 nvQuerySurfaceAttributes with 4 (8) (nil) -1431655766
      2159.738566527 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c:1765 nvQuerySurfaceAttributes with 4 (8) 0xed4016889c0 8
      2159.741035175 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c:1838 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      2159.741055317 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c: 648           nvCreateConfig got profile: 6 with 0 attributes
      2159.741092711 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c: 580    nvGetConfigAttributes Got here with profile: 7 == 4
      2159.741100230 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c: 648           nvCreateConfig got profile: 7 with 1 attributes
      2159.741106474 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c: 669           nvCreateConfig got config attrib: 0 0 1
      2159.741113690 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c:1765 nvQuerySurfaceAttributes with 4 (8) (nil) -1431655766
      2159.741120507 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c:1765 nvQuerySurfaceAttributes with 4 (8) 0xed4016889c0 8
      2159.743360693 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c:1838 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      2159.743372865 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c: 648           nvCreateConfig got profile: 7 with 0 attributes
      2159.743387471 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c: 580    nvGetConfigAttributes Got here with profile: 13 == 4
      2159.743394669 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c: 648           nvCreateConfig got profile: 13 with 1 attributes
      2159.743401582 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c: 669           nvCreateConfig got config attrib: 0 0 1
      2159.743408699 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c:1765 nvQuerySurfaceAttributes with 4 (8) (nil) -1431655766
      2159.743415132 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c:1765 nvQuerySurfaceAttributes with 4 (8) 0xed4016889c0 8
      2159.745633611 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c:1838 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      2159.745646122 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c: 648           nvCreateConfig got profile: 13 with 0 attributes
      2159.745656918 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c: 580    nvGetConfigAttributes Got here with profile: 17 == 8
      2159.745662585 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c: 648           nvCreateConfig got profile: 17 with 1 attributes
      2159.745667233 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c: 669           nvCreateConfig got config attrib: 0 0 1
      2159.745672451 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c:1765 nvQuerySurfaceAttributes with 8 (8) (nil) -1431655766
      2159.745677665 [10647-10647] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c:1765 nvQuerySurfaceAttributes with 8 (8) 0xed4016889c0 8
[... cut ...]
      2253.312140926 [10647-11034] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c:2081              nvTerminate Terminating 0xed40788d600
      2253.312167137 [10647-11034] ../nvidia-vaapi-driver-0.0.10/src/export-buf.c:  61      egl_releaseExporter Releasing exporter, 0 outstanding frames
      2253.312175328 [10647-11034] ../nvidia-vaapi-driver-0.0.10/src/export-buf.c:  78      egl_releaseExporter Done releasing frames
      2253.312183298 [10647-11034] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c:2095              nvTerminate Now have 0 (0 max) instances
</details>

gilvbp, 2023-08-30:
> Weird. Intel team seems to contribute to chromium but not Nvidia & AMD.
Hi there, I Uploaded the last chromium-wayland-vaapi nvidia working version.
https://www.dropbox.com/scl/fi/6x65jhgdpo8yhj4bcgl9v/chromium-wayland-vaapi-111.0.5563.146-1-x86_64.pkg.tar.zst?rlkey=xipgffq1kzldft7wcwlhpk631&dl=0

gilvbp, 2023-09-01:
> /usr/lib/dri/nvidia_drv_video.so

@asklow  check this issue: https://github.com/intel/libva/issues/575

maybe it's env variable problem or a driver problem:

BTW, my vainfo is fine: 

` vainfo
Trying display: wayland
vainfo: VA-API version: 1.19 (libva 2.19.0)
vainfo: Driver version: VA-API NVDEC driver [egl backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
`

gilvbp, 2023-09-01:
> @asklow

Yes, I answered some questions, like I already said and provided the old package, so to get work hw video acceleration, you have to use the chromium-wayland-vaapi-111.0.5563.146  version ( above that version, ex: 112 - 116, which doesn't work because of gbm_wrapper.cc(258)]` Failed to export buffer to dma_buf: bug).

You just have to install the chromium-wayland-vaapi-111.0.5563.146-1-x86_64.pkg.tar.zst package to get downgraded.

PS: You need to install l icu72-bin  (aur) and re2 1:20230301-1 (archive) packages because of chromium 111.0.5563-146v dependencies.

gilvbp, 2023-10-03:
> Welp. In that case, it's better for me to wait for the regression fix. I hope you & the chromium team find the core issue & revert that commit whenever possible...👍
> 
> For now, I'll continue using firefox for streaming stuff...

It's finally working with the last aur version and add this patch: https://gist.github.com/thubble/235806c4c64b159653de879173d24d9f?permalink_comment_id=4712393

99linesofcode, 2023-10-09:
While not necessarily related to nvidia-vaapi-driver, I've managed to enable hardware acceleration on hyprland-nvidia using Google Chrome 117.0.5938.149-1 and the following `chrome-flags.conf`:

```
--enable-features=CanvasOopRasterization,VaapiVideoDecoder,VaapiVideoEncoder
```

[image] 

[image]

99linesofcode, 2023-10-09:
> > I've managed to enable hardware acceleration
> 
> Have you checked that it actually uses GPU for decoding?

Unfortunately you were right. `chrome://gpu` reported that hw acceleration was enabled but it still uses VpxVideoDecoder and Dav1dVideoDecoder when playing 4k and 8k video on YouTube. I haven't found a fix just yet while some people report using an older version of chrome does work. Version 109 I believe.

retrixe, 2023-10-09:
Chrome seems to need patches for biplanar support
There is some ongoing discussion here https://gist.github.com/thubble/235806c4c64b159653de879173d24d9f

gilvbp, 2023-11-01:
> Chrome seems to need patches for biplanar support. There is some ongoing discussion here https://gist.github.com/thubble/235806c4c64b159653de879173d24d9f

It's a Nvidia driver bug:

My Brazilian buddy Nick fixed the issue: 

"
Good news! Luckily, while setting up a new laptop with a recent enough Nvidia GPU, I was able to repro this reliably and could confirm that:

- There is a bug in the Nvidia proprietary driver, as initially pointed out by reporters of [issue 1463851](https://bugs.chromium.org/p/chromium/issues/detail?id=1463851). It prevents from getting a bo's plane fd after having destroyed a corresponding imported bo for it.
- For some reason, bisect didn't help, but the chromium change that unveiled the bug was https://crrev.com/1124657, which added the BO creation/import loop to verify modifiers support. That'd explain the recent regression reports here.

The best workarounds I could come up with are:
- https://chromium-review.googlesource.com/c/chromium/src/+/4989782
- https://chromium-review.googlesource.com/c/chromium/src/+/4989962

I've tested both, but I would appreciate it if others experiencing this could try as well. Hopefully, reviewers will point out the best approach, and it will land soon.

Possible follow-ups:
- jfroy@ pchang@ Thanks for the minimal sample to repro the bug. Have any of you reported it or are following up with Nvidia? If not, I believe we should, and I'll be happy to help with that.
I couldn't (yet) determine if that's regressed in any specific Nvidia driver version and, if so, which version. Actually, it may be been there since the first GBM-enable version. I need to check this.
-
"

jfroy, 2023-11-01:
@gilvbp It's not clear at all that solving the GBM buffer import issue in the ozone wayland platform is going to make this work. We need to check if hardware codecs are actually used. At the very least, the "Video Acceleration Information" section of chrome://gpu/ needs to show some codecs.

gilvbp, 2023-11-01:
> @gilvbp It's not clear at all that solving the GBM buffer import issue in the ozone wayland platform is going to make this work. We need to check if hardware codecs are actually used. At the very least, the "Video Acceleration Information" section of chrome://gpu/ needs to show some codecs.

Hi there, it seems ok to me. Here´s my chrome://GPU/ report 
[about-gpu-2023-11-01T15-42-23-888Z.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/13228896/about-gpu-2023-11-01T15-42-23-888Z.txt)

Also, I did the test that I used to do (since old working chromium versions 114), using videos from chromium-wayland-vaapi (aur) link website reference: https://test-videos.co.uk/bigbuckbunny/mp4-h264.

jfroy, 2023-11-01:
> Hi there, it seems ok to me. Here´s my chrome://GPU/ report
> [about-gpu-2023-11-01T15-42-23-888Z.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/13228896/about-gpu-2023-11-01T15-42-23-888Z.txt)

There are no codecs listed.

```
Video Acceleration Information
==============================
Decoding: 
Encoding:
```

So definitely not working.

gilvbp, 2023-11-01:
> There are no codecs listed.
>
> ```
> Video Acceleration Information
> ==============================
> Decoding:
> Encoding:
> ```
>
> So definitely not working.

Maybe that's because im using pure chromium arch package. I couldn't get chromium-wayland-vaapi working. Getting elf header tag not found error

[182 more replies]