elFarto/nvidia-vaapi-driver #6: Firefox: Sandbox: seccomp sandbox violation

hellobbn, 2022-01-04:
- Firefox version: 97.0a1 (2022-01-04) (64-bit)
   Some troubleshooting info (See [attachment](https://github.com/elFarto/nvidia-vaapi-driver/files/7809451/firefox-troubleshooting.txt) for full info)
```
gfx.x11-egl.force-enabled: true
security.sandbox.content.level: 0
security.sandbox.content.syscall_whitelist: 41,49,50,332
```
LOG:
```
Sandbox: seccomp sandbox violation: pid 32109, tid 32118, syscall 146, args 2 140628402060476 0 140628376061040 140628897165472 1.  Killing process.
libva info: VA-API version 1.13.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/dri/nvidia_drv_video.so
```
Adding system call 146 to whitelist does not seem to work
- `vainfo`:
```
[30042-30042] ../src/vabackend.c:1342       __vaDriverInit_1_0 Initing NVIDIA VA-API Driver
[30042-30042] ../src/export-buf.c:  56                reconnect Reconnecting to stream
vainfo: VA-API version: 1.13 (libva 2.13.0)
vainfo: Driver version: VA-API NVDEC driver
[30042-30042] ../src/vabackend.c: 200              vaToCuCodec vaToCuCodec: Unknown codec: 2
[30042-30042] ../src/vabackend.c: 200              vaToCuCodec vaToCuCodec: Unknown codec: 3
[30042-30042] ../src/vabackend.c: 200              vaToCuCodec vaToCuCodec: Unknown codec: 4
[30042-30042] ../src/vabackend.c: 200              vaToCuCodec vaToCuCodec: Unknown codec: 12
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileAV1Profile0            :	VAEntrypointVLD
[30042-30042] ../src/vabackend.c:1332              nvTerminate In nvTerminate
[30042-30042] ../src/export-buf.c:  36                    debug [EGL] eglDebugMessageControlKHR: EGL_BAD_STATE_KHR error: In EGL Access Table::stream2.consumer.disconnect: Consumer handle does not match reservation (0xf550ab1 vs 0xf550ab9).
```
There seems to be some errors?

- System Info:
```
Kernel: 5.15.11
DE: GNOME 41.0
WM: Mutter
XDG_SESSION_TYPE: X11
CPU: AMD Ryzen 9 5900HX with Radeon Graphics (16) @ 2.600GHz
GPU: NVIDIA GeForce RTX 3070 Mobile / Max-Q
GPU: AMD ATI 04:00.0 Cezanne
```
The system is running in "nvidia only" mode, as instructed [here](https://wiki.archlinux.org/title/NVIDIA_Optimus#Use_NVIDIA_graphics_only)

NXij, 2022-01-04:
`libva info: VA-API version 1.13.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/dri/nvidia_drv_video.so
Sandbox: seccomp sandbox violation: pid 3376, tid 3514, syscall 146, args 2 140595888217340 0 140595852133488 140596377415840 1.  Killing process.
[Child 3240, MediaDecoderStateMachine #1] WARNING: Decoder=7f6bef0b5100 Decode error: NS_ERROR_DOM_MEDIA_FATAL_ERR (0x806e0005) - auto mozilla::MediaChangeMonitor::CreateDecoderAndInit(mozilla::MediaRawData *)::(anonymous class)::operator()()::(anonymous class)::operator()(const mozilla::MediaResult &) const: Unable to initialize decoder: file /builds/worker/checkouts/gecko/dom/media/MediaDecoderStateMachine.cpp:3568
`

helloer, 2022-01-04:
You can set env var `MOZ_DISABLE_RDD_SANDBOX=1`, but that's a **unsafe** workaround

elFarto, 2022-01-04:
@helloer Thanks for that.

I've just tracked down what's happened. There was a [patch](https://bugzilla.mozilla.org/show_bug.cgi?id=1698778) to allow VA-API in the RDD process, but the change happens to force VA-API to *only* work in the RDD process. AFAICS this then ignores the whitelist we've configured.

rmader, 2022-01-05:
> You can set env var `MOZ_DISABLE_RDD_SANDBOX=1`, but that's an **unsafe** workaround

From a FF dev on Matrix:
> even running in the rdd with the sandbox completely disabled wouldn't be the worst option in the world
> definitely better than disabling the sandbox in the content processes

So if the combination of `MOZ_DISABLE_RDD_SANDBOX=1` + `media.rdd-*.enabled:true` works, that would be preferred (and should allow to drop the `security.sandbox.content.*` entries IIUC). Well, it's currently the only option for FF >= 97.

Darkspirit, 2022-01-06:
Can you remove
* security.sandbox.content.level (Disabling the content process sandbox removes the barrier between JS and the rest of the OS.)
* security.sandbox.content.syscall_whitelist
* media.ffvpx.enabled
* media.rdd-vpx.enabled 

from README.md
and instead recommend media.rdd-ffmpeg.enabled=true for 96 (it's default in 97) and MOZ_DISABLE_RDD_SANDBOX=1 ?

https://bugzilla.mozilla.org/show_bug.cgi?id=1743926 has been fixed in 97, but if you disable the media process (RDD) sandbox, they don't affect 96.

elFarto, 2022-01-07:
@Darkspirit What's the reasoning behind media.rdd-ffmpeg.enabled=true? It seems to be working fine without that on 96.

rmader, 2022-01-07:
@elFarto: the combination of `media.rdd-ffmpeg.enabled=true` and `MOZ_DISABLE_RDD_SANDBOX=1` is just way safer than what's currently in the README - and will be required for 97 anyway. So better not recommend dangerous things to users if there are better alternatives (assuming it actually works, which I'm not sure).

Darkspirit, 2022-01-07:
This is not about whether it's working. It's about security.
security.sandbox.content.level=0 makes users vulnerable. Do not recommend it! Remove it immediately.

* media.rdd-ffvpx.enabled is already true in 96. VP8 and VP9 are software/hardware decoded in the media process.
* media.rdd-ffmpeg.enabled is false in 96. H264 is decoded in the content process where hostlie javascript may reside. One must not disable the content process sandbox. If a hostile script exploits a Firefox bug, the sandbox is the barrier that prevents system access. **Therefore, H264 decoding should be moved into the media process by enabling media.rdd-ffmpeg.enabled. This has been done in 97. 96 users must enable it manually.**
* Use MOZ_DISABLE_RDD_SANDBOX=1. The technical owner of Firefox Graphics recommended it in Matrix. https://bugzilla.mozilla.org/show_bug.cgi?id=1748460#c2
* https://bugzilla.mozilla.org/show_bug.cgi?id=1743647 disallowed VAAPI in the content process so that projects don't even think about recommending disabling the content process sandbox anymore.

rmader, 2022-01-07:
Thanks @elFarto! With that, we can close this issue I guess.

Darkspirit, 2022-01-07:
Thank you for this library!

elFarto, 2022-01-07:
Thanks for that. I understand the reasoning behind it, I was just uncertain what the rdd-ffmpeg option actually did.

closed 2022-01-07 by elFarto

rmader, 2022-01-07:
Nitpick about README: it's RDD process, not RDD thread ;)

Darkspirit, 2022-01-08:
And yes, media.navigator.mediadatadecoder_vpx_enabled=true (so far only enabled in Nightly) is required for WebRTC VAAPI with VP8/VP9.
https://bugzilla.mozilla.org/show_bug.cgi?id=1709009
https://bugzilla.mozilla.org/show_bug.cgi?id=1748862#c3

(media.navigator.mediadatadecoder_h264_enabled is already true for H264.)