elFarto/nvidia-vaapi-driver #7: Video is green in firefox, then crashes

dawidgarus, 2022-01-04:
I've installed the drivers and configured firefox according to the instructions, but I see green screen instead of the video and after a while firefox crashes.
Firefox logs the same message repeatedly:
```
[4720-4836] ../src/export-buf.c: 284            exportCudaPtr Unable to export image
IPDL protocol Error: Received an invalid file descriptor
IPDL protocol Error: Received an invalid file descriptor
[4720-4832] ../src/vabackend.c: 801             nvEndPicture cuvid decoded successful to idx: 22
[4720-4832] ../src/vabackend.c:1438    nvExportSurfaceHandle got 0x7f52cff86240
[4720-4832] ../src/vabackend.c:1460    nvExportSurfaceHandle got address 205828000 (1536) for surface 24 (picIdx: 22)
[4720-4832] ../src/export-buf.c: 279            exportCudaPtr eglExportDMABUFImageQueryMESA: 0x7f5297926561 NV12 (3231564e) planes:2 mods:300000000606014 300000000606014
[4720-4832] ../src/export-buf.c:  36                    debug [EGL] eglExportDMABUFImageMESA: EGL_BAD_MATCH error: In eglExportDMABUFImageMESA: EGLImage can't be exported to dma-buf
```
Also vainfo:
```
[4106-4106] ../src/vabackend.c:1522       __vaDriverInit_1_0 Initing NVIDIA VA-API Driver
[4106-4106] ../src/export-buf.c:  56                reconnect Reconnecting to stream
vainfo: VA-API version: 1.13 (libva 2.13.0)
vainfo: Driver version: VA-API NVDEC driver
[4106-4106] ../src/vabackend.c: 202              vaToCuCodec vaToCuCodec: Unknown codec: 2
[4106-4106] ../src/vabackend.c: 202              vaToCuCodec vaToCuCodec: Unknown codec: 3
[4106-4106] ../src/vabackend.c: 202              vaToCuCodec vaToCuCodec: Unknown codec: 4
[4106-4106] ../src/vabackend.c: 202              vaToCuCodec vaToCuCodec: Unknown codec: 12
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
[4106-4106] ../src/vabackend.c:1511              nvTerminate In nvTerminate
[4106-4106] ../src/export-buf.c:  36                    debug [EGL] eglDebugMessageControlKHR: EGL_BAD_STATE_KHR error: In EGL Access Table::stream2.consumer.disconnect: Consumer handle does not match reservation (0x8e2fdc11 vs 0x8e2fdc19)
```
arch, xorg
firefox-developer-edition: 96.0b10 (64-bit)
nvidia: 495.46-6
kernel: 5.15.12-arch1-1

Thanks for your effort.

helloer, 2022-01-04:
I had the same error, I fixed it by adding `nvidia-drm.modeset=1` to kernel parameters.

elFarto, 2022-01-04:
Yes that parameter is required. I'll update the README to make that clear.

dawidgarus, 2022-01-04:
Thanks. That worked for me.
Here's something you may want to remember if you set `nvidia-drm.modeset=1`. Setting it changed default session from wayland to xorg in my case. Keep that in mind.

closed 2022-01-04 by dawidgarus

tufei, 2022-08-28:
@elFarto just wanted to say thank you! Confirmed working on Fedora 35:

Firefox: 103.0.2
Nvidia Driver: 515.65.01
Kernel: 5.18.19-100.fc35.x86_64