elFarto/nvidia-vaapi-driver #8: [Q] use directly ffmpeg instead gstreamer

sl1pkn07, 2022-01-04:
why gstreamer and not directly ffmpeg?

greetings

elFarto, 2022-01-04:
I wish I didn't have to use either, but there's some information that isn't passed via VA-API that NVDEC needs, and ffmpeg doesn't expose the output of the codec parsing AFAICS. GStreamer happened to have a simple library that did exactly what I needed.

Ideally these fields would be added into VA-API, but it is what it is.

sl1pkn07, 2022-01-04:
and with mediainfo?

https://mediaarea.net/en/MediaInfo

elFarto, 2022-01-05:
I'm not sure that software can be used in the way I'm using the gstreamer library. I need something to parse the VP9 bitstream to recover some of the fields VA-API doesn't pass in. MediaInfo looks like a GUI application, not a library.

sl1pkn07, 2022-01-05:
in their github

https://github.com/MediaArea/MediaInfoLib

but i'm not sure what type of data you need

this library is used for retreive the info about media files, like bitrate, size, metadatas, sreams of container, encodeng profiles, color space, bitdepths, coded ID, etc etc

elFarto, 2022-01-09:
It seems that library doesn't actually support parsing VP9 bitstream. In any case I'm not really interested in replacing the gstreamer library, more on removing it's need completely.

closed 2022-01-09 by elFarto