elFarto/nvidia-vaapi-driver #12: `eglQueryStreamConsumerEventNV` can return `EGL_FALSE` without an error

openglfreak, 2022-01-05:
The driver was spinning on [this code](https://github.com/elFarto/nvidia-vaapi-driver/blob/master/src/export-buf.c#L347) on my machine.
Apparently buried somewhere in the spec, if you read carefully enough, there are the 5 words "or the stream becomes disconnected" that allow `eglQueryStreamConsumerEventNV` to return `EGL_FALSE` while setting the error to `EGL_SUCCESS` if "the stream becomes disconnected".

This patch fixes it for me:
```diff
diff --git a/src/export-buf.c b/src/export-buf.c
index bbc2f7c..dc35cfb 100644
--- a/src/export-buf.c
+++ b/src/export-buf.c
@@ -302,7 +302,7 @@ int exportCudaPtr(NVDriver *drv, CUdeviceptr ptr, NVSurface *surface, uint32_t p
         EGLAttrib aux = 0;
 //        LOG("eglQueryStreamConsumerEventNV: %p", drv->eglStream);
         EGLint eventRet = eglQueryStreamConsumerEventNV(drv->eglDisplay, drv->eglStream, 0, &event, &aux);
-        if (eventRet == EGL_TIMEOUT_EXPIRED_KHR) {
+        if (eventRet == EGL_TIMEOUT_EXPIRED_KHR || (eventRet == EGL_FALSE && eglGetError() == EGL_SUCCESS)) {
             break;
         }
 
```

elFarto, 2022-01-05:
Thanks for that, I've applied the fix. Are you seeing any visual artefacts when this scenario occurs?

openglfreak, 2022-01-05:
> Thanks for that, I've applied the fix. Are you seeing any visual artefacts when this scenario occurs?

The videos just didn't start

closed 2022-01-06 by openglfreak