elFarto/nvidia-vaapi-driver #13: Strange thing: extremely high frame loss rate in firefox, but not in mpv, with the same video

huyz-git, 2022-01-06:
For example, I just search youtube for 8k video, and pick the first one: https://www.youtube.com/watch?v=linlz7-Pnvw
If I just play this 8k video in firefox dev edition (96.0b10), I got around 80% frames dropped, and the video is visually very unsmooth.
But if I download this 8k video by youtube-dlp, and play it with `mpv --hwdec=vaapi`, then the video is visually smooth.

I uses mobile rtx 2070 gpu. When playing this video in firefox, the gpu consumes about 45-60 watts, very unstable, and its core frequency varies from 1.4GHz to 1.9GHz; When playing this video in mpv, the gpu consumes about 65-67 watts, very stable, and its core frequency is 1.9GHz.

elFarto, 2022-01-06:
I've just quickly tested it here, and I see the same issue. Interestingly the video freezes completely when the Youtube video controls disappear, and then continues playing as normal if i move the mouse to make them reappear.

elFarto, 2022-01-13:
I believe this issue lies within Firefox, but I've been unable to debug it fully. I can see that as soon as the control disappear, something causes compositing on vsync to be disabled.

NoXPhasma, 2022-01-18:
I have a lot of frame drops on Youtube in Firefox as well. Both Firefox 96 and Nightly. It does not matter what video (vp9 codec) and the Youtube controls do not matter for me. The frames do not drop faster when the controls are displayed or not.

elFarto, 2022-01-18:
@NoXPhasma is this with an 8k video?

NoXPhasma, 2022-01-18:
No, it happens on 1080p and 4k. Haven't even tried 8k. And I have 0 dropped frames on the same videos with the CPU being used.

NoXPhasma, 2022-01-18:
Apparently these drops only occur on videos which run at 60FPS.

Example videos on which frames drop (60FPS):
https://www.youtube.com/watch?v=LXb3EKWsInQ&
https://www.youtube.com/watch?v=VnLdOcDNTe0
https://www.youtube.com/watch?v=5CLMCpqJxzg

Example videos on which it does not (30FPS or less):
https://www.youtube.com/watch?v=PaXVZJFfgJA
https://www.youtube.com/watch?v=li-_BLtq58w
https://www.youtube.com/watch?v=0nTO4zSEpOs

Systeminfo, if it helps:
Arch Linux
Nvidia GTX 1080 (510.39.01)

aleasto, 2022-01-18:
Only happens at 8k on my system. GTX 1080 Ti.

NoXPhasma, 2022-01-19:
I took a deeper look into why I get those frame drops with 1080p60 or above and apparently it has nothing to do with package but WebRender. As long as Webrender runs in Hardware mode, it will drop frames. Only when I force WebRender in Software mode these drops stop, but of course then I'm unable to use video hardware acceleration.

Seems Firefox WebRender is cursed (for me).

elFarto, 2022-01-19:
@NoXPhasma I don't think it's the same issue as the original issue here, but there is some weirdness with Firefox and VSYNC. I can also see a gradual increase in dropped frames.

NoXPhasma, 2022-01-19:
@elFarto You are right with my issue being VSYNC, running Firefox with `__GL_SYNC_TO_VBLANK=0` stops these frame drops. Thanks for the heads up and sorry for making unrelated noise.

elFarto, 2022-01-23:
An update to the original issue. After a lot of debugging, I believe I've tracked the issue down to a fundamental limitation with how Firefox decodes videos, which effectively causes the process to become serialised, it decodes and then exports each frame, then decodes the next frame and exports is. This severely limits how much work the GPU can do.

Without changes to how Firefox works, I don't think decoding 8k video is going to be viable for a while.

This issue is log as [1750388](https://bugzilla.mozilla.org/show_bug.cgi?id=1750388) in Firefox's bugzilla.

qumaciel, 2022-01-27:
Just to mention that last commits fixed this issue for me!

elFarto, 2022-01-27:
I hadn't retested this with the recent changes, but it doesn't fix it for me. The export rewrite does reduce how much work is needed to export each frame, but the largest part is still there. So it's possible it'll fix it for some people, but it's not an ultimate fix.

huyz-git, 2022-01-27:
> I hadn't retested this with the recent changes, but it doesn't fix it for me. The export rewrite does reduce how much work is needed to export each frame, but the largest part is still there. So it's possible it'll fix it for some people, but it's not an ultimate fix.

I tested the latest commit, on some high bit-rate 4k video, this issue is reduced, but for 8k video, this issue still exist.
For the youtube link above, if I select 4k , version 5 days ago have some frames dropped, but the latest commit have nearly no frame dropped. But if I select 8k, both version have high frame drop rate.

krab, 2022-01-31:
For me that doesn't happen when i have "Force composition pipeline" enabled in Nvidia Settigs. But here on nvidia gtx 950 i can test only 4096x4096 video maximum.

I saw that stuttering on twitch on 1080@60 stream before i have enabled "Force composition pipeline".

From what i understand from my limited gaming experience on linux that option in nvidia locks FPS to display HZ like vsync?

krab, 2022-01-31:
> Only happens at 8k on my system. GTX 1080 Ti. Youtube is serving me VP9.

Can you test with "Force composition pipeline" enabled in Nvidia Settings? (XServer Display Configuration -> Advanced).

NoXPhasma, 2022-01-31:
Force composition pipeline shouldn't be used if you can avoid that. The whole purpose of that functionality is to move monitors in environments without a compositor, and it's doing that by adding a composition to the whole X screen. This leads to higher frame times and even input lag.

The only real benefit of that is, it will remove any tearing, but a a rather high cost.

krab, 2022-01-31:
> Force composition pipeline shouldn't be used if you can avoid that. The whole purpose of that functionality is to move monitors in environments without a compositor, and it's doing that by adding a [...]

That still better than builtin vsync in games for example like path of exile under steam/proton. KWin disables its own composition when fullscreen games runs.

Darkspirit, 2022-01-31:
"Force composition pipeline" must be enabled for Firefox!

Firefox needs to set Nvidia env var __GL_ALLOW_FXAA_USAGE=0 because some people enabled FXAA and suddenly had broken rendering when Firefox switched from GLX to EGL.
But setting that env var causes a performance regression in the driver.
Workaround: Enabling "Force composition pipeline" basically reverts the performance regression.
https://bugzilla.mozilla.org/show_bug.cgi?id=1736245

NoXPhasma, 2022-01-31:
I think the better workaround is to set `layout.frame_rate` from `-1` to `0` in `about:config`.

Darkspirit, 2022-01-31:
No,
* layout.frame_rate=0 renders every animation with max fps = hundreds of fps!
* people are falsely recommending it to fix a GLX performance problem (https://bugzilla.mozilla.org/show_bug.cgi?id=1716049) caused by gfx.swap-interval.glx being false. That pref should instead be set to true, but it's not needed for EGL. EGL already has the desired default.

NoXPhasma, 2022-01-31:
Yeah, I've just seen that the CPU load was cranking up with set to `0`.

elFarto, 2022-02-12:
I've just try retesting this with FF96, v0.0.5 and the Switzerland 8k video, and it seems to be playing back perfectly.

joebonrichie, 2022-02-12:
watched about 2mins from the linked video `Viewport / Frames 530 dropped of 7261` significantly better than it was but seemed to stutter every ~30secs or so and drop a bunch of frames with most of the frame drops being in the first 10secs or so of the video.

0.0.5, FF 97. NV driver 510, X11

elFarto, 2022-12-07:
Closing this issue as we've done all we can for it.

closed 2022-12-07 by elFarto