elFarto/nvidia-vaapi-driver #14: EGL_BAD_STATE_KHR error

miyuchina, 2022-01-06:
Thank you for this project -- it's great to see hwdec on nvidia moving along. I'm seeing similar errors as #4 and #11, but am using:

* X11;
* RTX 2060 Super;
* Ryzen 7 3700X (no integrated graphics, no Optimus)

Here are my logs:

<details open>
<summary>nvidia-drm.modeset=1</summary>

``` sh
$ cat /sys/module/nvidia_drm/parameters/modeset
Y
```
</details>

<details open>

<summary>vainfo (similar output to #4, despite running on X11 this time):</summary>

``` sh
$ NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia vainfo
[5033-5033] ../src/vabackend.c:1522       __vaDriverInit_1_0 Initing NVIDIA VA-API Driver
[5033-5033] ../src/export-buf.c:  90          findCudaDisplay Found 3 EGL devices
[5033-5033] ../src/export-buf.c:  94          findCudaDisplay Got EGL_CUDA_DEVICE_NV value '0' from device 0
[5033-5033] ../src/export-buf.c: 121             initExporter Got EGLDisplay from CUDA device
[5033-5033] ../src/export-buf.c:  56                reconnect Reconnecting to stream
vainfo: VA-API version: 1.13 (libva 2.13.0)
vainfo: Driver version: VA-API NVDEC driver
[5033-5033] ../src/vabackend.c: 202              vaToCuCodec vaToCuCodec: Unknown codec: 2
[5033-5033] ../src/vabackend.c: 202              vaToCuCodec vaToCuCodec: Unknown codec: 3
[5033-5033] ../src/vabackend.c: 202              vaToCuCodec vaToCuCodec: Unknown codec: 4
[5033-5033] ../src/vabackend.c: 202              vaToCuCodec vaToCuCodec: Unknown codec: 12
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      <unknown profile>               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
[5033-5033] ../src/vabackend.c:1511              nvTerminate In nvTerminate
[5033-5033] ../src/export-buf.c:  36                    debug [EGL] eglStreamImageConsumerConnectNV: EGL_BAD_STATE_KHR error: In EGL Access Table::stream2.consumer.disconnect: Consumer handle does not match reservation (0x422626b1 vs 0x422626b9).
```

</details>

<details>

<summary>mpv (video opens but only shows a green screen, same as any YouTube videos I tried)</summary>

``` sh
$ NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia mpv --hwdec=vaapi --opengl-debug --hwdec-codecs=all file_example_MP4_1920_18MG.mp4 > out.log
Warning: option --opengl-debug was replaced with --gpu-debug and might be removed in the future.
 (+) Video --vid=1 (*) (h264 1920x1080 30.000fps)
 (+) Audio --aid=1 (*) (aac 2ch 48000Hz)
[4685-4696] ../src/vabackend.c:1522       __vaDriverInit_1_0 Initing NVIDIA VA-API Driver
[4685-4696] ../src/export-buf.c:  90          findCudaDisplay Found 3 EGL devices
[4685-4696] ../src/export-buf.c:  94          findCudaDisplay Got EGL_CUDA_DEVICE_NV value '0' from device 0
[4685-4696] ../src/export-buf.c: 121             initExporter Got EGLDisplay from CUDA device
[4685-4696] ../src/export-buf.c:  56                reconnect Reconnecting to stream
[4685-4696] ../src/vabackend.c: 865      nvQueryImageFormats In nvQueryImageFormats
[4685-4696] ../src/vabackend.c: 202              vaToCuCodec vaToCuCodec: Unknown codec: 2
[4685-4696] ../src/vabackend.c: 202              vaToCuCodec vaToCuCodec: Unknown codec: 3
[... 1658 lines cut ...]
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (invalid VAContextID)
[vo/gpu/vaapi-egl] mapping VAAPI EGL image failed
[vo/gpu] Mapping hardware decoded surface failed.
[cplayer] Terminating demuxers...
[cplayer] Done terminating demuxers.
[cplayer] finished playback, success (reason 3)
[cplayer] 
[cplayer] Exiting... (Quit)
[cplayer] Run command: change-list, flags=64, args=[name="shared-script-properties", operation="remove", value="osc-margins"]
[ytdl_hook] Exiting...
[console] Exiting...
[cplayer] Set property: shared-script-properties -> 1
[stats] Exiting...
[osc] Exiting...
[vo/gpu/x11] Enabling screensaver.
[4652-4663] ../src/vabackend.c:1511              nvTerminate In nvTerminate
[4652-4663] ../src/export-buf.c:  36                    debug [EGL] eglSwapBuffers: EGL_BAD_STATE_KHR error: In EGL Access Table::stream2.consumer.disconnect: Consumer handle does not match reservation (0x58df1f11 vs 0x58df1f19).

[vo/gpu] flushing shader cache
[vo/gpu/x11] uninit ...
```

</details>

Let me know if there's anything else I should try. Happy to test any changes as well. Thanks again for your efforts.

miyuchina, 2022-01-06:
Never mind, `--hwdec=vaapi-copy` works for mpv -- didn't read far enough in the other issue. Still getting the same outputs from `vainfo` though. I can close this issue if it helps keeping all the info in #4.

thesword53, 2022-01-06:
I have the same issue (even with Wayland)
GPU: GTX 1060

elFarto, 2022-01-06:
vaapi-copy isn't needed, normal vaapi should work just fine. This error is subtly different from #11 as you are able to export the dma-buf, but it fails on importing.

These issues are difficult for me to debug as I can't see why the driver is failing to export/import the images.

elFarto, 2022-01-06:
Now I've had a chance to look at this properly, the EGL_BAD_STATE_KHR error is spurious on shutdown, and I've just fixed it (not releasing things in the correct order).

The main issue here is the `GL_INVALID_OPERATION error generated. <image> and <target> are incompatible` error.

Are you using the git master build of mpv? That's needed for this to work.

thesword53, 2022-01-06:
No.
I use `mpv 0.34.1` from Arch Linux.

miyuchina, 2022-01-07:
Just tested everything again with `mpv 0.34.0-121-g57bc5ba6d6` from the master branch, and `--hwdec=vaapi` works just fine now. I don't see `GL_INVALID_OPERATION` anymore.

The `EGL_BAD_STATE_KHR` is also no longer there on shutdown.

Closing this for now -- thanks!

closed 2022-01-07 by miyuchina