elFarto/nvidia-vaapi-driver #18: av1 support

krakow10, 2022-01-07:
Testing on mpv with vaapi-copy, the other codecs work fine.  Playing av1 videos does report video decode in nvtop and the log prints the images decoded without any errors.  However, the output is corrupted:
[image]
I don't know where to begin looking for problems, what can I look for?

krakow10, 2022-01-07:
I logged out to test on wayland, and when I logged back in the decode was working (albeit with a terrifying 2GB/s memory leak).  I can't reproduce the issue so far.

elFarto, 2022-01-07:
AV1 isn't supported at the moment, unfortunately I've just noticed that I didn't comment it out with the rest of the codecs. There is a partial implementation, which is likely what is producing the corrupted image, but it's still missing some logic, and testing.

elFarto, 2022-01-09:
Closing this issue as AV1 isn't supported at the moment.

closed 2022-01-09 by elFarto