elFarto/nvidia-vaapi-driver #23: Allow running in Flatpak Firefox

Gert-dev, 2022-01-11:
Hello

I did some cursory tests when running Firefox in a Flatpak, but it doesn't appear to work here. VA-API appears to return -1 when trying to select the `nvidia` driver when the first video is started.

I'm not too familiar with how Flatpak works, but IIRC it has its own VA-API packages, so it is possible that they cannot access a built version of this driver plug-in and a Flatpak extension might need to be built (but this might be complete nonsense, as I said I don't know too much about Flatpak).

I verified it did work with a native, non-flatpak-ed, Firefox.

In any case, great work on this project!

elFarto, 2022-01-19:
Sorry, I didn't ever get around to commenting on this. It's definitely something I'll look into, but it's not a priority for me.

Gert-dev, 2022-01-28:
On a related note, there is now [a feature request on the FreeDesktop SDK issue tracker](https://gitlab.com/freedesktop-sdk/freedesktop-sdk/-/issues/1385) asking for this to be packaged, which would/could make it available in Flatpaks using this runtime.

marioroy, 2022-08-30:
The VA-API NVDEC (used by Firefox, Mpv) and VDPAU (used by Chrome) backends are both needed. That issue request will break flatpaks running Chrome and derivatives. The reason is that the two backends have conflicting file names: nvidia_drv_video.so.

JCallicoat, 2023-01-03:
I am using a little script to set env variable overrides for the firefox flatpak, and copy the driver to where the flatpak can read it. After running the script you just run the firefox flatpak as you would normally. Copying the driver is kind of hacky -- the script must be run whenever you update `nvidia-vaapi-driver` on the host -- but thought it might be useful to share here as a stopgap.

```bash
#!/bin/sh

DRI_PATH=${HOME}/.var/app/org.mozilla.firefox/dri

echo "Clearing firefox flatpak overrides"
flatpak override --user --reset org.mozilla.firefox

echo "Adding firefox flatpak overrides"
flatpak override --user --env=LIBVA_DRIVER_NAME=nvidia \
                        --env=LIBVA_DRIVERS_PATH=${DRI_PATH} \
                        --env=MOZ_DISABLE_RDD_SANDBOX=1 \
                        --env=NVD_BACKED=direct \
                        org.mozilla.firefox

# uncomment the following if you use keepass running on the host
# flatpak override --user --filesystem=xdg-run/app/org.keepassxc.KeePassXC org.mozilla.firefox

echo "Copying nvidia vaapi driver into ${DRI_PATH}"
mkdir -p ${DRI_PATH}
cp /usr/lib/dri/nvidia_drv_video.so ${DRI_PATH}/nvidia_drv_video.so

cat <<"EOF"

    Now open about:config and change `gfx.webrender.all` and `media.ffmpeg.vaapi.enabled` to true.

EOF
```

You probably also need to install the ffmpeg-full platform extension for the version of the runtime firefox is using. You can check with `flatpak info --show-runtime org.mozilla.firefox`. The current stable version is using the 22.08 runtime so you can install the extension with `flatpak install --user org.freedesktop.Platform.ffmpeg-full//22.08`. I also have flatpak nvidia drivers installed (`org.freedesktop.Platform.GL.nvidia-525-60-11` as of writing) which may also be needed.

FranciscoDiMarzo, 2023-01-23:
> I am using a little script to set env variable overrides for the firefox flatpak, and copy the driver to where the flatpak can read it. After running the script you just run the firefox flatpak as you would normally. Copying the driver is kind of hacky -- the script must be run whenever you update `nvidia-vaapi-driver` on the host -- but thought it might be useful to share here as a stopgap.
> 
> ```shell
> #!/bin/sh
> 
> DRI_PATH=${HOME}/.var/app/org.mozilla.firefox/dri
> 
> echo "Clearing firefox flatpak overrides"
> flatpak override --user --reset org.mozilla.firefox
> 
> echo "Adding firefox flatpak overrides"
> flatpak override --user --env=LIBVA_DRIVER_NAME=nvidia \
>                         --env=LIBVA_DRIVERS_PATH=${DRI_PATH} \
>                         --env=LIBVA_MESSAGING_LEVEL=1 \
>                         --env=MOZ_DISABLE_RDD_SANDBOX=1 \
>                         --env=NVD_BACKEND=direct \
>                         org.mozilla.firefox
> 
> # uncomment the following if you use keepass running on the host
> # flatpak override --user --filesystem=xdg-run/app/org.keepassxc.KeePassXC org.mozilla.firefox
> 
> echo "Copying nvidia vaapi driver into ${DRI_PATH}"
> mkdir -p ${DRI_PATH}
> cp /usr/lib/dri/nvidia_drv_video.so ${DRI_PATH}/nvidia_drv_video.so
> 
> cat <<"EOF"
> 
>     Now open about:config and change `gfx.webrender.all` and `media.ffmpeg.vaapi.enabled` to true.
> 
> EOF
> ```
> 
> You probably also need to install the ffmpeg-full platform extension for the version of the runtime firefox is using. You can check with `flatpak info --show-runtime org.mozilla.firefox`. The current stable version is using the 22.08 runtime so you can install the extension with `flatpak install --user org.freedesktop.Platform.ffmpeg-full//22.08`. I also have flatpak nvidia drivers installed (`org.freedesktop.Platform.GL.nvidia-525-60-11` as of writing) which may also be needed.

Thanks, worked for me, just had to change the path from the driver from `cp /usr/lib/dri/nvidia_drv_video.so ${DRI_PATH}/nvidia_drv_video.so` to `cp /usr/lib64/dri/nvidia_drv_video.so ${DRI_PATH}/nvidia_drv_video.so` on line 21
shouldn't a symlink work too and prevent the need to re-run the script each time the api is updated?

JCallicoat, 2023-02-08:
> I don't know if giving access to that specific folder through flatseal and then creating a symlink would be any different.

Yeah linking or just allowing filesystem access is not possible because `/usr` is blacklisted, because the flatpak basically just bind-mounts an ostree repo into it's own mount namespace and has it's own system dirs like `/usr` that are provided by the runtimes. Which also means a symlink would just point to the non-existent file in the runtime `/usr` and a hardlink wouldn't be allowed to access the inode on underlying the host `/usr`.

I  think the best option for flatpak end-users would be to have it included in the runtime per the issue @Gert-dev linked. For folks who want to run nvidia-vaapi-driver git builds after that, we will still need to just copy the driver and force the vaapi driver search path it seems.

JCallicoat, 2023-03-16:
Did you make sure you have the correct nvidia flatpack drivers installed to match the host drivers? For 530.30.02 host drivers you would need the org.freedesktop.Platform.GL32.nvidia-530-30-02 flatpak drivers installed.

berkant, 2023-03-18:
Similarly, I think, Snap-installed Firefox doesn't work either.

JCallicoat, 2023-04-29:
@berkant snaps do something ~~crazy~~ different and mount all the packages on their own loopback devices, but in theory a similar workaround to the script I am using (posted above) would work for snaps? I.e., copy the driver somewhere the snap can read it, override env variables for the snap to use the driver. I have never really used snaps so not sure how feasible it is.

JCallicoat, 2023-08-07:
I ran into an issue that was a pain to figure out so documenting here. Arch uses a newer glibc (2.38 as of writing) but the current freedesktop platform runtime uses 2.35 and cannot load nvidia_drv_video.so built with system gcc.

You just get the log message:

```
[GFX1-]: vaapitest: ERROR
[GFX1-]: vaapitest: VA-API test failed: failed to initialise VAAPI connection.
```

None of the logging options for NVD, LIBVA, MOZ_LOG show anything because the loader was actually failing with ``version `GLIBC_2.38' not found`` because of the newer symbols from 2.38. I managed to work around this by "cross-compiling" to an older version of glibc using [zig cc](https://andrewkelley.me/post/zig-cc-powerful-drop-in-replacement-gcc-clang.html) to compile nvidia-vaapi-driver. I used these changes for the [libva-nvidia-driver-git](https://aur.archlinux.org/packages/libva-nvidia-driver-git) PKGBUILD.

```
 build() {
     cd ${pkgname}

+    export CC="zig cc"
+    export CFLAGS="-target x86_64-linux.6.4.8...6.4.8-gnu.2.34 -march=x86-64 -mtune=x86_64 -I/usr/include"
+
     arch-meson . build

     ninja -C build
```

Similar should work for other bleeding edge distros like gentoo and rawhide.

Maccer1, 2024-01-21:
> I am using a little script to set env variable overrides for the firefox flatpak, and copy the driver to where the flatpak can read it. After running the script you just run the firefox flatpak as you would normally. Copying the driver is kind of hacky -- the script must be run whenever you update `nvidia-vaapi-driver` on the host -- but thought it might be useful to share here as a stopgap.
> 
> ```shell
> #!/bin/sh
> 
> DRI_PATH=${HOME}/.var/app/org.mozilla.firefox/dri
> 
> echo "Clearing firefox flatpak overrides"
> flatpak override --user --reset org.mozilla.firefox
> 
> echo "Adding firefox flatpak overrides"
> flatpak override --user --env=LIBVA_DRIVER_NAME=nvidia \
>                         --env=LIBVA_DRIVERS_PATH=${DRI_PATH} \
>                         --env=LIBVA_MESSAGING_LEVEL=1 \
>                         --env=MOZ_DISABLE_RDD_SANDBOX=1 \
>                         --env=NVD_BACKEND=direct \
>                         org.mozilla.firefox
> 
> # uncomment the following if you use keepass running on the host
> # flatpak override --user --filesystem=xdg-run/app/org.keepassxc.KeePassXC org.mozilla.firefox
> 
> echo "Copying nvidia vaapi driver into ${DRI_PATH}"
> mkdir -p ${DRI_PATH}
> cp /usr/lib/dri/nvidia_drv_video.so ${DRI_PATH}/nvidia_drv_video.so
> 
> cat <<"EOF"
> 
>     Now open about:config and change `gfx.webrender.all` and `media.ffmpeg.vaapi.enabled` to true.
> 
> EOF
> ```
> 
> You probably also need to install the ffmpeg-full platform extension for the version of the runtime firefox is using. You can check with `flatpak info --show-runtime org.mozilla.firefox`. The current stable version is using the 22.08 runtime so you can install the extension with `flatpak install --user org.freedesktop.Platform.ffmpeg-full//22.08`. I also have flatpak nvidia drivers installed (`org.freedesktop.Platform.GL.nvidia-525-60-11` as of writing) which may also be needed.
> 
> Edit: Flatpak nvidia drivers are definitely needed and they need to match the host driver version. For 530.30.02 host drivers you would need the org.freedesktop.Platform.GL.nvidia-530-30-02 flatpak drivers installed for example.

Adding this for future refence. 
I was able to make the driver work under ubuntu 22.04 + firefox 121 flatpak + nvidia-535.146.02 flatpak + nvidia-vaapi-driver v0.0.11 compiled from source.  I had to change the  ` cp /usr/lib/dri/nvidia_drv_video.so ${DRI_PATH}/nvidia_drv_video.so ` to  `cp /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so ${DRI_PATH}/nvidia_drv_video.so`.

Thanks a lot for the script!

S7venLights, 2024-06-11:
Question: In flatseal or KDE settings you can set env variables there, would it be fine to just use that instead of your script?
[image]
I will try just add your variables there and copy the drivers to my librewolf directory

@JCallicoat

S7venLights, 2024-06-11:
Separate issue, I'm also having trouble playing local video with the non-free nvidia driver. It's only using the CPU. This is with any video player including MPV based Haruna.
But non-flatpak firefox plays video just fine with HW acceleration.

Would this be a nvidia-vaapi-driver issue or not?

JCallicoat, 2024-06-13:
@S7venLights Yes, using flatseal does the same as the cli command `flatpak override`. You can also find the overrides in the file `~/.local/share/flatpak/overrides/org.mozilla.firefox`

Also firefox doesn't currently support HEVC on linux at all, but it appears to be planned. Latest update I have seen about it is from 2 months ago:

```
We've actually added HEVC support (it's still off by default, check the pref media.wmf.hevc.enabled) on Windows, and we have a plan to extend it on other platforms as well.
```

https://bugzilla.mozilla.org/show_bug.cgi?id=1889731

S7venLights, 2024-06-15:
Hmm I just rebooted after Fedora updates and hardware acceleration in browser is not working again.
ran
`cp /usr/lib64/dri/nvidia_drv_video.so /var/home/user/.var/app/io.gitlab.librewolf-community/dri/nvidia_drv_video.so`
again and restarted librewolf to no avail.
Maybe it'll work after a reboot. 
Anyone else have issues?

holocronweaver, 2025-04-25:
On Kubuntu 24.04 I was able to get Flatpak Firefox acceleration working.

- installed the driver via `sudo apt install nvidia-vaapi-driver`
- set the environment variables in Flatseal as @S7venLights suggested
- set the Firefox config flags as in the README

I then closed Firefox and re-opened without restarting my PC, and like magic video acceleration was working. The reduction in stutter on certain streaming sites was massive, and my desktop CPU + GPU are no slouches.