elFarto/nvidia-vaapi-driver #24: Twitter videos crash tab

TheTimeWalker, 2022-01-12:
I currently have an issue where videos from for example Twitter would consistently crash the tab if it is being played. Nothing special is being done, just open Twitter and find a video to play. Firefox would then show the "Tab has crashed" info.

Looking at the NVD_LOG, I get an error fro cuda mentioning that OS call failed or is not support which sounds trange to me as I'm using the latest version:

```
[19978-19984] ../src/vabackend.c:1549       __vaDriverInit_1_0 Initing NVIDIA VA-API Driver
[19978-19984] ../src/export-buf.c:  90          findCudaDisplay Found 3 EGL devices
[19978-19984] ../src/export-buf.c:  94          findCudaDisplay Got EGL_CUDA_DEVICE_NV value '0' from device 0
[19978-19984] ../src/export-buf.c: 122             initExporter Got EGLDisplay from CUDA device
[19978-19984] ../src/export-buf.c:  56                reconnect Reconnecting to stream
[19978-19984] ../src/vabackend.c: 882      nvQueryImageFormats In nvQueryImageFormats
[19978-19989] ../src/vabackend.c:1549       __vaDriverInit_1_0 Initing NVIDIA VA-API Driver
[18876-19974] ../src/vabackend.c:1549       __vaDriverInit_1_0 Initing NVIDIA VA-API Driver
[18876-19974] ../src/vabackend.c:1568       __vaDriverInit_1_0 cuda error 'OS call failed or operation not supported on this OS' (304)
```

The corresponding vaainfo
```
vainfo: VA-API version: 1.13 (libva 2.13.0)
vainfo: Driver version: VA-API NVDEC driver
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      <unknown profile>               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
```
OS: Manjaro Linux
CUDA: 11.5.1
GPU: GTX 1070

elFarto, 2022-01-12:
I can reproduce this, but it makes absolutely no sense. I can literally go from Youtube playing a video correctly, to twitter, and have it crash saying it can't initialise CUDA, which it clearly can.

I can also get a 999 error in the same place. 

I'll take a look into it.

elFarto, 2022-01-12:
Can you check you have the `media.rdd-ffmpeg.enabled` config option set to `true` in Firefox?

TheTimeWalker, 2022-01-12:
[image]

Yes, I have set the config option to true according to the README

imatimba, 2022-01-12:
Also happens on twitch, youtube, etc. But it's pretty inconsistent, sometimes just works.  
On twitch instead of crashing the tab, the player can show error #3000 (Decoder error) showing this in terminal:  
```
[Child 2876853, MediaDecoderStateMachine #1] WARNING: Decoder=7ffe550ea300 Decode error: NS_ERROR_DOM_MEDIA_FATAL_ERR (0x806e0005) - auto mozilla::MediaChangeMonitor::CreateDecoderAndInit(mozilla::MediaRawData *)::(anonymous class)::operator()()::(anonymous class)::operator()(const mozilla::MediaResult &) const: Unable to initialize decoder: file /builds/worker/checkouts/gecko/dom/media/MediaDecoderStateMachine.cpp:3568
```

This issue started with the change to use nv-codec-headers.

<details>
<summary>Full log with NVD_LOG=1 MOZ_LOG="PlatformDecoderModule:5"</summary>

```
[GFX1-]: More than 1 GPU from same vendor detected via PCI, cannot deduce device
[Child 2875427: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in Content process
[Child 2875427: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 2875427: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 2875427: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 2875427: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 2875427: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 2875427: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[RDD 2876104: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in RDD process
[RDD 2876104: Main Thread]: D/PlatformDecoderModule Agnostic decoder rejects requested type
[RDD 2876104: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type
[RDD 2876104: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type
[RDD 2876104: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type
[RDD 2876104: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type
[RDD 2876104: MediaSupervisor #1]: D/PlatformDecoderModule FFmpegAudioDecoder, found encoder delay (2257) and padding values (290) in extra data
[RDD 2876104: MediaSupervisor #1]: D/PlatformDecoderModule Agnostic decoder supports requested type
[RDD 2876104: MediaSupervisor #1]: D/PlatformDecoderModule Agnostic decoder supports requested type
[RDD 2876104: MediaPDecoder #2]: D/PlatformDecoderModule Initialising FFmpeg decoder.
[RDD 2876104: MediaPDecoder #2]: D/PlatformDecoderModule FFmpeg init successful.
[Child 2875427: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdec85420] ::PrepareTrimmers: sample[0,26122] no trimming information
[Child 2875427: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdec85420] ::HandleDecodedResult: sample[0,26122] (decoded[0,0] no trimming needed
[Child 2875427: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdec85420] ::PrepareTrimmers: sample[26122,52244] no trimming information
[Child 2875427: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdec85420] ::HandleDecodedResult: sample[26122,52244] (decoded[26122,27187] no trimming needed
[Child 2875427: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdec85420] ::PrepareTrimmers: sample[52244,78366] no trimming information
[Child 2875427: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdec85420] ::HandleDecodedResult: sample[52244,78366] (decoded[52244,78366] no trimming needed
[Child 2875427: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdec85420] ::PrepareTrimmers: sample[78366,104488] no trimming information
[Child 2875427: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdec85420] ::HandleDecodedResult: sample[78366,104488] (decoded[78366,104488] no trimming needed
[Child 2875427: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdec85420] ::PrepareTrimmers: sample[104488,130610] no trimming information
[Child 2875427: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdec85420] ::HandleDecodedResult: sample[104488,130610] (decoded[104488,130610] no trimming needed
[Child 2875427: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdec85420] ::PrepareTrimmers: sample[130610,156732] no trimming information
[Child 2875427: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdec85420] ::HandleDecodedResult: sample[130610,156732] (decoded[130610,156732] no trimming needed
[Child 2875427: MediaPDecoder #3]: D/PlatformDecoderModule AudioTrimmer[7fffdec85420] ::PrepareTrimmers: sample[156732,182854] no trimming information
[Child 2875427: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdec85420] ::HandleDecodedResult: sample[156732,182854] (decoded[156732,182854] no trimming needed
[Child 2875427: MediaPDecoder #2]: D/PlatformDecoderModule AudioTrimmer[7fffdec85420] ::PrepareTrimmers: sample[182854,208976] no trimming information
[Child 2875427: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdec85420] ::HandleDecodedResult: sample[182854,208976] (decoded[182854,208976] no trimming needed
[Child 2875427: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdec85420] ::PrepareTrimmers: sample[208976,235098] no trimming information
[... cut ...]
[Child 2876853, MediaDecoderStateMachine #1] WARNING: Decoder=7ffe550ea300 Decode error: NS_ERROR_DOM_MEDIA_FATAL_ERR (0x806e0005) - auto mozilla::MediaChangeMonitor::CreateDecoderAndInit(mozilla::MediaRawData *)::(anonymous class)::operator()()::(anonymous class)::operator()(const mozilla::MediaResult &) const: Unable to initialize decoder: file /builds/worker/checkouts/gecko/dom/media/MediaDecoderStateMachine.cpp:3568
```

</details>

OS: Gentoo Linux
nvidia-drivers: 495.46
nv-codec-headers: 11.1.5.1
GPU: GTX 970

elFarto, 2022-01-12:
> This issue started with the change to use nv-codec-headers.

@philipl Could this be a race condition/threading issue with these headers?

philipl, 2022-01-12:
Maybe this goes back to the question of multiple initialisation. Perhaps firefox is doing just that as you move from site to site and maybe it's shutting down one instance after initialising another one. Then the statics come back to bite us. I can work on passing the function structs around everywhere they are needed.

elFarto, 2022-01-12:
@TheTimeWalker & @imatimba I've made a change to move the CUDA init into a constructor/destructor. Can you test that change, and see if that resolves the issue?

imatimba, 2022-01-12:
Fixed on my side.  
Tested opening several twitch streams at the same time, youtube, twitter videos. Closing and opening again, loading the tab from cache on startup.  
Thanks!

TheTimeWalker, 2022-01-12:
I can also confirm that everything is working! It fixed multiple issues I first had that seemingly were related to this:
1. Twitch streams now work with 1080p60 (before max was 720p60)
2. GIFs / Webm videos don't crash on Discord anymore
3. Twitter, Reddit and YouTube (1080p60) videos work fine
4. No random crashes from multiple running videos/streams

Thank you for the very fast fix! And thanks in general for the implementation :)

I will keep the issue open for a bit as I see that there's an open PR connected to that. Feel free to close it earlier if the solution has been settled.

elFarto, 2022-01-12:
Thanks for the feedback. We'll get the final fix for this in soon.