elFarto/nvidia-vaapi-driver #26: SPS decoding failure with nvidia-drivers 510.39.01

imatimba, 2022-01-13:
After upgrading to nvidia-drivers 510.39.01 every stream/video player shows some kind of error and aborts the playback. No tab crashes so far.  
Rolling back to 495.46 fixes the issue.  

Log is just from opening one youtube video until the player shows the generic playback error.

<details>
<summary>Full log with NVD_LOG=1 MOZ_LOG="PlatformDecoderModule:5"</summary>

```
[GFX1-]: More than 1 GPU from same vendor detected via PCI, cannot deduce device
[Child 27777: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in Content process
[Child 27777: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 27777: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 27777: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 27777: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 27777: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 27777: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[RDD 28426: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in RDD process
[RDD 28426: Main Thread]: D/PlatformDecoderModule Agnostic decoder rejects requested type
[RDD 28426: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type
[RDD 28426: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type
[RDD 28426: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type
[RDD 28426: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type
[RDD 28426: MediaSupervisor #1]: D/PlatformDecoderModule FFmpegAudioDecoder, found encoder delay (2257) and padding values (290) in extra data
[RDD 28426: MediaSupervisor #1]: D/PlatformDecoderModule Agnostic decoder supports requested type
[RDD 28426: MediaSupervisor #1]: D/PlatformDecoderModule Agnostic decoder supports requested type
[RDD 28426: MediaPDecoder #1]: D/PlatformDecoderModule Initialising FFmpeg decoder.
[RDD 28426: MediaPDecoder #1]: D/PlatformDecoderModule FFmpeg init successful.
[Child 27777: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::PrepareTrimmers: sample[0,26122] no trimming information
[Child 27777: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::HandleDecodedResult: sample[0,26122] (decoded[0,0] no trimming needed
[Child 27777: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::PrepareTrimmers: sample[26122,52244] no trimming information
[Child 27777: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::HandleDecodedResult: sample[26122,52244] (decoded[26122,27187] no trimming needed
[Child 27777: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::PrepareTrimmers: sample[52244,78366] no trimming information
[Child 27777: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::HandleDecodedResult: sample[52244,78366] (decoded[52244,78366] no trimming needed
[Child 27777: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::PrepareTrimmers: sample[78366,104488] no trimming information
[Child 27777: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::HandleDecodedResult: sample[78366,104488] (decoded[78366,104488] no trimming needed
[Child 27777: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::PrepareTrimmers: sample[104488,130610] no trimming information
[Child 27777: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::HandleDecodedResult: sample[104488,130610] (decoded[104488,130610] no trimming needed
[Child 27777: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::PrepareTrimmers: sample[130610,156732] no trimming information
[Child 27777: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::HandleDecodedResult: sample[130610,156732] (decoded[130610,156732] no trimming needed
[Child 27777: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::PrepareTrimmers: sample[156732,182854] no trimming information
[Child 27777: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::HandleDecodedResult: sample[156732,182854] (decoded[156732,182854] no trimming needed
[Child 27777: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::PrepareTrimmers: sample[182854,208976] no trimming information
[Child 27777: MediaPDecoder #2]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::HandleDecodedResult: sample[182854,208976] (decoded[182854,208976] no trimming needed
[Child 27777: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::PrepareTrimmers: sample[208976,235098] no trimming information
[Child 27777: MediaPDecoder #2]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::HandleDecodedResult: sample[208976,235098] (decoded[208976,235098] no trimming needed
[Child 27777: MediaPDecoder #2]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::PrepareTrimmers: sample[235098,261220] no trimming information
[Child 27777: MediaPDecoder #3]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::HandleDecodedResult: sample[235098,261220] (decoded[235098,261220] no trimming needed
[Child 27777: MediaPDecoder #3]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::PrepareTrimmers: sample[261220,287342] no trimming information
[Child 27777: MediaPDecoder #2]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::HandleDecodedResult: sample[261220,287342] (decoded[261220,287342] no trimming needed
[Child 27777: MediaPDecoder #2]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::PrepareTrimmers: sample[287342,313464] no trimming information
[Child 27777: MediaPDecoder #3]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::HandleDecodedResult: sample[287342,313464] (decoded[287342,313464] no trimming needed
[Child 27777: MediaPDecoder #3]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::PrepareTrimmers: sample[313464,339586] no trimming information
[Child 27777: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::HandleDecodedResult: sample[313464,339586] (decoded[313464,339586] no trimming needed
[Child 27777: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::PrepareTrimmers: sample[339586,365708] no trimming information
[Child 27777: MediaPDecoder #3]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::HandleDecodedResult: sample[339586,365708] (decoded[339586,365708] no trimming needed
[Child 27777: MediaPDecoder #3]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::PrepareTrimmers: sample[365708,391830] no trimming information
[Child 27777: MediaPDecoder #2]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::HandleDecodedResult: sample[365708,391830] (decoded[365708,391830] no trimming needed
[Child 27777: MediaPDecoder #2]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::PrepareTrimmers: sample[391830,417952] no trimming information
[Child 27777: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::HandleDecodedResult: sample[391830,417952] (decoded[391830,417952] no trimming needed
[Child 27777: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::PrepareTrimmers: sample[417952,444074] no trimming information
[Child 27777: MediaPDecoder #2]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::HandleDecodedResult: sample[417952,444074] (decoded[417952,444074] no trimming needed
[Child 27777: MediaPDecoder #2]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::PrepareTrimmers: sample[444074,470196] no trimming information
[Child 27777: MediaPDecoder #3]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::HandleDecodedResult: sample[444074,470196] (decoded[444074,470196] no trimming needed
[Child 27777: MediaPDecoder #3]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::PrepareTrimmers: sample[470196,496318] no trimming information
[Child 27777: MediaPDecoder #2]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::HandleDecodedResult: sample[470196,496318] (decoded[470196,496318] no trimming needed
[Child 27777: MediaPDecoder #2]: D/PlatformDecoderModule AudioTrimmer[7fffdb8f31a0] ::PrepareTrimmers: sample[496318,522440] no trimming information
[... cut ...]
      VAProfileVC1Advanced            : VAEntrypointVLD
      <unknown profile>               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointVLD

```

</details>

OS: Gentoo Linux
nvidia-drivers: 510.39.01
nv-codec-headers: 11.1.5.1
GPU: GTX 970

PD: Not sure which error is the most relevant one, feel free to change the title

elFarto, 2022-01-13:
Interesting, they've made a fix. Can you try replacing the occurrence of `CU_EGL_COLOR_FORMAT_YVU420_SEMIPLANAR` with `CU_EGL_COLOR_FORMAT_YUV420_SEMIPLANAR` in export-buf.c?

SupervisedThinking, 2022-01-13:
this works for me:
```
--- a/src/export-buf.c
+++ b/src/export-buf.c
@@ -192,7 +192,7 @@
 
     int bpp = 1;
     if (surface->format == cudaVideoSurfaceFormat_NV12) {
-        eglframe.eglColorFormat = CU_EGL_COLOR_FORMAT_YVU420_SEMIPLANAR;
+        eglframe.eglColorFormat = CU_EGL_COLOR_FORMAT_YUV420_SEMIPLANAR;
         eglframe.cuFormat = CU_AD_FORMAT_UNSIGNED_INT8;
     } else if (surface->format == cudaVideoSurfaceFormat_P016) {
         //TODO not working, produces this error in mpv:
```

qumaciel, 2022-01-13:
I had the same issue and worked for me with YT/Twitch.

OS: Garuda (KDE on X)
nvidia-drivers: 510.39.01
nv-codec-headers: 11.1.5.0.r4.g84483da-1
GPU: RTX2060M

imatimba, 2022-01-13:
Yeah, that change also fixes the issue for me.

elFarto, 2022-01-13:
Excellent. I'll see if I can work out a way to auto-detect which version driver we're on and pick the correct value.

commit 5b6ca08c (2022-01-13): Added NV12/NV21 work around.

The NVIDIA drivers have previous had an issue with the mapping between the CUDA image formats and
the DRM formats NV12 and NV21 being reversed. This bug has been fixed in the 510 series, however
this means we now need to detect if the driver we're running with has the fix or not and correct
our usage of the API.

This fix will detect when the wrong format is used and flip to using the correct value. At some
point the default should be changed to the correct value, likely after 510 has been released.

Fixes #26
closed 2022-01-13 by elFarto