elFarto/nvidia-vaapi-driver #27: [wayland] Kodi crashs if you play a fps test video

SupervisedThinking, 2022-01-13:
Sway https://github.com/swaywm/sway/releases/tag/1.7-rc2
Kodi https://github.com/xbmc/xbmc/commit/5ad708b518e6bb3e8ac31a9c7b5a8908573a409f
Nvidia [510.39.01](http://us.download.nvidia.com/XFree86/Linux-x86_64/510.39.01/README/index.html)
nvidia-vaapi-driver https://github.com/elFarto/nvidia-vaapi-driver/commit/461260bdc0490eb4f489111b835f6f6ef8193e8e + https://github.com/elFarto/nvidia-vaapi-driver/issues/26#issuecomment-1011930855
libva patched with https://github.com/intel/libva/pull/554

### Kodi log
[kodi_crashlog_20220113174044.log](https://github.com/elFarto/nvidia-vaapi-driver/files/7865010/kodi_crashlog_20220113174044.log)

### nvidia-vaapi-driver log
[vaapi.log](https://github.com/elFarto/nvidia-vaapi-driver/files/7865012/vaapi.log)

### test file
[FPS_test_1080p23.976_L4.1.mkv.zip](https://github.com/elFarto/nvidia-vaapi-driver/files/7865015/FPS_test_1080p23.976_L4.1.mkv.zip)

elFarto, 2022-01-13:
I've managed to reproduce this locally. Not entirely sure what's causing it, everything looks correct.

SupervisedThinking, 2022-01-13:
I'm not sure if it's related but this one https://download.blender.org/peach/bigbuckbunny_movies/big_buck_bunny_1080p_h264.mov fails too

### Kodi log
[kodi_crashlog_20220113194609.log](https://github.com/elFarto
### nvidia-vaapi-driver log
[vaapi.log](https://github.com/elFarto/nvidia-vaapi-driver/files/7865537/vaapi.log)
/nvidia-vaapi-driver/files/7865536/kodi_crashlog_20220113194609.log)

commit 3b8bec8a (2022-01-13): Fixed issue when surface has slightly different size from previous frame.

A frame can differ in just one dimension.

Fixes #27
closed 2022-01-13 by elFarto