elFarto/nvidia-vaapi-driver #28: [wayland] Kodi crashs if you play interlaced test videos

SupervisedThinking, 2022-01-13:
Let me know if you need any additional information, logs etc. (sorry for flooding the issue tracker 🙈 )

Sway https://github.com/swaywm/sway/releases/tag/1.7-rc2
Kodi https://github.com/xbmc/xbmc/commit/5ad708b518e6bb3e8ac31a9c7b5a8908573a409f
Nvidia [510.39.01](http://us.download.nvidia.com/XFree86/Linux-x86_64/510.39.01/README/index.html)
nvidia-vaapi-driver https://github.com/elFarto/nvidia-vaapi-driver/commit/3b8bec8a1da43d37a3729e197985e0081992acc0
libva patched with https://github.com/intel/libva/pull/554

### Kodi log
[kodi_crashlog_20220113205911.log](https://github.com/elFarto/nvidia-vaapi-driver/files/7866041/kodi_crashlog_20220113205911.log)

### nvidia-vaapi-driver log
[vaapi.log](https://github.com/elFarto/nvidia-vaapi-driver/files/7866045/vaapi.log)

### test file
http://kwiboo.libreelec.tv/test/samples/H264_1080i-25-interlace_Kaesescheibchen.mkv

elFarto, 2022-01-13:
It doesn't appear that there's an error in that vaapi.log file, other than the VAProfileNone not being supported. I can play that video back on my machine fine all the way to the end.

I don't have the mpeg2 file I was using for testing. Honestly I haven't run any mpeg2 videos through it for a while, my main concern was h.264, h.265 and vp9.

SupervisedThinking, 2022-01-13:
@jernejsk guessed `this one looks like null pointer dereference` but he wasn't sure about that since he lacks knowledge if it comes to CUDA but maybe he can further explain what he spotted.

> my main concern was h.264, h.265 and vp9.

Which play fine & it's great to have basic support for Nvidia cards on Wayland now 👍🏻

qumaciel, 2022-01-13:
I am getting a green screen when trying to play this test file with mpv --hwdec=vaapi on X

<details>
  <summary>vaapi.log</summary>
  
  ```
[263606-263676] ../src/vabackend.c:1584       __vaDriverInit_1_0 Initing NVIDIA VA-API Driver
[263606-263676] ../src/export-buf.c:  90          findCudaDisplay Found 4 EGL devices
[263606-263676] ../src/export-buf.c:  94          findCudaDisplay Got EGL_CUDA_DEVICE_NV value '0' from device 0
[263606-263676] ../src/export-buf.c: 122             initExporter Got EGLDisplay from CUDA device
[263606-263676] ../src/export-buf.c:  56                reconnect Reconnecting to stream
[263606-263676] ../src/vabackend.c: 908      nvQueryImageFormats In nvQueryImageFormats
[263606-263676] ../src/vabackend.c: 427           nvCreateConfig In nvCreateConfig with profile: 0 with 0 attributes
[263606-263676] ../src/vabackend.c:1283 nvQuerySurfaceAttributes with 1 (nil) 0
[263606-263676] ../src/vabackend.c:1283 nvQuerySurfaceAttributes with 1 0x7fb2b4bd2100 5
[263606-263676] ../src/vabackend.c: 508        nvCreateSurfaces2 creating 1 surface(s) 128x128, format 1
[263606-263676] ../src/vabackend.c:1000            nvDeriveImage In nvDeriveImage
[263606-263676] ../src/export-buf.c:  35                    debug [EGL] eglDestroyImageKHR: EGL_BAD_PARAMETER error: In eglDestroyImage: Invalid EGLImage (0x1)

[263606-263676] ../src/export-buf.c:  35                    debug [EGL] eglDestroyImageKHR: EGL_BAD_PARAMETER error: In eglDestroyImage: Invalid EGLImage (0x7fb2b49287f0)

[263606-263676] ../src/export-buf.c:  35                    debug [EGL] eglDestroyImageKHR: EGL_BAD_PARAMETER error: In eglDestroyImage: Invalid EGLImage (0x7fb2b4bcfcb0)

[263606-263676] ../src/vabackend.c:1488    nvExportSurfaceHandle got 0x7fb2b43298d0
[263606-263676] ../src/export-buf.c: 302            exportCudaPtr eglExportDMABUFImageQueryMESA: 0x7fb2b4adbc51 NV21 (3132564e) planes:2 mods:300000000606014 300000000606013
[263606-263676] ../src/export-buf.c: 310            exportCudaPtr eglExportDMABUFImageMESA: 53 54 0 0, strides: 128 128 0 0, offsets: 0 0 0 0
[263606-263676] ../src/vabackend.c:1519    nvExportSurfaceHandle Detected NV12/NV21 NVIDIA driver bug, attempting to work around
[263606-263676] ../src/export-buf.c: 302            exportCudaPtr eglExportDMABUFImageQueryMESA: 0x7fb2b4bd1801 NV12 (3231564e) planes:2 mods:300000000606014 300000000606013
[263606-263676] ../src/export-buf.c: 310            exportCudaPtr eglExportDMABUFImageMESA: 55 56 0 0, strides: 128 128 0 0, offsets: 16384 8192 0 0
[263606-263676] ../src/vabackend.c:1527    nvExportSurfaceHandle Work around worked!
[263606-263676] ../src/vabackend.c: 427           nvCreateConfig In nvCreateConfig with profile: 1 with 0 attributes
[263606-263676] ../src/vabackend.c:1283 nvQuerySurfaceAttributes with 1 (nil) 0
[263606-263676] ../src/vabackend.c:1283 nvQuerySurfaceAttributes with 1 0x7fb2b4adc040 5
[263606-263676] ../src/vabackend.c: 427           nvCreateConfig In nvCreateConfig with profile: 8 with 0 attributes
[263606-263676] ../src/vabackend.c:1283 nvQuerySurfaceAttributes with 3 (nil) 0
[263606-263676] ../src/vabackend.c:1283 nvQuerySurfaceAttributes with 3 0x7fb2b4bd1e80 5
[263606-263676] ../src/vabackend.c: 427           nvCreateConfig In nvCreateConfig with profile: 9 with 0 attributes
[263606-263676] ../src/vabackend.c:1283 nvQuerySurfaceAttributes with 3 (nil) 0
[263606-263676] ../src/vabackend.c:1283 nvQuerySurfaceAttributes with 3 0x7fb2b4ae0980 5
[263606-263676] ../src/vabackend.c: 427           nvCreateConfig In nvCreateConfig with profile: 10 with 0 attributes
[263606-263676] ../src/vabackend.c:1283 nvQuerySurfaceAttributes with 3 (nil) 0
[263606-263676] ../src/vabackend.c:1283 nvQuerySurfaceAttributes with 3 0x7fb2b4ae0740 5
[263606-263676] ../src/vabackend.c: 427           nvCreateConfig In nvCreateConfig with profile: 5 with 0 attributes
[263606-263676] ../src/vabackend.c:1283 nvQuerySurfaceAttributes with 4 (nil) 0
[263606-263676] ../src/vabackend.c:1283 nvQuerySurfaceAttributes with 4 0x7fb2b4ae0c80 5
[263606-263676] ../src/vabackend.c: 427           nvCreateConfig In nvCreateConfig with profile: 6 with 0 attributes
[263606-263676] ../src/vabackend.c:1283 nvQuerySurfaceAttributes with 4 (nil) 0
[263606-263676] ../src/vabackend.c:1283 nvQuerySurfaceAttributes with 4 0x7fb2b4ae0700 5
[263606-263676] ../src/vabackend.c: 427           nvCreateConfig In nvCreateConfig with profile: 7 with 0 attributes
[263606-263676] ../src/vabackend.c:1283 nvQuerySurfaceAttributes with 4 (nil) 0
[263606-263676] ../src/vabackend.c:1283 nvQuerySurfaceAttributes with 4 0x7fb2b4ae0c80 5
[263606-263676] ../src/vabackend.c: 427           nvCreateConfig In nvCreateConfig with profile: 13 with 0 attributes
[263606-263676] ../src/vabackend.c:1283 nvQuerySurfaceAttributes with 4 (nil) 0
[263606-263676] ../src/vabackend.c:1283 nvQuerySurfaceAttributes with 4 0x7fb2b4ae0780 5
[263606-263676] ../src/vabackend.c: 427           nvCreateConfig In nvCreateConfig with profile: 17 with 0 attributes
[263606-263676] ../src/vabackend.c:1283 nvQuerySurfaceAttributes with 8 (nil) 0
[263606-263676] ../src/vabackend.c:1283 nvQuerySurfaceAttributes with 8 0x7fb2b4ae0c80 5
[263606-263676] ../src/vabackend.c: 427           nvCreateConfig In nvCreateConfig with profile: 14 with 0 attributes
[263606-263676] ../src/vabackend.c:1283 nvQuerySurfaceAttributes with 9 (nil) 0
[263606-263676] ../src/vabackend.c:1283 nvQuerySurfaceAttributes with 9 0x7fb2b4ae0780 5
[... cut ...]
[statusline] AV: 00:00:22 / 00:00:33 (67%) A-V:  0.000
[cplayer] Set property: shared-script-properties -> 1
[statusline] AV: 00:00:23 / 00:00:33 (71%) A-V:  0.000
[cplayer] Saving state.
[cplayer] EOF code: 5
[vo/gpu/x11] Enabling screensaver.
[cplayer] finished playback, success (reason 3)
[cplayer]
[cplayer] Exiting... (Quit)
[cplayer] Set property: shared-script-properties -> 1

   ```
</details>

OS: Garuda (KDE on X)
nvidia-drivers: 510.39.01
nv-codec-headers: 11.1.5.0.r4.g84483da-1
GPU: RTX2060M

elFarto, 2022-01-14:
My mistake, I missed the stack traces in the kodi log file. I'm not sure why a buffer would ever null, likely a threading issue. Unfortunately I still can't reproduce it. I'll add some more logging, and checks and see what that produces.

@qumaciel A green screen almost certainly means you're not using the latest git version of mpv.

qumaciel, 2022-01-14:
> @qumaciel A green screen almost certainly means you're not using the latest git version of mpv.

oh, my bad! Now it is reproducing fine.

elFarto, 2022-01-14:
I've pushed a couple of changes that might help, or should at least log when there's an issue.

SupervisedThinking, 2022-01-14:
fixed by: https://github.com/elFarto/nvidia-vaapi-driver/commit/00fac0bb2b74db64b4eff89da59b5670f0aed3b5

I guess you can draft a new release soon ^^

closed 2022-01-14 by SupervisedThinking