elFarto/nvidia-vaapi-driver #30: Have no idea where to give positive feedback, feel free to close!

smumriak, 2022-01-15:
Hey! 
I just wanted to say thank you for the great job on this project! With some bumps on the road I was able to build it on pop!_os (mainly hack around the missing meson feature because popos installs v0.56 instead of v0.58 and install cuda while not breaking my system lol) and use it with Firefox 96.0 on Nvidia 495 driver. Works like a charm!
GreenWithEnvy reports 6% decoder usage while playing 1440p VP9 youtube video (it bumps to 20% on start, but quickly goes does to 6%).
Awesome work!

elFarto, 2022-01-15:
Hey, thanks for the comment! The latest version doesn't need CUDA, so that should make it a little bit easier to build.

closed 2022-01-15 by elFarto

smumriak, 2022-01-15:
Btw, looks like AV1 codec is coming in 510: 
```
Added AV1 decode support to the NVIDIA VDPAU driver. See the VDPAU Support appendix in the README for supported AV1 decoder profiles.
```
https://www.nvidia.com/Download/driverResults.aspx/184911/en-us

SupervisedThinking, 2022-01-15:
@elFarto means we don't need the CUDA libs? 

> The CUDA library (/usr/lib/libcuda.so.510.39.01) which provides runtime support for CUDA (high-performance computing on 
the GPU) applications.

http://us.download.nvidia.com/XFree86/Linux-x86_64/510.39.01/README/installedcomponents.html

elFarto, 2022-01-16:
> @elFarto means we don't need the CUDA libs?

No, we still need libcuda, but that's installed by the NVIDIA driver. We don't need (and have never needed) libcudart which is normally installed by the CUDA packages. However, what we used to need was the header files to compile against, you can only get those by installing CUDA. We've now switched to the nv-codec-headers package which provides a set of headers, without the need to install CUDA.

> Btw, looks like AV1 codec is coming in 510:

We use NVDECODE, and that has supported AV1 for a long time now. I have most of the code written to support it in the library, but no way of testing it.

smumriak, 2022-01-16:
@elFarto how can I help you in testing? I've got 2070 super

elFarto, 2022-01-16:
> @elFarto how can I help you in testing? I've got 2070 super

You'll need a 30-series card to test AV1, they're the only ones that do.