elFarto/nvidia-vaapi-driver #31: I have a 3060Ti and want to help support AV1.

johnjohnlin, 2022-01-16:
Thanks for the great work for supporting hardware decoding on Linux. I have a 3060Ti (support AV1) desktop running ArchLinux, I have libraries/headers new enough to compile this project. How do I help test or develop the AV1 support? (I can write C, but I have zero knowledge about VAAPI).

elFarto, 2022-01-16:
Excellent, that would be a great help. I'm going to make a couple of changes to let AV1 support be a compile time option. You'll need a nightly build of MPV to do the testing (it's far easier than using Firefox). Also, the ffmpeg source code is very useful to reference.

The library has partial support for AV1, but it likely won't successfully decode a frame correctly.

elFarto, 2022-01-16:
There's now a new build time option `av1`:
```
meson setup --wipe -Dav1=true build
```
This will allow the library to report that it supports AV1, if you're hardware supports it. I've also added a command to the samples/gensamples.sh script to create a simple test video.
If you could test the new version with that file, and report back how badly it's broken that would be great.

johnjohnlin, 2022-01-16:
I ran this command.
```bash
./test.sh ./samples/smptebars_av1.mp4
```
This is my current vainfo.
```text
vainfo: VA-API version: 1.13 (libva 2.13.0)
vainfo: Driver version: VA-API NVDEC driver
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : [VAEntrypointVLD](url)
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      <unknown profile>               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileAV1Profile0            : VAEntrypointVLD
```

This is the log file.
[av1-fd2b6f-log.tar.gz](https://github.com/elFarto/nvidia-vaapi-driver/files/7876841/av1-fd2b6f-log.tar.gz)

I tried to capture a continuous frames from around 1 sec, and here are the screenshots. (Can I use ffmpeg or something to dump the resulting frames?)
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

elFarto, 2022-01-16:
Hmm, that's not quite as terrible as I expected it to be. I-Frames seem to be ok, but P-Frames are having a problem. That would track with what's missing in the decoder.

Thanks for testing this. I'm not sure there's too much you'll be able to do. I've already done all the easy mapping bits for the codec. What's left is trying to supply NVDECODE information that isn't passed to us via the VA-API.
If you really want to get stuck in, the vaapi_av1.c and nvdec_av1.c files in ffmpeg show you how it maps to both APIs. From that you can attempt to work out how you'd map from VA-API to NVDECODE. The biggest issue at the moment I think, is: https://github.com/elFarto/nvidia-vaapi-driver/blob/fd2b6f07231da52041e17a7531e1d767a283416e/src/av1.c#L22

There's no information passed in that I can see to fill that field in.

elFarto, 2022-01-17:
I made a couple of tweaks to the AV1 code if you want to give it another go. Not sure how much of a difference it'll make.

elFarto, 2022-01-19:
I've committed some more fixes. There's still some values missing so it's not complete, but they're optional values so not all video use them.

Draghtnod, 2022-01-20:
Checked out the current master. Works fine for me:
[image]

mpv logpart from your code:
```
[65646-65657] ../src/vabackend.c:1610       __vaDriverInit_1_0 Initing NVIDIA VA-API Driver
[65646-65657] ../src/export-buf.c:  90          findCudaDisplay Found 5 EGL devices
[65646-65657] ../src/export-buf.c:  94          findCudaDisplay Got EGL_CUDA_DEVICE_NV value '1' from device 0
[65646-65657] ../src/export-buf.c:  94          findCudaDisplay Got EGL_CUDA_DEVICE_NV value '0' from device 1
[65646-65657] ../src/export-buf.c: 122             initExporter Got EGLDisplay from CUDA device
[65646-65657] ../src/export-buf.c:  56                reconnect Reconnecting to stream
[vo/gpu/vaapi-egl/vaapi] libva: va_openDriver() returns 0
[vo/gpu/vaapi-egl/vaapi] Initialized VAAPI: version 1.13
[65646-65657] ../src/vabackend.c: 928      nvQueryImageFormats In nvQueryImageFormats
[ffmpeg] AVHWDeviceContext: VAAPI driver: VA-API NVDEC driver.
[ffmpeg] AVHWDeviceContext: Driver not found in known nonstandard list, using standard behaviour.
[vo/gpu/vaapi-egl] Going to probe surface formats (may log bogus errors)...
[65646-65657] ../src/vabackend.c: 429           nvCreateConfig got profile: 0 with 0 attributes
[65646-65657] ../src/vabackend.c:1303 nvQuerySurfaceAttributes with 1 (nil) 0
[65646-65657] ../src/vabackend.c:1303 nvQuerySurfaceAttributes with 1 0x7f85d0fae300 5
[65646-65657] ../src/vabackend.c: 516        nvCreateSurfaces2 creating 1 surface(s) 128x128, format 1
[65646-65657] ../src/vabackend.c:1020            nvDeriveImage In nvDeriveImage
[65646-65657] ../src/export-buf.c:  35                    debug [EGL] eglDestroyImageKHR: EGL_BAD_PARAMETER error: In eglDestroyImage: Invalid EGLImage (0x2)

[65646-65657] ../src/export-buf.c:  35                    debug [EGL] eglDestroyImageKHR: EGL_BAD_PARAMETER error: In eglDestroyImage: Invalid EGLImage (0x7f85d0d02f50)

[65646-65657] ../src/export-buf.c:  35                    debug [EGL] eglDestroyImageKHR: EGL_BAD_PARAMETER error: In eglDestroyImage: Invalid EGLImage (0x7f85d0fac900)

[65646-65657] ../src/vabackend.c:1508    nvExportSurfaceHandle got 0x7f85d0526110
[65646-65657] ../src/export-buf.c: 302            exportCudaPtr eglExportDMABUFImageQueryMESA: 0x7f85d0fae511 NV12 (3231564e) planes:2 mods:300000000606014 300000000606013
[65646-65657] ../src/export-buf.c: 310            exportCudaPtr eglExportDMABUFImageMESA: 58 59 0 0, strides: 128 128 0 0, offsets: 0 0 0 0
[vo/gpu/opengl] GL: GL_INVALID_OPERATION error generated. <image> and <target> are incompatible
[vo/gpu/opengl] GL: GL_INVALID_OPERATION error generated. <image> and <target> are incompatible
[65646-65657] ../src/vabackend.c: 429           nvCreateConfig got profile: 1 with 0 attributes
[65646-65657] ../src/vabackend.c:1303 nvQuerySurfaceAttributes with 1 (nil) 0
[65646-65657] ../src/vabackend.c:1303 nvQuerySurfaceAttributes with 1 0x7f85d0ebc980 5
[65646-65657] ../src/vabackend.c: 429           nvCreateConfig got profile: 8 with 0 attributes
[65646-65657] ../src/vabackend.c:1303 nvQuerySurfaceAttributes with 3 (nil) 0
[65646-65657] ../src/vabackend.c:1303 nvQuerySurfaceAttributes with 3 0x7f85d0ebce00 5
[65646-65657] ../src/vabackend.c: 429           nvCreateConfig got profile: 9 with 0 attributes
[65646-65657] ../src/vabackend.c:1303 nvQuerySurfaceAttributes with 3 (nil) 0
[65646-65657] ../src/vabackend.c:1303 nvQuerySurfaceAttributes with 3 0x7f85d0ebcf80 5
[65646-65657] ../src/vabackend.c: 429           nvCreateConfig got profile: 10 with 0 attributes
[65646-65657] ../src/vabackend.c:1303 nvQuerySurfaceAttributes with 3 (nil) 0
[65646-65657] ../src/vabackend.c:1303 nvQuerySurfaceAttributes with 3 0x7f85d0ebcf80 5
[65646-65657] ../src/vabackend.c: 429           nvCreateConfig got profile: 5 with 0 attributes
[65646-65657] ../src/vabackend.c:1303 nvQuerySurfaceAttributes with 4 (nil) 0
[65646-65657] ../src/vabackend.c:1303 nvQuerySurfaceAttributes with 4 0x7f85d0ebd000 5
[65646-65657] ../src/vabackend.c: 429           nvCreateConfig got profile: 6 with 0 attributes
[65646-65657] ../src/vabackend.c:1303 nvQuerySurfaceAttributes with 4 (nil) 0
[65646-65657] ../src/vabackend.c:1303 nvQuerySurfaceAttributes with 4 0x7f85d0eb9400 5
[65646-65657] ../src/vabackend.c: 429           nvCreateConfig got profile: 7 with 0 attributes
[65646-65657] ../src/vabackend.c:1303 nvQuerySurfaceAttributes with 4 (nil) 0
[65646-65657] ../src/vabackend.c:1303 nvQuerySurfaceAttributes with 4 0x7f85d0eb9480 5
[65646-65657] ../src/vabackend.c: 429           nvCreateConfig got profile: 13 with 0 attributes
[65646-65657] ../src/vabackend.c:1303 nvQuerySurfaceAttributes with 4 (nil) 0
[65646-65657] ../src/vabackend.c:1303 nvQuerySurfaceAttributes with 4 0x7f85d0eb9500 5
[65646-65657] ../src/vabackend.c: 429           nvCreateConfig got profile: 17 with 0 attributes
[65646-65657] ../src/vabackend.c:1303 nvQuerySurfaceAttributes with 8 (nil) 0
[65646-65657] ../src/vabackend.c:1303 nvQuerySurfaceAttributes with 8 0x7f85d0eb9580 5
[65646-65657] ../src/vabackend.c: 429           nvCreateConfig got profile: 14 with 0 attributes
[65646-65657] ../src/vabackend.c:1303 nvQuerySurfaceAttributes with 9 (nil) 0
[65646-65657] ../src/vabackend.c:1303 nvQuerySurfaceAttributes with 9 0x7f85d0eb9480 5
[65646-65657] ../src/vabackend.c: 429           nvCreateConfig got profile: 19 with 0 attributes
[65646-65657] ../src/vabackend.c:1303 nvQuerySurfaceAttributes with 10 (nil) 0
[65646-65657] ../src/vabackend.c:1303 nvQuerySurfaceAttributes with 10 0x7f85d0ebaac0 5

```

elFarto, 2022-01-20:
Correct, only the 30-series can do AV-1, your 1080 and 2080 Ti can't.

johnjohnlin, 2022-01-21:
Still looks the same, I attach the log for this run, hopefullly this help.
[b9c0d35.tar.gz](https://github.com/elFarto/nvidia-vaapi-driver/files/7913994/b9c0d35.tar.gz)

elFarto, 2022-02-03:
Ok, I've had another go at trying to get this working. I've managed to implement the largest missing piece of logic, and it seems to agree with what ffmpeg produces. I also filled out a couple of other missing values.

GageBerz, 2022-02-03:
Tried to run the sample video didn't even bring up mpv to see the video before it crashed.
[av1_playback_log.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/7998634/av1_playback_log.txt)

elFarto, 2022-02-04:
That crash should be fixed now.

struq, 2022-02-04:
I tested fefc5ff with Firefox nightly 98 and it works fine for the sample video. 
[av1_sample.log](https://github.com/elFarto/nvidia-vaapi-driver/files/8002606/av1_sample.log)

For [AV1 videos on YouTube](https://www.youtube.com/watch?v=h3fUgOKFMNU), 4k and below also works fine, 8k has a lot of dropped frames. 
[av1_youtube_4k.log](https://github.com/elFarto/nvidia-vaapi-driver/files/8002614/av1_youtube_4k.log)
[av1_youtube_8k.log](https://github.com/elFarto/nvidia-vaapi-driver/files/8002615/av1_youtube_8k.log)

All above are on X11. On Wayland it falls back into software decoding (for VP9 and H264 as well). Is this expected? 
[wayland.log](https://github.com/elFarto/nvidia-vaapi-driver/files/8002638/wayland.log)

philipl, 2022-02-04:
Still doesn't work in mpv, although that's obviously not the main use-case.

elFarto, 2022-02-04:
I honestly didn't expect any of that to work. The 8k issue is likely issue #13, not much I can do about that.

Not sure what's going on with it on Wayland. @struq Can you get another log file with NVD_LOG=1 set, and raise it as a separate issue?

@philipl Any specific issue?

philipl, 2022-02-04:
It just looks the same as it always did - green screen with grey box in the top left. My firefox is refusing to use vaapi for av1, but it's still 96 so maybe that's a reason.

struq, 2022-02-05:
@philipl It seems Firefox AV1 hardware decoding is only enable after version 98 https://bugzilla.mozilla.org/show_bug.cgi?id=1745285

@elFarto Done. #52

struq, 2022-02-05:
Tried mpv [7849a36](https://github.com/mpv-player/mpv/commit/7849a36beb458985d1e8b15bd2706a6e084a5f33). It works just as well, for both the sample file and YouTube. 8k still has frame drops but better than Firefox, about the same as mpv+nvdec.

Logs:
[mpv_av1.log](https://github.com/elFarto/nvidia-vaapi-driver/files/8007470/mpv_av1.log)
[mpv_av1_youtube.log](https://github.com/elFarto/nvidia-vaapi-driver/files/8007471/mpv_av1_youtube.log)

philipl, 2022-02-05:
Ok wild. I can play back av1 streams now, but only if the resolution is larger than 1080p. Sound crazy, but it is what I'm seeing here.

If I take @struq 's example clip (https://www.youtube.com/watch?v=nFd-_Q7fpxE) and I play it at 1440p, 4k or 8k, it plays and looks fine. If I play at 1080p or 720p or 480p, I get the green screen and gray blocks.

elFarto, 2022-02-05:
Right, I've pushed another batch of changes. This gets the code pretty close to what ffmpeg calls NVDEC with. There's really just one (important) field left, `ref_pic_flag`. As far as I can see, this field cannot be derived from what VA-API passes in. It's possible that NVDEC doesn't care about it for AV1 as it's in a structure common to all decoding.

GageBerz, 2022-02-05:
Experiencing the same issue that @philipl is detailing. Only 1440p and higher works on Firefox w/ av1.

philipl, 2022-02-05:
> Right, I've pushed another batch of changes. This gets the code pretty close to what ffmpeg calls NVDEC with. There's really just one (important) field left, `ref_pic_flag`. As far as I can see, [...]

Thanks for continuing to plug away at this. The fact that it's working for any resolutions is a huge milestone! The lower resolution problem has to be something small.

elFarto, 2022-02-05:
The issue is, there's not much left to change. This last round of changes was done by logging out every field of the struct we pass to NVDEC each frame from both this library and ffmpeg (plus a few hacks to let me run it without a GPU with AV1 support) then diff'ing the outputs. It's really just that `ref_pic_flag` field that's different.

If you want to try, you can hack-up ffmpeg to pass the `refresh_frame_flags` field through one of the reserved fields in the VA-API structure, and stick it into the `ref_pic_flag` field and see if that fixes the issue. I'm doubtful that will actually fix anything though.

struq, 2022-02-05:
I didn't test 1080p and below last time and indeed, I'm having the same issue too, both Firefox and mpv. Haven't tested latest commit yet.
[mpv_av1_1080p.log](https://github.com/elFarto/nvidia-vaapi-driver/files/8008893/mpv_av1_1080p.log)
[image]

Given that higher resolution is working and the sample file smptebars_av1.mp4 also works despite being 640x360, could it be possible that the YouTube low resolution files are somehow not encoded correctly?

elFarto, 2022-02-05:
I'll run a comparison using a video from Youtube. It's possible they're doing something unexpected with the headers.

GageBerz, 2022-02-05:
Seems to work flawlessly for all video resolutions on Firefox, and no regression that I observed for mpv.

philipl, 2022-02-06:
Yep. The low resolutions play correctly now!

struq, 2022-02-06:
Yes it works now.

johnjohnlin, 2022-02-08:
It works now. Thanks for the hardwork~

elFarto, 2022-02-08:
Excellent, I'll close this issue then. Thanks for all the testing you guys have done, I couldn't have done it without you!

closed 2022-02-08 by elFarto