elFarto/nvidia-vaapi-driver #32: VP8 issue with GTX 970

imatimba, 2022-01-18:
When I try to open a gfycat embed on reddit I get served a webm/VP8 file and it looks like is not decoding correctly on my gpu. Also happens with Coursera videos but you can force mp4/h264 changing video quality.  
I assume you tested VP8 on your 1060 so maybe is something particular about maxwell 2nd gen.

Test webm: https://giant.gfycat.com/FatDefensiveFlee.webm

https://user-images.githubusercontent.com/7434335/149846312-339699ba-887c-4b94-a7cc-d019d91447a0.mp4

[vp8 bug log](https://github.com/elFarto/nvidia-vaapi-driver/files/7885288/vp8_bug.txt)

<details>
<summary>vainfo</summary>

```
libva info: VA-API version 1.13.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.13 (libva 2.13.0)
vainfo: Driver version: VA-API NVDEC driver
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      <unknown profile>               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointVLD
```

</details>

elFarto, 2022-01-18:
Fun fact, the 1060 doesn't support VP8. Not entirely why, but that means I was unable to test it. I had completely forgotten about this codec. It's a pretty simple codec, so I'm not sure how I messed that up.

elFarto, 2022-01-18:
I've push a potential fix to the repo. There was one field that wasn't passed to VA-API, so I've added code to reparse the frame to extract it.

imatimba, 2022-01-18:
Oh yeah, just checking the NVDEC table on wikipedia, only 1070 and up have VP8 (not including the 1080ti and Titans? lol).  
Anyways, same behavior with the new commit.  
New logs, separated just in case you don't care about MOZ_LOG.  
[nvd_log_vp8.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/7890489/nvd_log-1_vp8.txt)
[nvd_log_and_moz_log_vp8.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/7890493/nvd_log_and_moz_log_vp8.txt)

elFarto, 2022-01-18:
Ok, I think I've spotted the root cause of the issue and implemented an awful hack to try to make it work. Let me know if it's successful.

imatimba, 2022-01-18:
Almost there

https://user-images.githubusercontent.com/7434335/150019229-2912c40d-2844-4400-8f71-b0e9e580b968.mp4

[nvd_log-2_vp8.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/7892405/nvd_log-2_vp8.txt)

[nvd_log_and_moz_log-2_vp8.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/7892406/nvd_log_and_moz_log-2_vp8.txt)

elFarto, 2022-01-18:
Ok, new commit. You might have to play with line 49 in vp8.c. I think it's correct, but it might be off by 1. Or we might have to subtract the buffer offset, not sure.

imatimba, 2022-01-19:
So this is the best I could get

<details>
<summary>picParams->CodecSpecific.vp8.first_partition_size += buf->offset - 3;</summary>

[nvd_log_-3_vp8.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/7894216/nvd_log_-3_vp8.txt)
[nvd_log_and_moz_log_-3_vp8.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/7894219/nvd_log_and_moz_log_-3_vp8.txt)

https://user-images.githubusercontent.com/7434335/150067523-63f330bd-388c-4e7a-9e2c-8ef72d8f79c0.mp4

</details>

Subtracting the buffer offset (-=) and all other hardcoded offsets from - 20 to + 20 gave similar output to this second video except the cat wasn't visible past the halfway point, just garbage patterns/pixels moving.  

<details>
<summary>picParams->CodecSpecific.vp8.first_partition_size -= buf->offset - 3;</summary>

[nvd_log_-=-3_vp8.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/7894253/nvd_log_-.-3_vp8.txt)
[nvd_log_and_moz_log_-=-3_vp8.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/7894256/nvd_log_and_moz_log_-.-3_vp8.txt)

https://user-images.githubusercontent.com/7434335/150068243-05110346-60f1-49f4-945b-7d5f42cc28f7.mp4

</details>

elFarto, 2022-01-19:
> picParams->CodecSpecific.vp8.first_partition_size += buf->offset - 3;

This probably means that first_partition_size is already correct, as buf->offset will mostly be 3, except for keyframes where it's 10. Which is odd, because that video looks better than the one you posted previously.

Anyway, I've posted another attempted fix.

imatimba, 2022-01-19:
Jackpot! It's fixed.  
Thanks a lot for your work.

elFarto, 2022-01-19:
Oh wow, I wasn't expecting that one to completely fix it 😄, but I'll take it. Thanks for all the testing.

closed 2022-01-19 by elFarto