elFarto/nvidia-vaapi-driver #33: Green or black video

multiflexi, 2022-01-18:
OS: Ubuntu 21.10
GPU: NVIDIA GTX 960
Driver: 495.46

Firefox STDOUT: 
```
[Child 33388, MediaDecoderStateMachine #1] WARNING: Decoder=7f4a85e14d00 Decode error: NS_ERROR_OUT_OF_MEMORY (0x8007000e) - mozilla::MediaResult mozilla::FFmpegVideoDecoder<46465650>::CreateImageVAAPI(int64_t, int64_t, int64_t, MediaDataDecoder::DecodedData &): VAAPI dmabuf allocation error: file /build/firefox-zTOn5e/firefox-96.0+build2/dom/media/MediaDecoderStateMachine.cpp:3568
IPDL protocol Error: Received an invalid file descriptor
```

Debug output:
```
[33597-35455] ../src/export-buf.c: 307            exportCudaPtr Unable to export image
[33597-35458] ../src/vabackend.c: 853             nvEndPicture cuvid decoded successful to idx: 19
[33597-35458] ../src/vabackend.c:1497    nvExportSurfaceHandle got 0x7efd38be7920
[33597-35458] ../src/vabackend.c:1519    nvExportSurfaceHandle got address B075E0000 (2048) for surface 17 (picIdx: 15)
[33597-35458] ../src/export-buf.c: 302            exportCudaPtr eglExportDMABUFImageQueryMESA: 0x7efd38c240f1 NV12 (3231564e) planes:2 mods:3000000004fe014 3000000004fe014
[33597-35458] ../src/export-buf.c:  35                    debug [EGL] eglExportDMABUFImageMESA: EGL_BAD_MATCH error: In eglExportDMABUFImageMESA: EGLImage can't be exported to dma-buf

[33597-35455] ../src/export-buf.c: 307            exportCudaPtr Unable to export image
[33597-35459] ../src/vabackend.c:1612       __vaDriverInit_1_0 cuda error 'OS call failed or operation not supported on this OS' (304)

```

Both repeat many times.
Video is either black or green and it either plays - I can hear the sound, or it just won't do anything. I tried to purge and install NVIDIA drivers again but the result is the same.

elFarto, 2022-01-18:
Are you able to post a link to the video that is failing, or is it all videos? Also, do you happen to have an Intel GPU in the same machine?

multiflexi, 2022-01-18:
It is all videos. When I try mpv with --hwdec=vaapi, it just ignores it, nvdec works fine. I have Intel GPU, the CPU is i7-2600K, but I do not see it in lspci so I think it is disabled in BIOS.

elFarto, 2022-01-18:
I'm not sure I can do much to help with this issue at the moment. I have no idea why the eglExportDMABUFImageMESA function throws an EGL_BAD_MATCH error. The documentation for that extension doesn't mention anything about it, and NVIDIA haven't answered my query about.

It would be really nice if someone like @cubanismo or @aaronp24 could chime in and answer why that method is generating that error.

Darkspirit, 2022-01-18:
Which desktop environment are you using?
Does this problem also occur with https://nightly.mozilla.org?
Please open about:support in Nightly, click on "Copy text to clipboard" and paste/post it here.

cubanismo, 2022-01-18:
Yeah, the extension spec doesn't specify any error conditions, so we just made up our own error codes when we implemented it.

You'll get BAD_MATCH if pretty much anything fails:

1. You have an image in a format we can't export to a DMA-BUF
2. The image layout can't be represented with existing DRM format modifiers. Generally this is a side effect of (1)
3. The driver can't create a DMA-BUF FD for the specified image's memory. The most obvious cause of this would be nvidia-drm being loaded with the modeset=1 parameter, but it could be a number of other things as well.

To understand what is happening in this specific case, we'd need a lot more details about the environment. Generally, to rule out configuration issues, we request users run nvidia-bug-report.sh and send us the log file it generates. For info specific to this issue, I'd need to know exactly how the EGLImage in question was created, what format it used, etc.

elFarto, 2022-01-18:
Thanks for replying James. The code we use to export the images is [here](https://github.com/elFarto/nvidia-vaapi-driver/blob/b2a01a13992be03954c5931c6250d4ad00d4fae5/src/export-buf.c#L140).
The short version is we use CUDA to copy to output of NVDEC to a CUarray, and export that to EGL, then attempt to export that to a DMA-BUF. This is the only method I've found to work correctly.

I don't think it's 1 or 2 on your list, as we always export an R8 and RG88 image. And the modifiers are being returned by the eglExportDMABUFImageQueryMESA call.

For option 3, I assume you meant to say if the driver *wasn't* loaded with the modeset=1 parameter, which is a possibility, however we've had someone else with the issue (#11) and they did confirm they had it set.

multiflexi, 2022-01-18:
> Which desktop environment are you using? Does this problem also occur with https://nightly.mozilla.org? Please open about:support in Nightly, click on "Copy text to clipboard" and paste/post it here.

I use Gnome. The Nightly version behaves exactly the same. Grub is set with `nvidia-drm.modeset=1`.      
   
[nvidia-bug-report.log.gz](https://github.com/elFarto/nvidia-vaapi-driver/files/7891421/nvidia-bug-report.log.gz)
[firefox_support.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/7891438/firefox_support.txt)

multiflexi, 2022-01-18:
I tried some stuff from #14 and `sudo cat /sys/module/nvidia_drm/parameters/modeset` outputs `N`, but I do not understand why, I have set it in `/etc/default/grub`. 

`NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia mpv --hwdec=vaapi --opengl-debug --hwdec-codecs=all` ouputs:
```
[592727-592738] ../src/vabackend.c:1599       __vaDriverInit_1_0 Initing NVIDIA VA-API Driver
[592727-592738] ../src/export-buf.c:  90          findCudaDisplay Found 3 EGL devices
[592727-592738] ../src/export-buf.c:  94          findCudaDisplay Got EGL_CUDA_DEVICE_NV value '0' from device 0
[592727-592738] ../src/export-buf.c: 122             initExporter Got EGLDisplay from CUDA device
[592727-592738] ../src/export-buf.c:  56                reconnect Reconnecting to stream
[592727-592738] ../src/vabackend.c: 917      nvQueryImageFormats In nvQueryImageFormats
[592727-592738] ../src/vabackend.c: 429           nvCreateConfig got profile: 0 with 0 attributes
[592727-592738] ../src/vabackend.c:1292 nvQuerySurfaceAttributes with 1 (nil) 0
[592727-592738] ../src/vabackend.c:1292 nvQuerySurfaceAttributes with 1 0x7f492c7091c0 5
[592727-592738] ../src/vabackend.c: 516        nvCreateSurfaces2 creating 1 surface(s) 128x128, format 1
[592727-592738] ../src/vabackend.c:1009            nvDeriveImage In nvDeriveImage
[592727-592738] ../src/export-buf.c:  35                    debug [EGL] eglDestroyImageKHR: EGL_BAD_PARAMETER error: In eglDestroyImage: Invalid EGLImage (0x6972642f7665642f)

[592727-592738] ../src/export-buf.c:  35                    debug [EGL] eglDestroyImageKHR: EGL_BAD_PARAMETER error: In eglDestroyImage: Invalid EGLImage (0x447265646e65722f)

[592727-592738] ../src/export-buf.c:  35                    debug [EGL] eglDestroyImageKHR: EGL_BAD_PARAMETER error: In eglDestroyImage: Invalid EGLImage (0x383231)

[592727-592738] ../src/vabackend.c:1497    nvExportSurfaceHandle got 0x7f492c214af0
[592727-592738] ../src/export-buf.c: 302            exportCudaPtr eglExportDMABUFImageQueryMESA: 0x7f492c7fc601 NV12 (3231564e) planes:2 mods:3000000004fe014 3000000004fe013
[592727-592738] ../src/export-buf.c:  35                    debug [EGL] eglExportDMABUFImageMESA: EGL_BAD_MATCH error: In eglExportDMABUFImageMESA: EGLImage can't be exported to dma-buf

[592727-592738] ../src/export-buf.c: 307            exportCudaPtr Unable to export image
[592727-592738] ../src/export-buf.c:  35                    debug [EGL] eglCreateImageKHR: EGL_BAD_ACCESS error: In eglCreateImageKHR: requested buffer attributes are not supported

[592727-592738] ../src/vabackend.c: 429           nvCreateConfig got profile: 1 with 0 attributes
[592727-592738] ../src/vabackend.c:1292 nvQuerySurfaceAttributes with 1 (nil) 0
[592727-592738] ../src/vabackend.c:1292 nvQuerySurfaceAttributes with 1 0x7f492c70a3c0 5
[592727-592738] ../src/vabackend.c: 516        nvCreateSurfaces2 creating 1 surface(s) 128x128, format 1
[592727-592738] ../src/vabackend.c:1009            nvDeriveImage In nvDeriveImage
[592727-592738] ../src/export-buf.c:  35                    debug [EGL] eglDestroyImageKHR: EGL_BAD_PARAMETER error: In eglDestroyImage: Invalid EGLImage (0x31)
```

elFarto, 2022-01-18:
> I tried some stuff from #14 and `sudo cat /sys/module/nvidia_drm/parameters/modeset` outputs `N`, but I do not understand why, I have set it in `/etc/default/grub`.

Did you update grub afterwards? I don't use Ubuntu, but that's just the defaults, not the actual grub config file.

multiflexi, 2022-01-18:
OK, I see now. I did `sudo update-grub` AGAIN and now the modeset says `Y.` The decoding now works in X11. Great job guys, this makes browsing the web 1000times better. I am just a stupid user I guess.

closed 2022-01-18 by multiflexi

cubanismo, 2022-01-18:
Thanks for double-checking. We're working to flip the default for this parameter in the future, but for the time being, this is the number one cause of most reports that dma-buf/GBM/etc. stuff isn't working.

elFarto, 2022-01-18:
Is there any chance of getting an extra/different debug message if modeset isn't enabled? That would at least point anyone else getting those errors in the right direction.