elFarto/nvidia-vaapi-driver #34: Cannot export surface formats needed for 10/12bit video

philipl, 2022-01-18:
I just wanted to create a venue to discuss this problem with @cubanismo

https://github.com/elFarto/nvidia-vaapi-driver/blob/master/src/export-buf.c#L198

We currently haven't found a way to successfully export P010/P016 surfaces - there seems to be a gap in the supported formats. Do you have any thoughts on this?

elFarto, 2022-01-18:
To clarify, exporting them works. Importing them does not work, specifically the R16 and RG1616 DRM formats.

cubanismo, 2022-01-18:
From a quick look at the EGL driver code, the following formats should be fully supported in the 495 series drivers:

- DRM_FORMAT_ABGR8888
- DRM_FORMAT_XBGR8888
- DRM_FORMAT_R8
- DRM_FORMAT_RG88
- DRM_FORMAT_ABGR16161616F
- DRM_FORMAT_ARGB8888
- DRM_FORMAT_XRGB8888
- DRM_FORMAT_RGB565
- DRM_FORMAT_ARGB2101010

And the following should be supported as external textures only (E.g., GL_TEXTURE_EXTERNAL_OES). For more info on that that means, see my comment regarding pitch/linear formats on #15:

- DRM_FORMAT_UYVY
- DRM_FORMAT_YUV420
- DRM_FORMAT_NV12
- DRM_FORMAT_NV21
- DRM_FORMAT_NV16
- DRM_FORMAT_NV24
- DRM_FORMAT_P010
- DRM_FORMAT_P210
- DRM_FORMAT_P012

Hope that helps.

elFarto, 2022-01-18:
Most things that export the surfaces via the VA-API do so in 'separate' mode, so each plane gets an fd in the format of that plane (R8, etc...). The other method is 'composed' mode which would give you one fd with the format of the overall surface (NV12, etc...). The issue here is that you can't use (AFAIK) YUV surfaces in OpenGL, so no one ever asks for it in that format, so we have to export them separately. In Vulkan it looks possible.

Is there a technical limit against R16 and RG1616, or did they just add in the formats off the top of their head?

cubanismo, 2022-01-20:
Thanks for the feedback. I've created an internal issue to investigate the feasibility of supporting R16 and RG1616 DRM fourcc formats in EGL.

philipl, 2022-10-16:
The new 520.xx drivers have includes R16 and RG1616. I uncommented the profiles and format list entries and I got one hevc10 sample to play successfully (the famous jelly fish), but others that don't. 10bit vp9 is corrupted but recognisable as an actual image. Still, I think that's enough to prove the fundamental surface support is present, and the rest is just getting the driver to support 10/23bit properly now that it's possible.

elFarto, 2022-10-16:
I had been checking the release notes for that feature for a while, I guess it slipped through :). I don't have 520 installed yet, but I've made a couple of changes to the code on the more-bits branch that should fix the VP9 playback. Still need to look at AV1.

I'm slightly limited for testing here, as my card can only do HEVC 10 and 12-bit, not VP9 10-bit, but hopefully that should be enough.

elFarto, 2022-10-23:
Reopening this although not for the 10/12-bit formats exactly, but for the 444 chroma formats. I'm not entirely sure how we should be exporting those via DMA-BUF. It doesn't seem like mpv or Firefox have support for these formats yet so I can't deduce what they're expecting.

Logically it should either be an single RGB plane, or three separate planes for each of the YUV components (which I believe is the format NVDEC wants to output). However attempting to export three separate planes gives us the same error as the R16/RG32 formats did, so this might need another change on the NVIDIA side.

philipl, 2022-10-23:
Hmm. So, with "real" VAAPI drivers, the only examples we have are Intel which use weird packed formats (Y410 and Y416). I say weird - they are [recommended](https://learn.microsoft.com/en-us/windows/win32/medfound/10-bit-and-16-bit-yuv-video-formats#444-formats) by Microsoft but have no standard usage in the wider video processing landscape.

Attempting to wrangle the nvidia output into these formats would be expensive, so three plane export that preserves the layout from nvdec makes total sense. I'm slightly surprised it doesn't work as the individual plane formats are all normal. You'd need to use the right cuEGL colour format, which is [documented](https://docs.nvidia.com/cuda/cuda-driver-api/group__CUDA__TYPES.html#group__CUDA__TYPES_1g38337000e43e400e77ad36c7e197a9f2) but it also says:

```
Three channel formats are currently not supported for [CU_EGL_FRAME_TYPE_ARRAY](https://docs.nvidia.com/cuda/cuda-driver-api/group__CUDA__TYPES.html#group__CUDA__TYPES_1ggec3f4a4e1a5785b1aa0fcc209cd47c38c02019e2bb4a56d31db30925d567d101)
```

so that's probably the issue - assuming you wired all of this up. I think the dmabuf interop part would be fine once you had an EGL image.

philipl, 2022-10-23:
Internally, there is evidence from `strings` that nvidia support NV24 which is the 444 version of NV12. But nvdec doesn't provide a documented way to use that. The three plane 444 format is the only one they mention. Good times.

elFarto, 2022-10-23:
I did just retest this in my test app, but got the same results. Also tested it with 16-bit surfaces. I created three arrays with this:
```
    CUDA_ARRAY3D_DESCRIPTOR arrDesc = {
        .Width = width,
        .Height = height,
        .Depth = 0,
        .NumChannels = 1,
        .Flags = 0,
        .Format = CU_AD_FORMAT_UNSIGNED_INT8 //or _INT16
    };
```
Then used this to export them.
```
    CUeglFrame eglframe = {
        .frame = {
           .pArray = { arr, arr2, arr3 }
        },
        .width = width,
        .height = height,
        .depth = 1,
        .pitch = 0,
        .planeCount = 3,
        .numChannels = 1,
        .frameType = CU_EGL_FRAME_TYPE_ARRAY,
        .eglColorFormat = CU_EGL_COLOR_FORMAT_YUV444_PLANAR,
        .cuFormat = CU_AD_FORMAT_UNSIGNED_INT8 //or _INT16
    };
```
Which I believe is all correct.

philipl, 2022-10-23:
Yes. The problem is the color format. That comment says cuda doesn't actually support three plane formats.

elFarto, 2022-10-23:
That comment says *3 channels* are not supported, not planes. The MAX_PLANES define is set to 3, so I would be surprised if it isn't supported.

philipl, 2022-10-23:
Fair. I misread it. But still,given how the Y12 format doesn't export correctly, I fully expect that the format is the problem here. It's not handled by cuda/the driver.

elFarto, 2022-10-23:
I had complete forgot to check the list of supported DRM formats. I assume it's the YU24 or YV24 ones we're looking for.

philipl, 2022-10-23:
That would only be true if exporting all the planes together  - which would be equivalent to doing an export as NV12 or P010. But in the same way that we export those as R8 + RG88 and R16 + RG1616, we would expect to export three plane 444 as R8+R8+R8 and R16+R16+R16, so there shouldn't be any new requirements at the dmabuf export or import level. So that's why I think it's related to the cuda color format.

thesword53, 2022-12-07:
I forked this project here https://github.com/thesword53/nvidia-vaapi-driver/tree/direct-backend and I successfully added YUV444 support on `direct-backend`. The latest stable FFMPEG version doesn't support YUV444 with VAAPI: this commit [`d3f48e68b3236bc3fbf75cc489d53e9f397f5e0a`](https://github.com/FFmpeg/FFmpeg/commit/d3f48e68b3236bc3fbf75cc489d53e9f397f5e0a) is needed

elFarto, 2022-12-07:
That's awesome. However I have just rebased the direct-backend branch in preparation for merging it into master, so integrating those changes might be a bit of a pain. I didn't expect anyone to be crazy enough to branch off *that* branch 😀.

thesword53, 2022-12-08:
I opened a pull request synced with the `master` branch https://github.com/elFarto/nvidia-vaapi-driver/pull/134.

PR #120 (2022-10-20): vabackend: support 10/12 bit formats
The latest nvidia 520 drivers include support for the required DRM
formats to allow exporting of 10/12 bit surfaces.

The driver currently has incomplete support and various bits and
pieces commented out.

This change is an initial attempt to implement the required support for
the relevant profiles.

Things are simple where the profile and bit depth have a 1:1
correspondence, but AV1Profile0 includes both 8 and 10 bit content,
while VP9Profile2 includes 10 and 12 bit.

In these cases, we have to have a way to learn from the caller what
format is desired. In [... cut]
closed 2023-06-16 by elFarto