elFarto/nvidia-vaapi-driver #35: Firefox 96 + 4K VP9 + 2080 Ti + 495.44

Draghtnod, 2022-01-20:
The driver used to crash Firefox but since the last version I get some maybe useful log output. See below.

I'm using nvidia-vaapi-driver version 0.0.2.r10.gf5924d5-1 from AUR. The line that cought my eye is:

`[59935-60317] ../src/export-buf.c:  35                    debug [EGL] eglExportDMABUFImageMESA: EGL_BAD_MATCH error: In eglExportDMABUFImageMESA: EGLImage can't be exported to dma-buf`

- OS: Manjaro
- Display server: X11
- GPU: 2080 Ti primary with monitors, 1080 secondary without monitors
- Driver: 495.44
- Kernel: 5.15.12
- Firefox 96 (stable)
- Firefox config:about settings according to Readme.md
- Test video: https://www.youtube.com/watch?v=li-_BLtq58w

<details>
  <summary>Click here for full log output.</summary>

```
$LIBVA_DRIVER_NAME=nvidia MOZ_DISABLE_RDD_SANDBOX=1 NVD_LOG=1 MOZ_LOG="PlatformDecoderModule:5" firefox  
[GFX1-]: More than 1 GPU from same vendor detected via PCI, cannot deduce device
[Child 59639: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in Content process
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[RDD 59935: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in RDD process
[RDD 59935: Main Thread]: D/PlatformDecoderModule Agnostic decoder rejects requested type
[RDD 59935: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type
[RDD 59935: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type
[RDD 59935: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type
[RDD 59935: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59639: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[... 567 lines cut ...]
[RDD 59935: MediaPDecoder #2]: D/PlatformDecoderModule VideoFrameSurfaceVAAPI: VAAPI releasing dmabuf surface UID = 2
[RDD 59935: MediaPDecoder #2]: D/PlatformDecoderModule Reusing VA-API DMABufSurface UID = 2
[RDD 59935: MediaPDecoder #2]: D/PlatformDecoderModule VideoFrameSurfaceVAAPI: VAAPI locking dmabuf surface UID = 2
IPDL protocol Error: Received an invalid file descriptor
IPDL protocol Error: Received an invalid file descriptor
[Child 59639: MediaPDecoder #2]: D/PlatformDecoderModule AudioTrimmer[7f542909df60] ::PrepareTrimmers: sample[2321000,2341000] no trimming information
[Child 59639: MediaPDecoder #1]: D/PlatformDecoderModule AudioTrimmer[7f542909df60] ::HandleDecodedResult: sample[2321000,2341000] (decoded[2314500,2334500] no trimming needed
[vp9 @ 0x7f0058950000] Param buffer (type 0, 92 bytes) is 0x40.
[vp9 @ 0x7f0058950000] Slice 0 param buffer (316 bytes) is 0x41.
[vp9 @ 0x7f0058950000] Slice 0 data buffer (578829 bytes) is 0x42.
[vp9 @ 0x7f0058950000] Decode to surface 0x11.
[59935-60321] ../src/vabackend.c: 864             nvEndPicture cuvid decoded successful to idx: 15
[vp9 @ 0x7f0058950000] Param buffer (type 0, 92 bytes) is 0x43.
[vp9 @ 0x7f0058950000] Slice 0 param buffer (316 bytes) is 0x44.
[vp9 @ 0x7f0058950000] Slice 0 data buffer (69934 bytes) is 0x45.
[vp9 @ 0x7f0058950000] Decode to surface 0x12.
[59935-60321] ../src/vabackend.c: 864             nvEndPicture cuvid decoded successful to idx: 16
[RDD 59935: MediaPDecoder #3]: D/PlatformDecoderModule VideoFrameSurfaceVAAPI: VAAPI releasing dmabuf surface UID = 3
[RDD 59935: MediaPDecoder #3]: D/PlatformDecoderModule VA-API Got one frame output with pts=434000dts=434000 duration=33000 opaque=-9223372036854775808
[59935-60321] ../src/vabackend.c:1508    nvExportSurfaceHandle got 0x7f005896d150
[59935-60321] ../src/vabackend.c:1530    nvExportSurfaceHandle got address 205E00000 (4096) for surface 18 (picIdx: 16)
[59935-60321] ../src/export-buf.c: 302            exportCudaPtr eglExportDMABUFImageQueryMESA: 0x7f00402e79d1 NV12 (3231564e) planes:2 mods:300000000606014 300000000606014
[59935-60321] ../src/export-buf.c:  35                    debug [EGL] eglExportDMABUFImageMESA: EGL_BAD_MATCH error: In
```
</details>

elFarto, 2022-01-20:
Hi, Can you paste the output of `cat /sys/module/nvidia_drm/parameters/modeset` and `nvidia-smi`?

Draghtnod, 2022-01-20:
```
$sudo cat /sys/module/nvidia_drm/parameters/modeset
N
$nvidia-smi                                  
Thu Jan 20 09:39:26 2022       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 495.44       Driver Version: 495.44       CUDA Version: 11.5     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:17:00.0 Off |                  N/A |
|  0%   42C    P8     9W / 230W |      8MiB /  8119MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   1  NVIDIA GeForce ...  Off  | 00000000:65:00.0  On |                  N/A |
|  0%   38C    P0    56W / 250W |   2207MiB / 11011MiB |      2%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1623      G   /usr/lib/Xorg                       4MiB |
|    1   N/A  N/A      1623      G   /usr/lib/Xorg                    1167MiB |
|    1   N/A  N/A      1738      G   xfwm4                               4MiB |
|    1   N/A  N/A      2065      G   ...AAAAAAAAA= --shared-files       21MiB |
|    1   N/A  N/A      2159      G   ...AAAAAAAAA= --shared-files       20MiB |
|    1   N/A  N/A     59368      G   /usr/lib/firefox/firefox          138MiB |
+-----------------------------------------------------------------------------+
```

elFarto, 2022-01-20:
The nvidia_drm kernel module needs to be configured with the parameter nvidia-drm.modeset=1

elFarto, 2022-01-20:
You may also have some issues with it using the 1080 to decode video on, as it's hard coded to pick device 0 (I'm assuming the 8GiB card is the 1080, nvidia-smi chopped off the only important bit of the name). I'll have to change the logic to detect the first CUDA device that EGL returns and use that, rather than forcing device 0.

Draghtnod, 2022-01-20:
Sorry i overread the modeset part in the readme. It works now!
nvidia-smi outputs device 1 as my 2080Ti, but the video is decoded on it anyway. This is with playing 4K video:
```
$nvidia-smi
Thu Jan 20 09:57:26 2022       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 495.44       Driver Version: 495.44       CUDA Version: 11.5     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:17:00.0 Off |                  N/A |
| 33%   43C    P8     9W / 230W |      8MiB /  8119MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   1  NVIDIA GeForce ...  Off  | 00000000:65:00.0  On |                  N/A |
|  0%   40C    P0    61W / 250W |   2161MiB / 11011MiB |     18%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1606      G   /usr/lib/Xorg                       4MiB |
|    1   N/A  N/A      1606      G   /usr/lib/Xorg                    1134MiB |
|    1   N/A  N/A      1714      G   xfwm4                               4MiB |
|    1   N/A  N/A      2066      G   ...AAAAAAAAA= --shared-files       21MiB |
|    1   N/A  N/A      2117      G   ...AAAAAAAAA= --shared-files       22MiB |
|    1   N/A  N/A      3318      G   /usr/lib/firefox/firefox          176MiB |
|    1   N/A  N/A      3856      C   /usr/lib/firefox/firefox          731MiB |
+-----------------------------------------------------------------------------+
$nvidia-smi --query-gpu=index,gpu_name --format=csv
index, name
0, NVIDIA GeForce GTX 1080
1, NVIDIA GeForce RTX 2080 Ti
$nvidia-smi dmon -s um                                                       
# gpu    sm   mem   enc   dec    fb  bar1
# Idx     %     %     %     %    MB    MB
    0     0     0     0     0     8     5
    1    19     6     0    12  2162     9
    0     0     0     0     0     8     5
    1    18     6     0    16  2162     9
    0     0     0     0     0     8     5
    1    18     6     0    16  2162     9
    0     0     0     0     0     8     5
    1    18     6     0    16  2162     9
    0     0     0     0     0     8     5
    1    18     6     0    17  2162     9
    0     0     0     0     0     8     5
    1    18     6     0    16  2162     9
    0     0     0     0     0     8     5
    1    18     6     0    16  2162     9
    0     0     0     0     0     8     5
    1    19     6     0    16  2169     9
    0     0     0     0     0     8     5
    1    19     6     0    12  2168     9
```

elFarto, 2022-01-20:
Excellent. I think I see what's going on with the device ordering. I was assuming that the CUDA device index match the index nvidia-smi printed, but it seems not. Your 2080 has the CUDA id of 0, even though it's second in the list.

closed 2022-01-20 by elFarto