elFarto/nvidia-vaapi-driver #36: Video speeds up while rewinding

linxlan, 2022-01-20:
As you can see in the demo, I rewind video by clicking on the timeline bar. During this, the video plays at fast speed:
https://user-images.githubusercontent.com/29060641/150300827-a3177c49-45f8-48ba-a197-409eef309234.mp4

Also I recorded demo of the expected behavior (had to disable vaapi):
https://user-images.githubusercontent.com/29060641/150300866-02f01548-d61a-4d98-a05f-9fe6ffbccd51.mp4

Log:
[out.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/7903452/out.txt)

elFarto, 2022-01-20:
Yes, I've noticed this too and it's almost certainly a Firefox issue. This library doesn't have any control over the actual display of the frames. It's on my list of things to investigate, but I've been side tracked by other issues.

I'll leave this issue open to track the issue.

elFarto, 2022-01-26:
@linxlan I'm not sure how easy it is for you to compile from a branch, but I've push some new code to the export-rewrite branch that should hopefully help with the issue you're seeing.

linxlan, 2022-01-27:
Issue is fixed, thanks!

closed 2022-01-27 by linxlan