elFarto/nvidia-vaapi-driver #37: YT stream drops frames every minute

nightshiba, 2022-01-20:
Hello, thank you a lot for that fantastic driver. Video playback works just fine for me.
Still, I've encountered an annoying problem with the YT streams since there are noticeable drops of 3-50 frames every 30-50 seconds at any stream quality (240/480/1080/2160). It was reproducible with any live stream, even though I noticed that dynamic playback was more affected.
I've tried the `__GL_SYNC_TO_VBLANK=0` setting, but it doesn't seem to be fixing the problem.

Error log when the frame rate gets dropped:
```
8832-9138] ../src/vabackend.c:1525    nvExportSurfaceHandle got 0x7fbb4358c6a0
[8832-9138] ../src/vabackend.c:1547    nvExportSurfaceHandle got address 205828000 (1024) for surface 13 (picIdx: 11)
[8832-9138] ../src/export-buf.c: 302            exportCudaPtr eglExportDMABUFImageQueryMESA: 0x7fbb4a8f4711 NV12 (3231564e) planes:2 mods:300000000606014 300000000606014
[8832-9138] ../src/export-buf.c: 310            exportCudaPtr eglExportDMABUFImageMESA: 67 68 0 0, strides: 854 854 0 0, offsets: 458752 229376 0 0
[8832-9139] ../src/vabackend.c: 881             nvEndPicture cuvid decoded successful to idx: 17
[8832-9139] ../src/vabackend.c:1525    nvExportSurfaceHandle got 0x7fbb4358c6a0
[8832-9139] ../src/vabackend.c:1547    nvExportSurfaceHandle got address 205828000 (1024) for surface 19 (picIdx: 17)
[8832-9139] ../src/export-buf.c: 302            exportCudaPtr eglExportDMABUFImageQueryMESA: 0x7fbb4a8f48c1 NV12 (3231564e) planes:2 mods:300000000606014 300000000606014
[8832-9139] ../src/export-buf.c: 310            exportCudaPtr eglExportDMABUFImageMESA: 53 55 0 0, strides: 854 854 0 0, offsets: 0 0 0 0
[8832-9139] ../src/vabackend.c: 685         nvDestroyContext In nvDestroyContext
[8832-9139] ../src/vabackend.c:1615              nvTerminate In nvTerminate
[8832-9144] ../src/vabackend.c:1627       __vaDriverInit_1_0 Initing NVIDIA VA-API Driver
[8832-9144] ../src/export-buf.c:  90          findCudaDisplay Found 4 EGL devices
[8832-9144] ../src/export-buf.c:  94          findCudaDisplay Got EGL_CUDA_DEVICE_NV value '0' from device 0
[8832-9144] ../src/export-buf.c: 122             initExporter Got EGLDisplay from CUDA device
[8832-9144] ../src/export-buf.c:  56                reconnect Reconnecting to stream
[8832-9144] ../src/vabackend.c: 945      nvQueryImageFormats In nvQueryImageFormats
[8832-9142] ../src/vabackend.c: 446           nvCreateConfig got profile: 19 with 0 attributes
[8832-9142] ../src/vabackend.c:1320 nvQuerySurfaceAttributes with 10 (nil) 0
[8832-9142] ../src/vabackend.c:1320 nvQuerySurfaceAttributes with 10 0x7fbb48ac9800 5
[8832-9142] ../src/vabackend.c:1320 nvQuerySurfaceAttributes with 10 (nil) 32699
[8832-9142] ../src/vabackend.c:1320 nvQuerySurfaceAttributes with 10 0x7fbb48ac9200 5
[8832-9142] ../src/vabackend.c: 533        nvCreateSurfaces2 creating 1 surface(s) 854x480, format 1
[8832-9142] ../src/vabackend.c:1037            nvDeriveImage In nvDeriveImage
[8832-9142] ../src/vabackend.c: 533        nvCreateSurfaces2 creating 1 surface(s) 854x480, format 1
[8832-9142] ../src/vabackend.c: 533        nvCreateSurfaces2 creating 1 surface(s) 854x480, format 1
[8832-9142] ../src/vabackend.c: 533        nvCreateSurfaces2 creating 1 surface(s) 854x480, format 1
[8832-9142] ../src/vabackend.c: 533        nvCreateSurfaces2 creating 1 surface(s) 854x480, format 1
[8832-9142] ../src/vabackend.c: 533        nvCreateSurfaces2 creating 1 surface(s) 854x480, format 1
[8832-9142] ../src/vabackend.c: 533        nvCreateSurfaces2 creating 1 surface(s) 854x480, format 1
[8832-9142] ../src/vabackend.c: 533        nvCreateSurfaces2 creating 1 surface(s) 854x480, format 1
[8832-9142] ../src/vabackend.c: 533        nvCreateSurfaces2 creating 1 surface(s) 854x480, format 1
[8832-9142] ../src/vabackend.c: 533        nvCreateSurfaces2 creating 1 surface(s) 854x480, format 1
[8832-9142] ../src/vabackend.c: 533        nvCreateSurfaces2 creating 1 surface(s) 854x480, format 1
[8832-9142] ../src/vabackend.c: 533        nvCreateSurfaces2 creating 1 surface(s) 854x480, format 1
[8832-9142] ../src/vabackend.c: 533        nvCreateSurfaces2 creating 1 surface(s) 854x480, format 1
[8832-9142] ../src/vabackend.c: 533        nvCreateSurfaces2 creating 1 surface(s) 854x480, format 1
[8832-9142] ../src/vabackend.c: 533        nvCreateSurfaces2 creating 1 surface(s) 854x480, format 1
[8832-9142] ../src/vabackend.c: 533        nvCreateSurfaces2 creating 1 surface(s) 854x480, format 1
[8832-9142] ../src/vabackend.c: 533        nvCreateSurfaces2 creating 1 surface(s) 854x480, format 1
[8832-9142] ../src/vabackend.c: 533        nvCreateSurfaces2 creating 1 surface(s) 854x480, format 1
[8832-9142] ../src/vabackend.c: 446           nvCreateConfig got profile: 19 with 0 attributes
[8832-9142] ../src/vabackend.c:1320 nvQuerySurfaceAttributes with 10 (nil) 0
[8832-9142] ../src/vabackend.c:1320 nvQuerySurfaceAttributes with 10 0x7fbb48ac9c80 5
[8832-9142] ../src/vabackend.c:1320 nvQuerySurfaceAttributes with 10 (nil) 0
[8832-9142] ../src/vabackend.c:1320 nvQuerySurfaceAttributes with 10 0x7fbb48ac9b00 5
[8832-9142] ../src/vabackend.c: 607          nvCreateContext with 18 render targets, at 854x480
[8832-9142] ../src/vabackend.c: 881             nvEndPicture cuvid decoded successful to idx: 17
[8832-9142] ../src/vabackend.c: 881             nvEndPicture cuvid decoded successful to idx: 16
[8832-9142] ../src/vabackend.c:1525    nvExportSurfaceHandle got 0x7fbb4358c510
[8832-9142] ../src/vabackend.c:1547    nvExportSurfaceHandle got address 205828000 (1024) for surface 18 (picIdx: 16)
[8832-9142] ../src/export-buf.c: 302            exportCudaPtr eglExportDMABUFImageQueryMESA: 0x7fbb4a8f4ca1 NV12 (3231564e) planes:2 mods:300000000606014 300000000606014
[8832-9142] ../src/export-buf.c: 310            exportCudaPtr eglExportDMABUFImageMESA: 51 52 0 0, strides: 854 854 0 0, offsets: 0 0 0 0
[8832-9144] ../src/vabackend.c: 881             nvEndPicture cuvid decoded successful to idx: 17
```

mohad12211, 2022-01-21:
I'm also facing this issue

elFarto, 2022-01-21:
It seems as if Firefox is just terminating the decoding and restarting it. I see it doing that when you pause, but never during playback.

Could you both capture some logs of this happening with MOZ_LOG="Dmabuf:5, PlatformDecoderModule:5" set?

nightshiba, 2022-01-21:
Sure :)
[mozlog.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/7912089/mozlog.txt)
I was wrong about any stream since it happens only with "vp9 / mp4a" streams in my case.
Could you please see if it's reproducible with that specific stream? Link: https://www.youtube.com/watch?v=DiCuCCEzJ0k

nightshiba, 2022-01-21:
Also, I forgot to mention that the log was taken just by starting the browser with the pre-opened YT stream tab and closing it without pausing.

elFarto, 2022-01-21:
I can reproduce this issue locally, but I don't think it's our fault. Everything is decoding fine, then Firefox just decides to kill the decoder and create a new one. There's not much I can do to fix that.

@Darkspirit could this be a Firefox issue?

pc00per, 2022-07-06:
It either could be firefox issue or youtube ingeneral. Try checking for dropped frames on other platforms such as on chromium vaapi-vdpau, or windows, etc. If you see the same numbers, that would mean issue ain't in here.

ValZapod, 2022-08-04:
> or youtube ingeneral

You can download mp4 file and open it directly in chrome to check.

pc00per, 2022-08-05:
> > or youtube ingeneral
> 
> You can download mp4 file and open it directly in chrome to check.

That must be chrome then I guess

xstaticxgpx, 2023-03-15:
In Firefox FWIW I think it's related to this: https://bugzilla.mozilla.org/show_bug.cgi?id=1812290

I can see every minute that like you mentioned firefox is re-initializing the VA-API backend for some reason - and it forces that check to happen every time:

https://gist.github.com/xstaticxgpx/93ed7bc29d477aa500acd588ac3458a8

Note the following lines which are also seen when setting `media.ffmpeg.vaapi.force-surface-copy` to `true`

```
src/gbm_drv_common.c:56: GBM-DRV error (get_bytes_per_component): Unknown or not supported format: 538982482

src/gbm_drv_common.c:56: GBM-DRV error (get_bytes_per_component): Unknown or not supported format: 538982482
```

Side note, looks like Firefox 112 beta broke this completely and regressed from Firefox 111 which has that patch linked above (it seems)

elFarto, 2023-03-16:
I'm not sure what it's trying to do. 538982482 is R8, which should be supported by the NVIDIA drivers.

thesword53, 2024-02-04:
> I'm not sure what it's trying to do. 538982482 is R8, which should be supported by the NVIDIA drivers.

The NVIDIA GBM implementation doesn't support the `DRM_FORMAT_R8` format.
I previously tried to implement that driver with a GBM implementation to have a stable API unlike the direct backend but the lacks of `DRM_FORMAT_R8` support blocked me.

elFarto, 2024-02-04:
I believe the new 550 series of drivers supports R8 in GBM.

marceldev89, 2024-05-14:
Same issue here. It's kinda weird this only happens with livestreams (for me anyway) though.

closed 2026-09-26 by elFarto