elFarto/nvidia-vaapi-driver #38: Unable to start driver despite nvidia_drm.modeset=1 being set.

ProjectSynchro, 2022-01-21:
Hi, the title of the issue is the behaviour I'm seeing

I have nvidia_drm.modeset=1 set in GRUB, and the output of  `cat /sys/module/nvidia_drm/parameters/modeset` is 'Y'.

The offending commit is 34e7b7c170add540983e876b7979022e342c7f0a.

Let me know how I can help you debug this :smile: 

Vague log info:
```
libva info: VA-API version 1.13.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
[4991-5010] ../src/vabackend.c:1627       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver
[4991-5010] ../src/export-buf.c: 110          findCudaDisplay Found 3 EGL devices
[4991-5010] ../src/export-buf.c: 114          findCudaDisplay Got EGL_CUDA_DEVICE_NV value '0' from device 0
[4991-5010] ../src/export-buf.c: 122          findCudaDisplay Checking device file: /dev/dri/card0
[4991-5010] ../src/export-buf.c: 131          findCudaDisplay ERROR: This driver requires the nvidia_drm.modeset kernel module parameter set to 1
libva error: /usr/lib64/dri/nvidia_drv_video.so init failed
libva info: va_openDriver() returns 1
```

elFarto, 2022-01-21:
I've made a change to which ioctl we're using, and how we're checking it. I've also added some additional logging just in case. Can you retest the latest version?

Jeglerjeg, 2022-01-21:
I had the same issue on the previous commit, https://github.com/elFarto/nvidia-vaapi-driver/commit/b9c0d352260880677eb1b5b3198c058e9c8b97cc fixed it for me.

ProjectSynchro, 2022-01-22:
> I've made a change to which ioctl we're using, and how we're checking it. I've also added some additional logging just in case. Can you retest the latest version?

Seems that change has resolved the issue :smile: 

Closing as this is fixed.

closed 2022-01-22 by ProjectSynchro