elFarto/nvidia-vaapi-driver #39: Not working for VP9 but for AVC1

Munzu, 2022-01-21:
* Arch 
* Kernel 5.16.1-arch1-1
* X11
* GTX 970
* Proprietary driver version 495.46
* Firefox 96.0.2
* `cat /sys/module/nvidia_drm/parameters/modeset` gives `Y`
* I have configured the four Firefox options as described in the README

I've tested multiple videos, incl. these two:
* https://www.youtube.com/watch?v=li-_BLtq58w (VP9)
* https://www.youtube.com/watch?v=l6jaKKiLbZQ (AVC1)
* Twitch (AVC1)

This is the output I get when I start Firefox with the first video (VP9). When I play the video, nothing new is being output.
```
$ LIBVA_DRIVER_NAME=nvidia MOZ_DISABLE_RDD_SANDBOX=1 NVD_LOG=1 firefox
[10658-10714] ../nvidia-vaapi-driver-0.0.2/src/vabackend.c:1592       __vaDriverInit_1_0 Initing NVIDIA VA-API Driver
[10658-10714] ../nvidia-vaapi-driver-0.0.2/src/export-buf.c:  90          findCudaDisplay Found 3 EGL devices
[10658-10714] ../nvidia-vaapi-driver-0.0.2/src/export-buf.c:  94          findCudaDisplay Got EGL_CUDA_DEVICE_NV value '0' from device 0
[10658-10714] ../nvidia-vaapi-driver-0.0.2/src/export-buf.c: 122             initExporter Got EGLDisplay from CUDA device
[10658-10714] ../nvidia-vaapi-driver-0.0.2/src/export-buf.c:  56                reconnect Reconnecting to stream
[10658-10714] ../nvidia-vaapi-driver-0.0.2/src/vabackend.c: 916      nvQueryImageFormats In nvQueryImageFormats
[10658-10713] ../nvidia-vaapi-driver-0.0.2/src/vabackend.c:1580              nvTerminate In nvTerminate
```

To confirm that it's not working, I put `nvidia-smi dmon` in the console and the `dec` column was consistently at 0% while the video's playing.

```
$ nvidia-smi dmon
# gpu   pwr gtemp mtemp    sm   mem   enc   dec  mclk  pclk
# Idx     W     C     C     %     %     %     %   MHz   MHz
    0    47    51     -    14     6     0     0  3505   847
...
```

Most of the random YouTube videos I clicked on seemed to be VP9. 
I only accidentally came across the second video which happened to be AVC1. It *does* seem to be working for the second video and Twitch streams, which are both AVC1 according to the "Stats for nerds" when I right-click the YouTube video or when I press the cog icon on a Twitch stream and select "Advanced" > "Video stats". When I play any of these, I get a lot of output to the terminal and the `dec` column in `nvidia-smi dmon` is >0. 

Please let me know if I can provide more information.

elFarto, 2022-01-21:
The 970 doesn't support VP9 decoding. Can you provide the output of `vainfo`?

Munzu, 2022-01-21:
As a workaround, I installed the Firefox extension "h264ify" that forces YouTube videos to be AVC1 and that seems to do the trick.

commit 3044e421 (2022-01-21): Update README.md

Added information on determining supported codec, and link to NVIDIA's website

Fixes #39
closed 2022-01-21 by elFarto