elFarto/nvidia-vaapi-driver #40: VRAM Memory Leaks

Jeglerjeg, 2022-01-21:
After having had the same browser instance open for a long time, VRAM usage keeps increasing to a very large amount. Closing a video temporarily frees the memory, but opening a new one populates this memory again. Compared to a newly opened browser instance it's a very high amount.

While watching:
```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 510.39.01    Driver Version: 510.39.01    CUDA Version: 11.6     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:07:00.0  On |                  N/A |
| 28%   60C    P2    45W / 230W |   3758MiB /  8192MiB |     11%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      2334      G   /usr/lib/Xorg                     309MiB |
|    0   N/A  N/A      2403      G   /usr/bin/gnome-shell               73MiB |
|    0   N/A  N/A      2719      G   /usr/bin/nextcloud                 11MiB |
|    0   N/A  N/A      3220      G   /usr/lib/firefox/firefox          331MiB |
|    0   N/A  N/A      3889      C   /usr/lib/firefox/firefox         2692MiB |
|    0   N/A  N/A     13354      G   ...gl=desktop --shared-files       63MiB |
|    0   N/A  N/A    155428      G   ...e/Steam/ubuntu12_32/steam       40MiB |
|    0   N/A  N/A    155518      G   ./steamwebhelper                    2MiB |
|    0   N/A  N/A    155599      G   ...ef_log.txt --shared-files      125MiB |
|    0   N/A  N/A    278299      G   /usr/lib/firefox/firefox           97MiB |
+-----------------------------------------------------------------------------+
```

Closing the video:
```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 510.39.01    Driver Version: 510.39.01    CUDA Version: 11.6     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:07:00.0  On |                  N/A |
| 25%   58C    P0    44W / 230W |   3210MiB /  8192MiB |      1%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      2334      G   /usr/lib/Xorg                     317MiB |
|    0   N/A  N/A      2403      G   /usr/bin/gnome-shell               97MiB |
|    0   N/A  N/A      2719      G   /usr/bin/nextcloud                 11MiB |
|    0   N/A  N/A      3220      G   /usr/lib/firefox/firefox          315MiB |
|    0   N/A  N/A     13354      G   ...gl=desktop --shared-files       63MiB |
|    0   N/A  N/A    155428      G   ...e/Steam/ubuntu12_32/steam       40MiB |
|    0   N/A  N/A    155518      G   ./steamwebhelper                    2MiB |
|    0   N/A  N/A    155599      G   ...ef_log.txt --shared-files      125MiB |
|    0   N/A  N/A    278299      G   /usr/lib/firefox/firefox           97MiB |
+-----------------------------------------------------------------------------+
```
Opening the video again:
```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 510.39.01    Driver Version: 510.39.01    CUDA Version: 11.6     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
[... 31 lines cut ...]
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:07:00.0  On |                  N/A |
| 29%   59C    P2    44W / 230W |   1648MiB /  8192MiB |      6%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      2334      G   /usr/lib/Xorg                     329MiB |
|    0   N/A  N/A      2403      G   /usr/bin/gnome-shell              104MiB |
|    0   N/A  N/A      2719      G   /usr/bin/nextcloud                 11MiB |
|    0   N/A  N/A     13354      G   ...gl=desktop --shared-files       55MiB |
|    0   N/A  N/A    155428      G   ...e/Steam/ubuntu12_32/steam       28MiB |
|    0   N/A  N/A    155518      G   ./steamwebhelper                    2MiB |
|    0   N/A  N/A    155599      G   ...ef_log.txt --shared-files      125MiB |
|    0   N/A  N/A    290485      G   /usr/lib/firefox/firefox          270MiB |
|    0   N/A  N/A    290741      G   /usr/lib/firefox/firefox           97MiB |
|    0   N/A  N/A    290941      C   /usr/lib/firefox/firefox          612MiB |
+-----------------------------------------------------------------------------+
```

elFarto, 2022-01-21:
I can confirm this. Seems to be about 28MiB per 4k video, and 12MiB per 1080p video. That is suspiciously video frame sized, so something is dropping the frame, but I'm not sure what.

I tried adding some extra clean-up code, but that didn't seem to help even though it was freeing objects that weren't being free before. I destroy the CUDA context when we terminate, so there should be nothing left.

I'll need to try running apitrace on Firefox to see if there's a missing eglDestroyImage or something.

Darkspirit, 2022-01-21:
ffvpx (vp8,vp9) is affected by https://bugzilla.mozilla.org/show_bug.cgi?id=1743750.

elFarto, 2022-01-22:
@Darkspirit that bug indicates it's leaking system RAM, not VRAM?

elFarto, 2022-01-22:
I've committed some fixes to properly release the EGLImages that we create. That seems to have resolved the issue on my machine.

Jeglerjeg, 2022-01-22:
Seems resolved on my end as well.

closed 2022-01-22 by Jeglerjeg