elFarto/nvidia-vaapi-driver #41: Nvidia modeset warning everytime I open a video

NoXPhasma, 2022-01-22:
Everytime I open a video in Firefox, I get this dmesg error message:
```
[drm:drm_new_set_master] *ERROR* [nvidia-drm] [GPU ID 0x00000a00] Failed to grab modeset ownership
```
An explanation why this message pops up: https://forums.developer.nvidia.com/t/ubuntu-21-10-failed-to-grab-modeset-ownership-with-495-44/193867/37

This has probably introduced with https://github.com/elFarto/nvidia-vaapi-driver/commit/34e7b7c170add540983e876b7979022e342c7f0a or  https://github.com/elFarto/nvidia-vaapi-driver/commit/b9c0d352260880677eb1b5b3198c058e9c8b97cc

elFarto, 2022-01-23:
Thanks for reporting that, I hadn't noticed it on my machine. It's more a warning than an error in our case as we don't need to become a DRM master, just query some capabilities.

I've got a change queued up to use the render node if it exists, but we might not be able to retrieve that on the older 470 drivers. You can change `EGL_DRM_DEVICE_FILE_EXT` to `EGL_DRM_RENDER_NODE_FILE_EXT` if it's an urgent issue for you.

NoXPhasma, 2022-01-23:
Changing the Extension in the file fixes the log spam :+1:

guihkx, 2022-02-04:
Sorry to bump a closed issue, but having that message spammed in my journal every time I watch any video is a bit annoying, especially when seeing hundreds of them when I run `journalctl -p 3 -xb` to view actual  useful error messages that might have happened.

I think I understand it's a technical limitation of the R470 drivers not having `EGL_DRM_RENDER_NODE_FILE_EXT` support, but I was wondering if I could manually set the render node device using an environment variable, e.g. `NVD_RENDER_NODE`, and then your driver could try it first instead of querying  the master node.

Is this a reasonable request? 🤔

elFarto, 2022-02-04:
It's a reasonable request. I think we can just guess at the render node, I believe it's just 128 + the card index. I'll double check that and add it in.

elFarto, 2022-02-05:
I've committed a change that will either use the render node provided by the driver, or it will derive the render node from the master node and use that. This effectively stops it from ever using the master node, so should stop the error messages you're seeing.

guihkx, 2022-02-05:
Just tested it, works great and no more complaints in the journal.

Thank you so very much!

commit adb3574e (2022-01-23): Try to use the render node, instead of the main drm device node.

Opening a DRM master node attempts to make you the master of it, which in the case of the NVIDIA drivers, logs an error
if there's an existing master. We don't actually need to be a master, and the same caps are available through the render
node, so we should try to use that if it exists. It might not be available on older drivers, so we should still fallback
to the device node.

Fixes #41
closed 2022-02-05 by elFarto