elFarto/nvidia-vaapi-driver #44: Video might crash after returning to a suspended tab

guihkx, 2022-01-24:
Firefox seems to suspend background tabs playing videos after 30 seconds or so, so upon returning to such tab, the video sometimes crashes.

Log: https://gist.githubusercontent.com/guihkx/2cfa9d5131ba801e55b3d873add760fa/raw/c91e345428b47939eafb515eb278cbdaa271014a/log.txt

Video for repro:

https://www.youtube.com/watch?v=RoVKSAuX9r8

elFarto, 2022-01-24:
I think that might be an issue with the latest commit, try rolling back to commit: 71f2e0a for the moment. It seems to be trying initalising the driver 3 or 4 times and breaking it.

elFarto, 2022-01-24:
Spotted the issue. I haven't seen Firefox destroy the context, but not terminate the library before. This left all the surfaces we were counting still around, and these got added to the new surfaces. Trying to allocate 50 1080p surfaces was breaking it.

closed 2022-01-24 by elFarto

guihkx, 2022-01-24:
Thank you for this project, you're my hero. <3

commit 441c03d6 (2022-01-24): When creating a context, reset the surfaceCount to 0.

It's possible that the context will be destroyed, but the library will *not* be terminated. If another context was to be created afterwards,
it would attempt to allocate all the new surfaces and all the previous contexts surfaces.

Fixes #44