elFarto/nvidia-vaapi-driver #45: export-buf.c:152:81: error: ‘EGL_DRM_RENDER_NODE_FILE_EXT’ undeclared

freakedenough, 2022-01-26:
```
besitzer@MSI:~/nvidia-vaapi-driver$ sudo meson install -C build
ninja: Entering directory `/home/besitzer/nvidia-vaapi-driver/build'
[4/12] Compiling C object nvidia_drv_video.so.p/src_export-buf.c.o
FAILED: nvidia_drv_video.so.p/src_export-buf.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I/usr/local/include -I/usr/include/gstreamer-1.0 -I/usr/include/x86_64-linux-gnu -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -g -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_export-buf.c.o -MF nvidia_drv_video.so.p/src_export-buf.c.o.d -o nvidia_drv_video.so.p/src_export-buf.c.o -c ../src/export-buf.c
../src/export-buf.c: In function ‘findCudaDisplay’:
../src/export-buf.c:152:81: error: ‘EGL_DRM_RENDER_NODE_FILE_EXT’ undeclared (first use in this function); did you mean ‘EGL_DRM_DEVICE_FILE_EXT’?
  152 |                 const char* drmDeviceFile = eglQueryDeviceStringEXT(devices[i], EGL_DRM_RENDER_NODE_FILE_EXT);
      |                                                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                                 EGL_DRM_DEVICE_FILE_EXT
../src/export-buf.c:152:81: note: each undeclared identifier is reported only once for each function it appears in
[6/12] Compiling C object nvidia_drv_video.so.p/src_h264.c.o
../src/h264.c: In function ‘computeH264CudaCodec’:
../src/h264.c:115:5: warning: ‘VAProfileH264Baseline’ is deprecated [-Wdeprecated-declarations]
  115 |     if (profile == VAProfileH264Baseline || profile == VAProfileH264ConstrainedBaseline || profile == VAProfileH264Main || profile == VAProfileH264High) {
      |     ^~
In file included from /usr/include/va/va_backend.h:32,
                 from ../src/vabackend.h:5,
                 from ../src/h264.c:1:
/usr/include/va/va.h:503:5: note: declared here
  503 |     VAProfileH264Baseline va_deprecated_enum = 5,
      |     ^~~~~~~~~~~~~~~~~~~~~
../src/h264.c: At top level:
../src/h264.c:127:5: warning: ‘VAProfileH264Baseline’ is deprecated [-Wdeprecated-declarations]
  127 |     VAProfileH264Baseline,
      |     ^~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/va/va_backend.h:32,
                 from ../src/vabackend.h:5,
                 from ../src/h264.c:1:
/usr/include/va/va.h:503:5: note: declared here
  503 |     VAProfileH264Baseline va_deprecated_enum = 5,
      |     ^~~~~~~~~~~~~~~~~~~~~
[11/12] Compiling C object nvidia_drv_video.so.p/src_vabackend.c.o
../src/vabackend.c: In function ‘nvQueryConfigProfiles’:
../src/vabackend.c:303:9: warning: ‘VAProfileH264Baseline’ is deprecated [-Wdeprecated-declarations]
  303 |         profile_list[profiles++] = VAProfileH264Baseline;
      |         ^~~~~~~~~~~~
In file included from /usr/include/va/va_backend.h:32,
                 from ../src/vabackend.h:5,
                 from ../src/vabackend.c:3:
/usr/include/va/va.h:503:5: note: declared here
  503 |     VAProfileH264Baseline va_deprecated_enum = 5,
      |     ^~~~~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
Could not rebuild /home/besitzer/nvidia-vaapi-driver/build

```

elFarto, 2022-01-26:
This build error means that your EGL header files are a bit old. I've committed a fix to master to include a local definition of that, as it is a fairly recent extension.

elFarto, 2022-01-27:
I don't use Ubuntu, but a quick search suggests that it's the `libegl-dev` package that needs to be updated.

freakedenough, 2022-01-27:
libegl-dev is version 1.3.3-1 in (k)ubuntu 21.10, i see 1.4.0 is available, but only for ubuntu jelly 22.04; a yet unreleased version.
thats pity. i tried to manually install 1.4.0 with dpkg and ended up with a horrible dependency mess, so i rolled back.

freakedenough, 2022-01-27:
Sorry, my fault, I didnt understand your sentence in that way. I really worked now:

```
besitzer@MSI:~/nvidia-vaapi-driver$ meson setup build
The Meson build system
Version: 0.61.1
Source dir: /home/besitzer/nvidia-vaapi-driver
Build dir: /home/besitzer/nvidia-vaapi-driver/build
Build type: native build
Project name: nvidia-vaapi-driver
Project version: 0.1
C compiler for the host machine: cc (gcc 11.2.0 "cc (Ubuntu 11.2.0-7ubuntu2) 11.2.0")
C linker for the host machine: cc ld.bfd 2.37
Host machine cpu family: x86_64
Host machine cpu: x86_64
Library m found: YES
Library dl found: YES
Library EGL found: YES
Found pkg-config: /usr/bin/pkg-config (0.29.2)
Run-time dependency gstreamer-codecparsers-1.0 found: YES 1.18.5
Run-time dependency ffnvcodec found: YES 11.1.5.1
Build targets in project: 1

Found ninja-1.10.1 at /usr/bin/ninja
besitzer@MSI:~/nvidia-vaapi-driver$ sudo meson install -C build
[sudo] Passwort für besitzer: 
ninja: Entering directory `/home/besitzer/nvidia-vaapi-driver/build'
[6/12] Compiling C object nvidia_drv_video.so.p/src_h264.c.o
../src/h264.c: In function ‘computeH264CudaCodec’:
../src/h264.c:115:5: warning: ‘VAProfileH264Baseline’ is deprecated [-Wdeprecated-declarations]
  115 |     if (profile == VAProfileH264Baseline || profile == VAProfileH264ConstrainedBaseline || profile == VAProfileH264Main || profile == VAProfileH264High) {
      |     ^~
In file included from /usr/include/va/va_backend.h:32,
                 from ../src/vabackend.h:5,
                 from ../src/h264.c:1:
/usr/include/va/va.h:503:5: note: declared here
  503 |     VAProfileH264Baseline va_deprecated_enum = 5,
      |     ^~~~~~~~~~~~~~~~~~~~~
../src/h264.c: At top level:
../src/h264.c:127:5: warning: ‘VAProfileH264Baseline’ is deprecated [-Wdeprecated-declarations]
  127 |     VAProfileH264Baseline,
      |     ^~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/va/va_backend.h:32,
                 from ../src/vabackend.h:5,
                 from ../src/h264.c:1:
/usr/include/va/va.h:503:5: note: declared here
  503 |     VAProfileH264Baseline va_deprecated_enum = 5,
      |     ^~~~~~~~~~~~~~~~~~~~~
[11/12] Compiling C object nvidia_drv_video.so.p/src_vabackend.c.o
../src/vabackend.c: In function ‘nvQueryConfigProfiles’:
../src/vabackend.c:299:9: warning: ‘VAProfileH264Baseline’ is deprecated [-Wdeprecated-declarations]
  299 |         profile_list[profiles++] = VAProfileH264Baseline;
      |         ^~~~~~~~~~~~
In file included from /usr/include/va/va_backend.h:32,
                 from ../src/vabackend.h:5,
                 from ../src/vabackend.c:3:
/usr/include/va/va.h:503:5: note: declared here
  503 |     VAProfileH264Baseline va_deprecated_enum = 5,
      |     ^~~~~~~~~~~~~~~~~~~~~
[12/12] Linking target nvidia_drv_video.so
Installing nvidia_drv_video.so to /usr/lib/x86_64-linux-gnu/dri
besitzer@MSI:~/nvidia-vaapi-driver$ 
```

Thank you very much for your efforts and assistance!

freakedenough, 2022-01-27:
```
ibva info: VA-API version 1.12.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
[4800-4810] ../src/vabackend.c:1625       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7f19b9ce86a0
[4800-4810] ../src/export-buf.c: 132          findCudaDisplay Found 3 EGL devices
[4800-4810] ../src/export-buf.c: 136          findCudaDisplay Got EGL_CUDA_DEVICE_NV value '0' from device 0
[4800-4810] ../src/export-buf.c: 148          findCudaDisplay Checking device file: /dev/dri/renderD128
[4800-4810] ../src/export-buf.c: 155          findCudaDisplay Got DRM_IOCTL_GET_CAP ioctl response: 0 1
[4800-4810] ../src/export-buf.c: 195             initExporter Got EGLDisplay from CUDA device
[4800-4810] ../src/export-buf.c:  95                reconnect Reconnecting to stream
libva info: va_openDriver() returns 0
```

that's just great :-)