elFarto/nvidia-vaapi-driver #47: Kodi support

Tatsh, 2022-01-26:
Can Kodi be supported? Currently Kodi uses the (not working for me) libva-vdpau-driver package. What would it need?

elFarto, 2022-01-27:
It should work. I've had it working locally. You should just need to launch kodi with the LIBVA_DRIVER_NAME environment variable set.

Tatsh, 2022-01-27:
Seems to work. I had to clear out my `~/.kodi` to really make sure. Tested MPEG-4 and VP9 files.

Some log messages that might be of interest?

```
22-01-27 12:58:04.159 T:7669     INFO <general>: VAAPI::SupportsFilter vaDeriveImage not supported by driver - ffmpeg postprocessing and CPU-copy rendering will not be available
2022-01-27 12:58:04.159 T:7669    ERROR <general>: VAAPI/vpp vaCreateConfig error: the requested VAProfile is not supported (12)
```

The above message appear for every file.

Sometimes it seems the video player crashes (but Kodi does not) if I try to seek the video ahead.

Tatsh, 2022-01-27:
Downside is with hardware accelerated MPEG-2, I cannot do deinterlacing. It is disabled in the video options.

It would be nice if the deinterlacing from NVDEC would be made available, but really Kodi should just support NVDEC directly.

elFarto, 2022-01-27:
The MPEG-4 there almost certainly means MPEG-4 not H.264.

elFarto, 2022-01-28:
Fiddled  around with the interlace support. There were a couple of issues with MPEG-2 videos specifically, but it should be working now.

closed 2022-01-28 by elFarto

philipl, 2022-01-28:
I worry about this solution. Both bob and weave deinterlacing will double the frame rate by producing a whole frame from each field. The client needs to know this is happening or you're either going to drop the extra frames (and then be unable to deinterlace to field rate, which is essentially for highest quality deinterlacing) or you're going to get half speed video if it plays the extra frames at the original frame rate.

The most robust solution here is to implement the va_vpp entry points and say you can do deinterlacing, and use that to toggle the flag. Then the client can deinterlace via the vpp. mpv and kodi both know how to do this. Then they will expect the doubled frame rate and can display correctly.

elFarto, 2022-01-28:
Hmm, I was unaware of that. I do have it on adaptive mode at the moment, and that seems to be ok. The test videos I've got are at the correct frame rate with no interlacing artifacts.

VPP is on the list to implement as Chrome needs it, for some reason.

philipl, 2022-01-28:
When you say correct frame rate, do you mean the original fps (eg: 30) or the field rate (eg: 60)? We had this problem in ffmpeg's nvdec hwaccel because it cannot control the frame rate (as it's a plugin into the decoder) and so the only possible deinterlacing is at the original frame rate and dropping the extra frames. I wonder where the extra frames are going for you.

Tatsh, 2022-02-04:
@elFarto It seems this change makes deinterlacing enabled at all times. The option in Kodi is still disabled compared to VDPAU (`nvidia_drv_video.so` from libva-vdpau-driver) which gives multiple options. I don't think deinterlacing should be force enabled at all times.

elFarto, 2022-02-04:
Yes, It's not ideal to have it permanently enabled like this, but I currently don't have any other mechanism to let the client enable it. That would likely require implementing all the post-processing interfaces in VA-API, which is quite the undertaking and not something I'm prioritising at the moment.

commit 154abaa2 (2022-01-28): Fixed up interlaced video support.

Fix interlaced video support by correctly detecting when a frame is interlaced, even if both fields are in a single frame.
Changed the de-interlace mode to adaptive. We don't currently support selecting the de-interlace mode.

Fixes #47