elFarto/nvidia-vaapi-driver #48: High RAM usage and frame drops in Firefox

mustafademiral, 2022-01-28:
Firstly thank you for the driver. This is a very good project. 

- OS: Arch
- Kernel: 5.16.2
- Display Server: X11
- Firefox Ver: 96.0.2
- Driver: nvidia 495.46-10
- GPU: GTX1050Ti [Desktop]

Test video: https://youtu.be/BGGs3a_DbPw

1th minute of video: 1.64 GB RAM usage
12th minute of video: 4.06 GB RAM usage

Frame drops and video details: 
[image]
Nvidia-smi output in 12th minute of video:
[image]
Nvidia Log: https://gist.githubusercontent.com/mustafademiral/4a75679bf21d7034a7ce099a0b4fb533/raw/36d7e91bcde96378e949294ad86fe34e53be9d97/gistfile1.txt

  ```
  $vainfo
vainfo: VA-API version: 1.13 (libva 2.13.0)
vainfo: Driver version: VA-API NVDEC driver
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD

  ```
  ```
$cat /proc/cmdline
BOOT_IMAGE=/boot/vmlinuz-linux root=UUID=20a39f93-61e4-4ef4-a033-107c400af242 rw loglevel=3 quiet nvidia-drm.modeset=1 nvidia nvidia_modeset nvidia_uvm nvidia_drm

  ```

elFarto, 2022-01-28:
By increasing memory usage, are you referring to VRAM or system RAM? Your screenshot suggests that the video decode is using about 500MB of VRAM at the 12th minute (the firefox process with type 'c'), which is about what I would expect for a 4k video.

The dropped frames are an ongoing issue, however we're not entirely sure where they're happening. I done captures of firefox where the Youtube player says it's dropped frames, but firefox says it hasn't.

mustafademiral, 2022-01-28:
Sorry i referring to system RAM. When i continue watch video, RAM usage can reach up to 6-7GB. Frame drops increase with RAM usage. 
[image]

elFarto, 2022-01-28:
The memory usage is probably related to this Firefox issue: [1743750](https://bugzilla.mozilla.org/show_bug.cgi?id=1743750)

joebonrichie, 2022-01-28:
I´d also try setting `media.ffvpx.enabled` to false in about:config - video decoding should still work.

mustafademiral, 2022-01-28:
> The memory usage is probably related to this Firefox issue: [1743750](https://bugzilla.mozilla.org/show_bug.cgi?id=1743750). Looks like there's some nightly builds available with that fix in which you could test with.

Yes i tried in nightly build and problem fixed and frame drops lesser. 

> I´d also try setting `media.ffvpx.enabled` to false in about:config - video decoding should still work.

Also that fixed RAM usage in stable version. 

Thank you for your helps.

closed 2022-01-28 by mustafademiral