elFarto/nvidia-vaapi-driver #52: The decoder is not working with Firefox on Wayland

struq, 2022-02-05:
The VA-API decoder works fine with Firefox on X11 but on Wayland Firefox will fall back to software decoding.

Firefox version: 98.0a1 (2022-02-03) 
Nvidia driver version: 510.47.03
nvidia-vaapi-driver: fefc5ff

Environment variables and Firefox about:config
```
LIBVA_DRIVER_NAME="nvidia"
MOZ_ENABLE_WAYLAND="1"
MOZ_DISABLE_RDD_SANDBOX="1"
MOZ_LOG="PlatformDecoderModule:5"
NVD_LOG="1"
```
```
media.ffmpeg.vaapi.enabled=true
media.rdd-ffmpeg.enabled=true
media.av1.enabled=true
gfx.x11-egl.force-enabled=true
widget.dmabuf.force-enabled=true
```

Logs for sample video smptebars_h264.mp4:
[wayland_h264.log](https://github.com/elFarto/nvidia-vaapi-driver/files/8007405/wayland_h264.log)
[x11_h264.log](https://github.com/elFarto/nvidia-vaapi-driver/files/8007406/x11_h264.log)

```
[32522-32559] ../src/export-buf.c:  52                    debug [EGL] eglGetDisplay: X screen is not using the NVIDIA driver.
libEGL warning: DRI2: failed to create dri screen
[32522-32559] ../src/export-buf.c:  52                    debug [EGL] eglCreateImageKHR: _eglParseImageAttribList
[RDD 32522: MediaPDecoder #1]: D/PlatformDecoderModule   failed to create texture over DMABuf memory!
```

elFarto, 2022-02-05:
You wouldn't happen to have an Intel graphics card in your computer, would you? The log file has references to MESA being used, which means the GPU Firefox is using is not the NVIDIA one. Check the `about:support` page to see which GPU Firefox is using.

You can try running firefox with `__NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia` set to try and force it to use the NVIDIA GPU.

struq, 2022-02-05:
Nope, Nvidia is the only available GPU. (There's another AMD integrated GPU but disabled in BIOS, so not visible to the OS.)

Both the compositor and Firefox are definitely using the Nvidia.
`about:support`
```
Graphics
Features
Compositing	WebRender
Asynchronous Pan/Zoom	wheel input enabled; scrollbar drag enabled; keyboard enabled; autoscroll enabled; smooth pinch-zoom enabled
WebGL 1 Driver WSI Info	EGL_VENDOR: NVIDIA
EGL_VERSION: 1.5
EGL_EXTENSIONS: EGL_EXT_buffer_age EGL_EXT_client_sync EGL_EXT_create_context_robustness EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_MESA_image_dma_buf_export EGL_EXT_output_base EGL_EXT_output_drm EGL_EXT_protected_content EGL_EXT_stream_consumer_egloutput EGL_EXT_stream_acquire_mode EGL_EXT_sync_reuse EGL_IMG_context_priority EGL_KHR_config_attribs EGL_KHR_create_context_no_error EGL_KHR_context_flush_control EGL_KHR_create_context EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_partial_update EGL_KHR_swap_buffers_with_damage EGL_KHR_no_config_context EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_reusable_sync EGL_KHR_stream EGL_KHR_stream_attrib EGL_KHR_stream_consumer_gltexture EGL_KHR_stream_cross_process_fd EGL_KHR_stream_fifo EGL_KHR_stream_producer_eglsurface EGL_KHR_surfaceless_context EGL_KHR_wait_sync EGL_NV_nvrm_fence_sync EGL_NV_quadruple_buffer EGL_NV_stream_consumer_eglimage EGL_NV_stream_cross_display EGL_NV_stream_cross_object EGL_NV_stream_cross_process EGL_NV_stream_cross_system EGL_NV_stream_dma EGL_NV_stream_flush EGL_NV_stream_metadata EGL_NV_stream_remote EGL_NV_stream_reset EGL_NV_stream_socket EGL_NV_stream_socket_inet EGL_NV_stream_socket_unix EGL_NV_stream_sync EGL_NV_stream_fifo_next EGL_NV_stream_fifo_synchronous EGL_NV_stream_consumer_gltexture_yuv EGL_NV_stream_attrib EGL_NV_stream_origin EGL_NV_system_time EGL_NV_output_drm_flip_event EGL_NV_triple_buffer EGL_NV_robustness_video_memory_purge EGL_WL_bind_wayland_display EGL_WL_wayland_eglstream
EGL_EXTENSIONS(nullptr): EGL_EXT_platform_base EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_KHR_platform_x11 EGL_EXT_platform_x11 EGL_EXT_platform_device EGL_KHR_platform_wayland EGL_EXT_platform_wayland EGL_KHR_platform_gbm EGL_MESA_platform_gbm EGL_MESA_platform_xcb EGL_MESA_platform_surfaceless
IsWebglOutOfProcessEnabled: 0
WebGL 1 Driver Renderer	NVIDIA Corporation -- NVIDIA GeForce RTX 3060 Laptop GPU/PCIe/SSE2
WebGL 1 Driver Version	4.6.0 NVIDIA 510.47.03
WebGL 1 Driver Extensions	GL_AMD_multi_draw_indirect GL_AMD_seamless_cubemap_per_texture GL_AMD_vertex_shader_viewport_index GL_AMD_vertex_shader_layer GL_ARB_arrays_of_arrays GL_ARB_base_instance GL_ARB_bindless_texture GL_ARB_blend_func_extended GL_ARB_buffer_storage GL_ARB_clear_buffer_object GL_ARB_clear_texture GL_ARB_clip_control GL_ARB_color_buffer_float GL_ARB_compatibility GL_ARB_compressed_texture_pixel_storage GL_ARB_conservative_depth GL_ARB_compute_shader GL_ARB_compute_variable_group_size GL_ARB_conditional_render_inverted GL_ARB_copy_buffer GL_ARB_copy_image GL_ARB_cull_distance GL_ARB_debug_output GL_ARB_depth_buffer_float GL_ARB_depth_clamp GL_ARB_depth_texture GL_ARB_derivative_control GL_ARB_direct_state_access GL_ARB_draw_buffers GL_ARB_draw_buffers_blend GL_ARB_draw_indirect GL_ARB_draw_elements_base_vertex GL_ARB_draw_instanced GL_ARB_enhanced_layouts GL_ARB_ES2_compatibility GL_ARB_ES3_compatibility GL_ARB_ES3_1_compatibility GL_ARB_ES3_2_compatibility GL_ARB_explicit_attrib_location GL_ARB_explicit_uniform_location GL_ARB_fragment_coord_conventions GL_ARB_fragment_layer_viewport GL_ARB_fragment_program GL_ARB_fragment_program_shadow GL_ARB_fragment_shader GL_ARB_fragment_shader_interlock GL_ARB_framebuffer_no_attachments GL_ARB_framebuffer_object GL_ARB_framebuffer_sRGB GL_ARB_geometry_shader4 GL_ARB_get_program_binary GL_ARB_get_texture_sub_image GL_ARB_gl_spirv GL_ARB_gpu_shader5 GL_ARB_gpu_shader_fp64 GL_ARB_gpu_shader_int64 GL_ARB_half_float_pixel GL_ARB_half_float_vertex GL_ARB_imaging GL_ARB_indirect_parameters GL_ARB_instanced_arrays GL_ARB_internalformat_query GL_ARB_internalformat_query2 GL_ARB_invalidate_subdata GL_ARB_map_buffer_alignment GL_ARB_map_buffer_range GL_ARB_multi_bind GL_ARB_multi_draw_indirect GL_ARB_multisample GL_ARB_multitexture GL_ARB_occlusion_query GL_ARB_occlusion_query2 GL_ARB_parallel_shader_compile GL_ARB_pipeline_statistics_query GL_ARB_pixel_buffer_object GL_ARB_point_parameters GL_ARB_point_sprite GL_ARB_polygon_offset_clamp GL_ARB_post_depth_coverage GL_ARB_program_interface_query GL_ARB_provoking_vertex GL_ARB_query_buffer_object GL_ARB_robust_buffer_access_behavior GL_ARB_robustness GL_ARB_sample_locations GL_ARB_sample_shading GL_ARB_sampler_objects GL_ARB_seamless_cube_map GL_ARB_seamless_cubemap_per_texture GL_ARB_separate_shader_objects GL_ARB_shader_atomic_counter_ops GL_ARB_shader_atomic_counters GL_ARB_shader_ballot GL_ARB_shader_bit_encoding GL_ARB_shader_clock GL_ARB_shader_draw_parameters GL_ARB_shader_group_vote GL_ARB_shader_image_load_store GL_ARB_shader_image_size GL_ARB_shader_objects GL_ARB_shader_precision GL_ARB_shader_storage_buffer_object GL_ARB_shader_subroutine GL_ARB_shader_texture_image_samples GL_ARB_shader_texture_lod GL_ARB_shading_language_100 GL_ARB_shader_viewport_layer_ [... cut ...]
unavailable by env: OS not supported
WEBRENDER_DCOMP_PRESENT	
available by default
disabled by user: User disabled via pref
unavailable by env: Requires Windows 10 or later
unavailable by runtime: Requires ANGLE
WEBRENDER_SOFTWARE	
available by default
WEBGPU	
disabled by default: Disabled by default
X11_EGL	
available by default
force_enabled by user: Force enabled by pref
DMABUF	
available by default
force_enabled by user: Force enabled by pref
```

And can see firefox in `nvidia-smi` output.

elFarto, 2022-02-05:
That's odd. MESA is definitely being called from somewhere in Firefox, which means something is picking the wrong driver. 
 
Could you try disabling/removing the `gfx.x11-egl.force-enabled` pref in Firefox while under Wayland? I have a feeling that might be confusing matters, esp. if you have XWayland running aswell.

struq, 2022-02-06:
I tried setting `gfx.x11-egl.force-enabled=false` (the default setting) or even `gfx.x11-egl.force-enabled=true`. It doesn't seem to make a difference at all.
[wayland_h264_2.log](https://github.com/elFarto/nvidia-vaapi-driver/files/8009472/wayland_h264_2.log)
(I actually couldn't find any when searching for "mesa" in the logs, which part are you referring to?)

Something that might be relevant: I kill Xwayland process before starting Firefox, but when the video is opened, a new Xwayland process will be launched. This doesn't happen with `media.ffmpeg.vaapi.enabled=false` and doesn't happen with mpv/nvidia-vaapi.

elFarto, 2022-02-06:
It's MESA's EGL library that's printing out these lines (although two of them are via our logger):
```
[32522-32559] ../src/export-buf.c:  52                    debug [EGL] eglGetDisplay: X screen is not using the NVIDIA driver.
libEGL warning: DRI2: failed to create dri screen
[32522-32559] ../src/export-buf.c:  52                    debug [EGL] eglCreateImageKHR: _eglParseImageAttribList
```
Honestly not sure what's going on here. Firefox has no problem getting the correct EGLDisplay for normal rendering, but somehow trips up when importing a frame and chooses a different one.

Could you run `eglinfo` and paste the output here?

elFarto, 2022-02-06:
@Darkspirit, any idea what's going on here. Firefox seems to get a second, incorrect EGL context when importing an DMA-BUF image. My guess is it's a llvmpipe context sitting off the XWayland instance.

elFarto, 2022-02-07:
Interesting, I've just updated my Firefox source to the latest version and rebuilt, now I'm seeing the same issue. At least this means I can debug it now. Here's the stacktrace from the first debug line `X screen is not using the NVIDIA driver`:

[image]

elFarto, 2022-02-07:
@struq Can you test with the environment variable `EGL_PLATFORM=wayland` set?

struq, 2022-02-08:
Yes I can confirm that with `EGL_PLATFORM=wayland` hardware decoding is working.
[wayland_h264_EGL_PLATFORM.log](https://github.com/elFarto/nvidia-vaapi-driver/files/8020703/wayland_h264_EGL_PLATFORM.log)
[eglinfo.log](https://github.com/elFarto/nvidia-vaapi-driver/files/8020711/eglinfo.log)

threedope, 2022-02-08:
Same issue here.
Both the libva-vdpau-driver and this driver are returning null to vainfo on nvidia proprietary 510.47.03 and kernel 5.16.7-arch1-1.
I've found similar issue reports on other project pages made over the last 2 days, which aligns with when I updated to those versions.

elFarto, 2022-02-08:
I don't think it's an issue with the NVIDIA driver version. I did test by rolling back my driver to 495 and retesting, but I got the same issue.

Also, I'm not sure what 'returning null to vainfo', that shouldn't happen.

slagiewka, 2022-02-12:
I see that the OP uses `MOZ_DISABLE_RDD_SANDBOX=1`.

For what is worth, I used `media.rdd-process.enabled` set to `false` in FF 96 to get acceleration running on my AMD GPU. After updating to FF 97 I'd noticed that it no longer worked. Setting the RDD process flag back to default (enabled=true) made VA-API work again. Maybe trying to revert any RDD changes may work?

My environment uses only `MOZ_ENABLE_WAYLAND=1` and changed flags are:
```
media.ffmpeg.vaapi.enabled=true
media.ffvpx.enabled=false // My RX 550 doesn't support VPx anyway
```

So for me as the [Arch Wiki correctly states](https://wiki.archlinux.org/title/Firefox#Hardware_video_acceleration) only the `media.ffmpeg.vaapi.enabled` flag is required. That's of course for VA-API supported by the driver. 

Hope this helps!

elFarto, 2022-02-12:
@slagiewka This driver is different to the AMD one. We're trying to run CUDA inside the sandboxed process, and it does not play nicely with the sandbox, so we have to disable it.

I am working to remove that requirement, but CUDA was never designed to work in such a restricted environment.

rkoot, 2022-02-13:
I can confirm that the 'EGL_PLATFORM=wayland' environment variable does indeed enable hw video decode with firefox 97.0 on wayland. Maybe add this option to the  README.md so wayland users can enjoy your excellent work as well.

Darkspirit, 2022-05-16:
https://bugzilla.mozilla.org/show_bug.cgi?id=1769499

struq, 2022-06-30:
`EGL_PLATFORM=wayland` seems not necessary anymore on Firefox 103 beta.

closed 2022-06-30 by struq