elFarto/nvidia-vaapi-driver #53: error while install on Elementary OS

mustafademiral, 2022-02-05:
I've tried install different vaapi versions from github but it didn't work. 

-Driver Version: 495.46
-Kernel: 5.13.0-28

```
mustafa@elementarymsi:~/nvidia-vaapi-driver$ meson build
The Meson build system
Version: 0.61.1
Source dir: /home/mustafa/nvidia-vaapi-driver
Build dir: /home/mustafa/nvidia-vaapi-driver/build
Build type: native build
Project name: nvidia-vaapi-driver
Project version: 0.1
C compiler for the host machine: cc (gcc 9.3.0 "cc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0")
C linker for the host machine: cc ld.bfd 2.34
Host machine cpu family: x86_64
Host machine cpu: x86_64
Library m found: YES
Library dl found: YES
Library EGL found: YES
Found pkg-config: /usr/bin/pkg-config (0.29.1)
Run-time dependency gstreamer-codecparsers-1.0 found: YES 1.16.2
Run-time dependency ffnvcodec found: YES 11.1.5.1
Build targets in project: 1

Found ninja-1.10.2.git.kitware.jobserver-1 at /home/mustafa/.local/bin/ninja
```

```
mustafa@elementarymsi:~/nvidia-vaapi-driver$ meson install -C build
ninja: Entering directory `/home/mustafa/nvidia-vaapi-driver/build'
[2/12] Compiling C object nvidia_drv_video.so.p/src_h264.c.o
../src/h264.c: In function ‘computeH264CudaCodec’:
../src/h264.c:116:5: warning: ‘VAProfileH264Baseline’ is deprecated [-Wdeprecated-declarations]
  116 |     if (profile == VAProfileH264Baseline || profile == VAProfileH264ConstrainedBaseline || profile == VAProfileH264Main || profile == VAProfileH264High) {
      |     ^~
In file included from /usr/include/va/va_backend.h:32,
                 from ../src/vabackend.h:5,
                 from ../src/h264.c:1:
/usr/include/va/va.h:501:5: note: declared here
  501 |     VAProfileH264Baseline va_deprecated_enum = 5,
      |     ^~~~~~~~~~~~~~~~~~~~~
../src/h264.c: At top level:
../src/h264.c:128:5: warning: ‘VAProfileH264Baseline’ is deprecated [-Wdeprecated-declarations]
  128 |     VAProfileH264Baseline,
      |     ^~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/va/va_backend.h:32,
                 from ../src/vabackend.h:5,
                 from ../src/h264.c:1:
/usr/include/va/va.h:501:5: note: declared here
  501 |     VAProfileH264Baseline va_deprecated_enum = 5,
      |     ^~~~~~~~~~~~~~~~~~~~~
[10/12] Compiling C object nvidia_drv_video.so.p/src_vabackend.c.o
../src/vabackend.c: In function ‘nvQueryConfigProfiles’:
../src/vabackend.c:307:9: warning: ‘VAProfileH264Baseline’ is deprecated [-Wdeprecated-declarations]
  307 |         profile_list[profiles++] = VAProfileH264Baseline;
      |         ^~~~~~~~~~~~
In file included from /usr/include/va/va_backend.h:32,
                 from ../src/vabackend.h:5,
                 from ../src/vabackend.c:3:
/usr/include/va/va.h:501:5: note: declared here
  501 |     VAProfileH264Baseline va_deprecated_enum = 5,
      |     ^~~~~~~~~~~~~~~~~~~~~
[12/12] Linking target nvidia_drv_video.so
Installation failed due to insufficient permissions.
Attempting to use polkit to gain elevated privileges...
ninja: Entering directory `/home/mustafa/nvidia-vaapi-driver/build'
ninja: no work to do.
Installing nvidia_drv_video.so to /usr/lib/x86_64-linux-gnu/dri

```

elFarto, 2022-02-05:
There is no error there, it built and installed correctly. There are a lot of warnings (that are expected), and it failed to install the first time before gaining permissions and retrying successfully.

mustafademiral, 2022-02-05:
> There is no error there, it built and installed correctly. There are a lot of warnings (that are expected), and it failed to install the first time before gaining permissions and retrying successfully.

Thank you. I thought there was a error.

closed 2022-02-05 by mustafademiral