elFarto/nvidia-vaapi-driver #54: Unknown options: "av1"

Munzu, 2022-02-06:
I'm using the AUR git-version package of this driver and I'm getting the following error when I tried to update my system:

```
meson.build:1:0: ERROR: Unknown options: "av1"

A full log can be found at /home/tai/.cache/paru/clone/nvidia-vaapi-driver-git/src/nvidia-vaapi-driver/build/meson-logs/meson-log.txt
==> ERROR: A failure occurred in prepare().
    Aborting...
error: failed to build 'nvidia-vaapi-driver-git-0.0.2.r10.gf5924d5-1': 

error: packages failed to build: nvidia-vaapi-driver-git-0.0.2.r10.gf5924d5-1
```

Here's the log:
[meson-log.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/8010060/meson-log.txt)

- Arch
- Kernel 5.16.2-arch1-1
- GTX 970
- Proprietary driver version 495.46
- X11
- KDE Plasma
- Firefox 96.0.2
- `cat /sys/module/nvidia_drm/parameters/modeset` gives Y
- I have configured the Firefox options as described in the README, excluding the last one which doesn't apply to my GPU driver version

elFarto, 2022-02-06:
I've just removed that option, and enabled AV1 by default so it's no longer required to pass that option in.

Munzu, 2022-02-06:
Oh my bad, I should have looked first. Others have reported it on the AUR already. Sorry about that. Thanks, I'm closing this.

closed 2022-02-06 by Munzu

eli-schwartz, 2022-02-08:
It is possible to keep the option in meson_options.txt, but warn when users try to define it.

https://mesonbuild.com/Build-options.html#deprecated-options

elFarto, 2022-02-09:
Is there any particular reason to keep it? There are very few people using it, and the AUR has already been updated.

eli-schwartz, 2022-02-09:
The capability is there in case people want it. Largely because there are people that actually do want to do staged deprecation followed by removal across multiple releases.

So basically, if you have to ask whether it's worth it then you probably don't want it. That's fine too. :)