elFarto/nvidia-vaapi-driver #57: fatal error: va/va_backend.h: No such file or directory / Ubuntu 22.04

Slackd, 2022-02-14:
Hey,

I am not able to install this is Ubuntu 22.04. I have installed all the dependencies (at-least I hope I have). I am able to run the command `meson setup build` with the following output:

```
The Meson build system
Version: 0.61.1
Source dir: /home/budhaditya/nvidia-vaapi-driver-0.0.5
Build dir: /home/budhaditya/nvidia-vaapi-driver-0.0.5/build
Build type: native build
Project name: nvidia-vaapi-driver
Project version: 0.1
C compiler for the host machine: cc (gcc 11.2.0 "cc (Ubuntu 11.2.0-16ubuntu1) 11.2.0")
C linker for the host machine: cc ld.bfd 2.37.90.20220130
Host machine cpu family: x86_64
Host machine cpu: x86_64
Library m found: YES
Library dl found: YES
Library EGL found: YES
Found pkg-config: /usr/bin/pkg-config (0.29.2)
Run-time dependency gstreamer-codecparsers-1.0 found: YES 1.20.0
Run-time dependency ffnvcodec found: YES 11.1.5.1
Build targets in project: 1

Found ninja-1.10.1 at /usr/bin/ninja
```

However, whenever I run the command `meson install -C build`, I get a bunch of errors spewed at me. I have attached the log from the terminal (build.log), along with the meson build.log.

Please let me know how I can help diagnose and fix this issue? 

[build.log](https://github.com/elFarto/nvidia-vaapi-driver/files/8061307/build.log)
[meson-log.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/8061313/meson-log.txt)

elFarto, 2022-02-14:
You'll need libva-dev installed to compile it.

I seem to have missed that dependency in the meson file, surprised it's taken this long for someone to have this issue.

Slackd, 2022-02-15:
Hey,

You suggestion fixed the issue. Glad I'll be able to test this now. You may also want to update your README.md.

`sudo apt install libffmpeg-nvenc-dev libva-dev meson libwayland-egl-backend-dev libnvidia-egl-wayland-dev libegl-dev cmake libgstreamer-plugins-bad1.0-dev`

Closing this as resolved.

Thanks,
B.Saha.

closed 2022-02-15 by Slackd