elFarto/nvidia-vaapi-driver #58: Building the driver fails on ubuntu 20.04

erod20, 2022-02-19:
ninja: Entering directory `/home/lalo/apps/nvidia-vaapi-driver/build'
[3/12] Compiling C object nvidia_drv_video.so.p/src_vabackend.c.o
FAILED: nvidia_drv_video.so.p/src_vabackend.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I/usr/local/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -g -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_vabackend.c.o -MF nvidia_drv_video.so.p/src_vabackend.c.o.d -o nvidia_drv_video.so.p/src_vabackend.c.o -c ../src/vabackend.c
../src/vabackend.c: In function ‘nvQueryConfigProfiles’:
../src/vabackend.c:307:9: warning: ‘VAProfileH264Baseline’ is deprecated [-Wdeprecated-declarations]
  307 |         profile_list[profiles++] = VAProfileH264Baseline;
      |         ^~~~~~~~~~~~
In file included from /usr/include/va/va_backend.h:32,
                 from ../src/vabackend.h:5,
                 from ../src/vabackend.c:3:
/usr/include/va/va.h:370:5: note: declared here
  370 |     VAProfileH264Baseline va_deprecated_enum = 5,
      |     ^~~~~~~~~~~~~~~~~~~~~
../src/vabackend.c: In function ‘nvCreateImage’:
../src/vabackend.c:1006:63: error: ‘VA_FOURCC_P012’ undeclared (first use in this function); did you mean ‘VA_FOURCC_P010’?
 1006 |     if (format->fourcc == VA_FOURCC_P010 || format->fourcc == VA_FOURCC_P012) {
      |                                                               ^~~~~~~~~~~~~~
      |                                                               VA_FOURCC_P010
../src/vabackend.c:1006:63: note: each undeclared identifier is reported only once for each function it appears in
../src/vabackend.c: In function ‘nvGetImage’:
../src/vabackend.c:1122:67: error: ‘VA_FOURCC_P012’ undeclared (first use in this function); did you mean ‘VA_FOURCC_P010’?
 1122 |     if (imageObj->format == VA_FOURCC_P010 || imageObj->format == VA_FOURCC_P012) {
      |                                                                   ^~~~~~~~~~~~~~
      |                                                                   VA_FOURCC_P010
../src/vabackend.c: In function ‘nvQuerySurfaceAttributes’:
../src/vabackend.c:1365:48: error: ‘VA_FOURCC_P012’ undeclared (first use in this function); did you mean ‘VA_FOURCC_P010’?
 1365 |                 attrib_list[0].value.value.i = VA_FOURCC_P012;
      |                                                ^~~~~~~~~~~~~~
      |                                                VA_FOURCC_P010
[6/12] Compiling C object nvidia_drv_video.so.p/src_h264.c.o
../src/h264.c: In function ‘computeH264CudaCodec’:
../src/h264.c:116:5: warning: ‘VAProfileH264Baseline’ is deprecated [-Wdeprecated-declarations]
  116 |     if (profile == VAProfileH264Baseline || profile == VAProfileH264ConstrainedBaseline || profile == VAProfileH264Main || profile == VAProfileH264High) {
      |     ^~
In file included from /usr/include/va/va_backend.h:32,
                 from ../src/vabackend.h:5,
                 from ../src/h264.c:1:
/usr/include/va/va.h:370:5: note: declared here
  370 |     VAProfileH264Baseline va_deprecated_enum = 5,
      |     ^~~~~~~~~~~~~~~~~~~~~
../src/h264.c: At top level:
../src/h264.c:128:5: warning: ‘VAProfileH264Baseline’ is deprecated [-Wdeprecated-declarations]
  128 |     VAProfileH264Baseline,
      |     ^~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/va/va_backend.h:32,
                 from ../src/vabackend.h:5,
                 from ../src/h264.c:1:
/usr/include/va/va.h:370:5: note: declared here
  370 |     VAProfileH264Baseline va_deprecated_enum = 5,
      |     ^~~~~~~~~~~~~~~~~~~~~
[11/12] Compiling C object nvidia_drv_video.so.p/src_vp9.c.o
ninja: build stopped: subcommand failed.
Could not rebuild /home/lalo/apps/nvidia-vaapi-driver/build

elFarto, 2022-02-19:
You have a really old version of libva, is it possible for you to update it?

erod20, 2022-02-19:
I installed Ubuntu 21.10 and it builds there. So I'll use it until 22.04 comes out.

jeffshuai, 2022-02-24:
yes ,I have the some issue ,I fix it ,down the master  libva from intel github, buil and install to replace .

marioroy, 2022-03-08:
I switched to Ubuntu Linux recently. Then created a [repository](https://github.com/marioroy/nvidia-acceleration-on-ubuntu) for installing both NVDEC-enabled and VDPAU-enabled VA drivers. They co-exist without issues. Firefox uses nvdec and Google Chrome and derivatives use vdpau.
```bash
$ ls -l /usr/local/lib/dri
total 1216
-rwxr-xr-x 1 root root 514648 Mar  6 11:43 nvdec_drv_video.so
lrwxrwxrwx 1 root root     18 Mar  6 11:51 nvidia_drv_video.so -> vdpau_drv_video.so
-rwxr-xr-x 1 root root 726456 Mar  6 11:51 vdpau_drv_video.so
```

closed 2022-04-09 by elFarto