elFarto/nvidia-vaapi-driver #60: Video playback green with multiple NVIDIA GPU's, single GPU works

ASeagullFan, 2022-02-22:
First time posting an issue to a bug tracker. Thank you for your work. :)

Firefox 97.0.1
Manjaro 21.2.2
GPU Configuration: 1080Ti (Display) / 3060Ti (Compute)

Video plays green with multiple GPU's attached

[image]

After isolating the 3060Ti by binding it to vfio at boot, thus making the 1080Ti the only visible GPU

[image]

Logs

[NVD.log](https://github.com/elFarto/nvidia-vaapi-driver/files/8113712/NVD.log) [Multi GPU]

[NVD_single.log](https://github.com/elFarto/nvidia-vaapi-driver/files/8113715/NVD_single.log) [Single GPU]

Thanks.

elFarto, 2022-02-22:
It looks like it's picking the 'wrong' device. The driver is hardcoded to pick CUDA device 0, and it looks like the GPU Firefox is using is device 1. I'll see if I can add an environment variable to control that.

elFarto, 2022-02-22:
That was easier than I thought. I've push a fix to master that adds the `NVD_GPU` environment variable. Try setting that to `1` and see if that fixes your issue.

This isn't the proper solution for the issue. The proper solution would be to use the device that VA-API passes to us. However that's complicated by the fact we still need LIBVA_DRIVER_NAME set, which could lead to being initialised on a different GPU. Hopefully this hack should suffice for the short term.

ASeagullFan, 2022-02-22:
Thank you. That has fixed the issue. ❤️

elFarto, 2022-03-21:
Hey, I've finally got around to making the proper fix. Could you retest using the latest commit and see if it works without the NVD_GPU variable set?

elFarto, 2022-03-22:
Thanks for that, and the log file. That looks to be doing exactly what I wanted.

closed 2022-03-22 by elFarto