elFarto/nvidia-vaapi-driver #61: gst-inspect-1.0vaapi/ libva-utils  demo/mpv/firfox test fail

jeffshuai, 2022-02-24:
In MY linux deepin kernel  5.10  PC with RX3060([10de:2504]/sub[1462:1462]    0000:01:00.0 /driver:  nvidia)  
first   $export LIBVA_DRIVER_NAME=nvidia  LIBVA_DRIVERS_PATH=/usr/lib/x86_64-linux-gnu/dri  
test vainfo  sucess :  
test  gst-inspect-1.0vaapi/ libva-utils  demo/mpv/firfox test fail ;the detail log blow:

$ gst-inspect-1.0 vaapi
free(): double free detected in tcache 2
free(): double free detected in tcache 2　　　　　　　　　　　　　　
No such element or plugin 'vaapi'
------------------

elFarto, 2022-02-24:
Can you rerun those tests with the `NVD_LOG` environment variable set to `1`?

MPV specifically requires the latest version from master to work. GStreamer hasn't been tested.

jeffshuai, 2022-02-25:
ok , I'll fowllow it  ,build new MPV for test. and update the statuation later..

jeffshuai, 2022-03-01:
hi  @elFarto 
 update the log with  export NVD_LOG=1

$ NVD_LOG=1  mpv --hwdec=vaapi --vo=vaapi 1920x1080.mp4 
Playing: dde-introduction.mp4
 (+) Video --vid=1 (*) (h264 1920x1080 30.000fps)
 (+) Audio --aid=1 --alang=eng (*) (aac 2ch 48000Hz)
[7161-7168] ../src/vabackend.c:1658       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7fe8a820b9c0
[7161-7168] ../src/export-buf.c: 171          findCudaDisplay Found 3 EGL devices
[7161-7168] ../src/export-buf.c: 175          findCudaDisplay Got EGL_CUDA_DEVICE_NV value '0' from EGLDevice 0
[7161-7168] ../src/export-buf.c: 123    checkModesetParameter Unable to retrieve render node, deriving render node from master node
[7161-7168] ../src/export-buf.c: 135    checkModesetParameter Checking device file: /dev/dri/renderD128
[7161-7168] ../src/export-buf.c: 145    checkModesetParameter ERROR: This driver requires the nvidia_drm.modeset kernel module parameter set to 1
[vaapi] libva: /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so init failed
[vaapi] Failed to initialize VAAPI: operation failed                                                                     
Error opening/initializing the selected video_out (--vo) device.                                                         
Video: no video

elFarto, 2022-03-01:
Here's your issue:
```
[7161-7168] ../src/export-buf.c: 145 checkModesetParameter ERROR: This driver requires the nvidia_drm.modeset kernel module parameter set to 1
```
You need to set the modeset parameter to 1 to allow this driver to work. I'm not sure how that's done specifically on your distro.

elFarto, 2022-03-02:
Ok, you're attempting to use other applications with this driver, specifically ones that use vaPutSurface and some stream out features, which are not supported yet.
This driver is specifically designed to support Firefox's use of VA-API. It may gain those features later, but at the moment I'm focused on getting the basics of the driver working correctly.

jeffshuai, 2022-03-02:
I think  the driver isn't compatiable with  API vaputsurface .

elFarto, 2022-03-02:
--vo=vaapi won't work, --vo=gpu is the one to use.

As for vaPutSurface, I've already had a look at implementing it, but I ran into problems attempting to get the buffer into CUDA, so it's likely sitting in system RAM, not GPU RAM. I'll need to use another technique to get that to work, likely a temporary buffer somewhere.

jeffshuai, 2022-03-04:
I find install the  nvidia-driver-470 ;  sh hwinfo    show : Driver Status: "nvidia_drm is not 
 active", in my options , it's mean: nvidia_drm.modeset awlays equal 0 , set 1  in grrub file ,but don't work;   how about your nvidia-driver  vesion ?

rkoot, 2022-03-04:
> I find install the nvidia-driver-470 ; sh hwinfo show : Driver Status: "nvidia_drm is not active", in my options , it's mean: nvidia_drm.modeset awlays equal 0 , set 1 in grrub file ,but don't work; how about your nvidia-driver vesion ?

try this:
create a new file /etc/modprobe.d/nvidia.conf
put this line in nvidia.conf:

options nvidia-drm modeset=1

save nvidia.conf
now, run 'depmod -a'

recreate your initramdisk (depends on distro you use. for instance, opensuse uses mkinitrd)

reboot.

philipl, 2022-03-07:
The funny thing about vaPutSurface is that because it's X11 specific, you're better off using the vdpau vaapi wrapper because vdpau has a dedicated X11 rendering path that nvdec lacks.

jeffshuai, 2022-03-11:
thanks  full ! I got it and make it well.

closed 2022-03-11 by jeffshuai