elFarto/nvidia-vaapi-driver #62: Video artifacts in low quality H264 stream

guihkx, 2022-02-25:
I'm experiencing video artifacts in this livestream (it's probably region locked though):

https://g1.globo.com/mundo/ao-vivo/guerra-ucrania-russia-putin-invasao.ghtml

The stream quality is really low when you begin streaming, but it self-adjusts after a few seconds.

The artifacts are only experienced when the stream is on low quality:

https://user-images.githubusercontent.com/626206/155768129-2d20a810-4434-4bd7-b026-b825e2b57a9b.mp4

Log file: https://gist.githubusercontent.com/guihkx/ee155125acbdef8035888b0d693b922e/raw/f00ebfc93dbab808a6e966da473dbb35430ac315/nvidia-vaapi-log.txt

Command used to run Firefox: 

```
$ NVD_LOG=1 MOZ_DISABLE_RDD_SANDBOX=1 LIBVA_DRIVER_NAME=nvidia MOZ_LOG="PlatformDecoderModule:5,Dmabuf:5" firefox
```

- OS: Arch Linux
- GPU: GTX 660
- Driver: 470.103.01
- Firefox: 97.0.1
- nvidia-vaapi-driver: v0.0.5 (21d0803)

elFarto, 2022-02-25:
Known issue with Firefox, [1750389](https://bugzilla.mozilla.org/show_bug.cgi?id=1750389). Will be fixed in Firefox 98.

guihkx, 2022-02-25:
Welp, that was quick. 😄 

Thanks for the information!

closed 2022-02-25 by guihkx