elFarto/nvidia-vaapi-driver #63: Decode fails at 3x playback speed in Firefox

seonwoolee, 2022-03-01:
I'm not exactly sure if this is something that you can fix, but if I play back videos in Firefox with the playback speed set to 3x or faster, the decoding fails. Specifically, the video is frozen while the audio keeps going, and I get the following message in the logs

```
[Child 142955, MediaDecoderStateMachine #3] WARNING: Decoder=66189fc7ba00 Decode error: NS_ERROR_DOM_MEDIA_DECODE_ERR (0x806e0004) - virtual mozilla::MediaResult mozilla::FFmpegVideoDecoder<58>::DoDecode(mozilla::MediaRawData *, uint8_t *, int, bool *, MediaDataDecoder::DecodedData &): avcodec_send_packet error: -12: file /build/firefox/src/firefox-97.0.1/dom/media/MediaDecoderStateMachine.cpp:3568
[Child 142955: MediaPDecoder #10]: D/PlatformDecoderModule AudioTrimmer[6618a3aaa7a0] ::HandleDecodedResult: sample[97221000,97241000] (decoded[97214500,97234500] no trimming needed
```

I've installed this via the AUR on my Arch Linux system, with a RTX 3090.

elFarto, 2022-03-01:
This sounds exactly like #13, which is a Firefox issue. The root cause is that Firefox is attempting to decode and display each frame one-by-one, which is the worst way possible to drive the hardware. Hardware decoders have bad latency, but great through-put, you need to have multiple frames being decoded in parallel to get the best performance.

There's not too much I can do about it, it requires a Firefox change. I'm working one some multi-threading changes to help speed decoding up as much as possible, but that change is becoming incredibly complex, so no ETA on that.

seonwoolee, 2022-03-01:
Gotcha. Hopefully Firefox will address this.

closed 2022-03-01 by seonwoolee