elFarto/nvidia-vaapi-driver #64: Should av1 work in Firefox 98?

philipl, 2022-03-10:
I just upgraded to Firefox 98 and av1 is still playing using the software decoder. Did I misinterpret "98+" and it actually means newer than 98? or is something else missing. I don't see any new av1 related config items in Firefox.

Thanks.

elFarto, 2022-03-10:
It should work with Firefox 98. Can you post the output with `NVD_LOG=1 MOZ_LOG="PlatformDecoderModule:5,Dmabuf:5"` set?

philipl, 2022-03-10:
There's a lot of log here, so I'm going to guess what is significant. This output appears to show that the Firefox ffmpeg code doesn't understand that av1 is an acceleratable codec.

```
[RDD 226523: MediaSupervisor #1]: D/Dmabuf nsDMABufDevice::IsDMABufVAAPIEnabled: EGL 1 media_ffmpeg_vaapi_enabled 1 CanUseHardwareVideoDecoding 1 XRE_IsRDDProcess 1
[RDD 226523: MediaPDecoder #1]: D/PlatformDecoderModule FFMPEG: Initialising FFmpeg decoder
[RDD 226523: MediaPDecoder #1]: D/PlatformDecoderModule FFMPEG:   codec aac : AAC (Advanced Audio Coding)
[RDD 226523: MediaPDecoder #2]: D/PlatformDecoderModule FFMPEG: Initialising VA-API FFmpeg decoder
[RDD 226523: MediaPDecoder #1]: D/PlatformDecoderModule FFMPEG:   FFmpeg decoder init successful.
[RDD 226523: MediaPDecoder #2]: D/PlatformDecoderModule FFMPEG:   codec vp9 : Google VP9
libva info: VA-API version 1.14.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
[Child 226596: MediaPDecoder #3]: V/PlatformDecoderModule AudioTrimmer[7f041f07a7a0] ::PrepareTrimmers: sample[1225177687,1225200907] no trimming information
libva info: Found init function __vaDriverInit_1_0
[226523-227202] ../src/vabackend.c:1646       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7fdfef0e33e0
[Child 226596: MediaPDecoder #2]: V/PlatformDecoderModule AudioTrimmer[7f041f07a7a0] ::HandleDecodedResult: sample[1225177687,1225200907] (decoded[1225177687,1225200906] no trimming needed
[Child 226596: MediaPDecoder #2]: V/PlatformDecoderModule AudioTrimmer[7f041f07a7a0] ::PrepareTrimmers: sample[1225200907,1225224126] no trimming information
[Child 226596: MediaPDecoder #1]: V/PlatformDecoderModule AudioTrimmer[7f041f07a7a0] ::HandleDecodedResult: sample[1225200907,1225224126] (decoded[1225200907,1225224126] no trimming needed
[Child 226596: MediaPDecoder #1]: V/PlatformDecoderModule AudioTrimmer[7f041f07a7a0] ::PrepareTrimmers: sample[1225224126,1225247346] no trimming information
[Child 226596: MediaPDecoder #3]: V/PlatformDecoderModule AudioTrimmer[7f041f07a7a0] ::HandleDecodedResult: sample[1225224126,1225247346] (decoded[1225224126,1225247345] no trimming needed
[226523-227202] ../src/export-buf.c: 171          findCudaDisplay Found 4 EGL devices
[226523-227202] ../src/export-buf.c: 175          findCudaDisplay Got EGL_CUDA_DEVICE_NV value '0' from EGLDevice 0
[226523-227202] ../src/export-buf.c: 135    checkModesetParameter Checking device file: /dev/dri/renderD129
[226523-227202] ../src/export-buf.c: 209             initExporter Got EGLDisplay from CUDA device 0
[226523-227202] ../src/export-buf.c: 239             initExporter Driver doesn't support 16-bit surfaces
[226523-227202] ../src/export-buf.c:  95                reconnect Reconnecting to stream
libva info: va_openDriver() returns 0
[226523-227202] ../src/vabackend.c: 976      nvQueryImageFormats In nvQueryImageFormats
[AVHWDeviceContext @ 0x7fdfe3a652c0] Format 0x3231564e -> nv12.
[AVHWDeviceContext @ 0x7fdfe3a652c0] VAAPI driver: VA-API NVDEC driver.
[AVHWDeviceContext @ 0x7fdfe3a652c0] Driver not found in known nonstandard list, using standard behaviour.
[RDD 226523: MediaPDecoder #2]: D/PlatformDecoderModule FFMPEG: FFmpegVideoDecoder::GetAcceleratedFormats()
[RDD 226523: MediaPDecoder #2]: D/PlatformDecoderModule FFMPEG:   Profile H264Main:
[226523-227202] ../src/vabackend.c: 461           nvCreateConfig got profile: 6 with 0 attributes
[226523-227202] ../src/vabackend.c:1350 nvQuerySurfaceAttributes with 4 (nil) 0
[226523-227202] ../src/vabackend.c:1350 nvQuerySurfaceAttributes with 4 0x7fdfd8223100 5
[RDD 226523: MediaPDecoder #2]: D/PlatformDecoderModule FFMPEG:     codec h264 format nv12              3         12
[RDD 226523: MediaPDecoder #2]: D/PlatformDecoderModule FFMPEG:   Profile H264High:
[226523-227202] ../src/vabackend.c: 461           nvCreateConfig got profile: 7 with 0 attributes
[226523-227202] ../src/vabackend.c:1350 nvQuerySurfaceAttributes with 4 (nil) 0
[226523-227202] ../src/vabackend.c:1350 nvQuerySurfaceAttributes with 4 0x7fdfd8223100 5
[RDD 226523: MediaPDecoder #2]: D/PlatformDecoderModule FFMPEG:     codec h264 format nv12              3         12
[RDD 226523: MediaPDecoder #2]: D/PlatformDecoderModule FFMPEG:   Profile H264ConstrainedBaseline:
[226523-227202] ../src/vabackend.c: 461           nvCreateConfig got profile: 13 with 0 attributes
[226523-227202] ../src/vabackend.c:1350 nvQuerySurfaceAttributes with 4 (nil) 0
[226523-227202] ../src/vabackend.c:1350 nvQuerySurfaceAttributes with 4 0x7fdfd8223500 5
[RDD 226523: MediaPDecoder #2]: D/PlatformDecoderModule FFMPEG:     codec h264 format nv12              3         12
[RDD 226523: MediaPDecoder #2]: D/PlatformDecoderModule FFMPEG:   Profile VP8Version0_3:
[226523-227202] ../src/vabackend.c: 461           nvCreateConfig got profile: 14 with 0 attributes
[226523-227202] ../src/vabackend.c:1350 nvQuerySurfaceAttributes with 9 (nil) 0
[226523-227202] ../src/vabackend.c:1350 nvQuerySurfaceAttributes with 9 0x7fdfd8223a00 5
[RDD 226523: MediaPDecoder #2]: D/PlatformDecoderModule FFMPEG:     codec vp8 format nv12              3         12
[RDD 226523: MediaPDecoder #2]: D/PlatformDecoderModule FFMPEG:   Profile VP9Profile0:
[226523-227202] ../src/vabackend.c: 461           nvCreateConfig got profile: 19 with 0 attributes
[226523-227202] ../src/vabackend.c:1350 nvQuerySurfaceAttributes with 10 (nil) 0
[... cut ...]
Codec 0xe2 is not in the full list.
[RDD 226523: MediaPDecoder #2]: D/PlatformDecoderModule FFMPEG:       unknown_codec
[RDD 226523: MediaPDecoder #2]: D/PlatformDecoderModule FFMPEG:   VA-API FFmpeg init successful
[226523-227204] ../src/vabackend.c:1630              nvTerminate Terminating 0x7fdfef0e33e0
[226523-227204] ../src/export-buf.c:  57          releaseExporter Releasing exporter, 0 outstanding frames
[226523-227204] ../src/export-buf.c:  74          releaseExporter Done releasing frames
```

philipl, 2022-03-10:
This is the official ubuntu firefox package. Maybe they build it without MOZ_WAYLAND which seems really weird. Also weird that av1 hwaccel would be linked to that at build time.

https://hg.mozilla.org/integration/autoland/rev/b7ff43bbee96

philipl, 2022-03-10:
https://bugzilla.mozilla.org/show_bug.cgi?id=1745285

Seems like lack of MOZ_WAYLAND is most likely cause :-(

elFarto, 2022-03-13:
Ok, there's not much I can do about that so I'm going to close this issue.

closed 2022-03-13 by elFarto