elFarto/nvidia-vaapi-driver #65: not work with firefox on GTX 750

KnightMurloc, 2022-03-11:
on firefox 96.0.3 decoding fails with the following errors:
<details>
<summary>log</summary>

```
[101811-101825] ../src/vabackend.c:1642       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7f8683be3200
[101811-101825] ../src/export-buf.c: 171          findCudaDisplay Found 3 EGL devices
[101811-101825] ../src/export-buf.c: 175          findCudaDisplay Got EGL_CUDA_DEVICE_NV value '0' from EGLDevice 0
[101811-101825] ../src/export-buf.c: 135    checkModesetParameter Checking device file: /dev/dri/renderD128
[101811-101825] ../src/export-buf.c: 209             initExporter Got EGLDisplay from CUDA device 0
[101811-101825] ../src/export-buf.c: 239             initExporter Driver doesn't support 16-bit surfaces
[101811-101825] ../src/export-buf.c:  95                reconnect Reconnecting to stream
[101811-101825] ../src/vabackend.c: 972      nvQueryImageFormats In nvQueryImageFormats
[101811-101825] ../src/vabackend.c: 457           nvCreateConfig got profile: 7 with 0 attributes
[101811-101825] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 4 (nil) 0
[101811-101825] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 4 0x7f8663f93480 5
[101811-101825] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 4 (nil) 32646
[101811-101825] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 4 0x7f8663f93480 5
[101811-101825] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 640x368, format 1
[101811-101825] ../src/vabackend.c:1064            nvDeriveImage In nvDeriveImage
[101811-101825] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 640x368, format 1
[101811-101825] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 640x368, format 1
[101811-101825] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 640x368, format 1
[101811-101825] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 640x368, format 1
[101811-101825] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 640x368, format 1
[101811-101825] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 640x368, format 1
[101811-101825] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 640x368, format 1
[101811-101825] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 640x368, format 1
[101811-101825] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 640x368, format 1
[101811-101825] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 640x368, format 1
[101811-101825] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 640x368, format 1
[101811-101825] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 640x368, format 1
[101811-101825] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 640x368, format 1
[101811-101825] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 640x368, format 1
[101811-101825] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 640x368, format 1
[101811-101825] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 640x368, format 1
[101811-101825] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 640x368, format 1
[101811-101825] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 640x368, format 1
[101811-101825] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 640x368, format 1
[101811-101825] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 640x368, format 1
[101811-101825] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 640x368, format 1
[101811-101825] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 640x368, format 1
[101811-101825] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 640x368, format 1
[101811-101825] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 640x368, format 1
[101811-101825] ../src/vabackend.c: 457           nvCreateConfig got profile: 7 with 0 attributes
[101811-101825] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 4 (nil) 0
[101811-101825] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 4 0x7f8663f93600 5
[101811-101825] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 4 (nil) 0
[101811-101825] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 4 0x7f8663f93600 5
[101811-101825] ../src/vabackend.c: 635          nvCreateContext with 25 render targets, 25 surfaces, at 640x368
[101811-101825] ../src/vabackend.c: 909             nvEndPicture cuvid decoded successful to idx: 0
[101811-101834] ../src/vabackend.c: 909             nvEndPicture cuvid decoded successful to idx: 1
[101811-101825] ../src/vabackend.c: 909             nvEndPicture cuvid decoded successful to idx: 2
[101811-101825] ../src/vabackend.c:1568    nvExportSurfaceHandle got address 5017A8000 (1024) for surface 26 (picIdx: 0)
[101811-101825] ../src/export-buf.c: 300          allocateSurface presenting frame 640x368 0x7f8662ff6040 0x7f8662ff6160
[101811-101825] ../src/export-buf.c: 317          allocateSurface Adding frame from EGLStream: 0x7f86630fd2d1
[101811-101825] ../src/export-buf.c: 330          allocateSurface Acquired image from EGLStream: 0x7f86630fd2d1
[101811-101834] ../src/vabackend.c: 909             nvEndPicture cuvid decoded successful to idx: 3
[101811-101834] ../src/vabackend.c:1567    nvExportSurfaceHandle cuda error 'unspecified launch failure' (719)

[101839-101845] ../src/vabackend.c:1642       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7effeb2e33e0
[101839-101845] ../src/export-buf.c: 171          findCudaDisplay Found 3 EGL devices
[101839-101845] ../src/export-buf.c: 175          findCudaDisplay Got EGL_CUDA_DEVICE_NV value '0' from EGLDevice 0
[101839-101845] ../src/export-buf.c: 135    checkModesetParameter Checking device file: /dev/dri/renderD128
[101839-101845] ../src/export-buf.c: 209             initExporter Got EGLDisplay from CUDA device 0
[101839-101845] ../src/export-buf.c: 239             initExporter Driver doesn't support 16-bit surfaces
[101839-101845] ../src/export-buf.c:  95                reconnect Reconnecting to stream
[101839-101845] ../src/vabackend.c: 972      nvQueryImageFormats In nvQueryImageFormats
[101839-101845] ../src/vabackend.c: 457           nvCreateConfig got profile: 7 with 0 attributes
[101839-101845] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 4 (nil) 0
[... 57 lines cut ...]
[99515-99515] ../src/export-buf.c: 239             initExporter Driver doesn't support 16-bit surfaces
[99515-99515] ../src/export-buf.c:  95                reconnect Reconnecting to stream
vainfo: VA-API version: 1.13 (libva 2.13.0)
vainfo: Driver version: VA-API NVDEC driver
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      <unknown profile>               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
[99515-99515] ../src/vabackend.c:1626              nvTerminate Terminating 0x564b5002ae60
[99515-99515] ../src/export-buf.c:  57          releaseExporter Releasing exporter, 0 outstanding frames
[99515-99515] ../src/export-buf.c:  74          releaseExporter Done releasing frames
```

nvidia-vaapi-driver version: 0.0.5.r8.g21d0803-1 (from aur)
OS: Manjaro Linux
cat /sys/module/nvidia_drm/parameters/modeset return: Y
NVIDIA Driver Version: 495.46

can this be fixed somehow?

elFarto, 2022-03-11:
I'm not entirely sure what's going on there, you seem to be getting a CUDA_ERROR_LAUNCH_FAILED error when calling cuvidMapVideoFrame for the second time, the first time works fine. 

If you're able to build the driver from source you can try changing this line in vabackend.c:
```
vdci.ulNumOutputSurfaces = 1;
```
to
```
vdci.ulNumOutputSurfaces = 2;
```
and see if that helps.

elFarto, 2022-03-11:
You could also try the 510 driver series.

KnightMurloc, 2022-03-12:
> I'm not entirely sure what's going on there, you seem to be getting a CUDA_ERROR_LAUNCH_FAILED error when calling cuvidMapVideoFrame for the second time, the first time works fine.
>
> If you're able [...]

did not help the same error

> You could also try the 510 driver series.

Unfortunately, I don't have the opportunity to do this. my last attempt to install this version of the driver ended up with the system not booting. maybe another time. thanks.

elFarto, 2022-03-13:
I can't seem to reproduce this on my 760 with FF96. The error is potentially related to an invalid memory access of some sort.

I've changed some of the logging to show which frame it's attempting to map, I'm not sure if this will show anything useful, but  I'm afraid I don't have any ideas on how to fix it.

buddyjojo, 2022-06-05:
I appear to have a similar issue with a gtx 750ti too.
I guess this log contains your logging changes.
<details><summary>log</summary>
<p>

```
libva info: VA-API version 1.14.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
      6651.768712698 [79809-79981] ../src/vabackend.c:1775       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7fda35f223e0 31
      6651.768721419 [79809-79981] ../src/vabackend.c:1778       __vaDriverInit_1_0 Now have 0 (0 max) instances
      6651.768733170 [79809-79981] ../src/export-buf.c: 161       findGPUIndexFromFd Looking for GPU index: 0
      6651.769970187 [79809-79981] ../src/export-buf.c: 172       findGPUIndexFromFd Found 4 EGL devices
      6651.770038126 [79809-79981] ../src/export-buf.c: 181       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
      6651.770049931 [79809-79981] ../src/export-buf.c: 205       findGPUIndexFromFd Selecting EGLDevice 0
      6651.809585034 [79809-79981] ../src/export-buf.c: 267             initExporter Driver doesn't support 16-bit surfaces
      6651.809598170 [79809-79981] ../src/export-buf.c:  99                reconnect Reconnecting to stream
libva info: va_openDriver() returns 0
      6651.809690874 [79809-79981] ../src/vabackend.c:1119      nvQueryImageFormats In nvQueryImageFormats
      6651.842870596 [79809-79982] ../src/vabackend.c: 556           nvCreateConfig got profile: 7 with 0 attributes
      6651.842889444 [79809-79982] ../src/vabackend.c:1493 nvQuerySurfaceAttributes with 4 (nil) 0
      6651.842896350 [79809-79982] ../src/vabackend.c:1493 nvQuerySurfaceAttributes with 4 0x7fda0c40ee80 5
      6651.845405044 [79809-79982] ../src/vabackend.c:1493 nvQuerySurfaceAttributes with 4 (nil) 0
      6651.845419366 [79809-79982] ../src/vabackend.c:1493 nvQuerySurfaceAttributes with 4 0x7fda0c40e780 5
      6651.847929074 [79809-79982] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 1920x1088, format 1 (0x7fda0c4c6200)
      6651.847942178 [79809-79982] ../src/vabackend.c:1211            nvDeriveImage In nvDeriveImage
      6651.847950120 [79809-79982] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 1920x1088, format 1 (0x7fda0c4c67a0)
      6651.847955745 [79809-79982] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 1920x1088, format 1 (0x7fda0c4c6340)
      6651.847968779 [79809-79982] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 1920x1088, format 1 (0x7fda0c4c68e0)
      6651.847974280 [79809-79982] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 1920x1088, format 1 (0x7fda0c4c6de0)
      6651.847979567 [79809-79982] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 1920x1088, format 1 (0x7fda0c4c6840)
      6651.847984751 [79809-79982] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 1920x1088, format 1 (0x7fda0c4c6980)
      6651.847989759 [79809-79982] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 1920x1088, format 1 (0x7fda0c4c6c00)
      6651.847994866 [79809-79982] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 1920x1088, format 1 (0x7fda0c4c71a0)
      6651.847999946 [79809-79982] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 1920x1088, format 1 (0x7fda0c4c6e80)
      6651.848004996 [79809-79982] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 1920x1088, format 1 (0x7fda0c4c6f20)
      6651.848016344 [79809-79982] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 1920x1088, format 1 (0x7fda0c4c6520)
      6651.848021743 [79809-79982] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 1920x1088, format 1 (0x7fda0c4c6020)
      6651.848028772 [79809-79982] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 1920x1088, format 1 (0x7fda0c4c6fc0)
      6651.848034115 [79809-79982] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 1920x1088, format 1 (0x7fda0c4c7060)
      6651.848039255 [79809-79982] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 1920x1088, format 1 (0x7fda0c4c6a20)
      6651.848044827 [79809-79982] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 1920x1088, format 1 (0x7fda0c4c65c0)
      6651.848049900 [79809-79982] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 1920x1088, format 1 (0x7fda0c4c7100)
      6651.848055131 [79809-79982] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 1920x1088, format 1 (0x7fda0c4c7240)
      6651.848060273 [79809-79982] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 1920x1088, format 1 (0x7fda0c4c72e0)
      6651.848065260 [79809-79982] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 1920x1088, format 1 (0x7fda0c4c62a0)
      6651.848070441 [79809-79982] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 1920x1088, format 1 (0x7fda0c4c7380)
      6651.848077617 [79809-79982] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 1920x1088, format 1 (0x7fda0c4c63e0)
      6651.848086613 [79809-79982] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 1920x1088, format 1 (0x7fda0c4c6660)
[... cut ...]
      6651.897529054 [79809-79981] ../src/export-buf.c: 536           realiseSurface Detected NV12/NV21 NVIDIA driver bug, attempting to work around
      6651.897537534 [79809-79981] ../src/export-buf.c: 318      destroyBackingImage Destroying BackingImage: 0x7fda0c40e780
      6651.900475024 [79809-79981] ../src/export-buf.c: 436     allocateBackingImage cuda error 'unspecified launch failure' (719)
```
</p>
</details>

I'm currently on driver version 470.
Id rather not attempt to update to 510.

buddyjojo, 2022-06-07:
Ok so I updated to 510 and I still get the same error.

KnightMurloc, 2022-06-15:
tried on driver 515.48.07 same error. also tried on wayland, nothing changed.

elFarto, 2022-06-16:
Can you guys let me know which GTX 750 you have, there appears to be two different versions, one based off the GM107 and one off the GM206. I think this command should show it: `lspci -nn | grep 10de`

buddyjojo, 2022-06-16:
Mine is a GM107.
`01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GM107 [GeForce GTX 750 Ti] [10de:1380] (rev a2)`

KnightMurloc, 2022-06-17:
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GM107 [GeForce GTX 750] [10de:1381] (rev a2)

elFarto, 2022-06-19:
I've added in a bunch more checks to see if the error is actually occurring somewhere else, and being reported incorrectly. If you could retest with the latest git version of the driver and post the output, that would be great.

buddyjojo, 2022-06-19:
> I've added in a bunch more checks to see if the error is actually occurring somewhere else, and being reported incorrectly. If you could retest with the latest git version of the driver and post the output, that would be great.

Here: 

<details><summary>log</summary>
<p>

```
ATTENTION: default value of option mesa_glthread overridden by environment.
ATTENTION: default value of option mesa_glthread overridden by environment.
libva info: VA-API version 1.14.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
       150.472458273 [14583-14583] ../src/vabackend.c:1784       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7f720de24e20 31
       150.472465410 [14583-14583] ../src/vabackend.c:1787       __vaDriverInit_1_0 Now have 0 (0 max) instances
       150.478077429 [14583-14583] ../src/export-buf.c: 159       findGPUIndexFromFd Looking for DRM device index: 1
       150.481060325 [14583-14583] ../src/export-buf.c: 172       findGPUIndexFromFd Found 4 EGL devices
       150.481182369 [14583-14583] ../src/export-buf.c: 181       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
       150.481193040 [14583-14583] ../src/export-buf.c: 187       findGPUIndexFromFd Found drmDeviceIndex: 1
       150.481198974 [14583-14583] ../src/export-buf.c: 205       findGPUIndexFromFd Selecting EGLDevice 0
       150.512041918 [14583-14583] ../src/export-buf.c: 267             initExporter Driver doesn't support 16-bit surfaces
       150.512053199 [14583-14583] ../src/export-buf.c:  99                reconnect Reconnecting to stream
libva info: va_openDriver() returns 0
       150.534411756 [14583-14583] ../src/vabackend.c: 556           nvCreateConfig got profile: 6 with 0 attributes
       150.534424871 [14583-14583] ../src/vabackend.c: 556           nvCreateConfig got profile: 7 with 0 attributes
       150.534428691 [14583-14583] ../src/vabackend.c: 556           nvCreateConfig got profile: 13 with 0 attributes
       150.534432755 [14583-14583] ../src/vabackend.c:1761              nvTerminate Terminating 0x7f720de24e20
       150.534436149 [14583-14583] ../src/export-buf.c:  65          releaseExporter Releasing exporter, 0 outstanding frames
       150.535522467 [14583-14583] ../src/export-buf.c:  82          releaseExporter Done releasing frames
       150.552322432 [14583-14583] ../src/vabackend.c:1775              nvTerminate Now have 0 (0 max) instances
libva info: VA-API version 1.14.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
       154.551169449 [14961-15019] ../src/vabackend.c:1784       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7f57f5f22b60 31
       154.551177847 [14961-15019] ../src/vabackend.c:1787       __vaDriverInit_1_0 Now have 0 (0 max) instances
       154.558278080 [14961-15019] ../src/export-buf.c: 159       findGPUIndexFromFd Looking for DRM device index: 1
       154.561469571 [14961-15019] ../src/export-buf.c: 172       findGPUIndexFromFd Found 4 EGL devices
       154.561661193 [14961-15019] ../src/export-buf.c: 181       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
       154.561671823 [14961-15019] ../src/export-buf.c: 187       findGPUIndexFromFd Found drmDeviceIndex: 1
       154.561680229 [14961-15019] ../src/export-buf.c: 205       findGPUIndexFromFd Selecting EGLDevice 0
[2022-06-19T10:46:28Z ERROR mp4parse] Found 2 nul bytes in "\u{0}\u{0}"
[2022-06-19T10:46:28Z ERROR mp4parse] Found 2 nul bytes in "\u{0}\u{0}"
[2022-06-19T10:46:28Z ERROR mp4parse] Found 2 nul bytes in "\u{0}\u{0}"
[2022-06-19T10:46:28Z ERROR mp4parse] Found 2 nul bytes in "\u{0}\u{0}"
       154.607676959 [14961-15019] ../src/export-buf.c: 267             initExporter Driver doesn't support 16-bit surfaces
       154.607692020 [14961-15019] ../src/export-buf.c:  99                reconnect Reconnecting to stream
libva info: va_openDriver() returns 0
       154.607827749 [14961-15019] ../src/vabackend.c:1128      nvQueryImageFormats In nvQueryImageFormats
       154.637025964 [14961-15019] ../src/vabackend.c: 556           nvCreateConfig got profile: 6 with 0 attributes
       154.637041929 [14961-15019] ../src/vabackend.c:1502 nvQuerySurfaceAttributes with 4 (nil) 0
       154.637045833 [14961-15019] ../src/vabackend.c:1502 nvQuerySurfaceAttributes with 4 0x7f57dc018580 5
       154.639110890 [14961-15019] ../src/vabackend.c: 556           nvCreateConfig got profile: 7 with 0 attributes
       154.639139210 [14961-15019] ../src/vabackend.c:1502 nvQuerySurfaceAttributes with 4 (nil) 0
       154.639145438 [14961-15019] ../src/vabackend.c:1502 nvQuerySurfaceAttributes with 4 0x7f57dc018f00 5
       154.641413546 [14961-15019] ../src/vabackend.c: 556           nvCreateConfig got profile: 13 with 0 attributes
       154.641427837 [14961-15019] ../src/vabackend.c:1502 nvQuerySurfaceAttributes with 4 (nil) 0
       154.641434335 [14961-15019] ../src/vabackend.c:1502 nvQuerySurfaceAttributes with 4 0x7f57dc018580 5
       154.646903737 [14961-15030] ../src/vabackend.c:1761              nvTerminate Terminating 0x7f57f5f22b60
       154.646926882 [14961-15030] ../src/export-buf.c:  65          releaseExporter Releasing exporter, 0 outstanding frames
[... cut ...]
       164.171205972 [15162-15219] ../src/vabackend.c: 357          resolveSurfaces Reading from queue: 1 2
       164.171223027 [15162-15219] ../src/vabackend.c: 372          resolveSurfaces Mapping surface 1
       164.175912042 [15162-15219] ../src/vabackend.c: 374          resolveSurfaces Mapped surface 1 to 706780000 (2048)
       164.175928290 [15162-15219] ../src/export-buf.c: 527           realiseSurface No free surfaces found
       164.175937447 [15162-15219] ../src/export-buf.c: 438     allocateBackingImage cuda error 'unspecified launch failure' (719)
```
</p>
</details>

KnightMurloc, 2022-06-19:
<details open>
<summary>log</summary>

>       
      2270.143710080 [8779-8779] ../src/vabackend.c:1784       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7fb627b24a60 31
      2270.143728029 [8779-8779] ../src/vabackend.c:1787       __vaDriverInit_1_0 Now have 0 (0 max) instances
      2270.149457634 [8779-8779] ../src/export-buf.c: 159       findGPUIndexFromFd Looking for DRM device index: 0
      2270.151231130 [8779-8779] ../src/export-buf.c: 172       findGPUIndexFromFd Found 3 EGL devices
      2270.151351463 [8779-8779] ../src/export-buf.c: 181       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
      2270.151362707 [8779-8779] ../src/export-buf.c: 187       findGPUIndexFromFd Found drmDeviceIndex: 0
      2270.151368993 [8779-8779] ../src/export-buf.c: 205       findGPUIndexFromFd Selecting EGLDevice 0
      2270.185744460 [8779-8779] ../src/export-buf.c: 267             initExporter Driver doesn't support 16-bit surfaces
      2270.185761292 [8779-8779] ../src/export-buf.c:  99                reconnect Reconnecting to stream
      2270.211430679 [8779-8779] ../src/vabackend.c: 556           nvCreateConfig got profile: 6 with 0 attributes
      2270.211453516 [8779-8779] ../src/vabackend.c: 556           nvCreateConfig got profile: 7 with 0 attributes
      2270.211460151 [8779-8779] ../src/vabackend.c: 556           nvCreateConfig got profile: 13 with 0 attributes
      2270.211467274 [8779-8779] ../src/vabackend.c:1761              nvTerminate Terminating 0x7fb627b24a60
      2270.211473630 [8779-8779] ../src/export-buf.c:  65          releaseExporter Releasing exporter, 0 outstanding frames
      2270.212571990 [8779-8779] ../src/export-buf.c:  82          releaseExporter Done releasing frames
      2270.230345640 [8779-8779] ../src/vabackend.c:1775              nvTerminate Now have 0 (0 max) instances
      2275.799402217 [9093-9110] ../src/vabackend.c:1784       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7ff45ae2f7a0 31
      2275.799420235 [9093-9110] ../src/vabackend.c:1787       __vaDriverInit_1_0 Now have 0 (0 max) instances
      2275.807991461 [9093-9110] ../src/export-buf.c: 159       findGPUIndexFromFd Looking for DRM device index: 0
      2275.810061076 [9093-9110] ../src/export-buf.c: 172       findGPUIndexFromFd Found 3 EGL devices
      2275.810294060 [9093-9110] ../src/export-buf.c: 181       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
      2275.810307609 [9093-9110] ../src/export-buf.c: 187       findGPUIndexFromFd Found drmDeviceIndex: 0
      2275.810331215 [9093-9110] ../src/export-buf.c: 205       findGPUIndexFromFd Selecting EGLDevice 0
      2275.854287242 [9093-9110] ../src/export-buf.c: 267             initExporter Driver doesn't support 16-bit surfaces
      2275.854889465 [9093-9110] ../src/export-buf.c:  99                reconnect Reconnecting to stream
      2275.855134741 [9093-9110] ../src/vabackend.c:1128      nvQueryImageFormats In nvQueryImageFormats
      2275.892541667 [9093-9110] ../src/vabackend.c: 556           nvCreateConfig got profile: 6 with 0 attributes
      2275.892567088 [9093-9110] ../src/vabackend.c:1502 nvQuerySurfaceAttributes with 4 (nil) 0
      2275.892577913 [9093-9110] ../src/vabackend.c:1502 nvQuerySurfaceAttributes with 4 0x7ff43c3c6180 5
      2275.894632862 [9093-9110] ../src/vabackend.c: 556           nvCreateConfig got profile: 7 with 0 attributes
      2275.894654931 [9093-9110] ../src/vabackend.c:1502 nvQuerySurfaceAttributes with 4 (nil) 0
      2275.894664709 [9093-9110] ../src/vabackend.c:1502 nvQuerySurfaceAttributes with 4 0x7ff43c3c6580 5
      2275.896817921 [9093-9110] ../src/vabackend.c: 556           nvCreateConfig got profile: 13 with 0 attributes
      2275.896838803 [9093-9110] ../src/vabackend.c:1502 nvQuerySurfaceAttributes with 4 (nil) 0
      2275.896848860 [9093-9110] ../src/vabackend.c:1502 nvQuerySurfaceAttributes with 4 0x7ff43c3c6680 5
      2275.940260980 [9093-9110] ../src/vabackend.c: 556           nvCreateConfig got profile: 7 with 0 attributes
      2275.940290103 [9093-9110] ../src/vabackend.c:1502 nvQuerySurfaceAttributes with 4 (nil) 0
      2275.940300229 [9093-9110] ../src/vabackend.c:1502 nvQuerySurfaceAttributes with 4 0x7ff43c3c6e80 5
      2275.942595076 [9093-9110] ../src/vabackend.c:1502 nvQuerySurfaceAttributes with 4 (nil) 32756
      2275.942618821 [9093-9110] ../src/vabackend.c:1502 nvQuerySurfaceAttributes with 4 0x7ff43c3c6d00 5
      2275.944610426 [9093-9110] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 640x368, format 1 (0x7ff43840c9c0)
      2275.944629771 [9093-9110] ../src/vabackend.c:1220            nvDeriveImage In nvDeriveImage
      2275.944640666 [9093-9110] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 640x368, format 1 (0x7ff43840cba0)
      2275.944647790 [9093-9110] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 640x368, format 1 (0x7ff43840cf60)
      2275.944654774 [9093-9110] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 640x368, format 1 (0x7ff43840cc40)
      2275.944661618 [9093-9110] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 640x368, format 1 (0x7ff43840cce0)
      2275.944668113 [9093-9110] ../src/vabackend.c: 685        nvCreateSurfaces2 Creating surface 640x368, format 1 (0x7ff43840ce20)
[... cut ...]
      2275.996746999 [9093-9110] ../src/vabackend.c:1061            nvSyncSurface Syncing on surface: 0 (0x7ff4385b7de0)
      2275.996757475 [9093-9110] ../src/vabackend.c:1076            nvSyncSurface Surface 0 resolved (0x7ff4385b7de0)
      2275.999214767 [9093-9115] ../src/vabackend.c: 374          resolveSurfaces Mapped surface 1 to 5017A8000 (1024)
      2275.999232366 [9093-9115] ../src/export-buf.c: 527           realiseSurface No free surfaces found
      2275.999276994 [9093-9115] ../src/export-buf.c: 438     allocateBackingImage cuda error 'unspecified launch failure' (719)

</details>

KnightMurloc, 2022-06-19:
@buddyjojo who is the manufacturer of your GPU? I have MSI.

buddyjojo, 2022-06-19:
> @buddyjojo who is the manufacturer of your GPU? I have MSI.

Its some weird prebuilt dell thing.
I can't really find its manufacturer, perhaps its from dell themselves.

elFarto, 2022-06-19:
You can use `lspci -v` to see the subsystem for the device. Example from my card:
```
08:00.0 VGA compatible controller: NVIDIA Corporation GP106 [GeForce GTX 1060 6GB] (rev a1) (prog-if 00 [VGA controller])
	Subsystem: ZOTAC International (MCO) Ltd. Device 1438
```

KnightMurloc, 2022-06-19:
before I wrote that it works with mpv. but i checked now with getting the same error.

buddyjojo, 2022-06-19:
> You can use `lspci -v` to see the subsystem for the device. Example from my card:
> 
> ```
> 08:00.0 VGA compatible controller: NVIDIA Corporation GP106 [GeForce GTX 1060 6GB] (rev a1) (prog-if 00 [VGA controller])
> 	Subsystem: ZOTAC International (MCO) Ltd. Device 1438
> ```

I tried that already, it just says nvidia corporation twice.

KnightMurloc, 2022-06-19:
i ran mpv with cuda-memcheck and got this:
<details open>
<summary>log</summary>

```[cplayer] Command line options: '-v' '--hwdec=vaapi-copy' '--hwdec-codecs=all' '--vd-lavc-check-hw-profile=no' 'smptebars_h264_nv12.mp4'
[cplayer] mpv 0.34.1-dirty Copyright © 2000-2021 mpv/MPlayer/mplayer2 projects
[cplayer]  built on UNKNOWN
[cplayer] FFmpeg library versions:
[cplayer]    libavutil       57.17.100
[cplayer]    libavcodec      59.18.100
[cplayer]    libavformat     59.16.100
[cplayer]    libswscale      6.4.100
[cplayer]    libavfilter     8.24.100
[cplayer]    libswresample   4.3.100
[cplayer] FFmpeg version: n5.0
[cplayer] 
[cplayer] Configuration: /usr/bin/waf configure --prefix=/usr --confdir=/etc/mpv --enable-cdda --enable-dvb --enable-dvdnav --enable-libarchive --enable-libmpv-shared --disable-build-date --disable-caca
[cplayer] List of enabled features: alsa asm cdda cplayer cplugins cuda-hwaccel cuda-interop debug-build drm dvbin dvdnav egl egl-drm egl-helpers egl-x11 ffmpeg ffmpeg-aviocontext-bytes-read ffnvcodec gbm gbm.h gl gl-wayland glibc-thread-name glob glob-posix gpl iconv jack javascript jpeg lcms2 libarchive libass libavdevice libbluray libdl libm libmpv-shared libplacebo librt linux-fstatfs linux-input-event-codes lua luajit memfd_create optimize plain-gl posix posix-or-mingw pthreads pulse rubberband shaderc shaderc-static stdatomic uchardet vaapi vaapi-drm vaapi-egl vaapi-vulkan vaapi-wayland vaapi-x-egl vaapi-x11 vdpau vector vt.h vulkan wayland wayland-protocols x11 xv zimg zlib
[cplayer] Reading config file /etc/mpv/encoding-profiles.conf
[cplayer] Applying profile 'default'...
[cplayer] Reading config file /home/victor/.config/mpv/mpv.conf
[cplayer] Applying profile 'default'...
[cplayer] Setting option 'audio-file-auto' = 'exact' (flags = 4)
[cplayer] Setting option 'vo' = 'gpu' (flags = 4)
[cplayer] Setting option 'v' = '' (flags = 8)
[cplayer] Setting option 'hwdec' = 'vaapi-copy' (flags = 8)
[cplayer] Setting option 'hwdec-codecs' = 'all' (flags = 8)
[cplayer] Setting option 'vd-lavc-check-hw-profile' = 'no' (flags = 8)
[cplayer] Waiting for scripts...
[cplayer] Set property: shared-script-properties -> 1
[cplayer] Set property: shared-script-properties -> 1
[osd/libass] libass API version: 0x1502000
[osd/libass] libass source: commit: 0.15.2-0-gc967a5a3d9ec0d36af1148b3fdf2f307a21dd122-dirty
[osd/libass] Shaper: FriBidi 1.0.11 (SIMPLE) HarfBuzz-ng 4.3.0 (COMPLEX)
[osd/libass] Setting up fonts...
[osd/libass] Using font provider fontconfig
[osd/libass] Done.
[cplayer] Done loading scripts.
[cplayer] Running hook: ytdl_hook/on_load
[ytdl_hook] ytdl:// hook 
[cplayer] Set property: shared-script-properties -> 1
[ytdl_hook] not a ytdl:// url 
[ifo_dvdnav] Opening smptebars_h264_nv12.mp4
[bdmv/bluray] Opening smptebars_h264_nv12.mp4
[cplayer] Set property: shared-script-properties -> 1
[file] Opening smptebars_h264_nv12.mp4
[osd/libass] libass API version: 0x1502000
[osd/libass] libass source: commit: 0.15.2-0-gc967a5a3d9ec0d36af1148b3fdf2f307a21dd122-dirty
[osd/libass] Shaper: FriBidi 1.0.11 (SIMPLE) HarfBuzz-ng 4.3.0 (COMPLEX)
[demux] Trying demuxers for level=normal.
[osd/libass] Setting up fonts...
[lavf] Found 'mov,mp4,m4a,3gp,3g2,mj2' at score=100 size=2048.
[demux] Detected file format: mov,mp4,m4a,3gp,3g2,mj2 (libavformat)
[cplayer] Opening done: smptebars_h264_nv12.mp4
[osd/libass] Using font provider fontconfig
[osd/libass] Done.
[find_files] Loading external files in .
[cplayer] Running hook: ytdl_hook/on_preloaded
[lavf] select track 0
[cplayer]  (+) Video --vid=1 (*) (h264 640x360 30.000fps)
[vo/gpu] Probing for best GPU context.
[vo/gpu/opengl] Initializing GPU context 'wayland'
[vo/gpu/opengl] Initializing GPU context 'x11egl'
[vo/gpu/x11] X11 opening display: :0
[vo/gpu/x11] Display 0 (DVI-D-0): [0, 0, 1920, 1080] @ 60.000000 FPS
[vo/gpu/x11] Current display FPS: 60.000000
[vo/gpu/opengl] EGL_VERSION=1.5
[vo/gpu/opengl] EGL_VENDOR=NVIDIA
[vo/gpu/opengl] EGL_CLIENT_APIS=OpenGL_ES OpenGL
[vo/gpu/opengl] Trying to create Desktop OpenGL context.
[... 153 lines cut ...]
=========     Host Frame:mpv [0x107030]
=========     Host Frame:mpv [0xfe039]
=========     Host Frame:mpv [0x843db]
=========     Host Frame:mpv [0x82570]
=========     Host Frame:mpv [0x8288c]
=========     Host Frame:mpv [0x89c0d]
=========     Host Frame:mpv [0xe3f55]
=========     Host Frame:mpv [0xcf3b3]
=========     Host Frame:mpv [0xd0ccc]
=========     Host Frame:mpv [0x41fcd]
=========          7295.536859661 [54846-54846] ../src/vabackend.c:1299               nvGetImage got address 17A8000 for surface 21
      7295.537182955 [54846-54846] ../src/vabackend.c:1303               nvGetImage cuvidMapVideoFrame failed: 719
[ffmpeg] AVHWFramesContext: Failed to read image from surface 0x15: 23 (internal decoding error).
[vd] Could not copy back hardware decoded frame.
[vd] Error while decoding frame (hardware decoding)!
      7295.537588032 [54846-54846] ../src/vabackend.c: 830         nvDestroyContext Destroying context: 23
      7295.537661994 [54846-54846] ../src/vabackend.c: 249           destroyContext cuda error '(null)' (4)

 Host Frame:/usr/lib/libc.so.6 [0x29290]
=========     Host Frame:/usr/lib/libc.so.6 (__libc_start_main + 0x8a) [0x2934a]
=========     Host Frame:mpv [0x42085]
=========
========= ERROR SUMMARY: 4 errors
```
</details>

elFarto, 2022-06-19:
Oh, that's not right. It's not meant to be using vaGetImage, that's the old and slow way of getting the image out. No idea why mpv/ffmpeg is using that code path. I had assumed nothing was using it so I hadn't removed the old implementation I had.

I've removed that method and marked it as unimplemented in the latest git version.

philipl, 2022-06-19:
It's because mpv was configured to use vaapi-copy as the hwdec. That means it will try to copy back to system memory with vaGetImage. Use hwdec=vaapi.

KnightMurloc, 2022-06-19:
tried with latest version and vaapi instead of vaapi-copy.
<details>
<summary>logs</summary>

```
[cplayer] Command line options: '-v' '--hwdec=vaapi' '--hwdec-codecs=all' '--vd-lavc-check-hw-profile=no' 'smptebars_h264.mp4'
[cplayer] mpv 0.34.1-dirty Copyright © 2000-2021 mpv/MPlayer/mplayer2 projects
[cplayer]  built on UNKNOWN
[cplayer] FFmpeg library versions:
[cplayer]    libavutil       57.17.100
[cplayer]    libavcodec      59.18.100
[cplayer]    libavformat     59.16.100
[cplayer]    libswscale      6.4.100
[cplayer]    libavfilter     8.24.100
[cplayer]    libswresample   4.3.100
[cplayer] FFmpeg version: n5.0
[cplayer] 
[cplayer] Configuration: /usr/bin/waf configure --prefix=/usr --confdir=/etc/mpv --enable-cdda --enable-dvb --enable-dvdnav --enable-libarchive --enable-libmpv-shared --disable-build-date --disable-caca
[cplayer] List of enabled features: alsa asm cdda cplayer cplugins cuda-hwaccel cuda-interop debug-build drm dvbin dvdnav egl egl-drm egl-helpers egl-x11 ffmpeg ffmpeg-aviocontext-bytes-read ffnvcodec gbm gbm.h gl gl-wayland glibc-thread-name glob glob-posix gpl iconv jack javascript jpeg lcms2 libarchive libass libavdevice libbluray libdl libm libmpv-shared libplacebo librt linux-fstatfs linux-input-event-codes lua luajit memfd_create optimize plain-gl posix posix-or-mingw pthreads pulse rubberband shaderc shaderc-static stdatomic uchardet vaapi vaapi-drm vaapi-egl vaapi-vulkan vaapi-wayland vaapi-x-egl vaapi-x11 vdpau vector vt.h vulkan wayland wayland-protocols x11 xv zimg zlib
[cplayer] Reading config file /etc/mpv/encoding-profiles.conf
[cplayer] Applying profile 'default'...
[cplayer] Reading config file /home/victor/.config/mpv/mpv.conf
[cplayer] Applying profile 'default'...
[cplayer] Setting option 'audio-file-auto' = 'exact' (flags = 4)
[cplayer] Setting option 'vo' = 'gpu' (flags = 4)
[cplayer] Setting option 'v' = '' (flags = 8)
[cplayer] Setting option 'hwdec' = 'vaapi' (flags = 8)
[cplayer] Setting option 'hwdec-codecs' = 'all' (flags = 8)
[cplayer] Setting option 'vd-lavc-check-hw-profile' = 'no' (flags = 8)
[cplayer] Waiting for scripts...
[osd/libass] libass API version: 0x1502000
[osd/libass] libass source: commit: 0.15.2-0-gc967a5a3d9ec0d36af1148b3fdf2f307a21dd122-dirty
[osd/libass] Shaper: FriBidi 1.0.11 (SIMPLE) HarfBuzz-ng 4.3.0 (COMPLEX)
[osd/libass] Setting up fonts...
[osd/libass] Using font provider fontconfig
[osd/libass] Done.
[cplayer] Set property: shared-script-properties -> 1
[cplayer] Set property: shared-script-properties -> 1
[cplayer] Done loading scripts.
[cplayer] Running hook: ytdl_hook/on_load
[ytdl_hook] ytdl:// hook 
[ytdl_hook] not a ytdl:// url 
[cplayer] Set property: shared-script-properties -> 1
[ifo_dvdnav] Opening smptebars_h264.mp4
[bdmv/bluray] Opening smptebars_h264.mp4
[file] Opening smptebars_h264.mp4
[cplayer] Set property: shared-script-properties -> 1
[osd/libass] libass API version: 0x1502000
[osd/libass] libass source: commit: 0.15.2-0-gc967a5a3d9ec0d36af1148b3fdf2f307a21dd122-dirty
[demux] Trying demuxers for level=normal.
[osd/libass] Shaper: FriBidi 1.0.11 (SIMPLE) HarfBuzz-ng 4.3.0 (COMPLEX)
[osd/libass] Setting up fonts...
[lavf] Found 'mov,mp4,m4a,3gp,3g2,mj2' at score=100 size=2048.
[demux] Detected file format: mov,mp4,m4a,3gp,3g2,mj2 (libavformat)
[cplayer] Opening done: smptebars_h264.mp4
[osd/libass] Using font provider fontconfig
[osd/libass] Done.
[find_files] Loading external files in .
[cplayer] Running hook: ytdl_hook/on_preloaded
[lavf] select track 0
[cplayer]  (+) Video --vid=1 (*) (h264 640x360 30.000fps)
[vo/gpu] Probing for best GPU context.
[vo/gpu/opengl] Initializing GPU context 'wayland'
[vo/gpu/opengl] Initializing GPU context 'x11egl'
[vo/gpu/x11] X11 opening display: :0
[vo/gpu/x11] Display 0 (DVI-D-0): [0, 0, 1920, 1080] @ 60.000000 FPS
[vo/gpu/x11] Current display FPS: 60.000000
[vo/gpu/opengl] EGL_VERSION=1.5
[vo/gpu/opengl] EGL_VENDOR=NVIDIA
[vo/gpu/opengl] EGL_CLIENT_APIS=OpenGL_ES OpenGL
[... 105 lines cut ...]
========= Program hit CUDA_ERROR_LAUNCH_FAILED (error 719) due to "unspecified launch failure" on CUDA API call to cuStreamDestroy_v2.
=========     Saved host backtrace up to driver entry point at error
=========     Host Frame:/usr/lib/libcuda.so.1 [0x2aafd2]
=========     Host Frame:/usr/lib/libnvcuvid.so.1 [0x2ca4b]
=========     Host Frame:/usr/lib/libnvcuvid.so.1 [0x2ca69]
=========     Host Frame:/usr/lib/libnvcuvid.so.1 [0x1755d]
=========     Host Frame:/usr/lib/libnvcuvid.so.1 [0x2d6d9]
=========     Host Frame:/usr/lib/libnvcuvid.so.1 (cuvidGetDecoderCaps + 0xf4) [0x1bf54]
=========     Host Frame:/usr/lib/dri/nvidia_drv_video.so [0xe57f]
=========     Host Frame:/usr/lib/libva.so.2 (vaQuerySurfaceAttributes + 0x69) [0xa529]
=========     Host Frame:/usr/lib/libavutil.so.57 [0x35826]
=========     Host Frame:/usr/lib/libavutil.so.57 (av_hwdevice_get_hwframe_constraints + 0x68) [0x24298]
=========     Host Frame:mpv [0x1225f5]
=========     Host Frame:mpv [0x11d5ce]
=========     Host Frame:mpv [0x134be4]
=========     Host Frame:mpv [0x12e904]
=========     Host Frame:mpv [0x90ac8]
=========     Host Frame:mpv [0x1339ed]
=========     Host Frame:/usr/lib/libc.so.6 [0x8c54d]
=========     Host Frame:/usr/lib/libc.so.6 (clone + 0x44) [0x111874]
=========
========= ERROR SUMMARY: 3 errors
```

<\details>

philipl, 2022-06-19:
If it's crashing in `cuvidGetDecoderCaps`, then something is broken at a much lower level. Try out `nvdecinfo` from https://github.com/philipl/nv-video-info/ and see what it does. Did you try rebooting? Sometimes the driver can get into a bad state and basic stuff fails.

KnightMurloc, 2022-06-20:
> Did you try rebooting?

Yes, there is a problem with CUDA after sleep (they can be solved by unloading and loading the nvidia_uvm module). but yesterday I did all the tests on a freshly booted system. also hardware decoding via nvdec directly works.

> Try out nvdecinfo

tried it works without problems.

KnightMurloc, 2022-06-20:
i ran with cuda-gdb and in addition to the old errors i got this:
`warning: Cuda API error detected: cuCtxPushCurrent_v2 returned (0x2cf)`

KnightMurloc, 2022-06-20:
>it's crashing in `cuvidGetDecoderCaps`

It depends on which method is being used. With vaapi falls on the `cuvidGetDecoderCaps`. With vaapi-copy falls on `cuArray3DCreate`.

KnightMurloc, 2022-07-11:
just checked version 0.0.5 works with mpv vaapi-copy.

KnightMurloc, 2022-07-18:
tested direct backend. mpv with vaapi-copy works. mpv with vaapi green screen and sometimes this error:

`[vo/gpu/opengl] before video texture creation: OpenGL error INVALID_OPERATION.`

firefox not working.
<details>

<summary>
log
</summary>

```
      1259.145115910 [7990-7990] ../src/vabackend.c:1761       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7f3dc105b5c0 31
      1259.145132463 [7990-7990] ../src/vabackend.c:1782       __vaDriverInit_1_0 Now have 0 (0 max) instances
      1259.145138050 [7990-7990] ../src/vabackend.c:1807       __vaDriverInit_1_0 Selecting Direct backend
      1259.150350672 [7990-7990] ../src/direct/nv-driver.c: 187            init_nvdriver Initing nvdriver...
      1259.150368621 [7990-7990] ../src/direct/nv-driver.c: 192            init_nvdriver Got dev info: 100 1 0 fe
      1259.150470449 [7990-7990] ../src/direct/nv-driver.c: 253            init_nvdriver NVIDIA kernel driver version: 515.57
      1259.205769809 [7990-7990] ../src/vabackend.c: 564           nvCreateConfig got profile: 6 with 0 attributes
      1259.205789364 [7990-7990] ../src/vabackend.c: 564           nvCreateConfig got profile: 7 with 0 attributes
      1259.205797047 [7990-7990] ../src/vabackend.c: 564           nvCreateConfig got profile: 13 with 0 attributes
      1259.205808501 [7990-7990] ../src/vabackend.c:1735              nvTerminate Terminating 0x7f3dc105b5c0
      1259.221104917 [7990-7990] ../src/vabackend.c:1749              nvTerminate Now have 0 (0 max) instances
      1270.113864037 [8342-8355] ../src/vabackend.c:1761       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7fea4cb2f5c0 31
      1270.113883662 [8342-8355] ../src/vabackend.c:1782       __vaDriverInit_1_0 Now have 0 (0 max) instances
      1270.113890856 [8342-8355] ../src/vabackend.c:1807       __vaDriverInit_1_0 Selecting Direct backend
      1270.119777654 [8342-8355] ../src/direct/nv-driver.c: 187            init_nvdriver Initing nvdriver...
      1270.120566294 [8342-8355] ../src/direct/nv-driver.c: 192            init_nvdriver Got dev info: 100 1 0 fe
      1270.120881974 [8342-8355] ../src/direct/nv-driver.c: 253            init_nvdriver NVIDIA kernel driver version: 515.57
      1270.161667304 [8342-8355] ../src/vabackend.c:1137      nvQueryImageFormats In nvQueryImageFormats
      1270.199226486 [8342-8355] ../src/vabackend.c: 564           nvCreateConfig got profile: 6 with 0 attributes
      1270.199524635 [8342-8355] ../src/vabackend.c:1505 nvQuerySurfaceAttributes with 4 (nil) 0
      1270.199539651 [8342-8355] ../src/vabackend.c:1505 nvQuerySurfaceAttributes with 4 0x7fea2dfc1980 5
      1270.201734044 [8342-8355] ../src/vabackend.c: 564           nvCreateConfig got profile: 7 with 0 attributes
      1270.201754787 [8342-8355] ../src/vabackend.c:1505 nvQuerySurfaceAttributes with 4 (nil) 0
      1270.201765473 [8342-8355] ../src/vabackend.c:1505 nvQuerySurfaceAttributes with 4 0x7fea2dfc1980 5
      1270.203791061 [8342-8355] ../src/vabackend.c: 564           nvCreateConfig got profile: 13 with 0 attributes
      1270.203841137 [8342-8355] ../src/vabackend.c:1505 nvQuerySurfaceAttributes with 4 (nil) 0
      1270.203866978 [8342-8355] ../src/vabackend.c:1505 nvQuerySurfaceAttributes with 4 0x7fea2dfc1980 5
      1270.236987157 [8342-8355] ../src/vabackend.c: 564           nvCreateConfig got profile: 7 with 0 attributes
      1270.237013487 [8342-8355] ../src/vabackend.c:1505 nvQuerySurfaceAttributes with 4 (nil) 0
      1270.237022427 [8342-8355] ../src/vabackend.c:1505 nvQuerySurfaceAttributes with 4 0x7fea2dfc2180 5
      1270.239275207 [8342-8355] ../src/vabackend.c:1505 nvQuerySurfaceAttributes with 4 (nil) 32746
      1270.239301327 [8342-8355] ../src/vabackend.c:1505 nvQuerySurfaceAttributes with 4 0x7fea2dfc2080 5
      1270.241517091 [8342-8355] ../src/vabackend.c: 693        nvCreateSurfaces2 Creating surface 640x368, format 1 (0x7fea2a018f60)
      1270.241538323 [8342-8355] ../src/vabackend.c:1229            nvDeriveImage In nvDeriveImage
      1270.241556831 [8342-8355] ../src/vabackend.c: 693        nvCreateSurfaces2 Creating surface 640x368, format 1 (0x7fea2a018600)
      1270.241575269 [8342-8355] ../src/vabackend.c: 693        nvCreateSurfaces2 Creating surface 640x368, format 1 (0x7fea2a0186a0)
      1270.241583510 [8342-8355] ../src/vabackend.c: 693        nvCreateSurfaces2 Creating surface 640x368, format 1 (0x7fea2a018740)
      1270.241592519 [8342-8355] ../src/vabackend.c: 693        nvCreateSurfaces2 Creating surface 640x368, format 1 (0x7fea2a103160)
      1270.241599713 [8342-8355] ../src/vabackend.c: 693        nvCreateSurfaces2 Creating surface 640x368, format 1 (0x7fea2a103e80)
      1270.241607395 [8342-8355] ../src/vabackend.c: 693        nvCreateSurfaces2 Creating surface 640x368, format 1 (0x7fea2a103340)
      1270.241614519 [8342-8355] ../src/vabackend.c: 693        nvCreateSurfaces2 Creating surface 640x368, format 1 (0x7fea2a1030c0)
      1270.241621852 [8342-8355] ../src/vabackend.c: 693        nvCreateSurfaces2 Creating surface 640x368, format 1 (0x7fea2a1042e0)
      1270.241629116 [8342-8355] ../src/vabackend.c: 693        nvCreateSurfaces2 Creating surface 640x368, format 1 (0x7fea2a103c00)
      1270.241636868 [8342-8355] ../src/vabackend.c: 693        nvCreateSurfaces2 Creating surface 640x368, format 1 (0x7fea2a103ca0)
[... cut ...]
      1270.355263267 [8342-8359] ../src/vabackend.c: 727        nvDestroySurfaces Destroying surface -1 (0x7fea2a0186a0)
      1270.355277794 [8342-8359] ../src/vabackend.c: 727        nvDestroySurfaces Destroying surface -1 (0x7fea2a018600)
      1270.355291413 [8342-8359] ../src/vabackend.c: 727        nvDestroySurfaces Destroying surface -1 (0x7fea2a018f60)
      1270.355318162 [8342-8359] ../src/vabackend.c:1735              nvTerminate Terminating 0x7fea4cb2f5c0
      1270.376386919 [8342-8359] ../src/vabackend.c:1749              nvTerminate Now have 0 (0 max) instances

```
</details>

elFarto, 2022-07-18:
A green screen from mpv normally indicates that you don't have the latest master version of it.

KnightMurloc, 2022-07-18:
i tested version with last commit f7b08ef7ff9bb02ed49195d1035c1c608c482009 from direct-backend branch.

elFarto, 2022-07-18:
Sorry, I meant the master version of mpv. The last released version of mpv doesn't have the fix needed to import DMA-BUF from this library correctly.

KnightMurloc, 2022-07-18:
mpv vaapi works on git version. and firefox works on version 103.0beta (it doesn't work on version 102.0.1). Thanks!

closed 2022-07-18 by KnightMurloc