elFarto/nvidia-vaapi-driver #66: NS_ERROR_OUT_OF_MEMORY on firefox-99.0b

qumaciel, 2022-03-11:
Hi there,

I've upgraded to firefox-99.0b (beta1 and now beta2) some days ago and I am getting some wierd behavior that was not present before.

Now, whenever a click on a youtbe video, e.g., I get `NS_ERROR_OUT_OF_MEMORY` error and the player stops.
```
[94750-94750] ../src/vabackend.c:1642       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7f273ab2ba60
[94750-94750] ../src/export-buf.c: 171          findCudaDisplay Found 4 EGL devices
[94750-94750] ../src/export-buf.c: 175          findCudaDisplay Got EGL_CUDA_DEVICE_NV value '0' from EGLDevice 0
[94750-94750] ../src/export-buf.c: 135    checkModesetParameter Checking device file: /dev/dri/renderD129
[94750-94750] ../src/export-buf.c: 209             initExporter Got EGLDisplay from CUDA device 0
[94750-94750] ../src/export-buf.c: 239             initExporter Driver doesn't support 16-bit surfaces
[94750-94750] ../src/export-buf.c:  95                reconnect Reconnecting to stream
[94750-94750] ../src/vabackend.c: 457           nvCreateConfig got profile: 6 with 0 attributes
[94750-94750] ../src/vabackend.c: 457           nvCreateConfig got profile: 7 with 0 attributes
[94750-94750] ../src/vabackend.c: 457           nvCreateConfig got profile: 13 with 0 attributes
[94750-94750] ../src/vabackend.c: 457           nvCreateConfig got profile: 14 with 0 attributes
[94750-94750] ../src/vabackend.c: 457           nvCreateConfig got profile: 19 with 0 attributes
[94750-94750] ../src/vabackend.c:1626              nvTerminate Terminating 0x7f273ab2ba60
[94750-94750] ../src/export-buf.c:  57          releaseExporter Releasing exporter, 0 outstanding frames
[94750-94750] ../src/export-buf.c:  74          releaseExporter Done releasing frames
[95150-95170] ../src/vabackend.c:1642       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7efc8872b4c0
[95150-95170] ../src/export-buf.c: 171          findCudaDisplay Found 4 EGL devices
[95150-95170] ../src/export-buf.c: 175          findCudaDisplay Got EGL_CUDA_DEVICE_NV value '0' from EGLDevice 0
[95150-95170] ../src/export-buf.c: 135    checkModesetParameter Checking device file: /dev/dri/renderD129
[95150-95170] ../src/export-buf.c: 209             initExporter Got EGLDisplay from CUDA device 0
[95150-95170] ../src/export-buf.c: 239             initExporter Driver doesn't support 16-bit surfaces
[95150-95170] ../src/export-buf.c:  95                reconnect Reconnecting to stream
[95150-95170] ../src/vabackend.c: 972      nvQueryImageFormats In nvQueryImageFormats
[95150-95170] ../src/vabackend.c: 457           nvCreateConfig got profile: 6 with 0 attributes
[95150-95170] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 4 (nil) 0
[95150-95170] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 4 0x7efc61774480 5
[95150-95170] ../src/vabackend.c: 457           nvCreateConfig got profile: 7 with 0 attributes
[95150-95170] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 4 (nil) 0
[95150-95170] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 4 0x7efc61774480 5
[95150-95170] ../src/vabackend.c: 457           nvCreateConfig got profile: 13 with 0 attributes
[95150-95170] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 4 (nil) 0
[95150-95170] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 4 0x7efc61774380 5
[95150-95170] ../src/vabackend.c: 457           nvCreateConfig got profile: 14 with 0 attributes
[95150-95170] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 9 (nil) 0
[95150-95170] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 9 0x7efc61774880 5
[95150-95170] ../src/vabackend.c: 457           nvCreateConfig got profile: 19 with 0 attributes
[95150-95170] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 10 (nil) 0
[95150-95170] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 10 0x7efc61774080 5
[95150-95171] ../src/vabackend.c: 457           nvCreateConfig got profile: 19 with 0 attributes
[95150-95171] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 10 (nil) 0
[95150-95171] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 10 0x7efc61774a00 5
[95150-95171] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 10 (nil) 32508
[95150-95171] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 10 0x7efc61774200 5
[95150-95171] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 1920x1080, format 1
[95150-95171] ../src/vabackend.c:1064            nvDeriveImage In nvDeriveImage
[95150-95171] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 1920x1080, format 1
[95150-95171] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 1920x1080, format 1
[95150-95171] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 1920x1080, format 1
[95150-95171] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 1920x1080, format 1
[95150-95171] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 1920x1080, format 1
[95150-95171] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 1920x1080, format 1
[95150-95171] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 1920x1080, format 1
[95150-95171] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 1920x1080, format 1
[95150-95171] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 1920x1080, format 1
[95150-95171] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 1920x1080, format 1
[95150-95171] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 1920x1080, format 1
[95150-95171] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 1920x1080, format 1
[95150-95171] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 1920x1080, format 1
[95150-95171] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 1920x1080, format 1
[95150-95171] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 1920x1080, format 1
[95150-95171] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 1920x1080, format 1
[95150-95171] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 1920x1080, format 1
[95150-95171] ../src/vabackend.c: 457           nvCreateConfig got profile: 19 with 0 attributes
[95150-95171] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 10 (nil) 0
[... 398 lines cut ...]
[95618-95626] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 10 (nil) 32578
[95618-95626] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 10 0x7f42bac73180 5
[95618-95626] ../src/vabackend.c: 635          nvCreateContext with 18 render targets, 18 surfaces, at 1920x1080
[95618-95626] ../src/vabackend.c: 909             nvEndPicture cuvid decoded successful to idx: 0
[95618-95626] ../src/vabackend.c:1568    nvExportSurfaceHandle got address 205A00000 (2048) for surface 24 (picIdx: 0)
[95618-95626] ../src/export-buf.c: 300          allocateSurface presenting frame 1920x1080 0x7f42bacc5a60 0x7f42bacc5b80
[95618-95626] ../src/export-buf.c: 317          allocateSurface Adding frame from EGLStream: 0x7f42bad985c1
[95618-95626] ../src/export-buf.c: 330          allocateSurface Acquired image from EGLStream: 0x7f42bad985c1
[95618-95626] ../src/export-buf.c: 421            exportCudaPtr Detected NV12/NV21 NVIDIA driver bug, attempting to work around
[95618-95626] ../src/export-buf.c: 364              freeSurface Destroying EGLImage: 0x7f42bad985c1
[95618-95626] ../src/export-buf.c: 370              freeSurface Destroying CUarray: 0x7f42bacc5a60
[95618-95626] ../src/export-buf.c: 370              freeSurface Destroying CUarray: 0x7f42bacc5b80
[95618-95626] ../src/export-buf.c: 300          allocateSurface presenting frame 1920x1080 0x7f42bacc5a60 0x7f42bacc5b80
[95618-95626] ../src/export-buf.c: 317          allocateSurface Adding frame from EGLStream: 0x7f42bad980f1
[95618-95626] ../src/export-buf.c: 330          allocateSurface Acquired image from EGLStream: 0x7f42bad980f1
[95618-95626] ../src/export-buf.c: 431            exportCudaPtr Work around successful!
```

It works for youtube, but it is not working for twich, for example.
Do I need to change anything going from v98 -> v99?!

OS: Garuda
kwin X11
NVIDIA RTX 2060M, (510.54 driver)
nvidia-vaapi-driver: git

elFarto, 2022-03-11:
I've not tested on FF99 yet, but I don't believe there should be a problem. It's possible that there's some uninitialised bit of memory the new version is using. I'll try it out with the new version and see if I can reproduce it.

elFarto, 2022-03-12:
I've just tried out FF99 and I can't see any issue with it.

I see it's opening `/dev/dri/renderD129`, do you have a second GPU in your PC?

user0753, 2022-03-12:
I'm also experiencing this, but on nightly (2022-03-11). Using nvidia-vaapi-driver-git from AUR.
```
[arch@ArchPC ~]$ env LIBVA_DRIVER_NAME=nvidia NVD_LOG=1 MOZ_DISABLE_RDD_SANDBOX=1 firefox-nightly
[5164-5164] ../src/vabackend.c:1642       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7f630dc28f20
[5164-5164] ../src/export-buf.c: 171          findCudaDisplay Found 3 EGL devices
[5164-5164] ../src/export-buf.c: 175          findCudaDisplay Got EGL_CUDA_DEVICE_NV value '0' from EGLDevice 0
[5164-5164] ../src/export-buf.c: 135    checkModesetParameter Checking device file: /dev/dri/renderD128
[5164-5164] ../src/export-buf.c: 209             initExporter Got EGLDisplay from CUDA device 0
[5164-5164] ../src/export-buf.c: 239             initExporter Driver doesn't support 16-bit surfaces
[5164-5164] ../src/export-buf.c:  95                reconnect Reconnecting to stream
[5164-5164] ../src/vabackend.c: 457           nvCreateConfig got profile: 6 with 0 attributes
[5164-5164] ../src/vabackend.c: 457           nvCreateConfig got profile: 7 with 0 attributes
[5164-5164] ../src/vabackend.c: 457           nvCreateConfig got profile: 13 with 0 attributes
[5164-5164] ../src/vabackend.c: 457           nvCreateConfig got profile: 14 with 0 attributes
[5164-5164] ../src/vabackend.c: 457           nvCreateConfig got profile: 19 with 0 attributes
[5164-5164] ../src/vabackend.c:1626              nvTerminate Terminating 0x7f630dc28f20
[5164-5164] ../src/export-buf.c:  57          releaseExporter Releasing exporter, 0 outstanding frames
[5164-5164] ../src/export-buf.c:  74          releaseExporter Done releasing frames
[5560-5608] ../src/vabackend.c:1642       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7f842372b2e0
[5560-5608] ../src/export-buf.c: 171          findCudaDisplay Found 3 EGL devices
[5560-5608] ../src/export-buf.c: 175          findCudaDisplay Got EGL_CUDA_DEVICE_NV value '0' from EGLDevice 0
[5560-5608] ../src/export-buf.c: 135    checkModesetParameter Checking device file: /dev/dri/renderD128
[5560-5608] ../src/export-buf.c: 209             initExporter Got EGLDisplay from CUDA device 0
[5560-5608] ../src/export-buf.c: 239             initExporter Driver doesn't support 16-bit surfaces
[5560-5608] ../src/export-buf.c:  95                reconnect Reconnecting to stream
[5560-5608] ../src/vabackend.c: 972      nvQueryImageFormats In nvQueryImageFormats
[5560-5608] ../src/vabackend.c: 457           nvCreateConfig got profile: 6 with 0 attributes
[5560-5608] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 4 (nil) 0
[5560-5608] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 4 0x7f83fc84e400 5
[5560-5608] ../src/vabackend.c: 457           nvCreateConfig got profile: 7 with 0 attributes
[5560-5608] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 4 (nil) 0
[5560-5608] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 4 0x7f83fc84ef80 5
[5560-5608] ../src/vabackend.c: 457           nvCreateConfig got profile: 13 with 0 attributes
[5560-5608] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 4 (nil) 0
[5560-5608] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 4 0x7f83fc84e880 5
[5560-5608] ../src/vabackend.c: 457           nvCreateConfig got profile: 14 with 0 attributes
[5560-5608] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 9 (nil) 0
[5560-5608] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 9 0x7f83fc84ee00 5
[5560-5608] ../src/vabackend.c: 457           nvCreateConfig got profile: 19 with 0 attributes
[5560-5608] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 10 (nil) 0
[5560-5608] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 10 0x7f83fc84e880 5
[5560-5609] ../src/vabackend.c: 457           nvCreateConfig got profile: 19 with 0 attributes
[5560-5609] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 10 (nil) 0
[5560-5609] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 10 0x7f83fc84e100 5
[5560-5609] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 10 (nil) 32644
[5560-5609] ../src/vabackend.c:1346 nvQuerySurfaceAttributes with 10 0x7f83fc84f000 5
[5560-5609] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 3840x2160, format 1
[5560-5609] ../src/vabackend.c:1064            nvDeriveImage In nvDeriveImage
[5560-5609] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 3840x2160, format 1
[5560-5609] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 3840x2160, format 1
[5560-5609] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 3840x2160, format 1
[5560-5609] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 3840x2160, format 1
[5560-5609] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 3840x2160, format 1
[5560-5609] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 3840x2160, format 1
[5560-5609] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 3840x2160, format 1
[5560-5609] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 3840x2160, format 1
[5560-5609] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 3840x2160, format 1
[5560-5609] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 3840x2160, format 1
[5560-5609] ../src/vabackend.c: 552        nvCreateSurfaces2 creating 1 surface(s) 3840x2160, format 1
[... cut ...]
[Child 5426, MediaDecoderStateMachine #1] WARNING: Decoder=7f6fa6685200 Decode error: NS_ERROR_OUT_OF_MEMORY (0x8007000e): file /builds/worker/checkouts/gecko/dom/media/MediaDecoderStateMachine.cpp:3577
[Child 5426, MediaDecoderStateMachine #1] WARNING: Decoder=7f6fa6685200 Decode error: NS_ERROR_OUT_OF_MEMORY (0x8007000e): file /builds/worker/checkouts/gecko/dom/media/MediaDecoderStateMachine.cpp:3577
```
Mine also shows `checkModesetParameter Checking device file: /dev/dri/renderD128` but I'm using an AMD CPU w/ no iGPU and a single GTX 1080 w/ the latest driver.

elFarto, 2022-03-12:
Can you both try re-running the test with this variable set aswell: `MOZ_LOG="PlatformDecoderModule:5,Dmabuf:5"`

user0753, 2022-03-12:
From Firefox Nightly start to playback error displayed on YouTube:
```
[arch@ArchPC ~]$ env LIBVA_DRIVER_NAME=nvidia MOZ_LOG="PlatformDecoderModule:5,Dmabuf:5" NVD_LOG=1 MOZ_DISABLE_RDD_SANDBOX=1 firefox-nightly
[Parent 18459: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD128
[Parent 18459: Main Thread]: D/Dmabuf nsDMABufDevice::Configure()
[Parent 18459: Main Thread]: D/Dmabuf Loading DMABuf system library libgbm.so.1 ...
[Parent 18459: Main Thread]: D/Dmabuf DMABuf is enabled
[Parent 18459: Main Thread]: D/Dmabuf VAAPIGetAcceleratedProfiles()
[18459-18459] ../src/vabackend.c:1642       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7f306142c6a0
[18459-18459] ../src/export-buf.c: 171          findCudaDisplay Found 3 EGL devices
[18459-18459] ../src/export-buf.c: 175          findCudaDisplay Got EGL_CUDA_DEVICE_NV value '0' from EGLDevice 0
[18459-18459] ../src/export-buf.c: 135    checkModesetParameter Checking device file: /dev/dri/renderD128
[18459-18459] ../src/export-buf.c: 209             initExporter Got EGLDisplay from CUDA device 0
[18459-18459] ../src/export-buf.c: 239             initExporter Driver doesn't support 16-bit surfaces
[18459-18459] ../src/export-buf.c:  95                reconnect Reconnecting to stream
[Parent 18459: Main Thread]: D/Dmabuf   Initialised VAAPI connection: version 1.13
[18459-18459] ../src/vabackend.c: 457           nvCreateConfig got profile: 6 with 0 attributes
[Parent 18459: Main Thread]: D/Dmabuf   Supported profile H264Main:
[18459-18459] ../src/vabackend.c: 457           nvCreateConfig got profile: 7 with 0 attributes
[Parent 18459: Main Thread]: D/Dmabuf   Supported profile H264High:
[18459-18459] ../src/vabackend.c: 457           nvCreateConfig got profile: 13 with 0 attributes
[Parent 18459: Main Thread]: D/Dmabuf   Supported profile H264ConstrainedBaseline:
[18459-18459] ../src/vabackend.c: 457           nvCreateConfig got profile: 14 with 0 attributes
[Parent 18459: Main Thread]: D/Dmabuf   Supported profile VP8Version0_3:
[18459-18459] ../src/vabackend.c: 457           nvCreateConfig got profile: 19 with 0 attributes
[Parent 18459: Main Thread]: D/Dmabuf   Supported profile VP9Profile0:
[18459-18459] ../src/vabackend.c:1626              nvTerminate Terminating 0x7f306142c6a0
[18459-18459] ../src/export-buf.c:  57          releaseExporter Releasing exporter, 0 outstanding frames
[18459-18459] ../src/export-buf.c:  74          releaseExporter Done releasing frames
[Child 18722: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in Content process
[Child 18722: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD128
[Child 18722: Main Thread]: D/Dmabuf Failed to open drm render node /dev/dri/renderD128 error Permission denied
[Child 18722: Main Thread]: D/Dmabuf nsDMABufDevice::IsDMABufVAAPIEnabled: EGL 1 media_ffmpeg_vaapi_enabled 1 CanUseHardwareVideoDecoding 1 XRE_IsRDDProcess 0
[Child 18722: Main Thread]: D/Dmabuf nsDMABufDevice::IsDMABufVAAPIEnabled: EGL 1 media_ffmpeg_vaapi_enabled 1 CanUseHardwareVideoDecoding 1 XRE_IsRDDProcess 0
[Child 18722: Main Thread]: D/PlatformDecoderModule FFMPEG: Couldn't load function av_frame_get_colorspace
[Child 18722: Main Thread]: D/PlatformDecoderModule FFMPEG: Couldn't load function av_frame_get_color_range
[Child 18722: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 18722: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 18722: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 18722: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[RDD 18855: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in RDD process
[RDD 18855: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD128
[RDD 18855: Main Thread]: D/Dmabuf nsDMABufDevice::IsDMABufVAAPIEnabled: EGL 1 media_ffmpeg_vaapi_enabled 1 CanUseHardwareVideoDecoding 1 XRE_IsRDDProcess 1
[RDD 18855: Main Thread]: D/Dmabuf nsDMABufDevice::IsDMABufVAAPIEnabled: EGL 1 media_ffmpeg_vaapi_enabled 1 CanUseHardwareVideoDecoding 1 XRE_IsRDDProcess 1
[RDD 18855: Main Thread]: D/PlatformDecoderModule FFMPEG: Couldn't load function av_frame_get_colorspace
[RDD 18855: Main Thread]: D/PlatformDecoderModule FFMPEG: Couldn't load function av_frame_get_color_range
[RDD 18855: Main Thread]: D/PlatformDecoderModule Agnostic decoder rejects requested type
[RDD 18855: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type
[RDD 18855: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type
[RDD 18855: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type
[RDD 18855: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type
[Child 18722: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 18722: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 18722: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 18722: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 18722: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 18722: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 18722: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[... cut ...]
[RDD 19045: MediaPDecoder #1]: V/PlatformDecoderModule ReleaseVideoBufferWrapper: PlanarYCbCrImage=7f4f7fa6d4a0
[RDD 19045: MediaPDecoder #1]: V/PlatformDecoderModule ReleaseVideoBufferWrapper: PlanarYCbCrImage=7f4f7fa6d640
[RDD 19045: MediaPDecoder #1]: V/PlatformDecoderModule ReleaseVideoBufferWrapper: PlanarYCbCrImage=7f4f7fa6d840
```
I believe this started a couple days ago, like the 8th through 10th. Around then I started experiencing video playback issues (on twitter too) and figured it was just a regression in nightly. Sure enough, everything works fine with software decoding.

elFarto, 2022-03-12:
Hmm, that output is odd. It decodes one frame fine, then prints `D/PlatformDecoderModule FFMPEG: VAAPI is disabled by parent decoder module.` and falls back to software decoding, which then immediately throws the out of memory error. I'll have a rummage through the commit history and see if anything stands out.

elFarto, 2022-03-12:
Can you try this build of Firefox? It's the last one without those changes:
https://archive.mozilla.org/pub/firefox/nightly/2022/03/2022-03-07-09-38-30-mozilla-central/

user0753, 2022-03-12:
> Can you try this build of Firefox? It's the last one without those changes: https://archive.mozilla.org/pub/firefox/nightly/2022/03/2022-03-07-09-38-30-mozilla-central/

I can confirm it works, no errors from NVD_LOG and GreenWithEnvy reports GPU decoder activity.

qumaciel, 2022-03-13:
> I see it's opening `/dev/dri/renderD129`, do you have a second GPU in your PC?

Yeap! I have an intel iGPU. I'm running with a nvidia-only config though.

> Can you both try re-running the test with this variable set aswell: `MOZ_LOG="PlatformDecoderModule:5,Dmabuf:5"`

Sure, running using your last commit:

```
env LIBVA_DRIVER_NAME=nvidia NVD_LOG=1 MOZ_LOG="PlatformDecoderModule:5,Dmabuf:5" firefox-devel
oper-edition
[Parent 56884: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD129
[Parent 56884: Main Thread]: D/Dmabuf nsDMABufDevice::Configure()
[Parent 56884: Main Thread]: D/Dmabuf Loading DMABuf system library libgbm.so.1 ...
[Parent 56884: Main Thread]: D/Dmabuf DMABuf is enabled
[Parent 56884: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD129
[Parent 56884: Main Thread]: D/Dmabuf nsDMABufDevice::Configure()
[Parent 56884: Main Thread]: D/Dmabuf Loading DMABuf system library libgbm.so.1 ...
[Parent 56884: Main Thread]: D/Dmabuf DMABuf is enabled
[Parent 56884: Main Thread]: D/Dmabuf VAAPIGetAcceleratedProfiles()
[56884-56884] ../src/vabackend.c:1643       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7f96ca12a7a0
[56884-56884] ../src/export-buf.c: 176          findCudaDisplay Found 4 EGL devices
[56884-56884] ../src/export-buf.c: 180          findCudaDisplay Got EGL_CUDA_DEVICE_NV value '0' from EGLDevice 0
[56884-56884] ../src/export-buf.c: 140    checkModesetParameter Checking device file: /dev/dri/renderD129
[56884-56884] ../src/export-buf.c: 214             initExporter Got EGLDisplay from CUDA device 0
[56884-56884] ../src/export-buf.c: 244             initExporter Driver doesn't support 16-bit surfaces
[56884-56884] ../src/export-buf.c:  99                reconnect Reconnecting to stream
[Parent 56884: Main Thread]: D/Dmabuf   Initialised VAAPI connection: version 1.13
[56884-56884] ../src/vabackend.c: 457           nvCreateConfig got profile: 6 with 0 attributes
[Parent 56884: Main Thread]: D/Dmabuf   Supported profile H264Main:
[56884-56884] ../src/vabackend.c: 457           nvCreateConfig got profile: 7 with 0 attributes
[Parent 56884: Main Thread]: D/Dmabuf   Supported profile H264High:
[56884-56884] ../src/vabackend.c: 457           nvCreateConfig got profile: 13 with 0 attributes
[Parent 56884: Main Thread]: D/Dmabuf   Supported profile H264ConstrainedBaseline:
[56884-56884] ../src/vabackend.c: 457           nvCreateConfig got profile: 14 with 0 attributes
[Parent 56884: Main Thread]: D/Dmabuf   Supported profile VP8Version0_3:
[56884-56884] ../src/vabackend.c: 457           nvCreateConfig got profile: 19 with 0 attributes
[Parent 56884: Main Thread]: D/Dmabuf   Supported profile VP9Profile0:
[56884-56884] ../src/vabackend.c:1627              nvTerminate Terminating 0x7f96ca12a7a0
[56884-56884] ../src/export-buf.c:  61          releaseExporter Releasing exporter, 0 outstanding frames
[56884-56884] ../src/export-buf.c:  78          releaseExporter Done releasing frames
[Child 59680: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in Content process
[Child 59680: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD129
[Child 59680: Main Thread]: D/Dmabuf Failed to open drm render node /dev/dri/renderD129 error Permission denied
[Child 59680: Main Thread]: D/Dmabuf nsDMABufDevice::IsDMABufVAAPIEnabled: EGL 1 media_ffmpeg_vaapi_enabled 1 CanUseHardwareVideoDecoding 1 XRE_IsRDDProcess 0
[Child 59680: Main Thread]: D/Dmabuf nsDMABufDevice::IsDMABufVAAPIEnabled: EGL 1 media_ffmpeg_vaapi_enabled 1 CanUseHardwareVideoDecoding 1 XRE_IsRDDProcess 0
[Child 59680: Main Thread]: D/PlatformDecoderModule FFMPEG: Couldn't load function av_frame_get_colorspace
[Child 59680: Main Thread]: D/PlatformDecoderModule FFMPEG: Couldn't load function av_frame_get_color_range
[Child 59680: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59680: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59680: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 59680: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[RDD 59830: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in RDD process
[RDD 59830: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD129
[RDD 59830: Main Thread]: D/Dmabuf nsDMABufDevice::IsDMABufVAAPIEnabled: EGL 1 media_ffmpeg_vaapi_enabled 1 CanUseHardwareVideoDecoding 1 XRE_IsRDDProcess 1
[RDD 59830: Main Thread]: D/Dmabuf nsDMABufDevice::IsDMABufVAAPIEnabled: EGL 1 media_ffmpeg_vaapi_enabled 1 CanUseHardwareVideoDecoding 1 XRE_IsRDDProcess 1
[RDD 59830: Main Thread]: D/PlatformDecoderModule FFMPEG: Couldn't load function av_frame_get_colorspace
[RDD 59830: Main Thread]: D/PlatformDecoderModule FFMPEG: Couldn't load function av_frame_get_color_range
[RDD 59830: Main Thread]: D/PlatformDecoderModule Agnostic decoder rejects requested type
[RDD 59830: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type
[RDD 59830: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type
[RDD 59830: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type
[RDD 59830: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type
[Child 59680: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[... cut ...]
[Child 59680, MediaDecoderStateMachine #1] WARNING: Decoder=7fa9766dbf00 Decode error: NS_ERROR_OUT_OF_MEMORY (0x8007000e): file /build/firefox-developer-edition/src/firefox-99.0/dom/media/MediaDecoderStateMachine.cpp:3577
[56884-57111] ../src/export-buf.c:  56                    debug [EGL] eglCreateContext: EGL_BAD_MATCH error: In eglCreateContext: Failure in argument parsing

[56884-57111] ../src/export-buf.c:  56                    debug [EGL] eglCreateContext: EGL_BAD_ATTRIBUTE error: In eglCreateContext: Failure in argument parsing
```

elFarto, 2022-03-16:
Just getting back to this, but unfortunately I can't reproduce it. I've tried 99b2, 99b3, 99b4 and the latest 100 nightly, but all play back videos on Youtube, Twitter and Twitch just fine.

Is this still happening for you?

user0753, 2022-03-16:
> Is this still happening for you?
Unfortunately, yes. Currently on 99b4 and it's still happening. Tried "purging" it with pacman -Rscn and reinstalling but no luck. Are you also using Arch? Does the driver create any config or cache files I can delete? If nothing else, I may try nuking the nvidia driver and reinstalling it, currently using nvidia-dkms but I doubt it matters. Just so odd that it worked perfectly up until about a week ago and there was no version change for your driver or the nvidia driver.

hellobbn, 2022-03-17:
I am also having this problem. My system uses a optimus setup (AMD iGPU + Nvidia 3070 dGPU). The problem happens when
the system is running in nvidia-only mode with `LIBVA_DRIVER_NAME` set to `nvidia`

elFarto, 2022-03-17:
> > Is this still happening for you?
> 
> Are you also using Arch?

No, I run Fedora, but your comment has given me an idea. Can you try the [FF99b4 build](https://ftp.mozilla.org/pub/firefox/releases/99.0b4/linux-x86_64/) from Mozilla, rather than the Arch build?

hellobbn, 2022-03-17:
> > > Is this still happening for you?
> > 
> > 
> > Are you also using Arch?
> 
> No, I run Fedora, but your comment has given me an idea. Can you try the [FF99b4 build](https://ftp.mozilla.org/pub/firefox/releases/99.0b4/linux-x86_64/) from Mozilla, rather than the Arch build?

With this build, the problem persists:
```
[Child 145270, MediaDecoderStateMachine #1] WARNING: Decoder=7f2db3f3c700 Decode error: NS_ERROR_OUT_OF_MEMORY (0x8007000e): file /builds/worker/checkouts/gecko/dom/media/MediaDecoderStateMachine.cpp:3577
```

user0753, 2022-03-17:
> > > Is this still happening for you?
> > 
> > 
> > Are you also using Arch?
> 
> No, I run Fedora, but your comment has given me an idea. Can you try the [FF99b4 build](https://ftp.mozilla.org/pub/firefox/releases/99.0b4/linux-x86_64/) from Mozilla, rather than the Arch build?

No difference. Actually ran across even more weird behavior where on twitter, videos would endlessly load and the driver log would output nothing. Happened a few times but eventually it went back to showing media cant be played w/ out of memory error.

qumaciel, 2022-03-17:
> Is this still happening for you?

I've just remembered [this issue](https://bugzilla.mozilla.org/show_bug.cgi?id=1673184) back then, and used the same workaround.
Completely disabling the RDD process by setting `media.rdd-process.enabled` and `media.ffvpx.enabled` to `false` did the trick.

@elFarto, If I remember correctly, FF99 strengthened the sandbox, right?

elFarto, 2022-03-17:
> If I remember correctly, FF99 strengthened the sandbox, right?

That shouldn't matter as the `MOZ_DISABLE_RDD_SANDBOX=1` env var should disable the RDD sandbox completely. The other issue with that is it's working fine for me, with those two config options set to true.

I've gone back over the logs you've posted and noticed something:
```
[RDD 59830: MediaPDecoder #1]: D/PlatformDecoderModule VideoFrameSurfaceVAAPI: creating surface UID = 1
[RDD 59830: MediaPDecoder #1]: D/Dmabuf DMABufSurfaceYUV::VerifyTextureCreation() UID 1
[RDD 59830: MediaPDecoder #1]: D/Dmabuf DMABufSurfaceYUV::CreateEGLImage() UID 1 plane 0
[RDD 59830: MediaPDecoder #1]: D/Dmabuf   Success.
[RDD 59830: MediaPDecoder #1]: D/Dmabuf DMABufSurfaceYUV::CreateEGLImage() UID 1 plane 1
[RDD 59830: MediaPDecoder #1]: D/Dmabuf   Success.
[RDD 59830: MediaPDecoder #1]: D/Dmabuf DMABufSurfaceYUV::ReleaseEGLImages() UID 1
[RDD 59830: MediaPDecoder #1]: D/PlatformDecoderModule VideoFrameSurfaceVAAPI: VAAPI locking dmabuf surface UID = 1
```
That's actually incomplete, it's meant to be followed by something like this:
```
[RDD 8769: MediaSupervisor #1]: D/Dmabuf DMABufSurfaceYUV::Serialize() UID 1
[Parent 8357: Compositor]: D/Dmabuf DMABufSurfaceYUV::ImportSurfaceDescriptor() UID 1
[Parent 8357: Compositor]: D/Dmabuf     plane 0 fd 232 size 1280 x 720 format 20203852
[Parent 8357: Compositor]: D/Dmabuf     plane 1 fd 233 size 640 x 360 format 38384752
```
Anyway, I'll keep digging.

elFarto, 2022-03-19:
I've looked through the code, but I can't see an obvious cause for this. I've raise issue [1760414](https://bugzilla.mozilla.org/show_bug.cgi?id=1760414) to see if the FF team know anything.

LordTermor, 2022-03-20:
I just wanted to note that it's not specific to nvidia. I've got the exactly same issue on latest FF builds using Intel's GPU.

Celmor, 2022-03-21:
I'm having the same issue.
My setup consists of an nvidia GPU and AMD gpu though only the latter one has drivers loaded (my nvidia GPU is reserved for gpu passthrough for VMs with the vfio-pci stub driver loaded). Perhaps the vaapi stuff gets confused by this.
I've build firefox-developer-edition since I started having  issues after updating to 98 though 99 didn't help. 
Using MOZ_DISABLE_RDD_SANDBOX=1 made matters worse (twitch also became unplayable). To note though that YouTube always kept working (perhaps because it can't use HVA).

I could provide a trace back or put together more descriptive cases/results if needed.

huyz-git, 2022-03-23:
https://bugzilla.mozilla.org/show_bug.cgi?id=1759137
Perhaps this issue is related to this bug.

user0753, 2022-03-25:
> https://bugzilla.mozilla.org/show_bug.cgi?id=1759137 Perhaps this issue is related to this bug.

As of writing, this got marked resolved about 30 minutes ago so I'll test again with the next couple nightly updates.

elFarto, 2022-03-26:
There's now a new Firefox build available with a potential fix in available [here](https://archive.mozilla.org/pub/firefox/nightly/2022/03/2022-03-25-21-47-37-mozilla-central/).

huyz-git, 2022-03-26:
Unfortunately, this issue still exists in the latest firefox nightly.

qumaciel, 2022-03-27:
Yeah, the issue remains with nightly, although I am getting a somewhat different output this time:

```
env LIBVA_DRIVER_NAME=nvidia NVD_LOG=1 MOZ_LOG="PlatformDecoderModule:5,Dmabuf:5" firefox-nightly
[Parent 40143: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD129
[Parent 40143: Main Thread]: D/Dmabuf nsDMABufDevice::Configure()
[Parent 40143: Main Thread]: D/Dmabuf Loading DMABuf system library libgbm.so.1 ...
[Parent 40143: Main Thread]: D/Dmabuf DMABuf is enabled
[Parent 40143: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD129
[Parent 40143: Main Thread]: D/Dmabuf nsDMABufDevice::Configure()
[Parent 40143: Main Thread]: D/Dmabuf Loading DMABuf system library libgbm.so.1 ...
[Parent 40143: Main Thread]: D/Dmabuf DMABuf is enabled
[Parent 40143: Main Thread]: D/Dmabuf VAAPIGetAcceleratedProfiles()
[40143-40143] ../src/vabackend.c:1642       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7f6eb8c2a5c0 31
[40143-40143] ../src/export-buf.c: 162       findGPUIndexFromFd Looking for drmDeviceIndex: 1
[40143-40143] ../src/export-buf.c: 175       findGPUIndexFromFd Found 4 EGL devices
[40143-40143] ../src/export-buf.c: 184       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
[40143-40143] ../src/export-buf.c: 190       findGPUIndexFromFd Found drmDeviceIndex: 1
[40143-40143] ../src/export-buf.c: 208       findGPUIndexFromFd Selecting EGLDevice 0
[40143-40143] ../src/export-buf.c: 270             initExporter Driver doesn't support 16-bit surfaces
[40143-40143] ../src/export-buf.c: 103                reconnect Reconnecting to stream
[Parent 40143: Main Thread]: D/Dmabuf   Initialised VAAPI connection: version 1.13
[40143-40143] ../src/vabackend.c: 456           nvCreateConfig got profile: 6 with 0 attributes
[Parent 40143: Main Thread]: D/Dmabuf   Supported profile H264Main:
[40143-40143] ../src/vabackend.c: 456           nvCreateConfig got profile: 7 with 0 attributes
[Parent 40143: Main Thread]: D/Dmabuf   Supported profile H264High:
[40143-40143] ../src/vabackend.c: 456           nvCreateConfig got profile: 13 with 0 attributes
[Parent 40143: Main Thread]: D/Dmabuf   Supported profile H264ConstrainedBaseline:
[40143-40143] ../src/vabackend.c: 456           nvCreateConfig got profile: 14 with 0 attributes
[Parent 40143: Main Thread]: D/Dmabuf   Supported profile VP8Version0_3:
[40143-40143] ../src/vabackend.c: 456           nvCreateConfig got profile: 19 with 0 attributes
[Parent 40143: Main Thread]: D/Dmabuf   Supported profile VP9Profile0:
[40143-40143] ../src/vabackend.c:1626              nvTerminate Terminating 0x7f6eb8c2a5c0
[40143-40143] ../src/export-buf.c:  65          releaseExporter Releasing exporter, 0 outstanding frames
[40143-40143] ../src/export-buf.c:  82          releaseExporter Done releasing frames
[Child 40441: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in Content process
[Child 40441: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD129
[Child 40441: Main Thread]: D/Dmabuf Failed to open drm render node /dev/dri/renderD129 error Permission denied
[Child 40441: Main Thread]: D/Dmabuf nsDMABufDevice::IsDMABufVAAPIEnabled: EGL 1 media_ffmpeg_vaapi_enabled 1 CanUseHardwareVideoDecoding 1 XRE_IsRDDProcess 0
[Child 40441: Main Thread]: D/Dmabuf nsDMABufDevice::IsDMABufVAAPIEnabled: EGL 1 media_ffmpeg_vaapi_enabled 1 CanUseHardwareVideoDecoding 1 XRE_IsRDDProcess 0
[Child 40441: Main Thread]: D/PlatformDecoderModule FFMPEG: Couldn't load function av_frame_get_colorspace
[Child 40441: Main Thread]: D/PlatformDecoderModule FFMPEG: Couldn't load function av_frame_get_color_range
[Child 40441: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 40441: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[RDD 40564: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in RDD process
[RDD 40564: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD129
[RDD 40564: Main Thread]: D/Dmabuf nsDMABufDevice::IsDMABufVAAPIEnabled: EGL 1 media_ffmpeg_vaapi_enabled 1 CanUseHardwareVideoDecoding 1 XRE_IsRDDProcess 1
[RDD 40564: Main Thread]: D/Dmabuf nsDMABufDevice::IsDMABufVAAPIEnabled: EGL 1 media_ffmpeg_vaapi_enabled 1 CanUseHardwareVideoDecoding 1 XRE_IsRDDProcess 1
[RDD 40564: Main Thread]: D/PlatformDecoderModule FFMPEG: Couldn't load function av_frame_get_colorspace
[RDD 40564: Main Thread]: D/PlatformDecoderModule FFMPEG: Couldn't load function av_frame_get_color_range
[RDD 40564: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type
[RDD 40564: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type
[RDD 40564: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type
[RDD 40564: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type
[Child 40441: Main Thread]: D/Dmabuf nsDMABufDevice::Configure()
[Child 40441: Main Thread]: D/Dmabuf Loading DMABuf system library libgbm.so.1 ...
[Child 40441: Main Thread]: D/Dmabuf Failed to create drm render device
[Child 40441: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 40441: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 40441: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 40441: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[... cut ...]
[RDD 43523: MediaPDecoder #1]: V/PlatformDecoderModule ReleaseVideoBufferWrapper: PlanarYCbCrImage=7f2af5871a20
[RDD 43523: MediaPDecoder #1]: V/PlatformDecoderModule ReleaseVideoBufferWrapper: PlanarYCbCrImage=7f2af5871c20
[RDD 43523: MediaPDecoder #1]: V/PlatformDecoderModule ReleaseVideoBufferWrapper: PlanarYCbCrImage=7f2af5871be0

###!!! [Parent][PImageBridgeParent] Error: RunMessage(msgname=PImageBridge::Msg_WillClose) Channel closing: too late to send/recv, messages will be lost

[2022-03-27T13:27:39Z ERROR xulstore::persist] removeDocument error: unavailable
```

elFarto, 2022-03-28:
Can someone who's getting this issue check this page `about:crashes` and see if there's a crash that corresponds to this issue. If there is, can you submit it and post the link here?

HE7086, 2022-03-28:
It generates 3 crashes every time I tries to play a h.264 video, here's one of the report:

https://crash-stats.mozilla.org/report/index/3a24a04c-68d4-4e3e-8469-a82d80220328

qumaciel, 2022-03-28:
here is mine

[https://crash-stats.mozilla.org/report/index/bp-b798d868-df0a-4d47-b6ac-d834a0220328](https://crash-stats.mozilla.org/report/index/bp-b798d868-df0a-4d47-b6ac-d834a0220328)

elFarto, 2022-03-28:
Ok, those crash dumps indicate it's the same issue as [1759137](https://bugzilla.mozilla.org/show_bug.cgi?id=1759137).

elFarto, 2022-03-31:
Looks like there's a new build, with a new fix available [here](https://archive.mozilla.org/pub/firefox/nightly/2022/03/2022-03-29-20-42-47-mozilla-central/).

elFarto, 2022-03-31:
Is it still an issue with the latest [99 beta](https://archive.mozilla.org/pub/firefox/releases/99.0b8/) too?

qumaciel, 2022-03-31:
> Is it still an issue with the latest [99 beta](https://archive.mozilla.org/pub/firefox/releases/99.0b8/) too?
As far as I could test, yeah. 

The fix landed on nightly in these commits:
[https://hg.mozilla.org/mozilla-central/rev/c7ca5d4c890a](https://hg.mozilla.org/mozilla-central/rev/c7ca5d4c890a)
[https://hg.mozilla.org/mozilla-central/rev/a107df8ae87c](https://hg.mozilla.org/mozilla-central/rev/a107df8ae87c)

huyz-git, 2022-04-01:
> Is it still an issue with the latest [99 beta](https://archive.mozilla.org/pub/firefox/releases/99.0b8/) too?

Yes, 99b8 is released at 3.25. But and these fix commits were landed at 3.29.

After I cherry pick these commits into 99b8 and rebuild it, this issue is solved.

thesword53, 2022-04-06:
I have this issue only on H.264 videos. VP9 decoding works for me.

rkoot, 2022-04-09:
os:OpenSUSE tumbleweed, dGPU:GTX-1650, X11 and wayland, firefox 99.0 -> works

guihkx, 2022-04-13:
I just don't understand the logic of Mozilla not backporting this fix to the v99 series, especially because this is clearly a regression...?

jeois, 2022-04-13:
**99.0.1** released, and they didn't backport the commits.  We might have to wait until version 101 (possibly a month) or patch it ourselves, which is pretty bad for firefox variants built off **_stable_**.  

btw, ignore what I said about the Ubuntu binary package. While it has global menu support and contains the fix for this driver, it also plays with `apparmor` and messed up my profile when used in combination with `profile-sync-daemon`.  I _don't_ recommend it anymore.  I've downgraded to `firefox-appmenu` 98.0.2 until I find time to compile, smh.

guihkx, 2022-04-13:
> **99.0.1** released, and they didn't backport the commits. We might have to wait until version 101 (possibly a month) or patch it ourselves, which is pretty bad for firefox variants built off **_stable_**.

I think you meant v100, but yeah, still annoying.

Darkspirit, 2022-04-13:
> I just don't understand the logic of Mozilla not backporting this fix to the v99 series, especially because this is clearly a regression...?

Please use https://nightly.mozilla.org to try out experimental features that are not yet enabled by default in stable for a reason.

guihkx, 2022-04-13:
I suppose I could, but how is this considered an experimental feature and not just a fix for a regression? It worked in previous versions of Firefox... 🤔

jeois, 2022-04-14:
Video acceleration isn't new, and thanks to these drivers, it's been working since before v95. (probably been in Windows versions for years)  Moreover, some of us use patched binaries from repos built against **stable**, so `nightly` builds would definitely lack those extra features. 

I tried to patch and compile 100.0b5 but it's a struggle... worked around a python dependency issue, only to run out of memory after it used 17gb in tmpfs, and perhaps my build script is faulty too.

thesword53, 2022-05-03:
It seems to be fixed in Firefox 100.

rkoot, 2022-05-04:
Firefox 100 seems to be working *mostly*
I have found at least one youtube video in vp9 that results in the ff99 behavior. This one gives issues: https://www.youtube.com/watch?v=GeWkQHFrENA

jeois, 2022-05-04:
@rkoot 
that video is fine for me; no v99 behavior on youtube anymore.  It seem like it's reloading if skip through it, but it's only a few seconds delay, so not a problem.  however, trying to skip through twitch VODs will make that player crash most of the time.

Also, I find if I pause live streams, FF might be falling back to using software decoding for avc1 (H.264) upon resume/unpause.  in other words, the "compute" process displayed by `nvtop` disappears, and overall GPU utilization increases quite a bit while slightly increasing CPU utilization as well.  This is likely a separate issue from the v99 crashes though.