elFarto/nvidia-vaapi-driver #68: Not working with Firefox 98

KidoVin01, 2022-03-26:
Hello, I have an XPS 9500, and `nvidia-smi` and `nvidia-settings` seems to show 0% utilization:

`inxi -G`
```
Graphics:
  Device-1: Intel CometLake-H GT2 [UHD Graphics] driver: i915 v: kernel
  Device-2: NVIDIA TU117M [GeForce GTX 1650 Ti Mobile] driver: nvidia
    v: 510.54
  Device-3: Microdia Integrated_Webcam_HD type: USB driver: uvcvideo
  Display: x11 server: X.Org v: 1.21.1.3 driver: X: loaded: intel,nvidia
    unloaded: modesetting gpu: i915 resolution: 1920x1200~60Hz
  OpenGL: renderer: Mesa Intel UHD Graphics (CML GT2) v: 4.6 Mesa 21.3.7
```

`cat /sys/module/nvidia_drm/parameters/modeset`
```
Y
```

` NVD_LOG=1 MOZ_LOG="PlatformDecoderModule:5,Dmabuf:5" vainfo`
<details>
<summary>Output</summary>

```
[10124-10124] ../src/vabackend.c:1642       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x5581af236ba0 10
[10124-10124] ../src/vabackend.c:1645       __vaDriverInit_1_0 Non-DRM display type detected, defaulting to GPU ID 0. Use NVD_GPU to pick a specific GPU.
[10124-10124] ../src/export-buf.c: 164       findGPUIndexFromFd Looking for GPU index: 0
[10124-10124] ../src/export-buf.c: 175       findGPUIndexFromFd Found 4 EGL devices
[10124-10124] ../src/export-buf.c: 184       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
[10124-10124] ../src/export-buf.c: 208       findGPUIndexFromFd Selecting EGLDevice 0
[10124-10124] ../src/export-buf.c: 270             initExporter Driver doesn't support 16-bit surfaces
[10124-10124] ../src/export-buf.c: 103                reconnect Reconnecting to stream
vainfo: VA-API version: 1.13 (libva 2.14.0)
vainfo: Driver version: VA-API NVDEC driver
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      <unknown profile>               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
[10124-10124] ../src/vabackend.c:1626              nvTerminate Terminating 0x5581af236ba0
[10124-10124] ../src/export-buf.c:  65          releaseExporter Releasing exporter, 0 outstanding frames
[10124-10124] ../src/export-buf.c:  82          releaseExporter Done releasing frames
```

</details>

`NVD_LOG=1 MOZ_LOG="PlatformDecoderModule:5,Dmabuf:5"`
<details>
<summary>Output</summary>

```
[Parent 7993: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD129
[Parent 7993: Main Thread]: D/Dmabuf nsDMABufDevice::Configure()
[Parent 7993: Main Thread]: D/Dmabuf Loading DMABuf system library libgbm.so.1 ...
ATTENTION: default value of option mesa_glthread overridden by environment.
[Parent 7993: Main Thread]: D/Dmabuf DMABuf is enabled
[Child 8251: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in Content process
[Child 8251: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD129
[Child 8251: Main Thread]: D/Dmabuf nsDMABufDevice::IsDMABufVAAPIEnabled: EGL 1 media_ffmpeg_vaapi_enabled 1 CanUseHardwareVideoDecoding 1 XRE_IsRDDProcess 0
[Child 8251: Main Thread]: D/PlatformDecoderModule FFMPEG: VA-API FFmpeg is disabled by platform
[Child 8251: Main Thread]: D/Dmabuf nsDMABufDevice::IsDMABufVAAPIEnabled: EGL 1 media_ffmpeg_vaapi_enabled 1 CanUseHardwareVideoDecoding 1 XRE_IsRDDProcess 0
[Child 8251: Main Thread]: D/PlatformDecoderModule FFMPEG: VA-API FFmpeg is disabled by platform
[... 353 lines cut ...]
[vp9 @ 0x7fa1cec72f00] Format yuv420p chosen by get_format().
[RDD 9673: MediaPDecoder #2]: V/PlatformDecoderModule GetVideoBuffer: aCodecContext=7fa1cec72f00 aFrame=7fa1cec94a00
[RDD 9673: MediaPDecoder #1]: V/PlatformDecoderModule GetVideoBuffer: aCodecContext=7fa1cec76100 aFrame=7fa1cec9d100
[RDD 9673: MediaPDecoder #1]: D/PlatformDecoderModule FFVPX: Got one frame output with pts=0 dts=0 duration=33000 opaque=-9223372036854775808
[RDD 9673: MediaPDecoder #1]: V/PlatformDecoderModule GetVideoBuffer: aCodecContext=7fa1cec72f00 aFrame=7fa1cec94a00
[Child 9598: MediaPDecoder #1]: V/PlatformDecoderModule AudioTrimmer[7ff6372e3700] ::PrepareTrimmers: sample[21000,41000] no trimming information
[RDD 9673: MediaPDecoder #3]: V/PlatformDecoderModule GetVideoBuffer: aCodecContext=7fa1cec76100 aFrame=7fa1cec9d100
[Child 9598: MediaPDecoder #3]: V/PlatformDecoderModule AudioTrimmer[7ff6372e3700] ::HandleDecodedResult: sample[21000,41000] (decoded[14500,34500] no trimming needed
[RDD 9673: MediaPDecoder #3]: D/PlatformDecoderModule FFVPX: Got one frame output with pts=33000 dts=33000 duration=34000 opaque=-9223372036854775808
[Child 9598: MediaPDecoder #3]: V/PlatformDecoderModule AudioTrimmer[7ff6372e3700] ::PrepareTrimmers: sample[41000,61000] no trimming information
[Child 9598: MediaPDecoder #2]: V/PlatformDecoderModule AudioTrimmer[7ff6372e3700] ::HandleDecodedResult: sample[41000,61000] (decoded[34500,54500] no trimming needed
[RDD 9673: MediaPDecoder #2]: V/PlatformDecoderModule GetVideoBuffer: aCodecContext=7fa1cec72f00 aFrame=7fa1cec94a00
[Child 9598: MediaPDecoder #2]: V/PlatformDecoderModule AudioTrimmer[7ff6372e3700] ::PrepareTrimmers: sample[61000,81000] no trimming information
[Child 9598: MediaPDecoder #1]: V/PlatformDecoderModule AudioTrimmer[7ff6372e3700] ::HandleDecodedResult: sample[61000,81000] (decoded[54500,74500] no trimming needed
[RDD 9673: MediaPDecoder #2]: D/PlatformDecoderModule FFVPX: Got one frame output with pts=67000 dts=67000 duration=33000 opaque=-9223372036854775808
[Child 9598: MediaPDecoder #1]: V/PlatformDecoderModule AudioTrimmer[7ff6372e3700] ::PrepareTrimmers: sample[81000,101000] no trimming information
[Child 9598: MediaPDecoder #3]: V/PlatformDecoderModule AudioTrimmer[7ff6372e3700] ::HandleDecodedResult: sample[81000,101000] (decoded[74500,94500] no trimming needed
[Child 9598: MediaPDecoder #3]: V/PlatformDecoderModule AudioTrimmer[7ff6372e3700] ::PrepareTrimmers: sample[101000,121000] no trimming information
[Child 9598: MediaPDecoder #2]: V/PlatformDecoderModule AudioTrimmer[7ff6372e3700] ::HandleDecodedResult: sample[101000,121000] (decoded[94500,114500] no trimming needed
[RDD 9673: MediaPDecoder #3]: V/PlatformDecoderModule GetVideoBuffer: aCodecContext=7fa1cec76100 aFrame=7fa1cec9d100
[Child 9598: MediaPDecoder #2]: V/PlatformDecoderModule AudioTrimmer[7ff6372e3700] ::PrepareTrimmers: sample[121000,141000] no trimming information
[RDD 9673: MediaPDecoder #3]: D/PlatformDecoderModule FFVPX: Got one frame output with pts=100000 dts=100000 duration=33000 opaque=-9223372036854775808
[Child 9598: MediaPDecoder #1]: V/PlatformDecoderModule AudioTrimmer[7ff6372e3700] ::HandleDecodedResult: sample[121000,141000] (decoded[114500,134500] no trimming needed
```
</details>

elFarto, 2022-03-26:
Firefox doesn't seem to be trying to use VA-API or the driver at all. Do you have `LIBVA_DRIVER_NAME=nvidia` set?

Also, can you paste the about:support page from Firefox (there's a Copy text to Clipboard button)?

KidoVin01, 2022-03-26:
> Firefox doesn't seem to be trying to use VA-API or the driver at all. Do you have LIBVA_DRIVER_NAME=nvidia set?

Running `echo $LIBVA_DRIVER_NAME` gives me `nvidia` as expected

> Also, can you paste the about:support page from Firefox (there's a Copy text to Clipboard button)?
I removed some IDs just in case they are sensitive, but feel free to let me know if I missed anything.
<details>
<summary>about:support</summary>
```
Application Basics
------------------

Name: Firefox
Version: 98.0.2
Build ID: _
Distribution ID: archlinux
User Agent: Mozilla/5.0 (X11; Linux x86_64; rv:98.0) Gecko/20100101 Firefox/98.0
OS: Linux 5.15.31-1-lts #1 SMP Wed, 23 Mar 2022 09:55:25 +0000
OS Theme: Breeze / Adwaita-dark
Multiprocess Windows: 1/1
Fission Windows: 1/1 Enabled by default
Remote Processes: 6
Enterprise Policies: Inactive
Google Location Service Key: Found
Google Safebrowsing Key: Found
Mozilla Location Service Key: Found
Safe Mode: false

Crash Reports for the Last 3 Days
---------------------------------

Firefox Features
----------------

Name: DoH Roll-Out
Version: 2.0.0
ID: doh-rollout@mozilla.org

Name: Firefox Screenshots
Version: 39.0.1
ID: screenshots@mozilla.org

Name: Form Autofill
Version: 1.0.1
ID: formautofill@mozilla.org

Name: Picture-In-Picture
Version: 1.0.0
ID: pictureinpicture@mozilla.org

Name: Web Compatibility Interventions
Version: 30.0.0
ID: webcompat@mozilla.org

Name: WebCompat Reporter
Version: 1.4.2
ID: webcompat-reporter@mozilla.org

Remote Features
---------------

bug-1690367-rollout-moving-webrtc-networking-functionality-into-i-release-87-100: active

Remote Processes
----------------

Type: Privileged About
Count: 1

Type: Extension
[... 295 lines cut ...]

Application Settings
Requested Locales: ["en-US"]
Available Locales: ["en-US"]
App Locales: ["en-US"]
Regional Preferences: ["en-US"]
Default Locale: "en-US"
Operating System
System Locales: ["en-US"]
Regional Preferences: ["en-US"]

Remote Debugging (Chromium Protocol)
------------------------------------

Accepting Connections: false
URL:

Printing
--------

Modified print settings
-----------------------

```
</details>

elFarto, 2022-03-26:
Ok, Firefox has picked your Intel GPU to run on, and the driver doesn't currently support cross-GPU video decoding. Can you try running Firefox with this set: `__NV_PRIME_RENDER_OFFLOAD=1`.

KidoVin01, 2022-03-26:
I tried running `__NV_PRIME_RENDER_OFFLOAD=1 firefox` but it did not change anything.

I copied the about:support just in case:
<details>
<summary></summary>
Application Basics
------------------

Name: Firefox
Version: 98.0.2
Build ID: _
Distribution ID: archlinux
User Agent: Mozilla/5.0 (X11; Linux x86_64; rv:98.0) Gecko/20100101 Firefox/98.0
OS: Linux 5.15.31-1-lts #1 SMP Wed, 23 Mar 2022 09:55:25 +0000
OS Theme: Breeze / Adwaita-dark
Multiprocess Windows: 1/1
Fission Windows: 1/1 Enabled by default
Remote Processes: 6
Enterprise Policies: Inactive
Google Location Service Key: Found
Google Safebrowsing Key: Found
Mozilla Location Service Key: Found
Safe Mode: false

Crash Reports for the Last 3 Days
---------------------------------

Firefox Features
----------------

Name: DoH Roll-Out
Version: 2.0.0
ID: doh-rollout@mozilla.org

Name: Firefox Screenshots
Version: 39.0.1
ID: screenshots@mozilla.org

Name: Form Autofill
Version: 1.0.1
ID: formautofill@mozilla.org

Name: Picture-In-Picture
Version: 1.0.0
ID: pictureinpicture@mozilla.org

Name: Web Compatibility Interventions
Version: 30.0.0
ID: webcompat@mozilla.org

Name: WebCompat Reporter
Version: 1.4.2
ID: webcompat-reporter@mozilla.org

Remote Features
---------------

bug-1690367-rollout-moving-webrtc-networking-functionality-into-i-release-87-100: active

Remote Processes
----------------

Type: Privileged About
Count: 1

Type: Extension
Count: 1

Type: Preallocated
Count: 3
[... 289 lines cut ...]
-----------------------------------

Application Settings
Requested Locales: ["en-US"]
Available Locales: ["en-US"]
App Locales: ["en-US"]
Regional Preferences: ["en-US"]
Default Locale: "en-US"
Operating System
System Locales: ["en-US"]
Regional Preferences: ["en-US"]

Remote Debugging (Chromium Protocol)
------------------------------------

Accepting Connections: false
URL:

Printing
--------

Modified print settings
-----------------------

</details>

elFarto, 2022-03-26:
Ok, try with `__NV_PRIME_RENDER_OFFLOAD=1` and `__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/10_nvidia.json`.

KidoVin01, 2022-03-26:
Sadly still didn't work.

I do get the following error:
```
[GFX1-]: glxtest: libEGL missing eglGetDisplayDriverName
```

<details>
<summary>about:support</summary>
```
Application Basics
------------------

Name: Firefox
Version: 98.0.2
Build ID: _
Distribution ID: archlinux
User Agent: Mozilla/5.0 (X11; Linux x86_64; rv:98.0) Gecko/20100101 Firefox/98.0
OS: Linux 5.15.31-1-lts #1 SMP Wed, 23 Mar 2022 09:55:25 +0000
OS Theme: Breeze / Adwaita-dark
Multiprocess Windows: 1/1
Fission Windows: 1/1 Enabled by default
Remote Processes: 8
Enterprise Policies: Inactive
Google Location Service Key: Found
Google Safebrowsing Key: Found
Mozilla Location Service Key: Found
Safe Mode: false

Crash Reports for the Last 3 Days
---------------------------------

Firefox Features
----------------

Name: DoH Roll-Out
Version: 2.0.0
ID: doh-rollout@mozilla.org

Name: Firefox Screenshots
Version: 39.0.1
ID: screenshots@mozilla.org

Name: Form Autofill
Version: 1.0.1
ID: formautofill@mozilla.org

Name: Picture-In-Picture
Version: 1.0.0
ID: pictureinpicture@mozilla.org

Name: Web Compatibility Interventions
Version: 30.0.0
ID: webcompat@mozilla.org

Name: WebCompat Reporter
Version: 1.4.2
ID: webcompat-reporter@mozilla.org

Remote Features
---------------

bug-1690367-rollout-moving-webrtc-networking-functionality-into-i-release-87-100: active

Remote Processes
----------------

Type: Privileged About
Count: 1

[... 302 lines cut ...]

Application Settings
Requested Locales: ["en-US"]
Available Locales: ["en-US"]
App Locales: ["en-US"]
Regional Preferences: ["en-US"]
Default Locale: "en-US"
Operating System
System Locales: ["en-US"]
Regional Preferences: ["en-US"]

Remote Debugging (Chromium Protocol)
------------------------------------

Accepting Connections: false
URL:

Printing
--------

Modified print settings
-----------------------

```
</details>

elFarto, 2022-03-26:
Hmm, I'm not sure what's going on here. Firefox seems dead set against using the NVIDIA card. I'll do some more research into it and see if I can find out why.

KidoVin01, 2022-03-26:
> Can you check that you have the file /usr/share/glvnd/egl_vendor.d/10_nvidia.json?

Yeap, the content is as follows:
```
{
    "file_format_version" : "1.0.0",
    "ICD" : {
        "library_path" : "libEGL_nvidia.so.0"
    }
}
```

elFarto, 2022-03-26:
Hang on, that first test (with just `__NV_PRIME_RENDER_OFFLOAD=1`) did actually switch it over to the NVIDIA GPU. So that's the correct option. Can you run that again, and try setting this Firefox option in about:config `gfx.webrender.compositor.force-enabled=true`. You may have to restart FF after setting that.

KidoVin01, 2022-03-26:
I get this error now:
```
[GFX1-]: Failed to create EGLSurface!: 0x3009
[GFX1-]: Failed to create EGLSurface
[GFX1-]: Fallback WR to SW-WR
```

Running `nvidia-smi` shows very little usage despite loading a 4k video on youtube:
```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 510.54       Driver Version: 510.54       CUDA Version: 11.6     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0 Off |                  N/A |
| N/A   63C    P0    15W /  N/A |     18MiB /  4096MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A       545      G   /usr/lib/Xorg                      14MiB |
|    0   N/A  N/A      4015      G   /usr/lib/firefox/firefox            2MiB |
+-----------------------------------------------------------------------------+
```

<details>
<summary>about:support</summary>
```
Application Basics
------------------

Name: Firefox
Version: 98.0.2
Build ID: _
Distribution ID: archlinux
User Agent: Mozilla/5.0 (X11; Linux x86_64; rv:98.0) Gecko/20100101 Firefox/98.0
OS: Linux 5.15.31-1-lts #1 SMP Wed, 23 Mar 2022 09:55:25 +0000
OS Theme: Breeze / Adwaita-dark
Multiprocess Windows: 1/1
Fission Windows: 1/1 Enabled by default
Remote Processes: 6
Enterprise Policies: Inactive
Google Location Service Key: Found
Google Safebrowsing Key: Found
Mozilla Location Service Key: Found
Safe Mode: false

Crash Reports for the Last 3 Days
---------------------------------

Firefox Features
----------------

Name: DoH Roll-Out
Version: 2.0.0
ID: doh-rollout@mozilla.org

Name: Firefox Screenshots
Version: 39.0.1
ID: screenshots@mozilla.org

Name: Form Autofill
Version: 1.0.1
ID: formautofill@mozilla.org
[... 329 lines cut ...]

Application Settings
Requested Locales: ["en-US"]
Available Locales: ["en-US"]
App Locales: ["en-US"]
Regional Preferences: ["en-US"]
Default Locale: "en-US"
Operating System
System Locales: ["en-US"]
Regional Preferences: ["en-US"]

Remote Debugging (Chromium Protocol)
------------------------------------

Accepting Connections: false
URL:

Printing
--------

Modified print settings
-----------------------

```
</details>

elFarto, 2022-03-26:
I've had a play around with this on my Optimus setup, and the only way I could reproduce it is if the Intel GPU was the default. Attempting to force Firefox onto the NVIDIA GPU would always produce an error (usually `[GFX1-]: Failed to create EGLSurface!: 0x3009`). However, I couldn't find a way around it.

I'm not sure if anything has changed with Firefox recently as I seem to remember it working, but I could be mistaken.

Unfortunately there's not much more I do to help, as this is a Firefox issue. I believe as soon as you can get Firefox to initialise on the NVIDIA GPU with webrender compositing, it should start to use VA-API.

KidoVin01, 2022-03-26:
With your Optimus laptop, did you use [Optimus Manager](https://github.com/Askannz/optimus-manager)?

> it is if the Intel GPU was the default. 

How did you select the discrete GPU be the default instead?

elFarto, 2022-03-26:
I don't have an optimus laptop, just a computer with an Intel and NVIDIA GPU in it. That also lets me change which one is initialised first in the BIOS.

KidoVin01, 2022-03-26:
I think it might related to my hardware issue instead then.

closed 2022-03-26 by KidoVin01