elFarto/nvidia-vaapi-driver #69: Not working on Wayland with Firefox 98

thesword53, 2022-03-26:
```
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c:1634       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7fcabc5225c0
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/export-buf.c: 171          findCudaDisplay Found 4 EGL devices
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/export-buf.c: 175          findCudaDisplay Got EGL_CUDA_DEVICE_NV value '0' from device 0
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/export-buf.c: 135    checkModesetParameter Checking device file: /dev/dri/renderD129
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/export-buf.c: 210             initExporter Got EGLDisplay from CUDA device
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/export-buf.c: 240             initExporter Driver doesn't support 16-bit surfaces
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/export-buf.c:  95                reconnect Reconnecting to stream
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 964      nvQueryImageFormats In nvQueryImageFormats
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 449           nvCreateConfig got profile: 6 with 0 attributes
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c:1338 nvQuerySurfaceAttributes with 4 (nil) 0
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c:1338 nvQuerySurfaceAttributes with 4 0x7fca93bfab00 5
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 449           nvCreateConfig got profile: 7 with 0 attributes
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c:1338 nvQuerySurfaceAttributes with 4 (nil) 0
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c:1338 nvQuerySurfaceAttributes with 4 0x7fca93bfa500 5
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 449           nvCreateConfig got profile: 13 with 0 attributes
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c:1338 nvQuerySurfaceAttributes with 4 (nil) 0
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c:1338 nvQuerySurfaceAttributes with 4 0x7fca93bfaf80 5
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 449           nvCreateConfig got profile: 19 with 0 attributes
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c:1338 nvQuerySurfaceAttributes with 10 (nil) 0
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c:1338 nvQuerySurfaceAttributes with 10 0x7fca93bfab80 5
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 449           nvCreateConfig got profile: 6 with 0 attributes
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c:1338 nvQuerySurfaceAttributes with 4 (nil) 0
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c:1338 nvQuerySurfaceAttributes with 4 0x7fca93bfae80 5
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c:1338 nvQuerySurfaceAttributes with 4 (nil) 32714
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c:1338 nvQuerySurfaceAttributes with 4 0x7fca93bfad80 5
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 544        nvCreateSurfaces2 creating 1 surface(s) 1920x1088, format 1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c:1056            nvDeriveImage In nvDeriveImage
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 544        nvCreateSurfaces2 creating 1 surface(s) 1920x1088, format 1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 544        nvCreateSurfaces2 creating 1 surface(s) 1920x1088, format 1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 544        nvCreateSurfaces2 creating 1 surface(s) 1920x1088, format 1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 544        nvCreateSurfaces2 creating 1 surface(s) 1920x1088, format 1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 544        nvCreateSurfaces2 creating 1 surface(s) 1920x1088, format 1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 544        nvCreateSurfaces2 creating 1 surface(s) 1920x1088, format 1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 544        nvCreateSurfaces2 creating 1 surface(s) 1920x1088, format 1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 544        nvCreateSurfaces2 creating 1 surface(s) 1920x1088, format 1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 544        nvCreateSurfaces2 creating 1 surface(s) 1920x1088, format 1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 544        nvCreateSurfaces2 creating 1 surface(s) 1920x1088, format 1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 544        nvCreateSurfaces2 creating 1 surface(s) 1920x1088, format 1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 544        nvCreateSurfaces2 creating 1 surface(s) 1920x1088, format 1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 544        nvCreateSurfaces2 creating 1 surface(s) 1920x1088, format 1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 544        nvCreateSurfaces2 creating 1 surface(s) 1920x1088, format 1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 544        nvCreateSurfaces2 creating 1 surface(s) 1920x1088, format 1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 544        nvCreateSurfaces2 creating 1 surface(s) 1920x1088, format 1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 544        nvCreateSurfaces2 creating 1 surface(s) 1920x1088, format 1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 544        nvCreateSurfaces2 creating 1 surface(s) 1920x1088, format 1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 544        nvCreateSurfaces2 creating 1 surface(s) 1920x1088, format 1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 544        nvCreateSurfaces2 creating 1 surface(s) 1920x1088, format 1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 544        nvCreateSurfaces2 creating 1 surface(s) 1920x1088, format 1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 544        nvCreateSurfaces2 creating 1 surface(s) 1920x1088, format 1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 544        nvCreateSurfaces2 creating 1 surface(s) 1920x1088, format 1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 449           nvCreateConfig got profile: 6 with 0 attributes
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c:1338 nvQuerySurfaceAttributes with 4 (nil) 0
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c:1338 nvQuerySurfaceAttributes with 4 0x7fca93bfa880 5
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c:1338 nvQuerySurfaceAttributes with 4 (nil) 0
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c:1338 nvQuerySurfaceAttributes with 4 0x7fca93bfa500 5
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 627          nvCreateContext with 24 render targets, 24 surfaces, at 1920x1088
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c: 901             nvEndPicture cuvid decoded successful to idx: 0
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/vabackend.c:1560    nvExportSurfaceHandle got address 206C00000 (2048) for surface 29 (picIdx: 0)
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/export-buf.c: 301          allocateSurface presenting frame 1920x1088 0x7fca93bdeca0 0x7fca933d1280
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/export-buf.c: 318          allocateSurface Adding frame from EGLStream: 0x7fca933d2ca1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/export-buf.c: 331          allocateSurface Acquired image from EGLStream: 0x7fca933d2ca1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/export-buf.c: 422            exportCudaPtr Detected NV12/NV21 NVIDIA driver bug, attempting to work around
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/export-buf.c: 365              freeSurface Destroying EGLImage: 0x7fca933d2ca1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/export-buf.c: 371              freeSurface Destroying CUarray: 0x7fca93bdeca0
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/export-buf.c: 371              freeSurface Destroying CUarray: 0x7fca933d1280
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/export-buf.c: 301          allocateSurface presenting frame 1920x1088 0x7fca93bdeca0 0x7fca933d13a0
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/export-buf.c: 318          allocateSurface Adding frame from EGLStream: 0x7fca933d29d1
[32561-32706] ../nvidia-vaapi-driver-0.0.5/src/export-buf.c: 331          allocateSurface Acquired image from EGLStream: 0x7fca933d29d1
[... cut ...]
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
```

thesword53, 2022-03-26:
I found a workaround:
I have to set `EGL_PLATFORM=wayland` otherwise Firefox will try to use X11 with EGL even on native wayland.
On `README.md` you may change the line with `EGL_PLATFORM` from `This option is needed on current FF nightly builds, due to a regression that has been introduced.` to `This option is needed on FF98+, due to a regression that has been introduced.`

elFarto, 2022-03-27:
Yep, that's a known issue. I've updated the README as you suggested.

closed 2022-03-27 by elFarto

Darkspirit, 2022-05-16:
https://bugzilla.mozilla.org/show_bug.cgi?id=1769499