elFarto/nvidia-vaapi-driver #71: Optimus hardware + latest from git => segfault

MageSlayer, 2022-04-06:
Hi

I am trying to start vainfo, but it segfaults.
See log below.

My variables:
```
denis@devuan-nb:~$ set | grep "NV\|GLX\|LIBVA\|VDPA\|DRI"
DRI_PRIME=1
LIBVA_DRIVER_NAME=nvidia
VDPAU_DRIVER=nvidia
__GLX_VENDOR_LIBRARY_NAME=nvidia
__NV_PRIME_RENDER_OFFLOAD=1
```
Hardware:
```
denis@devuan-nb:~$ lspci | grep "VGA\|3D"
00:02.0 VGA compatible controller: Intel Corporation CometLake-H GT2 [UHD Graphics] (rev 05)
01:00.0 3D controller: NVIDIA Corporation TU117M [GeForce GTX 1650 Ti Mobile] (rev a1)
```

```
denis@devuan-nb:~$ NVD_LOG=1 gdb vainfo
GNU gdb (Debian 10.1-2+b1) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from vainfo...
(No debugging symbols found in vainfo)
(gdb) r
Starting program: /usr/bin/vainfo 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
libva info: VA-API version 1.10.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
[New Thread 0x7fffee885640 (LWP 10934)]
libva info: Found init function __vaDriverInit_1_0
[10928-10928] ../src/vabackend.c:1642       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x55555556b3d0 10
[10928-10928] ../src/vabackend.c:1645       __vaDriverInit_1_0 Non-DRM display type detected, defaulting to GPU ID 0. Use NVD_GPU to pick a specific GPU.
[10928-10928] ../src/export-buf.c: 164       findGPUIndexFromFd Looking for GPU index: 0
[10928-10928] ../src/export-buf.c: 175       findGPUIndexFromFd Found 4 EGL devices
[10928-10928] ../src/export-buf.c: 184       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
[10928-10928] ../src/export-buf.c: 136 checkModesetParameterFromFd Unable to check nvidia_drm modeset setting
[10928-10928] ../src/export-buf.c: 208       findGPUIndexFromFd Selecting EGLDevice 0
[New Thread 0x7fffed746640 (LWP 10935)]
[New Thread 0x7fffecd81640 (LWP 10936)]
[10928-10928] ../src/export-buf.c: 270             initExporter Driver doesn't support 16-bit surfaces
[10928-10928] ../src/export-buf.c: 103                reconnect Reconnecting to stream
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.10 (libva 2.12.0)
vainfo: Driver version: VA-API NVDEC driver

Thread 1 "vainfo" received signal SIGSEGV, Segmentation fault.
0x0000000000000000 in ?? ()
(gdb) bt
#0  0x0000000000000000 in  ()
#1  0x00007ffff7fb9801 in vaToCuCodec (profile=VAProfileMPEG2Simple) at ../src/vabackend.c:261
#2  0x00007ffff7fb9d98 in nvQueryConfigProfiles (ctx=0x55555556b3d0, profile_list=0x555555bff070, num_profiles=0x7fffffffde9c) at ../src/vabackend.c:376
#3  0x00007ffff7e224b5 in vaQueryConfigProfiles () at /usr/lib/x86_64-linux-gnu/libva.so.2
#4  0x00005555555564e3 in  ()
#5  0x00007ffff7c547fd in __libc_start_main (main=
    0x555555556370, argc=1, argv=0x7fffffffdfc8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffdfb8) at ../csu/libc-start.c:332
#6  0x00005555555568aa in  ()

```

Any help?

elFarto, 2022-04-07:
That's a very odd error, which shouldn't happen unless the compiler didn't build the driver correctly. Did you have any CFLAGS set while building it?

MageSlayer, 2022-04-07:
Nope.
```
denis@devuan-nb:~$ echo $CFLAGS

```

elFarto, 2022-04-07:
Can you send me the nvidia_drv_video.so binary that the build produced? It seems it somehow missing a critical section of the executable that's needed for it to work.

rkoot, 2022-04-07:
libva 2.10 might be too old?

MageSlayer, 2022-04-08:
[nvidia_drv_video.so.gz](https://github.com/elFarto/nvidia-vaapi-driver/files/8449217/nvidia_drv_video.so.gz)

MageSlayer, 2022-04-08:
> VA-API 1.10 might be too old?

Bingo. After upgrading to libva-dev to 2.14.0-1 it looks much better now. 
```
denis@devuan-nb:~/temp/nvidia-vaapi-driver$ vainfo 
libva info: VA-API version 1.14.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.14 (libva 2.12.0)
vainfo: Driver version: VA-API NVDEC driver
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
```

I guess it's now to check acceleration :)

MageSlayer, 2022-04-08:
Nope.
Still no good.
Firefox 99 still using software video decode - my laptop still produces sounds like a hurricane :)

```
denis@devuan-nb:~/temp/nvidia-vaapi-driver$ firefox
[GFX1-]: No GPUs detected via PCI
[GFX1-]: glxtest: process failed (received signal 11)
```
Does this log help somehow?

elFarto, 2022-04-08:
Firefox 99 has a separate issue that's preventing it from working at the moment. You'll need to try FF98 or one of the FF100 nightlies.

elFarto, 2022-04-08:
> > VA-API 1.10 might be too old?
> 
> Bingo. After upgrading to libva-dev to 2.14.0-1 it looks much better now.

Honestly, I'm surprised that was the issue. I can't see any changes to libva between 2.10 (1.10 is the API version, not the release version number) and 2.14 that would cause that sort of issue, which amounts to it failing to load/initialise the executable correctly.

MageSlayer, 2022-04-08:
> Firefox 99 has a separate issue that's preventing it from working at the moment. You'll need to try FF98 or one of the FF100 nightlies.

98 was removed from Debian repos and nightly does not work either :(

MageSlayer, 2022-04-08:
Hm. 
Perhaps I am doing something wrong, but FF98 also does not use acceleration with my prime configuration.

elFarto, 2022-04-08:
Are you starting firefox with `MOZ_DISABLE_RDD_SANDBOX=1` set? Either way, start Firefox with the follow variables and post the log file: `NVD_LOG=1 MOZ_LOG="PlatformDecoderModule:5,Dmabuf:5" firefox`.

MageSlayer, 2022-04-08:
> Are you starting firefox with MOZ_DISABLE_RDD_SANDBOX=1 set?
Nope. But enabling it for FF98 makes no difference.
Are you starting firefox with MOZ_DISABLE_RDD_SANDBOX=1 set?

elFarto, 2022-04-08:
Ok, it seems you're missing a bunch of the Firefox config options that are in the README (media.ffmpeg.vaapi.enabled, etc...), set those, restart Firefox and try again.

MageSlayer, 2022-04-08:
Hm.
It was a clear issue with that. However even after I disabled autoupdate which messed it, rechecked those settings, I can't still get low processor usage. Youtube video are still consume more that 1 core at a time.

Could you check my log again?
[ff.log.gz](https://github.com/elFarto/nvidia-vaapi-driver/files/8454977/ff.log.gz)

elFarto, 2022-04-08:
EGL isn't enabled, ensure that `gfx.x11-egl.force-enabled` is set to `true` (you'll need to restart FF on setting that).

MageSlayer, 2022-04-08:
[image]

elFarto, 2022-04-08:
Can you paste the output of `about:support`, there's a `Copy text to clipboard` button at the top.

MageSlayer, 2022-04-08:
<h2 xmlns="http://www.w3.org/1999/xhtml" class="major-section" id="app-basics" data-l10n-id="app-basics-title">Application Basics</h2>

      

Name | Firefox
-- | --
98.0
20220304153049
 
release
Mozilla/5.0 (X11; Linux x86_64; rv:98.0) Gecko/20100101 Firefox/98.0
Linux 5.10.0-13-amd64 #1 SMP Debian 5.10.106-1 (2022-03-17)
Adwaita / Adwaita
1/1
1/1               Enabled by default
5
Inactive
Found
Found
Found
false

    Application Basics
------------------

Name: Firefox
Version: 98.0
Build ID: 202203041[5](https://github.com/elFarto/nvidia-vaapi-driver/issues/71#remote-processes)3049
Distribution ID:
Update Channel: release
User Agent: Mozilla/5.0 (X11; Linux x86_64; rv:98.0) Gecko/20100101 Firefox/98.0
OS: Linux 5.10.0-13-amd64 #1 SMP Debian 5.10.106-1 (2022-03-17)
OS Theme: Adwaita / Adwaita
Multiprocess Windows: 1/1
Fission Windows: 1/1 Enabled by default
Remote Processes: 5
Enterprise Policies: Inactive
Google Location Service Key: Found
Google Safebrowsing Key: Found
Mozilla Location Service Key: Found
Safe Mode: false

Crash Reports for the Last 3 Days
---------------------------------

Firefox Features
----------------

Name: DoH Roll-Out
Version: 2.0.0
ID: doh-rollout@mozilla.org

Name: Firefox Screenshots
Version: 39.0.1
ID: screenshots@mozilla.org

Name: Form Autofill
Version: 1.0.1
ID: formautofill@mozilla.org

Name: Picture-In-Picture
Version: 1.0.0
ID: pictureinpicture@mozilla.org

Name: Web Compatibility Interventions
Version: 30.0.0
ID: webcompat@mozilla.org

Name: WebCompat Reporter
Version: 1.4.2
[... 357 lines cut ...]

Internationalization & Localization
-----------------------------------

Application Settings
Requested Locales: ["en-US"]
Available Locales: ["en-US"]
App Locales: ["en-US"]
Regional Preferences: ["en-US"]
Default Locale: "en-US"
Operating System
System Locales: ["en-US"]
Regional Preferences: ["en-US"]

Remote Debugging (Chromium Protocol)
------------------------------------

Accepting Connections: false
URL:

Printing
--------

Modified print settings
-----------------------

elFarto, 2022-04-08:
Looks like Firefox has given up trying to initialise the hardware and is falling back to software for everything. You can try removing the `gfx.blacklist.*` config items.

I'm assuming you've got Xorg running on your Intel card. Trying to get FF to initialise the correct card is a bit tricky. I'll have a play around with it on my other setup tomorrow to see if I can come up with a working solution.

MageSlayer, 2022-04-09:
I tried deleting every blacklist items and restart, but it does help acceleration and just recreated those blacklist records again.

As for hybrid graphics:
```
denis@devuan-nb:~$ xrandr --listproviders
Providers: number : 2
Provider 0: id: 0x1b8 cap: 0x1, Source Output crtcs: 0 outputs: 0 associated providers: 1 name:NVIDIA-0
Provider 1: id: 0x1df cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 4 associated providers: 1 name:modesetting
```

elFarto, 2022-04-09:
I've spent a while trying to get my Optimus machine to work correctly when running on the Intel GPU, but there doesn't seem to be a way to do it. I'm not quite sure what the root cause of the issue is, but it seems Firefox gets confused when there's multiple EGL implementations available.

It's possible that using this option `__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/10_nvidia.json` (assuming that file is in the same place) when starting Firefox might help, but it doesn't work on my PC likely because it's the older 470 driver series.

MageSlayer, 2022-04-09:
I have 460.91.03-1 nvidia-driver.
```
denis@devuan-nb:~/temp/ff98$ cat /usr/share/glvnd/egl_vendor.d/10_nvidia.json
{
    "file_format_version" : "1.0.0",
    "ICD" : {
        "library_path" : "libEGL_nvidia.so.0"
    }
}
```
However, FF98 just emits:
```
[GFX1-]: glxtest: libEGL no display
[GFX1-]: glxtest: libEGL no display
```
... and acceleration still does not work.

elFarto, 2022-04-09:
> I have 460.91.03-1 nvidia-driver.

That's quite an old driver series. With a 1650 card you can use the 510 series, which may work better especially with Firefox which really wants GBM.

MageSlayer, 2022-04-09:
Aha.
Looks upgrading to 510.47.03-1 really did the trick.
Now both FF98 & FF-nightly consume only ~20% CPU time.
Thanks a lot.

pobrn, 2022-04-11:
Sorry everyone, I am responsible for the initial segmentation fault. In libva 2.10.0, `VABufferTypeMax == 59` which makes the size of the `NVCodec` struct equal 496 bytes (on x86-64), but gcc decided to increase the alignment to 32 when laying out the objects in the executable file. That means there was 16 bytes of padding between the `NVCodec` structs in the executable that was not accounted for at runtime. So
```c
        cudaVideoCodec cvc = c->computeCudaCodec(profile);
```
jumped to a NULL pointer (a NULL pointer because probably because most of the `NVCodec::handlers` are NULL in all codecs) as can be seen in the initial stack trace.

Funnily, I have created #67 in order to solve exactly this kind of problem, but it was the wrong approach.

elFarto, 2022-04-11:
Ok, for some reason I was thinking that the whole section was missing/null, but just being off by a few bytes would also explain the null pointer.

closed 2022-04-15 by elFarto