elFarto/nvidia-vaapi-driver #73: Driver removal

get-coff3e, 2022-04-14:
Hello,

I've taken interest in this driver for a while now, but I suspect I'm having some video decoding issues among everything on my system.

I don't see a documented way to remove the driver, is there something im missing? or is it easier to simply reinstall my existing linux installation?

elFarto, 2022-04-15:
There's no need to reinstall the whole system. The driver only consists of a single file `nvidia_drv_video.so` normally located in `/usr/lib64/dri` (sometimes in a different lib directory depending on how the system is configured). Simply removing that file will uninstall it.

In addition, if you have the vdpau driver installed (`vdpau_drv_video.so` in the same directory), you should probably reinstate the link from `nvidia_drv_video.so` to `vdpau_drv_video.so` using this command: `ln -s vdpau_drv_video.so /usr/lib64/dri/nvidia_drv_video.so`

elFarto, 2022-06-05:
I've updated the README with instructions on how to remove the driver.

closed 2022-06-05 by elFarto