elFarto/nvidia-vaapi-driver #75: Failed to grab modeset ownership

jessestricker, 2022-04-19:
When I locked my desktop and came back 10 minutes later and unlocked it,
the running Firefox instance crashed. 

In the journal, there was this kernel error:
```
[drm:drm_new_set_master [drm]] *ERROR* [nvidia-drm] [GPU ID 0x00002700] Failed to grab modeset ownership
```

followed by a coredump ([firefox.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/8512859/firefox.txt)).

Is this an issue with this driver, Firefox, the Pantheon Desktop or something else?

## System Information

- NixOS unstable 
- Firefox 99.0.1 (64-bit)
- nvidia-vaapi-driver v0.0.5
- DE: Pantheon

## Similar Issues

- #41

elFarto, 2022-04-19:
v0.0.5 should have stopped causing the `Failed to grab modeset ownership` message to be printed, so the fact you're seeing it means something else is grabbing the ownership.

I don't see the nvidia_drv_video driver anywhere in the stacktrace, but the first Firefox thread is attempting to release some DMA-BUF EGLImages, and is causing an error somewhere in the NVIDIA EGL driver.

So my guess is that it's like an NVIDIA issue. It's possible that something cause Firefox to lose its GPU context and it's attempting to release some now invalid resources. Either way, the NVIDIA driver shouldn't crash like that.

elFarto, 2022-06-05:
Closing this issue as unable to reproduce, and it doesn't appear to be a driver issue

closed 2022-06-05 by elFarto