elFarto/nvidia-vaapi-driver #76: No GPUs detected via PCI when launching firefox

Nixuge, 2022-04-21:
Hey, i'm currently having an issue with firefox.
I'm using arch, so I installed nvidia-vaapi-driver from the aur
All the needed variables are set the variables in firefox (except media.av1.enabled and widget.dmabuf.force-enabled)
I also set the other environment variables, but when starting firefox i get the error:
`[GFX1-]: No GPUs detected via PCI`
`[GFX1-]: glxtest: process failed (received signal 11)`
Tried with firefox stable (99.0.1) and nightly (101.0a1), getting the same result from both

`lspci | grep "VGA\|3D"` results in:
`27:00.0 VGA compatible controller: NVIDIA Corporation TU104 [GeForce RTX 2080] (rev a1)`

elFarto, 2022-04-21:
Unfortunately that's a Firefox error, not one caused by this driver. I haven't seen that particular error before, so you will need to raise an issue with the Firefox team to get to try and get it resolved.

Just out of curiousity, could you paste the output of running: `NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia vainfo`?

Nixuge, 2022-04-21:
aw that's a shame, well just have to hope it'll magically work at some point 
here's the command output:
```
  ~ ❯ NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia vainfo
[5774-5774] ../src/vabackend.c:1642       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x5599fed04da0 10
[5774-5774] ../src/vabackend.c:1645       __vaDriverInit_1_0 Non-DRM display type detected, defaulting to GPU ID 0. Use NVD_GPU to pick a specific GPU.
[5774-5774] ../src/export-buf.c: 164       findGPUIndexFromFd Looking for GPU index: 0
[5774-5774] ../src/export-buf.c: 175       findGPUIndexFromFd Found 3 EGL devices
[5774-5774] ../src/export-buf.c: 184       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
[5774-5774] ../src/export-buf.c: 136 checkModesetParameterFromFd Unable to check nvidia_drm modeset setting
[5774-5774] ../src/export-buf.c: 208       findGPUIndexFromFd Selecting EGLDevice 0
[5774-5774] ../src/export-buf.c: 270             initExporter Driver doesn't support 16-bit surfaces
[5774-5774] ../src/export-buf.c: 103                reconnect Reconnecting to stream
vainfo: VA-API version: 1.14 (libva 2.14.0)
vainfo: Driver version: VA-API NVDEC driver
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
[5774-5774] ../src/vabackend.c:1626              nvTerminate Terminating 0x5599fed04da0
[5774-5774] ../src/export-buf.c:  65          releaseExporter Releasing exporter, 0 outstanding frames
[5774-5774] ../src/export-buf.c:  82          releaseExporter Done releasing frames
  ~ ❯
```

elFarto, 2022-04-22:
Well, for what it's worth, the nvidia-vaapi-driver part seems to be detecting the GPU just fine.

jeois, 2022-05-03:
I ran into this issue before, but I can't remember what I did to solve it.  I think it was due AUR's firefox-ubuntu-bin, and their weird apparmor setup.  Or, have you tried it without the "wayland" environment variable?  If you're using x11-egl, then using that might cause this error.

Nixuge, 2022-05-04:
I don't have anything like that installed unfortnately and running without the wayland var doesn't throw errors but doesn't use the gpu either, i don't i'm using x11-egl

jeois, 2022-05-04:
By "error" I meant the one you posted in the OP about Firefox not detecting the GPU.  The only way I can reproduce it is like this:

`MOZ_LOG="PlatformDecoderModule:5" EGL_PLATFORM="wayland" firefox-developer-edition`
`[GFX1-]: No GPUs detected via PCI`
`[GFX1-]: glxtest: process failed (received signal 11)`
Then, if I remove the "wayland" environment variable, it works (mostly).   

If I were you, I'd try using FF with wayland (MOZ_ENABLE_WAYLAND=1) instead of x11 or vice versa, and manually setting the variables for testing.  In my case, I think I forgot I had set it earlier in a different init script than the one I was editing.

mustafademiral, 2022-05-05:
I'm using Arch with X11 and i was also have same error. And now with FF101.0b2 beta, when i set EGL_PLATFORM=wayland i'm still getting no gpu detect error but when i set EGL_PLATFORM=x11 the error fixing and driver working.

king-gheedorah, 2022-05-14:
My Firefox was crashing at launch and then re-opening itself. I would get the attached crash report in journal: [journalctl-firefox-eglplatform-wayland.log](https://github.com/elFarto/nvidia-vaapi-driver/files/8691982/journalctl-firefox-eglplatform-wayland.log)

I reviewed all environment variables before trying it with and without `wayland`. For me, it seems that I get hardware acceleration either way, but if I use `EGL_PLATFORM=wayland`, Firefox crashes at launch and then re-opens.

As @mustafademiral recommended, I tried launching with `EGL_PLATFORM=x11` and it worked, no more crashing and hardware acceleration seems to be working fine now.

Perhaps a change occurred in a recent Firefox update?

reliant8307, 2022-05-19:
> [GFX1-]: glxtest: process failed (received signal 11)

This seems to be related to the following bug:

[https://bugzilla.mozilla.org/show_bug.cgi?id=1739611](https://bugzilla.mozilla.org/show_bug.cgi?id=1739611)

Since the Nvidia driver doesn't support eglGetDisplayDriverName, Firefox returns the glxtest error ("libEGL missing eglGetDisplayDriverName"). Firefox could retrieve the driver name from the PCI list, but this bug was opened 7 months ago and it's stalled, nobody seems interested.

It seems trivial to fix. Is anyone interested? I understand the issue, but I'm not proficient in C++. The issue is on the following code (from toolkit/xre/glxtest.cpp):

```
  if (eglGetDisplayDriverName) {
    // TODO(aosmond): If the driver name is empty, we probably aren't using Mesa
    // and instead a proprietary GL, most likely NVIDIA's. The PCI device list
    // in combination with the vendor name is very likely sufficient to identify
    // the device.
    const char* driDriver = eglGetDisplayDriverName(dpy);
    
    if (driDriver) {
      record_value("DRI_DRIVER\n%s\n", driDriver);
    }
  } else if (require_driver) {
    record_warning("libEGL missing eglGetDisplayDriverName");
    eglTerminate(dpy);
    dlclose(libegl);
    return false;
  }

```

In the case of Nvidia, it should get the driver name from the PCI list instead... Without fixing this bug, Firefox will be stuck in using GLX only, preventing it from using EGL.

jeois, 2022-05-20:
@reliant8307   Although this might be related, I don't think it is.  Firstly, the error messages from glxtest isn't the same as the one from this thread.  Secondly, we can only reproduce our issue if EGL_PLATFORM is set wrong (x11 or wayland), and it doesn't throw any errors if we set it correctly.  

For example, `about:support` shows Firefox detecting my Nvidia card and running x11_EGL, so clearly nothing is preventing most of us from using EGL with proprietary drivers, along with this vaapi driver, whether this other bug was fixed or not.

reliant8307, 2022-05-20:
> For example, `about:support` shows Firefox detecting my Nvidia card and running x11_EGL, so clearly nothing is preventing most of us from using EGL with proprietary drivers, along with this vaapi driver, whether this other bug was fixed or not.

Interesting because in my case EGL_PLATFORM doesn't seem to make any difference and I can only use EGL on Firefox fixing the bug I mentioned above (I tested with a ugly hack). Anyway, thanks for replying.

elFarto, 2022-06-05:
Closing this issue as it's not a driver issue.

closed 2022-06-05 by elFarto