elFarto/nvidia-vaapi-driver #77: The environment variable LIBVA_DRIVER_NAME=nvidia breaks the control panel

Alex-Bujorianu, 2022-04-23:
What the title says. Setting the LIBVA_DRIVER_NAME environment variable to nvidia breaks the Nvidia X Settings control panel. Even with all the environment variables and settings changed, I still have not been able to get this tool to work.

Firefox version: 99
nvidia-vaapi-driver: AUR version
Kernel: 5.15
Nvidia driver: 510
Display server: X11

elFarto, 2022-04-23:
I assume you're referring to the `nvidia-settings` application that comes with the NVIDIA X driver? How exactly is it breaking it? Is there an error printed?

Alex-Bujorianu, 2022-04-24:
> I assume you're referring to the `nvidia-settings` application that comes with the NVIDIA X driver? How exactly is it breaking it? Is there an error printed?

Yes, it is the nvidia-settings application. There is no error printed: running it in the terminal shows it just segfaults.

conorlmcbride, 2022-04-24:
I also have this issue. Took me awhile to figure out what was breaking nvidia control panel. I thought it had to do with upgrading to the 510 driver which I did at the same time.

Nvidia Driver Version: 510.60.02
Display Server: X11
Kernel: 5.15
Ubuntu 22.04 (Same error before updating from 21.10)

<details><summary>vaainfo output</summary>
> libva info: VA-API version 1.14.0
> libva info: Trying to open /usr/local/lib/dri/nvidia_drv_video.so
> libva info: Found init function __vaDriverInit_1_14
> libva info: va_openDriver() returns 0
> vainfo: VA-API version: 1.14 (libva 2.12.0)
> vainfo: Driver version: Splitted-Desktop Systems VDPAU backend for VA-API - 0.7.4
> vainfo: Supported profile and entrypoints
>       VAProfileMPEG2Simple            :	VAEntrypointVLD
>       VAProfileMPEG2Main              :	VAEntrypointVLD
>       VAProfileMPEG4Simple            :	VAEntrypointVLD
>       VAProfileMPEG4AdvancedSimple    :	VAEntrypointVLD
>       VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
>       VAProfileH264Main               :	VAEntrypointVLD
>       VAProfileH264High               :	VAEntrypointVLD
>       VAProfileVC1Simple              :	VAEntrypointVLD
>       VAProfileVC1Main                :	VAEntrypointVLD
>       VAProfileVC1Advanced            :	VAEntrypointVLD
>      VAProfileVP9Profile0            :	VAEntrypointVLD
</details>

rkoot, 2022-04-27:
Looks like the NVDEC driver isn't picked up.  I had this when I had the libva-vdpau driver installed (opensuse tumbleweed, your specific package name might vary)
Try uninstalling the libva-vdpau package, check whether /usr/lib64/dri/nvidia_drv_video.so exists and run vainfo again.

elFarto, 2022-04-27:
@conorlmcbride As rkoot mentioned, it looks like the VDPAU driver has overwritten the NVDEC one (you can tell by the `Splitted-Desktop Systems VDPAU backend for VA-API` in the vainfo output. Reinstall this driver and try again.

As for the crash, I'm not certain what's going on. As far as I can tell nvidia-settings does not interact with VA-API at all (and there's no reason for it to). If someone who's getting a crash could run it under gdb and get a backtrace, that would help immensely.

To run it under gdb use this command: `gdb nvidia-settings` (assuming that the LIBVA_DRIVER_NAME env var is already set), type `run` to start it, then when it crashes type `bt full` to get a backtrace.

conorlmcbride, 2022-04-28:
> 
nvidia-settings backtrace included here.

<details><summary>nvidia-settings backtrace</summary>
<p>

```
Starting program: /usr/bin/nvidia-settings 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff33b5640 (LWP 71307)]
[New Thread 0x7ffff2bb4640 (LWP 71308)]

Thread 1 "nvidia-settings" received signal SIGSEGV, Segmentation fault.
___pthread_mutex_lock (mutex=0xf3) at ./nptl/pthread_mutex_lock.c:80
80	./nptl/pthread_mutex_lock.c: No such file or directory.
(gdb) bt full
#0  ___pthread_mutex_lock (mutex=0xf3) at ./nptl/pthread_mutex_lock.c:80
        type = <optimized out>
        __PRETTY_FUNCTION__ = "___pthread_mutex_lock"
        id = <optimized out>
#1  0x00007ffff6549f65 in wl_proxy_create_wrapper () at /lib/x86_64-linux-gnu/libwayland-client.so.0
#2  0x00007ffff22a990e in  () at /lib/x86_64-linux-gnu/libnvidia-egl-wayland.so.1
#3  0x00007ffff0cac790 in  () at /lib/x86_64-linux-gnu/libEGL_nvidia.so.0
#4  0x00007ffff0c4c65c in  () at /lib/x86_64-linux-gnu/libEGL_nvidia.so.0
#5  0x00007ffff1080f65 in  () at /lib/x86_64-linux-gnu/libEGL.so.1
#6  0x0000555555571a85 in NvCtrlInitEglAttributes ()
#7  0x000055555556d9e0 in NvCtrlAttributeInit ()
#8  0x0000555555572f18 in  ()
#9  0x0000555555573c29 in NvCtrlConnectToSystem ()
#10 0x000055555555f468 in main ()
```
</p>
</details>

Thank you for your help! 
It was not easy removing vdpau and installing nvdec. I removed all packages referencing vdpau and removed the /usr/local/lib/dri/vdpau_drv_video.so file. Then I had to manually recreate the link from nvidia_drv_video.so to nvdec_drv_video.so. Not sure what caused vdpau drivers to be installed instead of nvdec.

elFarto, 2022-04-28:
It looks like you're trying to run nvidia-settings under wayland, which as far as I know doesn't work. If you're not under wayland, then check what environment variables are set, as something has convinced the NVIDIA EGL driver to use the wayland version rather than the X11 version.

conorlmcbride, 2022-05-10:
> 
Ah yes, I was blindly following the readme and set the `EGL_PLATFORM=wayland` env variable even though I use X11. Whoops. 
Thank you for helping me identify that.

I'm still definitely getting some weird errors when running apps trying to utilize hardware decoding, including firefox. But the error related to the control panel has been solved so I'll open another thread as to not hijack this one.

elFarto, 2022-06-19:
I'm going to close this issue. I've not seen any other reports of this happening, in addition to the fact that the nvidia-settings application doesn't interact with libva at all so shouldn't even be reading the LIBVA_DRIVER_NAME variable..

closed 2022-06-19 by elFarto